openwrt/mt76 #699: 5Ghz Wi-Fi doesn't work if DFS channel used on TP-Link Archer C6(EU) Ver3.20

kt315ua, 2022-09-12:
Basic working configuration:
[dmesg.txt](https://github.com/openwrt/mt76/files/9545890/dmesg.txt)
[wireless.txt](https://github.com/openwrt/mt76/files/9545902/wireless.txt)

Non DFS Channels: 36, 40, 44, 48, 149, 153, 157, and 161 

Steps to reproduce:

- Set Wi-Fi 5GHz channel to 52 (dfs channel)

Result:

- 5GHz Wi-Fi stops working:

[logread.txt](https://github.com/openwrt/mt76/files/9545947/logread.txt)

Reproduced  on versions:
<h3>System</h3><div>

Hostname | OpenWrt
-- | --
Model | TP-Link Archer C6 v3
Architecture | MediaTek MT7621 ver:1 eco:3
Target Platform | ramips/mt7621
Firmware Version | OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a
Kernel Version | 5.10.138

</div>

Hardware: TP-Link Archer C6(EU) Ver3.20

Borromini, 2022-09-15:
The MT7613BE radio does not support DFS by design.

sad-goldfish, 2022-09-30:
Not exactly a solution but setting list `hostapd_options 'ieee80211h=0'` in `/etc/config/wireless` should disable DFS and seems to avoid channels where it is necessary.

kt315ua, 2022-10-03:
> The MT7613BE radio does not support DFS by design.

Ok, maybe it's a fact, but it's not solution of a system bug.
If this chipset/platform doesn't support DFS, why it trying to use?