openwrt/mt76 #701: Rekeying PTK for STA <mac> but driver can't safely do that.

imwhocodes, 2022-09-19:
Hi

I'm using STA mode (connected to an AP on which I don't have control)
Connection is WPA2-Enterprise encrypted, every 15 minute I get this message on log `Rekeying PTK for STA XX:XX:XX:XX:XX:XX but driver can't safely do that.` and half the time the STA (so my router) disconnect going again through all connection process

Searching online and combing through the code it seem like to me that the driver do not have configured either `IEEE80211_KEY_FLAG_NO_AUTO_TX` and `NL80211_EXT_FEATURE_CAN_REPLACE_PTK0` flags

Is this a misconfiguration? or the driver is really unable to support this features?

For reference I'm running `OpenWrt SNAPSHOT r20686-f0ea3df439 / LuCI Master git-22.245.77360-10bcb22` on a `ASUS RT-AC85P`

Thank,
Luca

rany2, 2023-05-17:
Fixed by https://github.com/openwrt/mt76/commit/8ecf551c6d21be55cb0c8567542d1bb7c12f6410. This error should no longer appear on hardware that support this. If your device doesn't support, you could stop devices from disconnecting to rekey by seeing eap_reauth_period to 0 in OpenWRT BSS config.

imwhocodes, 2023-05-24:
Thanks

In my case I'm **STA** on a `802.1X` so I don't I have control over `eap_reauth_period` _(and on AP is set at 15 minutes)_
Last week I updated to openwrt-snapshot with these commits (with a lot of unrelated pain from switching to openssl3), connection is a **lot** more stable now

@nbd168 For context this is the timeline of what I noticed:

1. Initially connection was randomly dropping (_DE-AUTH_) on keys updates (more load on wifi = more probable)
2. After this [commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=d54c91bd9ab3c54ee06923eafbd67047816a37e4) connection was dropping **every** single key update with something on the line of "_invalid/expired key_" (I'm supposing that the CVE fixes made the management and data buffer order deterministic)
3. After https://github.com/openwrt/mt76/commit/8ecf551c6d21be55cb0c8567542d1bb7c12f6410 connections is stable

Felix I tagged you because I know you follow the mediatek drivers and I saw you committed the CVE fixes, hope this feedback was interesting,
Thanks again for your work,
Luca