openwrt/mt76 #702: Mediatek MT7986 / Banana Pi BPI-R3 Wifi not working

eckinga, 2022-09-23:
Someone opened a case in openwrt for this issue but i think its firmware related problem and better here.

```
[   10.127699] mt7986-wmac 18000000.wmac: HW/SW Version: 0x8a108a10, Build Time: 20220113162701a
[   10.127699] 
[   10.247829] mt7986-wmac 18000000.wmac: WM Firmware Version: ____000000, Build Time: 20220113162756
[   10.324188] mt7986-wmac 18000000.wmac: WA Firmware Version: DEV_000000, Build Time: 20220113163034
[   30.565222] mt7986-wmac 18000000.wmac: Message 00000002 (seq 4) timeout
[   30.571834] mt7986-wmac 18000000.wmac: Failed to start WA firmware
[   30.578095] mt7986-wmac: probe of 18000000.wmac failed with error -110'

frank-w, 2022-09-27:
maybe you miss eeprom-data?

daniel added it like me in the devicetree and with the firmware-files from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek probe works for me (creates 2 wifi-interfaces seen by "ip a").

eckinga, 2022-09-27:
Hello Frank, 
how to check and how to rewrite (is it a mtd device?)?
My experience so far with this problem:
I have 2 r3. One had the wifi working with image from sinovoip. after i installed the "official" snapshot the wifi stopped working.
That's why i thought it's firmware related. 
Today i reverted to the previous image and wifi didn't come back.
I also unpacked my second R3 today and installed the image from sinovoip but in contrast to the first one the wifi didnt work. Also with the snapshot i get the same error.

reddwarf69, 2022-09-29:
I have just tried with https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz, I didn't do anything else. It's worth mentioning that I'm running directly from the SD card, and it's a v1.1 board.

I guess the eeprom-data you are talking about is https://github.com/openwrt/openwrt/blob/master/target/linux/mediatek/dts/mt7986a-bananapi-bpi-r3.dts#L424, @frank-w?

The thing is that in https://openwrt.org/toh/sinovoip/bananapi_bpi-r3 there is a link to https://asciinema.org/a/518024 where it can be seen working just fine. And in https://github.com/openwrt/openwrt/commit/a96382c1bb204698cd43e82193877c10e4b63027 it says WiFi was working.

The eeprom-data in the HEAD of openwrt is still the same one that was added in https://github.com/openwrt/openwrt/commit/a96382c1bb204698cd43e82193877c10e4b63027, it has never changed.

frank-w, 2022-09-29:
Yes it's the portion i mean for eeprom-data. What kind of powersupply do you have (12v xA)? Are the sfp slots equipped (one/both) any other power consumers like m2 or mpcie card?

I guess it can be a current problem...

reddwarf69, 2022-09-29:
I was using a 12v x 2A power supply, it came with the board. But just to test, I have just tried with a 12v x 3.33A one, same problem.
I have just tried without anything in the SFP slot and, no, there has never been anything else plugged to the board.
I have yet to plug WiFi antennas! (That shouldn't make a difference, right?)

eckinga, 2022-09-29:
Now its a bit clearer. Just to say it's not a power supply issue.
Like i mentioned if have 2 BPI-R3, forgot to say they're both V1.1 and im booting from SD. Also tried from emmc and nand to make sure its not the sd card (sinovoip suggested to take a class 10 card low level formatted). SFP not equipped.

Have 2 power supplies from sinovoip and tested both, no difference. I just tested the board on a laboratory power supply with 12v@10a with no difference so there is no issue with that. Also not with the sd card.

What drives me crazy is the fact that with first board the wifi worked with the image from sinovoip and got broken as i installed the snapshot.

reddwarf69, 2022-09-29:
Is this supposed to be the same as https://github.com/openwrt/openwrt/blob/ec8fb542ec3e4f584444a97de5ac05dbc2a9cde5/target/linux/mediatek/dts/mt7986a-bananapi-bpi-r3.dts#L424 ?

```
root@OpenWrt:/tmp# hexdump /sys/firmware/devicetree/base/soc/wmac@18000000/media
tek,eeprom-data 
0000000 7986 0009 0c00 2643 0060 0000 0000 0000
0000010 0000 0000 0000 0000 0000 0000 0000 0000
*
0000050 0000 0000 0000 0000 0000 0000 0001 0000
0000060 0000 0000 0000 0000 0000 0000 0000 0000
*
0000100 0000 0000 0000 0008 0000 0000 0000 0000
0000110 0000 0000 0000 0000 0000 0000 0000 0000
*
0000190 6424 9090 2800 0000 1005 0000 0000 0000
00001a0 0000 0000 0000 0000 0000 0000 0000 0000
*
0000200 0200 001e 1e02 0200 001e 1e02 0200 0028
0000210 2802 0200 0028 2802 0000 0000 0000 0000
0000220 0000 0000 0000 0000 0000 0000 0000 0000
0000230 0000 0000 0000 0000 0000 8080 8080 f7fd
0000240 0309 0d15 8080 8080 8080 8080 0505 0d0d
0000250 1313 c6c6 c3c3 00c2 c200 c200 0000 8281
0000260 8585 c2c2 8282 8282 8585 c200 00c2 8100
0000270 8582 8785 c287 00c2 8181 8582 8785 c287
0000280 00c2 8100 8582 8785 c287 00c2 8100 8582
0000290 8785 c287 00c2 8100 8582 8785 c487 c2c4
00002a0 00c1 00c3 c3c3 00c1 8381 c383 c3c3 00c1
00002b0 8381 0083 c2c2 c0c2 8281 8484 0000 c300
00002c0 c3c3 00c1 8381 8683 c386 c3c3 00c1 8381
00002d0 8683 c286 c2c2 81c0 8482 8684 c386 c3c3
00002e0 00c1 8381 8683 c386 c3c3 00c1 8381 8683
00002f0 c386 c3c3 00c1 8381 8683 2286 0028 2822
0000300 2200 0028 2822 0000 dddd dddd dddd dddd
0000310 bbdd bbbb cccc ddcc dddd dddd dddd dddd
0000320 eeee ccee cccc dddd dddd dddd 4a00 6256
0000330 0000 4a00 6256 0000 4a00 6256 0000 4a00
0000340 6256 0000 8888 8888 3333 2633 2626 2626
0000350 2626 0026 3333 2633 2626 2626 2626 0026
0000360 3333 2633 2626 2626 2626 0026 3333 2633
0000370 2626 2626 2626 0026 0000 0000 f0f0 00cc
0000380 0000 0000 0000 aaaa bbaa bbbb cccc cccc
0000390 cccc bbbb bbbb bbbb bbbb aabb aaaa bbbb
00003a0 aabb aaaa 9999 aa99 aaaa bbbb ccbb cccc
00003b0 0000 0000 0000 aaaa 00aa 0000 bbbb bbbb
00003c0 bbbb aaaa 99aa 9999 aaaa aaaa aaaa aaaa
00003d0 aaaa aaaa aaaa aaaa aaaa bbbb bbbb bbbb
00003e0 0000 0000 0000 0000 0000 0000 9999 9999
00003f0 9999 aaaa aaaa aaaa 9999 aa99 aaaa aaaa
0000400 aaaa aaaa aaaa aaaa aaaa bbbb bbbb bbbb
0000410 0000 0000 0000 eeee ffee ffff cccc cccc
0000420 cccc dddd bbdd bbbb cccc bbcc bbbb bbbb
0000430 bbbb bbbb bbbb bbbb bbbb cccc ddcc dddd
0000440 5100 0062 6e68 5100 0062 6e68 5100 0062
0000450 6e68 5100 0062 6e68 5100 0062 6e68 5100
0000460 0062 6e68 5100 0062 6e68 5100 0062 6e68
0000470 5100 0062 6e68 5100 0062 6e68 5100 0062
0000480 6e68 5100 0062 6e68 5100 0062 6e68 5100
0000490 0062 6e68 5100 0062 6e68 5100 0062 6e68
00004a0 5100 0062 6e68 5100 0062 6e68 5100 0062
00004b0 6e68 5100 0062 6e68 5100 0062 6e68 5100
00004c0 0062 6e68 5100 0062 6e68 5100 0062 6e68
00004d0 5100 0062 6e68 5100 0062 6e68 5100 0062
00004e0 6e68 5100 0062 6e68 5100 0062 6e68 5100
00004f0 0062 6e68 5100 0062 6e68 5100 0062 6e68
0000500 8888 8888 8888 8888 8888 8888 8888 8888
*
0000520 0000 0000 0000 0000 0000 0000 0000 0000
*
0000650 0000 0100 0006 0001 0501 0200 ff00 0003
0000660 00f9 03fe 0000 0000 0000 0000 0000 9b00
0000670 376e 0000 0000 0000 fc00 0900 000a 00fe
0000680 0706 fe00 0700 0008 0005 0a0b 0000 0000
0000690 0000 0000 0000 e200 4696 0000 0000 0000
00006a0 0400 f700 00f8 0003 fefc 0300 fb00 00fc
00006b0 00ee f2e3 0000 0000 0000 0000 0000 1100
00006c0 55bb 0000 0000 0000 0600 f600 00fc 0003
00006d0 fefb 0400 fa00 00fe 00f6 f2ec 0000 0000
00006e0 0000 0000 0000 1f00 58bf 0000 0000 0000
00006f0 0600 f500 00f6 0004 f9f8 0400 f700 00f8
0000700 00f7 f4f0 0000 0000 0000 0000 0000 2400
0000710 57be 0000 0000 0000 0800 f800 00fe 0006
0000720 fdf8 0700 f900 00fe 00f5 f4f0 0000 0000
0000730 0000 0000 0000 2d00 61d6 0000 0000 0000
0000740 0400 f700 00fc 0005 fcf7 0500 f700 00fc
0000750 00f9 f8f5 0000 0000 0000 0000 0000 2600
0000760 6ed9 0000 0000 0000 0400 f700 00f9 0006
0000770 f7f5 0500 f500 00f8 00f9 f7f4 0000 0000
0000780 0000 0000 0000 1b00 69ce 0000 0000 0000
0000790 0300 f800 00f8 0006 f6f6 0400 f600 00f7
00007a0 00f9 f7f4 0000 0000 0000 0000 0000 1800
00007b0 72d8 0000 0000 0000 0000 0000 4002 0240
00007c0 0000 0000 0000 0000 0000 0000 0000 0000
*
0000900 c2c1 c2c1 c341 c341 c13f c13f c140 c23f
0000910 c23f c140 c041 c040 c23f c23f c140 c23f
0000920 c13f c040 c041 c040 c33f c33f c240 c23f
0000930 c23f c140 c041 c040 c23f c23f c240 c23f
0000940 c13f c140 c041 c040 0000 0000 0000 0000
0000950 c741 c741 c7c1 c7c1 0000 0000 0000 0000
0000960 c03f c03f c03f c03f c03f c03f c03f c03f
*
00009a0 a000 00ce 0000 0000 84b6 0000 0000 0000
00009b0 0000 0000 0000 0000 1818 1818 1818 1818
00009c0 0000 0000 0000 0000 0000 0000 0000 0000
*
0000a00 4b00 6357 0000 4b00 6357 0000 4b00 6357
0000a10 0000 4b00 6357 0000 8888 8888 4708 5947
0000a20 7869 4908 5949 7a6d 4908 5a49 796d 4808
0000a30 5948 786c 4808 5848 786a 4808 5848 786a
0000a40 4808 5848 786c 4a08 5b4a 796d 4708 5947
0000a50 7a69 4808 5948 796b 4808 5948 7a6c 4808
0000a60 5848 796c 4808 5748 7768 4808 5748 7768
0000a70 4808 5748 776a 4a08 5a4a 776a 4608 5946
0000a80 7969 4808 5848 796b 4808 5848 7a6c 4808
0000a90 5948 796c 4808 5748 7768 4808 5748 7768
0000aa0 4908 5849 7a6d 4b08 5c4b 776c 4708 5a47
0000ab0 7b6a 4908 5a49 7c6e 4908 5a49 7c6e 4908
0000ac0 5b49 7c6e 4908 5949 7a6a 4908 5949 7a6a
0000ad0 4a08 5a4a 7d6f 4b08 5c4b 7b6e 0000 0000
0000ae0 0000 0000 0000 0000 0000 0000 0000 0000
0000af0 0000 0000 0000 0000 0000 0000 8485 8484
0000b00 c4c3 c5c4 c3c4 3fc3 c3c3 c2c2 c2c2 3fc0
0000b10 c3c3 c4c3 c4c4 3fc3 c2c2 c2c2 c3c1 c1c1
0000b20 c0c0 8282 8483 8686 8888 0000 0000 0000
0000b30 0000 0000 0000 0000 0000 0000 0000 0000
*
0000d70 0000 0000 0000 0000 0000 1111 0000 0000
0000d80 8080 03f7 8010 8080 0580 130d 0000 0000
0000d90 0000 0000 0000 0000 0000 0000 0000 0000
*
0000e20 0000 0000 0000 0000 0000 0000 0000 a400
0000e30 00ce 0000 0000 84b6 0000 0000 0000 0000
0000e40 0000 0000 0000 0000 0000 0000 0000 0000
*
0001000
root@OpenWrt:/tmp#
```

eckinga, 2022-09-29:
it is the same as https://github.com/openwrt/openwrt/blob/master/target/linux/mediatek/dts/mt7986a-bananapi-bpi-r3.dts#L424
i diffed it.

eckinga, 2022-09-30:
```
[   29.264190] mt7986-wmac 18000000.wbsys: mcu clock not found
[   29.269766] mt7986-wmac 18000000.wbsys: ap2conn clock not found
[   29.657736] mt7986-wmac 18000000.wbsys: HW/SW Version: 0x8a108a10, Build Time: 20220113162701a
[   29.657736] 
[   30.018715] mt7986-wmac 18000000.wbsys: WM Firmware Version: ____000000, Build Time: 20220113162756
[   30.185391] mt7986-wmac 18000000.wbsys: WA Firmware Version: DEV_000000, Build Time: 20220113163034
[   52.247137] mt7986-wmac 18000000.wbsys: Message 00000002 (seq 4) timeout
[   52.253836] mt7986-wmac 18000000.wbsys: Fw is status(0)
[   52.259050] mt7986-wmac 18000000.wbsys: Failed to start WA firmware
[   52.265557] mt7986-wmac: probe of 18000000.wbsys failed with error -110

eckinga, 2022-10-02:
Hello ppl,

I have to tell you now, i noticed it 5min ago. I tested a bit beside the work and suddenly the wireless tab in the config appeared.
I don't know why, perhaps i disconnected the antennas cause wifi wasn't working... i reconnected it right now and it seems to work (just tested scanning). The only thing i noticed when i wanted to save the log from serial but it isn't really working.
There is something funky with that board...
The log locks like (I'll make a second post with log from menu):

```F0: 102B 0000

FA: 1040 0000

FA: 1040 0000 [0200]

F9: 103F 0000

F3: 1001 0000 [0200]

F3: 1001 0000

F6: 300C 0028

F5: 0000 0000

V0: 0000 0000 [0001]

00: 0000 0000

BP: 2400 0041 [0000]

G0: 1190 0000

EC: 0000 0000 [3000]

T0: 0000 01B1 [010F]

Jump to BL

NOTICE:  BL2: v2.7(release):OpenWrt v2022-08-31-75393484-1 (mt7986-sdmmc-ddr4)
NOTICE:  BL2: Built : 05:53:56, Sep 29 2022
NOTICE:  WDT: disabled
NOTICE:  CPU: MT7986 (2000MHz)
NOTICE:  EMI: Using DDR4 settings
NOTICE:  EMI: Detected DRAM size: 2048MB
NOTICE:  EMI: complex R/W mem test passed
NOTICE:  BL2: Booting BL31
NOTICE:  BL31: v2.7(release):OpenWrt v2022-08-31-75393484-1 (mt7986-sdmmc-ddr4)
NOTICE:  BL31: Built : 05:53:56, Sep 29 2022

U-Boot 2022.07-OpenWrt-r20784-ec8fb542ec (Sep 29 2022 - 05:53:56 +0000)

CPU:   MediaTek MT7986
Model: BananaPi BPi-R3
DRAM:  2 GiB
Core:  52 devices, 25 uclasses, devicetree: embed
jedec_spi_nor spi_nor@0: unrecognized JEDEC id bytes: ff, ef, aa
MMC:   mmc@11230000: 0
Loading Environment from MMC... OK
In:    serial@11002000
Out:   serial@11002000
Err:   serial@11002000
reset button found
Loading Environment from MMC... OK
Net:   eth0: ethernet@15100000
7[r[999;999H[6n8Cannot read EFI system partition
[?25l[2J[1;1H[1;1H[2K[2;3H      [0;34m( ( ( [1;39mOpenWrt[0;34m ) ) )  [0;36m[SD card][0m       [33mU-Boot 2022.07-OpenWrt-r20784-ec8fb542ec (Sep 29 2022 - 05:53:56 +0000)[0m[0K[3;1H[2K[16;1H[2K[17;3HPress UP/DOWN to move, ENTER to select, ESC/CTRL+C to quit[0K[18;1H[2K[4;7H[7m1. Run default boot command.[0m[5;7H2. Boot system via TFTP.[6;7H3. Boot production system from SD card.[7;7H4. Boot recovery system from SD card.[8;7H5. Load production system via TFTP then write to SD card.[9;7H6. Load recovery system via TFTP then write to SD card.[10;7H7. [31mInstall bootloader, recovery and production to NOR.[0m[11;7H8. [31mInstall bootloader, recovery and production to NAND.[0m[12;7H9. Reboot.[13;7Ha. Reset all settings to factory defaults.[14;7H0. U-Boot console[16;3HHit any key to stop autoboot: 3 [16;3HHit any key to stop autoboot: 2 [16;3HHit any key to stop autoboot: 1 [16;1H[2K[?25h[2J[1;1Hspi-nand: spi_nand spi_nand@1: Winbond SPI NAND was found.
spi-nand: spi_nand spi_nand@1: 128 MiB, block size: 128 KiB, page size: 2048, OOB size: 64
jedec_spi_nor spi_nor@0: unrecognized JEDEC id bytes: 00, ef, aa
'spi-nand0' is now active device
* spi-nand0
  - device: spi_nand@1
  - parent: spi@1100a000
  - driver: spi_nand
  - type: NAND flash
  
^[[24;80R

eckinga, 2022-10-02:
The log from menu:

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.69 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r20784-ec8fb542ec) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP Thu Sep 29 05:53:56 2022
[    0.000000] Machine model: Bananapi BPI-R3
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x000000004fbfffff]
[    0.000000]   node   0: [mem 0x000000004fc00000-0x000000004fcfffff]
[    0.000000]   node   0: [mem 0x000000004fd00000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 17 pages/cpu s30040 r8192 d31400 u69632
[    0.000000] pcpu-alloc: s30040 r8192 d31400 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516096
[    0.000000] Kernel command line: root=/dev/mmcblk0p65
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 2043176K/2097152K available (8320K kernel code, 894K rwdata, 2236K rodata, 448K init, 305K bss, 53976K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 640 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x000000000c080000
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[    0.000000] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns
[    0.000113] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=130000)
[    0.000120] pid_max: default: 32768 minimum: 301
[    0.000207] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.000220] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.001065] rcu: Hierarchical SRCU implementation.
[    0.001153] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.001324] smp: Bringing up secondary CPUs ...
[    0.001556] Detected VIPT I-cache on CPU1
[    0.001577] GICv3: CPU1: found redistributor 1 region 0:0x000000000c0a0000
[    0.001605] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.001855] Detected VIPT I-cache on CPU2
[    0.001867] GICv3: CPU2: found redistributor 2 region 0:0x000000000c0c0000
[    0.001878] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.002103] Detected VIPT I-cache on CPU3
[    0.002113] GICv3: CPU3: found redistributor 3 region 0:0x000000000c0e0000
[    0.002122] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.002150] smp: Brought up 1 node, 4 CPUs
[    0.002163] SMP: Total of 4 processors activated.
[    0.002166] CPU features: detected: 32-bit EL0 Support
[    0.002169] CPU features: detected: CRC32 instructions
[    0.002243] CPU: All CPU(s) started at EL2
[... 229 lines cut ...]
[   15.783739] device eth1 entered promiscuous mode
[   15.789706] br-wan: port 1(eth1) entered blocking state
[   15.794933] br-wan: port 1(eth1) entered forwarding state
[   15.802683] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[   15.812274] br-wan: port 2(wan) entered blocking state
[   15.813978] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[   15.817414] br-wan: port 2(wan) entered disabled state
[   15.818865] device wan entered promiscuous mode
[   15.837210] br-wan: port 2(wan) entered blocking state
[   15.842345] br-wan: port 2(wan) entered forwarding state
[   16.445902] br-wan: port 1(eth1) entered disabled state
[   18.171702] mt7530 mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control off
[   18.179047] br-lan: port 1(lan1) entered blocking state
[... cut ...]
[  353.520681] device wlan0 entered promiscuous mode
[  353.525473] br-lan: port 6(wlan0) entered blocking state
[  353.530781] br-lan: port 6(wlan0) entered forwarding state
[  353.536954] br-lan: port 6(wlan0) entered disabled state
[  353.545825] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  353.552310] br-lan: port 6(wlan0) entered blocking state
[  353.557614] br-lan: port 6(wlan0) entered forwarding state
[  366.435512] device wlan0 left promiscuous mode
[  366.440014] br-lan: port 6(wlan0) entered disabled state

eckinga, 2022-10-02:
Power cycle and away. Then removed the antennas and reattached like yesterday nothing. After i plugged out debug console and did a power cycle it is back.
@reddwarf69 Do you have a serial adapter plugged for debug console plugged in? Remove it and make a power cycle. A try.

frank-w, 2022-10-02:
you removed debug-console and then wifi works?? have you connected gnd/rx/tx to the board?

eckinga, 2022-10-02:
Hello frank,

just tested it again and yeah i plug out debug console, make a power cycle and there it is. Wifi working.
And yes gnd/rx/tx are plugged have to!?

frank-w, 2022-10-02:
i guess your adapter brings gnd to your board and shortens anything needed for wifi

to be precise all 3 have to be connected, but something there is strange...maybe try another adapter? is your host-system and the r3 connected to same power-source?

afaik it is UART0 yes

eckinga, 2022-10-02:
sure all 3, you need an reference... ;)
Or wifi shortens something for serial xD. I have power cycles where wifi is going but the serial console not.
And its the "same" plug. And i tested another adapter. Before it was a waveshare PL2303 and now its an FTDI dongle. same problems.

Here is another log, its the same position debug console stops working... and then wifi starts....
Why does it stop at this position?

```
root@OpenWrt:/#
F0: 102B 0000
FA: 1040 0000
FA: 1040 0000 [0200]
F9: 103F 0000
F3: 1006 0033 [0200]
F3: 4001 00E0 [0200]
F3: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 2400 0041 [0000]
G0: 1190 0000
EC: 0000 0000 [2000]
T0: 0000 020A [010F]
Jump to BL

NOTICE:  BL2: v2.7(release):OpenWrt v2022-08-31-75393484-1 (mt7986-emmc-ddr4)
NOTICE:  BL2: Built : 05:53:56, Sep 29 2022
NOTICE:  WDT: disabled
NOTICE:  CPU: MT7986 (1998MHz)
NOTICE:  EMI: Using DDR4 settings
NOTICE:  EMI: Detected DRAM size: 2048MB
NOTICE:  EMI: complex R/W mem test passed
NOTICE:  BL2: Booting BL31
NOTICE:  BL31: v2.7(release):OpenWrt v2022-08-31-75393484-1 (mt7986-emmc-ddr4)
NOTICE:  BL31: Built : 05:53:56, Sep 29 2022

U-Boot 2022.07-OpenWrt-r20784-ec8fb542ec (Sep 29 2022 - 05:53:56 +0000)

CPU:   MediaTek MT7986
Model: BananaPi BPi-R3
DRAM:  2 GiB
Core:  53 devices, 25 uclasses, devicetree: embed
jedec_spi_nor spi_nor@0: unrecognized JEDEC id bytes: ff, ef, aa
MMC:   mmc@11230000: 0
Loading Environment from MMC... OK
In:    serial@11002000
Out:   serial@11002000
Err:   serial@11002000
reset button found
Loading Environment from MMC... OK
Net:   eth0: ethernet@15100000
Cannot read EFI system partition

        ( ( ( OpenWrt ) ) )  [eMMC]       U-Boot 2022.07-OpenWrt-r20784-ec8fb542                                                                                                                                                             
      1. Run default boot command.
      2. Boot system via TFTP.
      3. Boot production system from eMMC.
      4. Boot recovery system from eMMC.
      5. Load production system via TFTP then write to eMMC.
      6. Load recovery system via TFTP then write to eMMC.
      7. Load BL31+U-Boot FIP via TFTP then write to eMMC.
      8. Load BL2 preloader via TFTP then write to eMMC.
      9. Reboot.
      a. Reset all settings to factory defaults.
      0. U-Boot console

  Press UP/DOWN to move, ENTER to select, ESC/CTRL+C to quit
spi-nand: spi_nand spi_nand@1: Winbond SPI NAND was found.
spi-nand: spi_nand spi_nand@1: 128 MiB, block size: 128 KiB, page size: 2048, OO                                                                                                                                                             B size: 64
jedec_spi_nor spi_nor@0: unrecognized JEDEC id bytes: 00, ef, aa
'spi-nand0' is now active device
* spi-nand0
  - device: spi_nand@1
  - parent: spi@1100a000
  - driver: spi_nand
  - type: NAND flash
  - block size:        0x20000 bytes
  - page size:         0x800 bytes
  - OOB size:          64 bytes
  - OOB available:     24 bytes
  - 0x000000000000-0x000008000000 : "spi-nand0"
          - 0x000000000000-0x000000080000 : "bl2"
          - 0x000000080000-0x000000380000 : "factory"
          - 0x000000380000-0x000000580000 : "fip"
          - 0x000000580000-0x000008000000 : "ubi"
* spi-nand0
  - device: spi_nand@1
  - parent: spi@1100a000
  - driver: spi_nand
  - type: NAND flash
  - block size:        0x20000 bytes
  - page size:         0x800 bytes
  - OOB size:          64 bytes
  - OOB available:     24 bytes
  - 0x000000000000-0x000008000000 : "spi-nand0"
          - 0x000000000000-0x000000080000 : "bl2"
          - 0x000000080000-0x000000380000 : "factory"
          - 0x000000380000-0x000000580000 : "fip"
          - 0x000000580000-0x000008000000 : "ubi"

MMC read: dev # 0, block # 131072, count 8 ... 8 blocks read: OK

MMC read: dev # 0, block # 131072, count 21712 ... 21712 blocks read: OK
## Loading kernel from FIT Image at 46000000 ...
   Using 'config-mt7986a-bananapi-bpi-r3-emmc-snand' configuration
   Trying 'kernel-1' kernel subimage
     Description:  ARM64 OpenWrt Linux-5.15.69
     Type:         Kernel Image
     Compression:  gzip compressed
     Data Start:   0x46001000
     Data Size:    5403649 Bytes = 5.2 MiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x44000000
     Entry Point:  0x44000000
     Hash algo:    crc32
     Hash value:   de683342
     Hash algo:    sha1
     Hash value:   fe98e48287d1851220edeaeeacef1393dfcca60a
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 46000000 ...
   Using 'config-mt7986a-bananapi-bpi-r3-emmc-snand' configuration
   Trying 'fdt-1' fdt subimage
     Description:  ARM64 OpenWrt bananapi_bpi-r3 device tree blob
     Type:         Flat Device
                              ^[[24;80

reddwarf69, 2022-10-03:
I can reproduce what @eckinga says with
```
067b:2303 Prolific Technology, Inc. PL2303 Serial Port / Mobile Action MA-8910P
10c4:ea60 Silicon Labs CP210x UART Bridge
```
With both the mt76 driver and the proprietary Mediatek driver. So... yeah, not a mt76 issue.

frank-w, 2022-10-03:
i do not have this issue with cp210x

```
$ udevadm info /dev/ttyUSB1
P: /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/ttyUSB1/tty/ttyUSB1
N: ttyUSB1
L: 0
S: serial/by-path/pci-0000:00:14.0-usb-0:3.2:1.0-port0
S: serial/by-id/usb-Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_010CB6FA-if00-port0
S: tty_r3
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/ttyUSB1/tty/ttyUSB1
E: DEVNAME=/dev/ttyUSB1
E: MAJOR=188
E: MINOR=1
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=4844533
E: ID_BUS=usb
E: ID_VENDOR_ID=10c4
E: ID_MODEL_ID=ea60
```
no stop of serial output and wifi is recognized

```
root@bpi-r3:~# modprobe mt7915e                                                 
[   68.949540] mt7986-wmac 18000000.wifi: HW/SW Version: 0x8a108a10, Build Timea
[   68.949540]                                                                  
[   69.103153] mt7986-wmac 18000000.wifi: WM Firmware Version: ____000000, Buil6
[   69.173764] mt7986-wmac 18000000.wifi: WA Firmware Version: DEV_000000, Buil4
root@bpi-r3:~# ip a s wlan0                                                     
12: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default q0
    link/ether 00:0c:43:26:60:00 brd ff:ff:ff:ff:ff:ff                          
root@bpi-r3:~# ip a s wlan1                                                     
13: wlan1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default q0
    link/ether 82:0c:43:26:60:00 brd ff:ff:ff:ff:ff:ff                          
root@bpi-r3:~# 
```
using linux 6.0.0-rc6...so there is something strange with your boards

reddwarf69, 2022-10-03:
@frank-w is your board a v1.1?

frank-w, 2022-10-03:
> @frank-w is your board a v1.1?

no, v1.0..so maybe something has changed here between those versions

eckinga, 2022-10-03:
Would be nice to see full schematic and pcb layout for both v1.0 and v1.1 to see what was changed.
As the general hardware is the same!?(dont know v1.0) there must be bug on layout.
I think we wont get the details here so will have to wait for sinovoip.

frank-w, 2022-10-03:
Have you reported it to sinovoip? I see no (public) Thread in forum.

Changes i know for v1.1 are adding hole for 2230 nvme, moving bootswitches together, add common usb2 and a sata power connector.

Reset bug schold be still there in 1.1 (when card in m2 slot)

eckinga, 2022-10-03:
yes i reported to them, due national day in china i guess respond will take a while...
Reset bug?

reddwarf69, 2022-10-03:
> Reset bug?

https://forum.banana-pi.org/t/bpi-r3-information/12897/42

hayzamjs, 2022-10-09:
I tried the snapshot from today (flashed on an SD card) and it seems to work well, just a note when I had the serial connected it wasn't working, not sure if it has anything to do with it.

reddwarf69, 2022-10-13:
> i do not have this issue with cp210x
>
> ```
> $ udevadm info /dev/ttyUSB1
> P: /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/ttyUSB1/tty/ttyUSB1
> N: ttyUSB1
> L: 0
> S: [...]

Bpi says "When powered on, if the uart tx is pulled high, it will cause this". I see you did a "modprobe mt7915e", way after powering it on. It doesn't happen to you, with the v1.0, even during boot?

frank-w, 2022-10-15:
simon says there are no differences regarding this between v1.0 and v1.1 and only PL2303 is affected, shematics look same too

added mt7915e module to /etc/modules-load.d/wifi.conf and see no issues with my cp210x. no broken serial output and wifi is initialized without problem (using my 6.1-rc tree, but basicly same patches as daniel used for openwrt)

```
root@bpi-r3:~# dmesg | grep -i mt7986-wmac                                      
[    6.762113] mt7986-wmac 18000000.wifi: HW/SW Version: 0x8a108a10, Build Time:
 20220113162701a                                                                
[    6.922936] mt7986-wmac 18000000.wifi: WM Firmware Version: ____000000, Build
 Time: 20220113162756                                                           
[    6.992604] mt7986-wmac 18000000.wifi: WA Firmware Version: DEV_000000, Build
 Time: 20220113163034
```

cp21x2, ch340 and ftdi should be not affected based on simons answer

Owersun, 2022-11-06:
I can confirm as well.
Connected console - wifi fails to load firmware, disconnected console - work just fine.
I have some cheap serial dongle, and completely build from scratch mainline u-boot/linux 6.1 with .dtb's of what I need.
I'd spend days trying to fix wifi if not that thread, but all in all it's that simple...

lorenz, 2022-11-12:
FYI from a quick look at the schematic I think that the problem is that UART0_TXD is also a strap pin. When attaching a serial adapter it will have its own pull (up) resistor, countering the built-in pull-down strap (RB8). This causes the second "A-Die" which I think are the RF frontend ICs to switch its crystal mode selection from crystal mode to "buffer mode". Most likely this causes the wifi firmware to fail to power up the second frontend.

The good thing is that this is not a fatal problem, it can be fixed by either decreasing RB6 to make for a stronger pull-down (probably something like 820Ohms -> ~4mA) or getting a serial adapter with a much higher pull-up (like 47kOhms). This stops the serial adapter's pull-up from overpowering the strapping resistor.

zamboni33, 2022-12-25:
I understand the UART problem I am reading about here but for my case I am not using a UART device. I boot the device with a clean openwrt build and wifi is not present. When I boot the device with the sdcard image found within the bpi-r3 forums I see wlan devices. Is there something within the build configuration I am missing to activate the wireless interfaces?

frank-w, 2023-02-04:
have now same error on new system (debian bullseye) with newest firmware-files and kernel 6.1.0

```
[Sat Feb  4 14:49:41 2023] mt7986-wmac 18000000.wifi: Message 00000002 (seq 9) timeout
[Sat Feb  4 14:49:41 2023] mt7986-wmac 18000000.wifi: Failed to start WA firmware
[Sat Feb  4 14:49:41 2023] mt7986-wmac: probe of 18000000.wifi failed with error -110
```

also if not connected uart (cp2102)...maybe it is firmware-issue?

lorenz, 2023-02-05:
Linux 6.2-rc6 works for me with a FT2232H serial console attached.

frank-w, 2023-02-05:
thanks for report...i tried the usb2uart i used on v1.0 and this works here...very strange...both are cp21xx...i try to find a difference. not-working one is a very small adapter with micro-usb jack, working one is larger with usb-A plug. need to find out, how i can get more information

this one is working:

```
$ udevadm info /dev/tty_r3
P: /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/ttyUSB1/tty/ttyUSB1
N: ttyUSB1
L: 0
S: serial/by-path/pci-0000:00:14.0-usb-0:3.2:1.0-port0
S: tty_r3
S: serial/by-id/usb-Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_010CB6FA-if00-port0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/ttyUSB1/tty/ttyUSB1
E: DEVNAME=/dev/ttyUSB1
E: MAJOR=188
E: MINOR=1
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=3923699
E: ID_BUS=usb
E: ID_VENDOR_ID=10c4
E: ID_MODEL_ID=ea60
E: ID_PCI_CLASS_FROM_DATABASE=Serial bus controller
E: ID_PCI_SUBCLASS_FROM_DATABASE=USB controller
E: ID_PCI_INTERFACE_FROM_DATABASE=XHCI
E: ID_VENDOR_FROM_DATABASE=Silicon Labs
E: ID_MODEL_FROM_DATABASE=CP210x UART Bridge
E: ID_VENDOR=Silicon_Labs
E: ID_VENDOR_ENC=Silicon\x20Labs
E: ID_MODEL=CP2104_USB_to_UART_Bridge_Controller
E: ID_MODEL_ENC=CP2104\x20USB\x20to\x20UART\x20Bridge\x20Controller
E: ID_REVISION=0100
E: ID_SERIAL=Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_010CB6FA
E: ID_SERIAL_SHORT=010CB6FA
E: ID_TYPE=generic
E: ID_USB_INTERFACES=:ff0000:
E: ID_USB_INTERFACE_NUM=00
E: ID_USB_DRIVER=cp210x
E: ID_PATH=pci-0000:00:14.0-usb-0:3.2:1.0
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_3_2_1_0
E: ID_MM_CANDIDATE=1
E: DEVLINKS=/dev/serial/by-path/pci-0000:00:14.0-usb-0:3.2:1.0-port0 /dev/tty_r3 /dev/serial/by-id/usb-Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_010CB6FA-if00-port0
E: TAGS=:systemd:snap_cups_cupsd:snap_cups_ippeveprinter:
E: CURRENT_TAGS=:systemd:snap_cups_cupsd:snap_cups_ippeveprinter:
```
this is the small one not working:

```
$ udevadm info /dev/ttyUSB2
P: /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/ttyUSB2/tty/ttyUSB2
N: ttyUSB2
L: 0
S: tty_r2
S: serial/by-path/pci-0000:00:14.0-usb-0:1:1.0-port0
S: serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/ttyUSB2/tty/ttyUSB2
E: DEVNAME=/dev/ttyUSB2
E: MAJOR=188
E: MINOR=2
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=709868665
E: ID_BUS=usb
E: ID_VENDOR_ID=10c4
E: ID_MODEL_ID=ea60
E: ID_PCI_CLASS_FROM_DATABASE=Serial bus controller
E: ID_PCI_SUBCLASS_FROM_DATABASE=USB controller
E: ID_PCI_INTERFACE_FROM_DATABASE=XHCI
E: ID_VENDOR_FROM_DATABASE=Silicon Labs
E: ID_MODEL_FROM_DATABASE=CP210x UART Bridge
E: ID_VENDOR=Silicon_Labs
E: ID_VENDOR_ENC=Silicon\x20Labs
E: ID_MODEL=CP2102_USB_to_UART_Bridge_Controller
E: ID_MODEL_ENC=CP2102\x20USB\x20to\x20UART\x20Bridge\x20Controller
E: ID_REVISION=0100
E: ID_SERIAL=Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001
E: ID_SERIAL_SHORT=0001
E: ID_TYPE=generic
E: ID_USB_INTERFACES=:ff0000:
E: ID_USB_INTERFACE_NUM=00
E: ID_USB_DRIVER=cp210x
E: ID_PATH=pci-0000:00:14.0-usb-0:1:1.0
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_1_1_0
E: ID_MM_CANDIDATE=1
E: DEVLINKS=/dev/tty_r2 /dev/serial/by-path/pci-0000:00:14.0-usb-0:1:1.0-port0 /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
E: TAGS=:snap_cups_cupsd:snap_cups_ippeveprinter:systemd:
E: CURRENT_TAGS=:snap_cups_cupsd:snap_cups_ippeveprinter:systemd:
```
so usb-id is same, but first (working) seems to be a cp2104 and second cp2102

lorenz, 2023-02-05:
If this is caused by the serial adapter fighting with the strapping it might not even be down to the USB-Serial chip, but any external pull-ups/pull-downs/buffers on the USB-Serial board.

OT but could you have a quick look at https://lore.kernel.org/linux-mediatek/SN1KPR.7R1MV34Z4HRM3@brun.one/T/#u? Have you encountered anything like it?

frank-w, 2023-02-12:
ch340G seems not working too:

```
$ udevadm info /dev/ttyUSB1
P: /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.0/ttyUSB1/tty/ttyUSB1
N: ttyUSB1
L: 0
S: tty_r2pro
S: serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0
S: serial/by-path/pci-0000:00:14.0-usb-0:3.1:1.0-port0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.0/ttyUSB1/tty/ttyUSB1
E: DEVNAME=/dev/ttyUSB1
E: MAJOR=188
E: MINOR=1
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=130857833
E: ID_BUS=usb
E: ID_VENDOR_ID=1a86
E: ID_MODEL_ID=7523
E: ID_PCI_CLASS_FROM_DATABASE=Serial bus controller
E: ID_PCI_SUBCLASS_FROM_DATABASE=USB controller
E: ID_PCI_INTERFACE_FROM_DATABASE=XHCI
E: ID_VENDOR_FROM_DATABASE=QinHeng Electronics
E: ID_MODEL_FROM_DATABASE=CH340 serial converter
E: ID_VENDOR=1a86
E: ID_VENDOR_ENC=1a86
E: ID_MODEL=USB2.0-Ser_
E: ID_MODEL_ENC=USB2.0-Ser\x21
E: ID_REVISION=0254
E: ID_SERIAL=1a86_USB2.0-Ser_
E: ID_TYPE=generic
E: ID_USB_INTERFACES=:ff0102:
E: ID_USB_INTERFACE_NUM=00
E: ID_USB_DRIVER=ch341
E: ID_USB_CLASS_FROM_DATABASE=Vendor Specific Class
E: ID_PATH=pci-0000:00:14.0-usb-0:3.1:1.0
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_3_1_1_0
E: ID_MM_CANDIDATE=1
E: DEVLINKS=/dev/tty_r2pro /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0 /dev/serial/by-path/pci-0000:00:14.0-usb-0:3.1:1.0-port0
E: TAGS=:snap_cups_cupsd:systemd:snap_cups_ippeveprinter:
E: CURRENT_TAGS=:snap_cups_cupsd:systemd:snap_cups_ippeveprinter:
```

frank-w, 2023-02-13:
> The good thing is that this is not a fatal problem, it can be fixed by either decreasing RB6 to make for a stronger pull-down (probably something like 820Ohms -> ~4mA) or getting a serial adapter [...]

@lorenz you mean decreasing RB8,right? RB6 is not connected and a pullup to 3v3

https://forum.banana-pi.org/t/bpi-r3-hw-issues/14027/2

lorenz, 2023-02-13:
Yeah, I meant RB8.

G-Pereira, 2024-03-21:
Hey, I am having rhis exact issue with ASUS RT-AX59U and I am not using any serial adapter. My device has the waranty seal intact.
Any idea how to fix?

Jagarana, 2024-04-28:
> Someone opened a case in openwrt for this issue but i think its firmware related problem and better here.
>
> ```
> [   10.127699] mt7986-wmac 18000000.wmac: HW/SW Version: 0x8a108a10, Build Time: [...]

I came accross the same problem today. I could solve it by reinstalling the following packages trough luci:
- mt7986-wo-firmware
- kmod-mt7986-firmware

Matador-67, 2024-08-10:
I had same Problem with CH340 and Board I don't know but i think it must be ver 1.1

Constantin07, 2024-08-17:
I confirm as well CH340 with V1.1 has the same issues - wifi won't initialize:
```
[    9.963576] mt798x-wmac 18000000.wifi: HW/SW Version: 0x8a108a10, Build Time: 20221012174743a
[   10.089068] mt798x-wmac 18000000.wifi: WM Firmware Version: ____000000, Build Time: 20221012174805
[   10.166796] mt798x-wmac 18000000.wifi: WA Firmware Version: DEV_000000, Build Time: 20221012174937
[   30.550216] mt798x-wmac 18000000.wifi: Message 00000002 (seq 9) timeout
[   30.556827] mt798x-wmac 18000000.wifi: Failed to start WA firmware
[   30.563086] mt798x-wmac: probe of 18000000.wifi failed with error -110
```

kaczor6418, 2025-01-03:
I have figured out that I don't understand why, but after I have installed OpenWRT on my router, my Wi-Fi settings were disabled in `wireless` config file. Maybe the reason was my strange infinity loops after I have installed OpenWRT (more about this you can read ***[here](https://forum.banana-pi.org/t/bpi-r3-infinity-reboot-while-boting-from-emmc/19209/10)***).

Anyway, for me, the solution was to modify *`etc/config/wireless`*. You have to make sure **wifi-device** → **option disabled** is set to **`'0'`**. If this option is set to **`'1'`** that means Wi-Fi is disabled.

Below you can find my  *`etc/config/wireless`*
```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/18000000.wifi'
        option band '2g'
        option channel '1'
        option htmode 'HE20'
        option disabled '0' 

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/soc/18000000.wifi+1'
        option band '5g'
        option channel '36'
        option htmode 'HE80'
        option disabled '0' 

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
```

Wendigogo, 2025-01-24:
Sorry to pop-up this thread but I got this issue too on my BPI-R3 : it leads to a kernel panic with this output :

```
[   11.873111] mt798x-wmac 18000000.wifi: WM Firmware Version: ____000000, Build Time: 20240823160804
[   11.911324] mt798x-wmac 18000000.wifi: WA Firmware Version: DEV_000000, Build Time: 20240823160840
[   32.471852] mt798x-wmac 18000000.wifi: Retry message 00000002 (seq 11)
[   52.951844] mt798x-wmac 18000000.wifi: Message 00000002 (seq 11) timeout
[   52.958532] mt798x-wmac 18000000.wifi: Failed to start WA firmware
[   52.958548] Unable to handle kernel write to read-only memory at virtual address 0000000000000008
[   52.973544] Mem abort info:
[   52.976322]   ESR = 0x0000000096000045
[   52.980053]   EC = 0x25: DABT (current EL), IL = 32 bits
[   52.985349]   SET = 0, FnV = 0
[   52.988387]   EA = 0, S1PTW = 0
[   52.991512]   FSC = 0x05: level 1 translation fault
[   52.996374] Data abort info:
[   52.999238]   ISV = 0, ISS = 0x00000045, ISS2 = 0x00000000
[   53.004706]   CM = 0, WnR = 1, TnD = 0, TagAccess = 0
[   53.009738]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[   53.015034] user pgtable: 4k pages, 39-bit VAs, pgdp=00000000468e9000
[   53.021454] [0000000000000008] pgd=080000004025d003, p4d=080000004025d003, pud=080000004025d003, pmd=0000000000000000
[   53.032045] Internal error: Oops: 0000000096000045 [#1] SMP
[   53.037597] Modules linked in: mt7915e(O+) mt76_connac_lib(O) mt76(O) mac80211(O) cfg80211(O) slhc sfp nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcrc32c ca
[   53.077538] CPU: 3 PID: 44 Comm: kworker/u8:3 Tainted: G           O       6.6.73 #0
[   53.085259] Hardware name: Bananapi BPI-R3 (DT)
[   53.089771] Workqueue: mt76 0xffffffc078ec7d60 [mt7915e@00000000634fb3ff+0x1a000]
[   53.097235] pstate: a0400005 (NzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   53.104175] pc : 0xffffffc078ea1c30 [mt76@000000006a3ad4bf+0xd000]
[   53.110335] lr : 0xffffffc078ea21f8 [mt76@000000006a3ad4bf+0xd000]
[   53.116496] sp : ffffffc080f8bc00
[   53.119793] x29: ffffffc080f8bc00 x28: ffffffc080f8bc70 x27: ffffff8007599fe0
[   53.126910] x26: ffffffffffff5af8 x25: ffffff8007591ff8 x24: ffffff8007595678
[   53.134025] x23: ffffff8007592790 x22: ffffff8007591fe0 x21: ffffff8007592008
[   53.141141] x20: ffffff8007591fe0 x19: ffffff8007591fe0 x18: 00000000cc9851c1
[   53.148257] x17: ffffffc080f8bc40 x16: ffffffc080f8bbd0 x15: 000000a41c4f1640
[   53.155372] x14: 0000000000000000 x13: 0000000000000020 x12: 0101010101010101
[   53.162488] x11: 7f7f7f7f7f7f7f7f x10: fefefefefefefeff x9 : 7f7f7f7f7f7f7f7f
[   53.169603] x8 : 8080808080808080 x7 : ffffff8000986200 x6 : ffffffffffff2460
[   53.176718] x5 : ffffffc080f8bc68 x4 : 0000000000000000 x3 : ffffff8007592010
[   53.183834] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000000
[   53.190950] Call trace:
[   53.193382]  0xffffffc078ea1c30 [mt76@000000006a3ad4bf+0xd000]
[   53.199196]  0xffffffc078ea21f8 [mt76@000000006a3ad4bf+0xd000]
[   53.205010]  0xffffffc078ec897c [mt7915e@00000000634fb3ff+0x1a000]
[   53.211169]  0xffffffc0800570d8
[   53.214295]  0xffffffc080057534
[   53.217419]  0xffffffc08005ebe4
[   53.220544]  0xffffffc080013024
[   53.223671] Code: f94037e3 eb00007f 54000120 a94306c2 (f900045c)
[   53.229743] ---[ end trace 0000000000000000 ]---
[   53.237172] pstore: backend (ramoops) writing error (-28)
[   53.242553] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[   53.249404] SMP: stopping secondary CPUs
[   53.253312] Kernel Offset: disabled
[   53.256784] CPU features: 0x0,00000000,00000000,1000400b
[   53.262077] Memory Limit: none
[   53.267839] Rebooting in 1 seconds..
```
I'm booting on SDCard to install 24.10-RC6 (got the same issue with RC5 too).

But what bother me is that I have NO issue at all with the stable 23.05 version with the same board, the same power plug and the same UART device (`ID 1a86:7523 QinHeng Electronics CH340 serial converter`). WTH!

I mean, I just have to boot on EMMC with 23.05 installed and got no kernel panic in the exact same configuration.

So in my case **it seems to be a software related issue**. Or am I wrong?

frank-w, 2025-01-25:
Please use ftdi uart device...most others have the described error... Except cp2104 can be used too.

Wendigogo, 2025-01-25:
@frank-w : Without my serial converter plugged in everything went fine. 👍 

Thanks for your suggestion.

But it works flawlessly in 23.05 and generate Kernel Panic in 24.10 with the exact same hardware: how the given solution could be to buy a new uart device ?

Here is what happens when 23.05 boots up (the relevant part I guess):
```
[    9.043532] mt798x-wmac 18000000.wifi: Direct firmware load for mediatek/mt7986_rom_patch_mt7975.bin failed with error -2
[    9.054501] mt798x-wmac 18000000.wifi: Falling back to sysfs fallback for: mediatek/mt7986_rom_patch_mt7975.bin
[    9.079947] mt798x-wmac: probe of 18000000.wifi failed with error -12
```
mt798x-wmac failed to load too but silently. Maybe there's an issue with it's "WM or WA" Firmware?
Is it embedded in kernel or is it a separated package?

frank-w, 2025-01-25:
it is hardware-issue see this: https://github.com/openwrt/mt76/issues/702#issuecomment-1312284560

Wendigogo, 2025-01-26:
That's the point I don't understand.

If I read it correctly (what I'm not sure about) my UART device causes the Wi-Fi firmware to fail to power up the second frontend in both OpenWRT version: here is the hardware issue right?

But in one case, it fails silently and in other case it generates a Kernel Panic bootloop. Imo it's a software issue here.

Finally, this Bug Report is not really concerned by this Kernel Panic. I'm coming from bug [#16924](https://github.com/openwrt/openwrt/issues/16924) and I don't think they address the same issue. Maybe it's there I should ask for a "solution" (ie no Wi-Fi in a running system).

Thanks for your patience.

Owersun, 2025-01-26:
Let me explain.

The issue this topic was started with after few messages of discussion was specifically and undobtfuly identified to be hardware issue. The hardware issue of the type that when a device of some physical properties is connected to UART pins due to it's electric properties related to board schematic affect the wireless chip the way, it gets its brain, lets call it "scrambled".
The issue was clearly identified to be present in A/B testing it, so there is no doubt that the issue this bug is opened about is hardware.
Now there are two things to consider relative to your question.
First would be, that if you have device that is mentioned as affected in this topic, the final result how it fails on different software is irrelevant. It's just the matter of hardware/software relation - when physical properties of the hardware exceed some adequate limits, software can behave whatever way. It can throw kernel panics, hand silently or do whatever. Difference in the outcome doesn't make it software problem in any way. It is just the way it is, if the hardware is faulty (which is the case and reason for this filed bug, I must remind), when the outcome is undoubtfuly relative to the used hardware, how the software fails doesn't matter, just change the hardware.
Second would be, that if you are saying that in your circumstances you have case A when your hardware _does_ work and case B when it doesn't, and you consider this to be software issue - then it is most certainly NOT the issue of this topic, and you probably have to file a different issue with different description, stating there that there is software A that works and software B that doesn't.

This is what most people are trying to tell you. In short - it is either irrelevant how your software fails, if the hardware is faulty, or your issue is different that what is described in this topic and you will not get any meaningful advice writing in it, as it will be irrelevant.

hboterman, 2025-01-26:
I'm using CP2102 and it doesn't work, I have no wifi initialized (I even have no "wifi" line in dmesg).
I tried to force Tx to GND at bootup, let R3 boot up completly and... no success, still no wifi initialized !!
R3 were working with OpenWrt 23... Electrical parameters can't have changed suddenly !? I don't understand the problem with pull up/pull down resistors @eckinga !?

eckinga, 2025-01-26:
@hboterman Why are you shorting Tx? So like @Owersun mentioned before it's a hardware failure and i think this thread make it clear how to handle with it. It's wrong here but still is here that others running in this problem can solve it.

Try another uart usb, i didn't read all here but i don't think shorten tx is a solution.

And don't mention me, this thread is clear for solution.

closed 2025-01-26 by eckinga

hboterman, 2025-01-26:
OK, seems pullup pulldown tips is true with OpenWRT23. If I boot without serial USB adpater, wifi radios are presents and initialized.  They're not if I boot with USB adapter connected.
So there is a problem with 24.10.rc5..6, radios are never detected.
Trying with rc1...

hboterman, 2025-01-26:
Tested again OW24.10r6 and it works. Please forget my posts, don't know what happened :)

Wendigogo, 2025-01-26:
@eckinga : what is your UART device please? Or do any CP2104 based one will work?

apple601601601, 2025-02-09:
Confirm having same problem when I have my serial plugged in. The system crashes with kernel panic during wifi driver init.
After removing the debug console connection it starts fine and syslog showed wifi initialized ok.
Thank you guys for the info, it had haunted me for several days ...