openwrt/mt76 #705: Please add support for wlan 2,4 and 5 GHz on Asus rt-ax-53u

lukjod, 2022-10-23:
I see, WLAN LEDs are controlled by mt7915 wireless chip. It's a pity but it seems that we can't control both of them now. #10400
```
GPIO PIN MAP in ASUS RT-AX53U:
WLAN2G -- MT7905 GPIO18
WLAN5G -- MT7905 GPIO26
```
Or maybe you can create an issue on [openwrt/mt76](https://github.com/openwrt/mt76) and provide the information above.

_Originally posted by @DragonBluep in https://github.com/openwrt/openwrt/issues/11068#issuecomment-1288178843_

Djfe, 2023-01-03:
pls close this issue. This support was added already added to the MT76 driver and will be activated by the following PR in openwrt for the actual device :)
It might be activated for all devices using the driver (generally), that's still being discussed.
https://github.com/openwrt/openwrt/pull/11601#discussion_r1059124071

closed 2023-01-03 by lukjod

Doldak, 2023-11-27:
not fixed!