openwrt/mt76 #707: Mediatek MT7915 - Mesh issues

Noki, 2022-11-04:
I have two devices with MT7915, a Zyxel NWA55AXE and a Netgear WAX202. Both devices show the following traces and 802.11s does not work properly:

```
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.475157] ------------[ cut here ]------------
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.475243] WARNING: CPU: 1 PID: 3911 at backports-5.15.74-1/net/mac80211/mesh_hwmp.c:348 0x82c6c444 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.475254] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY nfnetlink nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ipt_ECN ip_tables hwmon compat ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476021]  crc32c_generic
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476068] CPU: 1 PID: 3911 Comm: reader#0 Tainted: G        W         5.10.152 #0
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476081] Stack : 00000000 80726368 809b0000 00000580 00000000 00000000 00000000 00000000
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476160]         00000000 00000000 00000000 00000000 00000000 00000001 82763818 545c9e09
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476236]         827638b0 00000000 00000000 827636c0 00000038 80399d64 ffffffea 00000000
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476315]         827636cc 0000025a 807ca868 20202020 827637f8 806f513c 80000000 806f0000
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476385]         00000009 82b6cda0 816801f0 00000580 00000018 00000020 00000004 80990004
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476460]         ...
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476478] Call Trace:
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476591] [<80399d64>] 0x80399d64
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476643] [<80007b08>] 0x80007b08
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476657] [<80007b10>] 0x80007b10
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476670] [<8037ee68>] 0x8037ee68
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476703] [<82c6c444>] 0x82c6c444 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476714] [<8007fb60>] 0x8007fb60
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476738] [<82c6c444>] 0x82c6c444 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476750] [<8003007c>] 0x8003007c
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476778] [<82c6c444>] 0x82c6c444 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476794] [<80030144>] 0x80030144
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476838] [<82c6c444>] 0x82c6c444 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476863] [<82b05ed0>] 0x82b05ed0 [cfg80211@18be209c+0x45fe0]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476904] [<82c08ff4>] 0x82c08ff4 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476960] [<80379b68>] 0x80379b68
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477024] [<82c24010>] 0x82c24010 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477066] [<82b2dc20>] 0x82b2dc20 [cfg80211@18be209c+0x45fe0]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477079] [<800c154c>] 0x800c154c
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477092] [<8037896c>] 0x8037896c
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477112] [<803784cc>] 0x803784cc
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477129] [<800101d0>] 0x800101d0
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477152] [<80026237>] 0x80026237
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477196] [<80574ae8>] 0x80574ae8
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477237] [<80575e04>] 0x80575e04
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477252] [<8050e6c4>] 0x8050e6c4
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477284] [<80579790>] 0x80579790
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477307] [<804e3108>] 0x804e3108
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477354] [<82b2dcb0>] 0x82b2dcb0 [cfg80211@18be209c+0x45fe0]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477378] [<82b2db2c>] 0x82b2db2c [cfg80211@18be209c+0x45fe0]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477394] [<805797c0>] 0x805797c0
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477419] [<82b2db2c>] 0x82b2db2c [cfg80211@18be209c+0x45fe0]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477432] [<80578d2c>] 0x80578d2c
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477468] [<8057946c>] 0x8057946c
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477484] [<805783ac>] 0x805783ac
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477519] [<80578d14>] 0x80578d14
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477539] [<80577978>] 0x80577978
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477559] [<801a1410>] 0x801a1410
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477609] [<80577e10>] 0x80577e10
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477629] [<8017f9fc>] 0x8017f9fc
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477664] [<80577aac>] 0x80577aac
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477677] [<804d94c4>] 0x804d94c4
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477695] [<803233c0>] 0x803233c0
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477738] [<804db354>] 0x804db354
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477759] [<800101d0>] 0x800101d0
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477798] [<80174798>] 0x80174798
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477837] [<80174798>] 0x80174798
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477873] [<801cc3b8>] 0x801cc3b8
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477900] [<804db564>] 0x804db564
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477937] [<80014578>] 0x80014578
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477965]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.478014] ---[ end trace a5af265f4d7ee6f7 ]---
```

The problem exists with the following OpenWRT versions: 22.03.2, 22.03-SNAPSHOT, SNAPSHOT

Please let me know if I should provide any additional information to help track this down.

ryderlee1110, 2022-11-09:
i don't think that hurts anything as you can still play mesh.

ryderlee1110, 2022-11-09:
I found something in ieee80211s_update_metric. please do this -

-	sta_set_rate_info_tx(sta, &sta->tx_stats.last_rate, &rinfo);
+	if (st->rate)
+		rinfo = sta->tx_stats.last_rate_info;
+	else
+		sta_set_rate_info_tx(sta, &sta->tx_stats.last_rate, &rinfo);

nbd168, 2022-11-10:
Backported the fix to OpenWrt master and 22.03

Noki, 2022-11-11:
Thanks @nbd168 for backporting this. The warning is gone now, however my mesh performance problems still persist:

As you can see in the following screenshot it looks like the devices fail to negotiate the correct RX/TX rate / high throughput mode.

[image]

One direction is usually fine, the other one is not. Sometimes it works for a short periode of time, after a restart of the device.

The result is extremely poor throughput. Let me know if I can do anything to help track this down.

Noki, 2022-11-13:
After all the MT76 fixes I updated to SNAPSHOT today to get the latest patches, without much success. In addition the airtime_link_metric_get warning is back, simply with another line number:

```
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.852587] ------------[ cut here ]------------
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.852780] WARNING: CPU: 0 PID: 5329 at backports-5.15.74-1/net/mac80211/mesh_hwmp.c:353 airtime_link_metric_get+0x120/0x684 [mac80211]
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.852792] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY nfnetlink nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ipt_ECN ip_tables hwmon compat ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853184]  crc32c_generic
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853208] CPU: 0 PID: 5329 Comm: ip Tainted: G        W         5.10.154 #0
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853217] Stack : 80a40000 00000580 808582f8 80083410 808d0000 807bd220 00000000 00000000
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853261]         83c118f4 80a20000 8078b08c 83cf1b48 8085ae87 00000001 83c11898 5f93b6b4
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853303]         00000000 00000000 8078b08c 83c11738 c0003ad5 00000000 ffffffea 00000000
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853345]         83c11744 00000ad5 80860888 2e352020 80000000 00000001 00000000 80790000
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853386]         00000009 825d8da0 8142c1f0 00000580 00000001 ffffffff 00000000 80a20000
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853427]         ...
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853441] Call Trace:
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853471] [<800080e0>] show_stack+0x30/0x100
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853501] [<80381520>] dump_stack+0x9c/0xcc
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853531] [<8003007c>] __warn+0xc0/0x12c
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853543] [<80030144>] warn_slowpath_fmt+0x5c/0xac
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853611] [<8276c818>] airtime_link_metric_get+0x120/0x684 [mac80211]
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853678] [<82708ff4>] sta_set_sinfo+0x7e0/0xfec [mac80211]
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853745] [<827096e0>] sta_set_sinfo+0xecc/0xfec [mac80211]
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853781]
Sun Nov 13 12:13:05 2022 kern.warn kernel: [ 1954.853836] ---[ end trace e2410a1e84bec228 ]---
```

```
root@e16-core:~# opkg list | grep mt
kmod-mt76-connac - 5.10.154+2022-11-12-4bf26073-5
kmod-mt76-core - 5.10.154+2022-11-12-4bf26073-5
kmod-mt7615-common - 5.10.154+2022-11-12-4bf26073-5
kmod-mt7915e - 5.10.154+2022-11-12-4bf26073-5
mtd - 26
mtr-json - 0.95-2
root@e16-core:~# cat /etc/banner
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r21271-6f729163b1
 -----------------------------------------------------
root@e16-core:~#
```

ryderlee1110, 2022-11-13:
You didn't observe this issue before, right? My best guess is [mac80211: extend current rate control tx status API] breaks something rate reporting on mesh path.

Noki, 2022-11-13:
Both devices are new and never worked for me. The issues where there right from the first version I tried.

ryderlee1110, 2022-11-13:
Ah. Please ignore my words as owrt doesn't include that change.

Noki, 2022-11-28:
@nbd168 since the mesh problem still persists and there were various new fixes and even a firmware update for 7915E, can you please do a new driver release and also backport it to 22.03 if possible?

Noki, 2022-12-13:
I tested the latest mt76 update release on OpenWRT snapshot and the problems still persist.

Noki, 2022-12-14:
OpenWrt SNAPSHOT, r21452-1e240f60a5 now produces the following kernel warning:

```
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.423965] ------------[ cut here ]------------
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424159] WARNING: CPU: 1 PID: 10092 at backports-6.1-rc8/net/mac80211/mesh_hwmp.c:353 airtime_link_metric_get+0x120/0x684 [mac80211]
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424166] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY nfnetlink nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ipt_ECN ip_tables hwmon compat ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424541]  crc32c_generic
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424563] CPU: 1 PID: 10092 Comm: kworker/u8:1 Tainted: G        W         5.10.158 #0
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424615] Workqueue: phy0 ieee80211_ibss_leave [mac80211]
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424626] Stack : 80a40000 82a8c820 82707c84 8008358c 8078b07c 80850000 83789a00 80790000
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424669]         00000009 82a8c548 00000010 8008372c 8085ae87 00000001 82707c40 2f42e7fd
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424710]         00000000 00000000 8078b07c 82707ac8 fffff67f 00000000 ffffffea 00000000
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424750]         82707ad4 0000067f 80860888 31323038 00000000 00000001 00000000 80790000
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424791]         00000009 82a8c548 00000010 82a8c820 00000018 00000020 00000004 80a20004
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424831]         ...
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424845] Call Trace:
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424875] [<800080e0>] show_stack+0x30/0x100
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424906] [<80381a34>] dump_stack+0x9c/0xcc
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424935] [<8003007c>] __warn+0xc0/0x12c
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.424946] [<80030144>] warn_slowpath_fmt+0x5c/0xac
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.425000] [<82c73ec4>] airtime_link_metric_get+0x120/0x684 [mac80211]
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.425071] [<82c0a6a8>] sta_set_sinfo+0x7e0/0xfd4 [mac80211]
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.425142] [<82c0ad7c>] sta_set_sinfo+0xeb4/0xfd4 [mac80211]
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.425181]
Wed Dec 14 06:38:59 2022 kern.warn kernel: [165245.425189] ---[ end trace 62327d7b4e8e8251 ]---
```

Noki, 2022-12-19:
OpenWrt SNAPSHOT, r21516-1b8c8864c5 produces similar warnings with a slightly different trace:

```
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.364593] ------------[ cut here ]------------
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.364792] WARNING: CPU: 2 PID: 3930 at backports-6.1-rc8/net/mac80211/mesh_hwmp.c:353 airtime_link_metric_get+0x120/0x684 [mac80211]
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.364800] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY nfnetlink nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ipt_ECN ip_tables hwmon compat ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365195]  crc32c_generic
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365217] CPU: 2 PID: 3930 Comm: reader#0 Tainted: G        W         5.10.159 #0
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365226] Stack : 80a40000 826de820 808582f8 8008358c 808d0000 807be2e0 00000000 00000000
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365271]         837b787c 80a20000 8078c114 814f61c8 8085ae87 00000001 837b7820 1b2b5702
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365315]         00000000 00000000 8078c114 837b76c0 c0009393 00000000 ffffffea 00000000
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365358]         837b76cc 00000393 80860888 20202020 80000000 00000001 00000000 80790000
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365402]         00000009 826de548 00000010 826de820 00000018 ffffffff 00000008 80a20008
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365446]         ...
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365461] Call Trace:
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365492] [<800080e0>] show_stack+0x30/0x100
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365524] [<80381a5c>] dump_stack+0x9c/0xcc
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365554] [<8003007c>] __warn+0xc0/0x12c
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365566] [<80030144>] warn_slowpath_fmt+0x5c/0xac
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365624] [<82c73d94>] airtime_link_metric_get+0x120/0x684 [mac80211]
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365696] [<82c0a6a8>] sta_set_sinfo+0x7e0/0xfd4 [mac80211]
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365769] [<82c271ac>] ieee80211_nan_func_match+0x1a64/0x2130 [mac80211]
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365880] [<82b2fc14>] nl80211_put_sta_rate+0x14ec/0x16c0 [cfg80211]
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365919]
Mon Dec 19 16:49:07 2022 kern.warn kernel: [115566.365928] ---[ end trace 4c68210cd8780c15 ]---
```

Noki, 2023-01-10:
The problem still persists in OpenWrt SNAPSHOT, r21728-fc33c41c21.

From looking at other issues like #724 i think my mesh problems are related to MT7915E failing to negotiate proper tx/rx bitrates in high throughput modes in general.

I placed the two mentioned MT7915E routers about two meters from each other and this is the result:

```
iw dev wlan5-mesh station dump | grep -E '(Station|(tx|rx) bitrate)'
Station 34:98:b5:0f:37:62 (on wlan5-mesh)
        tx bitrate:     6.0 MBit/s
        rx bitrate:     54.0 MBit/s
```

```
iw dev wlan5-mesh station dump | grep -E '(Station|(tx|rx) bitrate)'
Station b8:ec:a3:e1:22:05 (on wlan5-mesh)
        tx bitrate:     54.0 MBit/s
        rx bitrate:     54.0 MBit/s
```

Noki, 2023-01-31:
Seems to be fixed with https://github.com/openwrt/openwrt/commit/a63430eac33ceb1dbf96d3667e2a0f2e04ba391f

closed 2023-01-31 by Noki

stweil, 2023-05-02:
I cannot confirm that this issue is fixed with that commit. Please reopen.

Noki, 2023-05-02:
@stweil There is a follow up issue here which is more precise for my case: https://github.com/openwrt/mt76/issues/755