openwrt/mt76 #708: Mediatek MT7922A22M - Mesh issues

Wang1998, 2022-11-11:
Nov 11 00:55:10 include kernel: mt7921e 0000:01:00.0: Message 00000010 (seq 2) timeout
Nov 11 00:55:10 include kernel: mt7921e 0000:01:00.0: Failed to get patch semaphore
Nov 11 00:55:10 include systemd[1]: Starting PVE guests...
Nov 11 00:55:11 include pve-guests[993]: <root@pam> starting task UPID:include:000003E2:00000485:636D2CEF:startall::root@pam:
Nov 11 00:55:11 include pve-guests[993]: <root@pam> end task UPID:include:000003E2:00000485:636D2CEF:startall::root@pam: OK
Nov 11 00:55:11 include systemd[1]: Finished PVE guests.
Nov 11 00:55:11 include systemd[1]: Starting Proxmox VE scheduler...
Nov 11 00:55:12 include pvescheduler[996]: starting server
Nov 11 00:55:12 include systemd[1]: Started Proxmox VE scheduler.
Nov 11 00:55:12 include systemd[1]: Reached target Multi-User System.
Nov 11 00:55:12 include systemd[1]: Reached target Graphical Interface.
Nov 11 00:55:12 include systemd[1]: Starting Update UTMP about System Runlevel Changes...
Nov 11 00:55:12 include systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
Nov 11 00:55:12 include systemd[1]: Finished Update UTMP about System Runlevel Changes.
Nov 11 00:55:12 include systemd[1]: Startup finished in 7.421s (firmware) + 5.883s (loader) + 2.907s (kernel) + 9.575s (userspace) = 25.788s.
Nov 11 00:55:13 include chronyd[753]: Selected source 185.209.85.222 (2.debian.pool.ntp.org)
Nov 11 00:55:13 include chronyd[753]: System clock TAI offset set to 37 seconds
Nov 11 00:55:13 include kernel: mt7921e 0000:01:00.0: Message 00000010 (seq 3) timeout
Nov 11 00:55:13 include kernel: mt7921e 0000:01:00.0: Failed to get patch semaphore
Nov 11 00:55:17 include kernel: mt7921e 0000:01:00.0: Message 00000010 (seq 4) timeout
Nov 11 00:55:17 include kernel: mt7921e 0000:01:00.0: Failed to get patch semaphore
Nov 11 00:55:20 include kernel: mt7921e 0000:01:00.0: Message 00000010 (seq 5) timeout
Nov 11 00:55:20 include kernel: mt7921e 0000:01:00.0: Failed to get patch semaphore
Nov 11 00:55:23 include kernel: mt7921e 0000:01:00.0: Message 00000010 (seq 6) timeout
Nov 11 00:55:23 include kernel: mt7921e 0000:01:00.0: Failed to get patch semaphore
Nov 11 00:55:27 include kernel: mt7921e 0000:01:00.0: Message 00000010 (seq 7) timeout
Nov 11 00:55:27 include kernel: mt7921e 0000:01:00.0: Failed to get patch semaphore
Nov 11 00:55:30 include kernel: mt7921e 0000:01:00.0: Message 00000010 (seq 8) timeout
Nov 11 00:55:30 include kernel: mt7921e 0000:01:00.0: Failed to get patch semaphore
Nov 11 00:55:33 include pvedaemon[976]: <root@pam> successful auth for user 'root@pam'
Nov 11 00:55:33 include kernel: mt7921e 0000:01:00.0: Message 00000010 (seq 9) timeout
Nov 11 00:55:33 include kernel: mt7921e 0000:01:00.0: Failed to get patch semaphore
Nov 11 00:55:33 include systemd[1]: systemd-fsckd.service: Succeeded.

dhewg, 2022-11-13:
I had `Failed to get patch semaphore` too. In my case it was a missing firmware.
Make sure you have:
```
/lib/firmware/mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
/lib/firmware/mediatek/WIFI_RAM_CODE_MT7961_1.bin
```

Wang1998, 2022-11-15:
> 
I have the 2 files above

Wang1998, 2022-11-15:
I have the file you're talking about.

deren, 2022-12-05:
Hi @Wang1998 

if you work with mt7922, please make sure the two files WIFI_RAM_CODE_MT7922_1.bin and WIFI_MT7922_patch_mcu_1_1_hdr.bin in your system. :)

antnyfls, 2023-06-23:
I can confirm this issue with the mt7922 wifi card , I am using the latest openwrt mt76 update
https://github.com/openwrt/openwrt/commit/901af2718957fc878279073e9b917458144c75a4 and the firmware files installed  on _/lib/firmware/mediatek_ 

It works ok as a client, when I try to use it as AP 
```
[ 1025.884263] br-lan: port 6(phy0-ap0) entered blocking state
84783] br-lan: port 6(phy0-ap0) entered disabled state
[ 1025.885591] device phy0-ap0 entered promiscuous mode
[ 1025.886548] br-lan: port 6(phy0-ap0) entered blocking state
[ 1025.887083] ered forwarding state
[ 1025.888146] br-lan: port 6(phy0-ap0) entered disabled state
[ 1029.350123] mt7921e 0001:01:00.0: Message 00020003 (seq 5) timeout
[ 1029.350742] phy0-ap0: ff:ff:ff:ff) to hardware (-110)
[ 1032.549934] mt7921e 0001:01:00.0: Message 00000010 (seq 7) timeout
[ 1032.550521] mt7921e 00tch semaphore
[ 1035.749862] mt7921e 0001:01:00.0: Message 00000010 (seq 8) ti1:01:00.0: Failed to get patch semaphore
[ 1038.949722] mt7921e 0001:01:00.0: Message 00000010 (seq 9) ti1] mt7921e 0001:01:00.0: Failed to get patch semaphore
[ 1042.149594] mt7921e 0001:01:00.0: Message 00000010 (seq 10) t[ 1042.150303] mt7921e 0001:01:0led to get patch semaphore
[ 1045.349528] mt7921e 0001:01:00.0: Message 00000010 (seq 11) timeout
[ 1045.350144] mt7921e 0:00.0: Failed to get patch semaphore
[ 1048.549392] mt7921e 0001:01:00.0: Message 00000010 (seq 12) timeout
[ 1048.550007] mt7921e 0001:01:00.0: Failed to get patch[ 1051.749231] mt7921e 0001:01:00.0: Message 00020002 (seq 13) t[ 1054.949127] mt7921e 0001:01:00.0: Message 00002ced (seq 14) timeout
[ 1058.148994] mt7921e 0001:01:00.0: Message 00000010 (seq 15) timeout
[ 1058.149613] mt7921e 0 patch semaphore
[ 1061.348970] mt7921e 0001:01:00.0: Message 00000010 (seq 1) ti 0001:01:00.0: Failed to get patch semaphore
[ 1064.548759] mt7921e 0001:01:00.0: Message 00020002 (seq 2) ti[ 1067.748657] mt7921e 0001:01:00.0: Message 00002ced (seq 3) ti[ 1070.948540] mt7921e 0001:01:00.0: Message 00000010 (seq 4) timeout
[ 1070.949148] mt7921e 0001:01:00.0: Failed to get patch semaphore
[ 1074.148419] mt7921e 0001:01:00.0: Message 00000010 (seq 5) timeout
[ 1074.149028] mt7921e 0001:01:00.0: Failed to get patch semaphore
[ 1074.149618] mt7921e 0001:01:00.0: chip reset failed
[ 1077.348321] mt7921e 0001:01:00.0: Message 00020001 (seq 6) ti[ 1080.548200] mt7921e 0001:01:00.0: Message 00020002 (seq 10) timeout
[ 1083.748104] mt7921e 0001:01:00.0: Message 00020003 (seq 11) timeout
[ 1086.948018] mt7921e 0001:01:00.0: Message 00002ced (seq 13) t[ 1090.147883] mt7921e 0001:01:00.0: Message 00002ced (seq 14) t[ 1093.347789] mt7921e 0001:01:00.0: Message 00000010 (seq 15) tut
[ 1093.348406] mt7921e 0001:01:00.0: Failed to get patch semaphore
```

lukasz1992, 2023-06-23:
try this: https://github.com/openwrt/mt76/commit/2f851902d6b1284ba7c326a685c96f5ba88dae1d

lukasz1992, 2023-06-23:
And this: https://patchwork.kernel.org/project/linux-wireless/patch/19f1aae1ab9ea867eb42742fc5b72ed4d7307b0a.1687159671.git.deren.wu@mediatek.com/

antnyfls, 2023-07-09:
Apparently this issue was related to hostapd , I'm using openwrt and with the latest hostap update https://github.com/openwrt/openwrt/commit/cd804c1ebb34c9740f8ad9fe393a81f3efb7b009, and my mt7922 card is working normally as AP now.