openwrt/mt76 #709: MT7622/MT7915E Slow 802.11ax speeds

d-me3, 2022-11-12:
OpenWrt on MT7622/MT7915E has issues with 802.11ax performance (specially upload speeds) when the client device has an obstacle (such as a brick wall) in between. In this situation 802.11ax performance degrades severely, while 802.11ac in the same situation has a much better performance. This has been reported in Redmi AX6S and Belkin RT3200 with 22.03.x builds and also in nightly builds. More details in the OpenWrt forum at https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/1 including test results and how to reproduce.

graysky2, 2022-11-12:
In all the forum posts about this, has anyone done the comparison of the proprietary 802.11ax driver (ie non-openwrt) vs this open source 802.11ax driver?

d-me3, 2022-11-12:
Yep, I did as @eginnc mentioned above (I'm dsouza user in the forum).

ryderlee1110, 2022-11-12:
I think this could be related to particular APs as everything goes well on my devices (as well as other users I know).

graysky2, 2022-11-16:
@ryderlee1110 - Can you share the exact settings you have configured in your `/etc/config/wireless` so that affected users can replicate and test?  Have you tried the iperf3 3-parallel streams upload test (on the phone) to the server with, 1) no walls between the AP and the phone, 2) 1 wall, 3) 2 walls, 4) 2 walls and some large object such as a piece of furniture in AX mode?

mdbecker, 2022-11-18:
Can confirm this exact issue on my brand new Belkin AX3200.

No brick walls required, if I just walk a few feet from the AP it will go from HE-MCS 11 to HE-MCS 0 on the RX. The issue never effects the TX. Speed test run from several places in my house on both the stock FW and openwrt. Channel 36, 80MHz in all tests. I ran all these tests on a new 2021 Macbook Pro. Once someone in the original thread pointed out that this issue seems to be limited to Apple devices I ran a speedtest on my Pixel 5 and saw that I was getting better speeds 
(294Mbps/244Mbps Pixel vs 296Mbps/183Mbps Macbook at location 1).

I noticed performance degrades even further when I switch from 80MHz to 160MHz windows (285Mbps/189Mbps Pixel vs 276Mbps/14Mbps Macbook at location 1).

## Internet speed tests
[image]

## iperf3 tests `iperf3 -c -d -P 10`
### Location 1
#### Stock AX
```
[SUM]   0.00-10.00  sec   317 MBytes   266 Mbits/sec                  sender
[SUM]   0.00-10.16  sec   302 MBytes   250 Mbits/sec                  receiver
```

#### OpenWRT AX
```
[SUM]   0.00-10.00  sec   203 MBytes   170 Mbits/sec                  sender
[SUM]   0.00-10.06  sec   198 MBytes   165 Mbits/sec                  receiver
```

#### OpenWRT AC
```
[SUM]   0.00-10.00  sec   379 MBytes   318 Mbits/sec                  sender
[SUM]   0.00-10.02  sec   374 MBytes   313 Mbits/sec                  receiver
```

### Location 2
#### Stock AX
```
[SUM]   0.00-10.00  sec  94.2 MBytes  79.0 Mbits/sec                  sender
[SUM]   0.00-10.05  sec  92.5 MBytes  77.2 Mbits/sec                  receiver
```

#### OpenWRT AX
```
[SUM]   0.00-10.00  sec  18.7 MBytes  15.7 Mbits/sec                  sender
[SUM]   0.00-10.71  sec  17.3 MBytes  13.6 Mbits/sec                  receiver
```

#### OpenWRT AC
```
[SUM]   0.00-10.00  sec   154 MBytes   130 Mbits/sec                  sender
[SUM]   0.00-10.11  sec   151 MBytes   125 Mbits/sec                  receiver
```

### Location 3
#### Stock AX
```
[SUM]   0.00-10.00  sec   102 MBytes  85.7 Mbits/sec                  sender
[SUM]   0.00-10.07  sec  99.4 MBytes  82.8 Mbits/sec                  receiver
```

#### OpenWRT AX
```
[SUM]   0.00-10.00  sec  14.7 MBytes  12.3 Mbits/sec                  sender
[SUM]   0.00-10.56  sec  13.5 MBytes  10.7 Mbits/sec                  receiver
```

#### OpenWRT AC
```
[SUM]   0.00-10.00  sec   261 MBytes   219 Mbits/sec                  sender
[SUM]   0.00-10.08  sec   256 MBytes   213 Mbits/sec                  receiver
```

### Location 4
#### Stock AX
```
[SUM]   0.00-10.00  sec  99.8 MBytes  83.7 Mbits/sec                  sender
[SUM]   0.00-10.12  sec  96.4 MBytes  79.9 Mbits/sec                  receiver
```

#### OpenWRT AX
```
[SUM]   0.00-10.00  sec  12.9 MBytes  10.8 Mbits/sec                  sender
[SUM]   0.00-10.59  sec  11.7 MBytes  9.28 Mbits/sec                  receiver
```

#### OpenWRT AC
```
[SUM]   0.00-10.00  sec   261 MBytes   219 Mbits/sec                  sender
[SUM]   0.00-10.05  sec   256 MBytes   214 Mbits/sec                  receiver
```

### Location 5
#### Stock AX
```
[SUM]   0.00-10.00  sec   208 MBytes   175 Mbits/sec                  sender
[SUM]   0.00-10.20  sec   199 MBytes   163 Mbits/sec                  receiver
```

#### OpenWRT AX
```
[SUM]   0.00-10.00  sec  16.5 MBytes  13.8 Mbits/sec                  sender
[SUM]   0.00-10.74  sec  15.3 MBytes  12.0 Mbits/sec                  receiver
```

#### OpenWRT AC
```
[SUM]   0.00-10.00  sec   193 MBytes   162 Mbits/sec                  sender
[SUM]   0.00-10.03  sec   190 MBytes   159 Mbits/sec                  receiver
```

graysky2, 2022-11-19:
@mdbecker - Thanks for doing those tests and for the nice summary.  Something is wrong with open source driver...
| condition |  % loss relative to stock ax driver |
| --- | --- |
| no barrier | 34-36 % |
| single wall | 80-82 % |
| two walls | 87-88 % |
| two walls and object | 92-93 % |

ernie-t, 2022-11-20:
> I think this could be related to particular APs as everything goes well on my devices (as well as other users I know).

I have tested on following Devices: Belkin AX3200, Xiaomi AX3200/AX6S - same problems. Also speed decreases in AC Mode after 24h extremely.

Gingernut1978, 2022-11-23:
I think this is not a mt76 driver problem as others have reported the same degrading performance on other platforms, ath11k for example.

Could it be possibly a problem with how mac80211/hostapd setup the radios?

ThiloteE, 2022-11-24:
> as others have reported the same degrading performance on other platforms, ath11k for example.

source?

Gingernut1978, 2022-11-24:
> > as others have reported the same degrading performance on other platforms, ath11k for example.
> 
> source?
At the bottom of this thread there are some wireless tests: 
https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/9077

ThiloteE, 2022-11-24:
> https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/9077

Not sure if those measurements really are related,

- Yes, [Measurement A](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/9073) and [Measurement B](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/9074) show very low values for uplink speed, but those were done via fast.com and purely from router to provider, so the uplink could contractually be limited by the internet provider and actually might have nothing to do with wifi.
- [Measurement C](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/9070) shows higher speed for ax as compared to ac, but the location is very close to the router (2 meter) without walls.
- None of the other actual wifi measurements in this thread (such as [Measurement D](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/9077) and [Measurement E](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/9078) compare wifi AC vs wifi AX (or uplink vs downlink), but rather compare stock firmware with openWRT firmware without ever mentioning wifi ax or ac...

Gingernut1978, 2022-11-24:
Well I did say it was just one example.

If you read through the thread you'll find more tests done by other users, some I actually did,  that are more wifi AX vs AC specific. Others and I found AX transfer speed heavily dropped when obstacles like thin walls, doors were in between client and station even though the signal quality wasn't bad between client and station.

ThiloteE, 2022-11-24:
@Gingernut1978 you understand that this thread has 8970 (!) comments?! Who has time to read all that. Please be precise with measurements and your statements and when stating a source. Thank you.

ThiloteE, 2022-11-24:
ok, i actually found an Ath11k driver comparison between wifi 802.11ac vs 802.11ax [in this comment](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/1431), but that was from March 2021... so might be outdated.

lukasz1992, 2022-12-10:
Do https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/170 or 
https://github.com/openwrt/openwrt/commit/590eaaeed59a9eb6637a1480587fc410de182523 help for the issue?

ThiloteE, 2022-12-10:
> Do https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/170 or [openwrt/openwrt@590eaae](https://github.com/openwrt/openwrt/commit/590eaaeed59a9eb6637a1480587fc410de182523) help for the issue?

Apparently not if this latest comment is to be believed: https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/177

graysky2, 2022-12-11:
See my post [here](https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/180) for details.  Highlight is that running a snapshot build from HEAD yesterday, my Redmi AX6000 finally gives high throughput from behind several walls and a couch (lots of signal blockage) in AX mode but only when using 160 MHz width/channel 100.  If I keep it on 80 MHz/channel 149, the throughput is terrible again.

Would love to hear @nbd168 insights into this interesting result.

Anteus, 2023-02-02:
Latest mt76 update seems to have fixed it on rt3200 at least.
https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/3394?u=anteus

graysky2, 2023-02-03:
The issue is still affecting Redmi AX6000.

darekxan, 2023-02-07:
According to my measurements this is no longer an issue with latest snapshots
https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/3419

taylorkline, 2023-02-07:
> According to my measurements this is no longer an issue with latest snapshots https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/3419

Chiming in to +1. Speeds now equivalent to 802.11ac on the troublesome 2021 MacBook Pro, and other devices are hitting 802.11ax-only speeds.

I'm on snapshot build `r22000-6bc675c0be`

FYI, I always need to SSH in and run `opkg update && opkg install luci` with snapshot builds.

taylorkline, 2023-03-03:
Just wanted to update that this issue seems to have regressed, it is being discussed in the forum thread starting from here: https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/334

taylorkline, 2023-03-03:
@nbd168 Would you mind if I politely tagged you to ask if you might know what caused this issue to regress to 22.03.3 behavior for 80 Mhz Channel Width, yet not in the 120 Mhz Channel Width?

At least, these are my findings here: https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/355?u=jkdf2

darekxan, 2023-03-04:
I've redone my tests, for me everything is ok for 5ghz ax radio:
https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/3507?

Anteus, 2023-03-16:
On rt3200 I found the line that triggers the slowdown https://github.com/openwrt/mt76/commit/f9ca70d6367a16fecd80b93fec60fb0d4e3b9fd0#commitcomment-104764055

Wepeemy, 2023-04-12:
Hi, just want to ask, has the speed issue solved, with the latest snapshot available for Redmi AX6s router based on MT7622B + MT7915E chips? Thank you.

Wepeemy, 2023-05-08:
> On my Xiaomi AX3200 router, iperf3 measurements show 1 Mbps from laptop (Intel AX201) to router and 400 Mbps from router to laptop. The router is on the table, the laptop is under the table. OpenWrt 22.03.5 and SNAPSHOT r22746-590d1fd0e6.
> 
> ```
> config wifi-device 'radio1'
>         option type 'mac80211'
>         option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
>         option channel '36'
>         option band '5g'
>         option htmode 'HE160'
>         option country 'RU'
>         option cell_density '0'
> 
> config wifi-iface 'default_radio1'
>         option device 'radio1'
>         option network 'lan'
>         option mode 'ap'
>         option ssid '***'
>         option encryption 'sae'
>         option key '***'
> ```

Have you try this workaround method:[](https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/50) and see if there is any improvement?

Anteus, 2023-05-08:
This fixes it for rt3200 which also have the mt7915. Easy to test if it works fine with AC instead of AX.

https://github.com/openwrt/mt76/pull/768

Aloki, 2023-05-09:
> Have you try this workaround method:https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/50 and see if there is any improvement?

I apologize for the false alarm. As I found out, the problem was in Hyper-V. The virtual switch was connected to an external network, so the Wi-Fi adapter was in bridge mode. And for some reason, in this mode, the upload speed dropped to 1 Mbps. I deleted the virtual switch and the speed became normal (950 Mbps).

taylorkline, 2023-05-31:
@Anteus is there any way I can try your patch via https://firmware-selector.openwrt.org/ or build it locally on macOS without extensive dev env setup? 

(PS I've moved to a Xiaomi Redmi Router AX6000 which has the same issue)

cryobry, 2023-05-31:
> @Anteus is there any way I can try your patch via https://firmware-selector.openwrt.org/ or build it locally on macOS without extensive dev env setup?
> 
> (PS I've moved to a Xiaomi Redmi Router AX6000 which has the same issue)

There's also an open PR I made for the AX6000 specifically that you can use. I'm also in the process of writing "openwrtbuilder", a program that can build from source and incorporate patches: https://git.bryanroessler.com/bryan/openwrtbuilder

It's not done and unfortunately I won't be able to help support you until tomorrow.

Anteus, 2023-05-31:
> @Anteus is there any way I can try your patch via https://firmware-selector.openwrt.org/ or build it locally on macOS without extensive dev env setup?
> 
> (PS I've moved to a Xiaomi Redmi Router AX6000 which has the same issue)

I build on my M2 MacBook Air, it takes only about ten minutes when the dev env is set up.
There's a section here for building on macOS https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem

Haven't gotten eBPF to build on Apple Silicon though so for example no Qosify but the normal SQM packages works fine.

Anteus, 2023-07-14:
I think this could be closed as of this commit https://github.com/openwrt/mt76/commit/bb3937d5c3e0b13c0d08747ec0fc9726fb4fd870 which removes VHT160 from mt7915 non-dbdc.

King-Of-Knights, 2023-07-14:
@Anteus Hi, does it mean mt7915e share the driver with mt7915? I found my mt7916 run about twice faster on ax 80MHz than ax 160MHz

Anteus, 2023-07-14:
I would think so. Maybe it's like mt7915 which can only can do 2x2 NSS at 160?

King-Of-Knights, 2023-07-14:
@Anteus ^_^ Does this commit mean mt7915 or mt7915e will run at normal speed?

Anteus, 2023-07-14:
@King-Of-Knights mt7915

King-Of-Knights, 2023-07-17:
@Anteus Ha, looks like this patch fix this issues after some test, thanks to let me know, bro!

King-Of-Knights, 2023-07-17:
@Anteus OK, after some test, I find [bb3937d](https://github.com/openwrt/mt76/commit/bb3937d5c3e0b13c0d08747ec0fc9726fb4fd870) will let me use more 5G channel. Like 149 153 channel (which signal cannot be shown in my list before). However, I still suffer slow speed in 5G 160MHz, while 80MHz works just fine.

castiel652, 2026-03-30:
Can you try a non DFS channel