openwrt/mt76 #711: mt7921u: kernel crashes while generating traffic

dhewg, 2022-11-13:
Device is in AP mode and operational. I can connect to it just fine, and it works a while, but the kernel crashes every now and then.
I don't have serial setup on the router it's connected to, so unfortunately the only log entry I have at this time is:

`kern.emerg kernel: [37856.956725] skbuff: skb_under_panic: text:834905b0 len:128 put:4 head:85b52ba0 data:85b52b9e tail:0x85b52c1e end:0x85b53400 dev:XXX`

0e8d:7961 Comfast CF-953AX (MT7921AU)
Version: https://github.com/openwrt/mt76/commit/4a27f23fc4f22510db81513680666be7e72ccb61
OpenWrt with a rather simple config:
```
config wifi-device 'radio2'                                                     
    option type 'mac80211'                                                      
    option path 'platform/1e1c0000.xhci/usb1/1-2/1-2:1.3'                       
    option band '5g'                                                            
    option country 'DE'                                                         
    option cell_density '0'                                                     
    option channel '149'                                                        
    option htmode 'HE80'                                                        
                                                                                
config wifi-iface 'wifinet1'                                                    
    option device 'radio2'                                                      
    option mode 'ap'                                                            
    option ssid 'XXX'                                                       
    option encryption 'sae'                                                     
    option ifname 'XXX'                                                     
    option key 'XXX'                                                
    option network 'lan'  
```

dhewg, 2022-11-13:
Another one:
`kern.emerg kernel: [13716.152655] skbuff: skb_under_panic: text:82a405b0 len:128 put:4 head:85730f80 data:85730f7e tail:0x85730ffe end:0x857317e0`

dhewg, 2022-11-17:
Updated with @nbd168 patch to use the mac80211 backports from 6.1-rc4
Which in return uses this repo's head of dec9b8181 (dev branch).

Crash happend again, but a bit more infos:
```
Thu Nov 17 17:37:38 2022 kern.emerg kernel: [ 7626.202853] skbuff: skb_under_panic: text:835005b0 len:128 put:4 head:822318e0 data:822318de tail:0x8223195e end:0x82232140 dev:XXX
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.227161] Kernel bug detected[#1]:
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.234282] CPU: 1 PID: 1161 Comm:mt76-tx phy2 Not tainted 5.15.79 #0
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.247272] $ 0   : 00000000 00000001 0000007b ee33509a
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.257692] $ 4   : ee33509a 8100b370 8100c92c 838fba78
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.268112] $ 8   : ffffefff 00000000 ffffffea 00000000
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.278525] $12   : 838fba80 80890000 ffffffff ffffffff
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.288935] $16   : 85a8b600 0000007c 85e7d950 00000001
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.299346] $20   : 834917a0 83a1ea08 00000002 85e7d7bc
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.309759] $24   : 00000018 80424528
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.320170] $28   : 838fa000 838fbc30 00000000 80505f7c
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.330581] Hi    : 00000125
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.336301] Lo    : 122f2000
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.342023] epc   : 80505f7c skb_panic+0x5c/0x60
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.351218] ra    : 80505f7c skb_panic+0x5c/0x60
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.360401] Status: 1100fc03     KERNEL EXL IE
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.368727] Cause : 50800024 (ExcCode 09)
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.376696] PrId  : 0001992f (MIPS1004Kc)
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.384837] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pl2303 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_natnf_flow_table nf_conntrack mt7921u mt7921_common mt7615e mt7615_common mt7603e mt76_usb mt76_connac_lib mt76 mac80211 ftdi_sio cfg80211 usbserial nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport zram zsmalloc crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.385306]  nls_base usb_common crc32c_generic
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.566995] Process mt76-tx phy2 (pid: 1161, threadinfo=0ebf360e, task=df4c45ec, tls=00000000)
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.584144] Stack : 838fbcfc 80820d5c 835005b0 00000080 00000004 822318e0 822318de 8223195e
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.600805]         82232140 83144000 85e7d950 80505fc4 834917a0 83a1ea08 00000002 85e7d7bc
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.617470]         00000000 835005b0 838fbd94 80890000 8100b31c 00006790 83a1ea08 00000001
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.634128]         00000002 00000000 83818680 834917a0 85a8b600 80890000 000000a7 00000002
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.650794]         85e7d950 82f0c140 80c53880 00000010 00000ff9 00000000 85e7d7bc 838fbcd0
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.667488]         ...
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.672371] Call Trace:
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.677226] [<80505f7c>] skb_panic+0x5c/0x60
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.685731] [<80505fc4>] mm_unaccount_pinned_pages+0x0/0x34
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.696819]
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.699774] Code: 01003825  0c021241  24840b18 <000c000d> 8c8200a4  8c890054  8c8800a0  00451023  01254821
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.719201]
Thu Nov 17 17:37:38 2022 kern.warn kernel: [ 7626.722416] ---[ end trace 3498e32e26a121d9 ]---
```

Let me know if more info is required

dhewg, 2022-11-22:
Meanwhile, I moved the device to a ipq4019 board. Same version of everything, and it's stable for days.
Maybe the ramips board couldn't keep up with the traffic? But it shouldn't crash I guess :)

esp0xdeadbeef, 2026-11-15:
Same `skb_under_panic`, hit it on mt7925u in AP mode too (`put:4`, right after `mt7925_usb_sdio_tx_prepare_skb`). It isn't the board being too slow.

`mt7925_usb_sdio_tx_prepare_skb()` writes the TXWI at `skb->data - MT_SDIO_TXD_SIZE` and then `skb_push()`es, without reserving headroom first. Its sibling `mt7921_usb_sdio_tx_prepare_skb()` does call `skb_cow_head()`. When mac80211 hands in an skb without enough headroom -- the beacon path in AP mode does -- `skb_push()` walks below `skb->head` and trips `skb_under_panic()`.

One-line fix, mirrors mt7921:

```c
err = skb_cow_head(skb, MT_SDIO_TXD_SIZE + MT_SDIO_HDR_SIZE);
if (err)
	return err;
```

Patch, applies to `mt7925/usb.c`:
https://github.com/esp0xdeadbeef/nixos/blob/cca1932f/patches/mt7925-tx-headroom.patch

The config that ran this in AP mode for months is pinned here (`patches/mt7925-*`): `github:esp0xdeadbeef/nixos/cca1932f`

esp0xdeadbeef, 2026-11-15:
Correction to my comment above: the function is in **`mt7925/mac.c`**, not `mt7925/usb.c`.

Verified against current mainline:

- `mt7921/mac.c:781` — `mt7921_usb_sdio_tx_prepare_skb()` calls `skb_cow_head(skb, MT_SDIO_TXD_SIZE + MT_SDIO_HDR_SIZE)` before the push.
- `mt7925/mac.c:1645` — `mt7925_usb_sdio_tx_prepare_skb()` goes straight from the `len <= ETH_HLEN` check to `if (!wcid)`, with no `skb_cow_head()`.

Same on v6.18. The patch linked above already targets the correct file; only my prose was wrong.

esp0xdeadbeef, 2026-11-15:
One more qualification on my first comment, after checking the code more carefully.

Your device is `mt7921u`, and `mt7921_usb_sdio_tx_prepare_skb()` in `mt7921/mac.c` **already** calls `skb_cow_head()` before the push — so the specific line my patch fixes in mt7925 is not obviously what bit you. Your crash is in the mt7921 path, and I can't tell from the trace which call site pushed below head.

What I'm confident about:
- the mt7925 analogue is genuinely missing that call (verified on mainline and v6.18)
- `skb_under_panic` with `put:4` in the tx path is the generic symptom of an skb lacking headroom, so the diagnosis is in the right area

What I'm *not* claiming: that this explains your crash, or that "the board couldn't keep up" is disproven. I don't know the cause of yours.