openwrt/mt76 #714: supporting multiple leds

frank-w, 2022-12-04:
Hi,

i have a mt7986a board (Bananapi-R3) which has 2 wifi-phys (2g4,5g). and leds connected to the responding pins

shematic:
https://drive.google.com/file/d/1mxKb8CBbnzfNSd_4esmcX_NovxaXjEb8/view

lower part of first page...WF2G_LED (GPIO1) / WF5G_LED (GPIO2)

both phys are registered, but only 1 led is visible in sysfs (mt76_phy0). this can be set to on, but not to off (brightness 0)

init:
https://elixir.bootlin.com/linux/v6.0.11/source/drivers/net/wireless/mediatek/mt76/mt7915/init.c#L1099
init seems to be done only for phy 0 (tried to replace 0 by dev->phy.band_idx in mt7915_led_set_config)
https://elixir.bootlin.com/linux/v6.0.11/source/drivers/net/wireless/mediatek/mt76/mt7915/init.c#L207

and added some debug-prints, but it seems the function is not called at all :(

one problem (not for the missing missing call, but multi-led-support itself) seems to be data-structure allow only 1 led
https://elixir.bootlin.com/linux/v6.0.11/source/drivers/net/wireless/mediatek/mt76/mt76.h#L786

led initialization itself (which creates the sysfs entry i see) seems to happen in mac80211.c mt76_led_init...but of course only for 1 led.

maybe you can help here to drive multiple leds, so i can see which phy is active. when should led lighten up? in normal setup (both phy in AP-Mode with hostapd running) does not bring the led(s) up (maybe traffic related? maybe additional options needed?).

regards Frank

ryderlee1110, 2022-12-06:
Can you help to test these changes? Just built pass without testing.

https://pastebin.mozilla.org/LWUeUWzH/raw
https://pastebin.mozilla.org/MSpxhuqK/raw

frank-w, 2022-12-06:
Hi,wrote with lorenzo yesterday gave me similar patches on mt76-next: https://github.com/LorenzoBianconi/wireless-drivers-next/commits/mt76-devel

These also converting dev-leds to phy-leds,but can try yours too...which base do yours have?

ryderlee1110, 2022-12-06:
ah you can use his series. I will add mt7986 part based on his change if that works.

frank-w, 2022-12-06:
Have already tested ryders patch on top of your last version...works except led-off (interface down,hostapd/connection terminated,poweroff/sysfs). Would be nice if we can fix this too.

LorenzoBianconi, 2022-12-06:
Have already tested ryders patch on top of your last version...works
except led-off (interface down,hostapd/connection
terminated,poweroff/sysfs). Would be nice if we can fix this too.

I think the mac is controlling the led according to the transmitted
packet (register MT_TMAC_TCR0, mask MT_TMAC_TCR0_TX_BLINK).
@ryder: is it possible to overwrite/avoid this behaviour?

Regards,
Lorenzo

frank-w, 2022-12-07:
do we want to use led or leds? imho led makes more sense as we drive only one led at this position, so i have changed leds struct in mt76_phy (mt76.h) to "led", but need to change other occurences too, before able to test. or i change it in the mt7915_led_set_config here to first test and then we refactore it afterwards

ryderlee1110, 2022-12-07:
You can use leds as his series does. This is just my local version.

LorenzoBianconi, 2022-12-07:
>
> I'm thinking switching blink mode to pwm mode as there's no way to turn LED off under blink mode.

ack, I agree, I tested it and it seems to be working fine (all modes).

@Frank can you please test my latest mt76-devl branch?

@Ryder: do you want to add mt7996 too?

Regards,
Lorenzo

>
> @LorenzoBianconi @frank-w Can you test this change? I think we can go with /* set PWM mode */ for all scenario and get rid off all "blink settings" ( that is, drop if (delay_on/ delay_off) check )
>
> #define MT_LED_STATUS_0(_n) MT_LED_PHYS(0x20 + ((_n) * 8))
> #define MT_LED_STATUS_1(_n) MT_LED_PHYS(0x24 + ((_n) * 8))
> #define MT_LED_STATUS_OFF GENMASK(31, 24)
> #define MT_LED_STATUS_ON GENMASK(23, 16)
> #define MT_LED_STATUS_DURATION GENMASK(15, 0)
>
> static void mt7915_led_set_config(struct led_classdev *led_cdev,
>  u8 delay_on, u8 delay_off)
> {
> struct mt7915_dev *dev;
> struct mt76_phy *mphy;
> u32 val;
>
> mphy = container_of(led_cdev, struct mt76_phy, led.cdev);
> dev = container_of(mphy, struct mt7915_dev, mphy);
>
> /* select TX blink mode, 2: only data frames */
> if (delay_on && (delay_off != 0xff))
> mt76_rmw_field(dev, MT_TMAC_TCR0(mphy->band_idx), MT_TMAC_TCR0_TX_BLINK, 2);
>
> /* enable LED */
> mt76_wr(dev, MT_LED_EN(mphy->band_idx), 1);
>
> if (!delay_on && delay_off == 0xff) {
> /* set PWM mode */
> val = FIELD_PREP(MT_LED_STATUS_DURATION, 0xffff) |
>      FIELD_PREP(MT_LED_STATUS_OFF, delay_off) |
>      FIELD_PREP(MT_LED_STATUS_ON, delay_on);
> mt76_wr(dev, MT_LED_STATUS_0(mphy->band_idx), val);
> mt76_wr(dev, MT_LED_STATUS_1(mphy->band_idx), val);
> } else {
> /* set LED Tx blink on/off time */
> val = FIELD_PREP(MT_LED_TX_BLINK_ON_MASK, delay_on) |
>      FIELD_PREP(MT_LED_TX_BLINK_OFF_MASK, delay_off);
> mt76_wr(dev, MT_LED_TX_BLINK(mphy->band_idx), val);
> }
>
> /* control LED */
> val = MT_LED_CTRL_KICK;
> if (mphy->led.al)
> val |= MT_LED_CTRL_POLARITY;
>
> if (delay_on && (delay_off != 0xff))
> val |= MT_LED_CTRL_BLINK_MODE;
>
> mt76_wr(dev, MT_LED_CTRL(mphy->band_idx), val);
> mt76_clear(dev, MT_LED_CTRL(mphy->band_idx), MT_LED_CTRL_KICK);
> }

frank-w, 2022-12-07:
i have added ryders patch directly, everytime i use your branch i need to many other patches to get my board itself working again and there is no simple rebase possible.

i have pushed my current state (last change not yet tested, but compiled and on card - ready to test when i'm finished wih work)

https://github.com/frank-w/BPI-R2-4.14/commits/6.1-r3-mt76devel2

if you have not added other changes we should have same state in mt76

also splitted out mt7996 changes so that i can apply changes to 6.1 (where this chip is not there yet)

LorenzoBianconi, 2022-12-07:
> i have added ryders patch directly, everytime i use your branch i need to many other patches to get my board itself working again and there is no simple rebase possible.
>
> i have pushed my current [...]

actually we can get rid of tx blink mode and just rely on pwm mode.
Can you please try this version? It works fine for me.

https://github.com/LorenzoBianconi/wireless-drivers-next/commit/a4d784681eaadd4709149b489d0c51f70f8484f6

We just need to set active low in dts (but this is not related to the patch).

Regards,
Lorenzo

>
> also splitted out mt7996 changes so that i can apply changes to 6.1 (where this chip is not there yet)

frank, 2022-12-07:
> I'm thinking switching blink mode to pwm mode as there's no way to turn LED off under blink mode.
> ack, I agree, I tested it and it seems to be working fine (all modes). @frank can you please test my latest mt76-devl branch? @ryder: do you want to add mt7996 too? Regards, Lorenzo

I am not the frank you're looking for
[image]

frank-w, 2022-12-07:
@LorenzoBianconi can you point me to necessary dts-changes? in my dts i have only a group defined which is handled by pinctrl driver

```
        wf_led_pins: wf-led-pins {
                mux {
                        function = "led";
                        groups = "wifi_led";
                };
        };
```

made again rebase with my patches on your last state

https://github.com/frank-w/BPI-R2-4.14/commits/6.1-r3-mt76devel3

my dts is arch/arm64/boot/dts/mediatek/mt7986a-bananapi-bpi-r3.dts there

LorenzoBianconi, 2022-12-07:
> @LorenzoBianconi can you point me to necessary dts-changes? in my dts i have only a group defined which is handled by pinctrl driver
>
> ```
>         wf_led_pins: wf-led-pins {
>                 mux [...]

does it work properly?

> 
> https://github.com/frank-w/BPI-R2-4.14/commits/6.1-r3-mt76devel3
> 
> my dts is arch/arm64/boot/dts/mediatek/mt7986a-bananapi-bpi-r3.dts there

I think you would need to add something like in mt7986a-bananapi-bpi-r3.dts

`

&wmac {
	...

	led {
		led-active-low;
	};
};
`

but you can even enable it through debugfs..
echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/led_active_low

frank-w, 2022-12-07:
tried this, seems only working for 2g-led...5g is blinking but with base state off like it looks

on poweroff 2g-led stays on, and 5g off

LorenzoBianconi, 2022-12-07:
> tried this, seems only working for 2g-led...5g is blinking but with base state off like it looks
> 
> on poweroff 2g-led stays on, and 5g off

ack, right. Can you please try to add this simple [patch](https://pastebin.com/54xm8Hb4)?

frank-w, 2022-12-07:
thanks now it looks great...on when interface active/hostapd running, blinking on traffic of the right phy, off when interface goes down and also on poweroff :)

good job +1

frank-w, 2022-12-07:
"backported" (basicly remove mt7996 from first 2 patches) to 6.1:

https://github.com/frank-w/BPI-R2-4.14/commits/6.1-r3

ryderlee1110, 2022-12-07:
we can close this ticket.

closed 2022-12-07 by frank-w

lukjod, 2022-12-11:
Any chance to add this support for Asus rt-ax53u

LorenzoBianconi, 2022-12-11:
> Any chance to add this support for Asus rt-ax53u

I think it is running mt7915, so it should be already supported.

lukjod, 2022-12-11:
Besides lan leds and power led 2 wlan leds and usb led are not supported AFAIK

LorenzoBianconi, 2022-12-11:
> Besides lan leds and power led 2 wlan leds and usb led are not supported
> AFAIK
>
> <https://github.com/openwrt/mt76/issues/714#issuecomment-1345614895>, or
> unsubscribe
What is the mt76 version you are running? You need to run the latest
version (nbd updated the OpenWrt package Makefile yesterday I think) and
enable mt76 leds afair.

frank-w, 2022-12-11:
Have you defined the led pinctrl in your dts?

Working Example from my r3:

https://github.com/frank-w/BPI-R2-4.14/blob/6.1-r3/arch/arm64/boot/dts/mediatek/mt7986a-bananapi-bpi-r3.dts#L347

https://elixir.bootlin.com/linux/v6.0.11/source/drivers/pinctrl/mediatek/pinctrl-mt7986.c#L749

https://elixir.bootlin.com/linux/v6.0.11/source/drivers/pinctrl/mediatek/pinctrl-mt7986.c#L641

lukjod, 2022-12-11:
I'm on  SNAPSHOT r21438-8327e0fb72, which files shoud I change to make it work after install?

rozhuk-im, 2022-12-12:
> Besides lan leds and power led 2 wlan leds and usb led are not supported AFAIK

All leds supported https://github.com/openwrt/openwrt/pull/10400 except two WiFi leds.

frank-w, 2022-12-18:
Have you got it working for your device? Have not found technical information which Wifi chip is used on this device.

Btw. you have to enable mac80211_leds too.