openwrt/mt76 #716: rcu_sched detected stalls

graphine27, 2022-12-09:
Hi,

I have one AW7915-NPD and one AW7915-NP1 inside a clearfog gt 8k and I'm seeing some stalls/crashes usually during high traffic but can also happen constantly after some time. This also seems to happen with only one card as well.
Device becomes unresponsive and can only be recovered by removing power
I got the following log over serial

`root@OpenWrt:/# [  184.045603] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  184.051575] rcu:     3-...0: (1 ticks this GP) idle=572/1/0x4000000000000000 softirq=5076/5076 fqs=935
[  184.060757]  (detected by 0, t=2102 jiffies, g=7101, q=25)
[  184.066272] Task dump for CPU 3:
[  184.069521] task:mt76-tx phy0    state:R  running task     stack:    0 pid: 1452 ppid:     2 flags:0x0000000a
[  184.079497] Call trace:
[  184.081972]  __switch_to+0x9c/0xfc
[  184.085419]  __mt76_worker_fn+0x0/0x100 [mt76]
[  186.025411] mt7915e 0001:01:00.0: Message 00005aed (seq 10) timeout
[  206.503270] mt7915e 0001:01:00.0: Message 000026ed (seq 11) timeout
[  247.089964] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  247.095932] rcu:     3-...0: (1 ticks this GP) idle=572/1/0x4000000000000000 softirq=5076/5076 fqs=3727
[  247.105201]  (detected by 0, t=8407 jiffies, g=7101, q=43)
[  247.110716] Task dump for CPU 3:
[  247.113965] task:mt76-tx phy0    state:R  running task     stack:    0 pid: 1452 ppid:     2 flags:0x0000000a
[  247.123940] Call trace:
[  247.126413]  __switch_to+0x9c/0xfc
[  247.129853]  __mt76_worker_fn+0x0/0x100 [mt76]
[  310.136842] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  310.142811] rcu:     3-...0: (1 ticks this GP) idle=572/1/0x4000000000000000 softirq=5076/5076 fqs=6244
[  310.152081]  (detected by 2, t=14713 jiffies, g=7101, q=178)
[  310.157771] Task dump for CPU 3:
[  310.161018] task:mt76-tx phy0    state:R  running task     stack:    0 pid: 1452 ppid:     2 flags:0x0000000a
[  310.170994] Call trace:
[  310.173467]  __switch_to+0x9c/0xfc
[  310.176910]  __mt76_worker_fn+0x0/0x100 [mt76]`

ryderlee1110, 2022-12-12:
echo 1 > sys/kernel/debug/ieee80211/phy0/mt76/fw_debug_wm
cat sys/kernel/debug/ieee80211/phy0/mt76/sys_recovery

graphine27, 2022-12-13:
I updated to latest openwrt master to see if anything changed and now I'm not even able to join the wireless network.
However somehow the device still froze, but without the rcu stalls, only the previous timeout from wifi.
```

[  650.638802] ieee80211 phy0: WM: ( 643.954285:88:BSS-E)BSSInfo 0 Active
[  650.640162] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[  650.645468] ieee80211 phy0: WM: ( 643.954346:89:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[  650.652273] br-lan: port 5(phy0-ap0) entered blocking state
[  650.661590] ieee80211 phy0: WM: ( 643.954712:90:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[  650.667189] br-lan: port 5(phy0-ap0) entered forwarding state
[  927.391586] ieee80211 phy0: WM: ( 920.660492:91:BSS-E)BSSInfo 0 Active
[  927.398255] ieee80211 phy0: WM: ( 920.660553:92:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[  927.407744] ieee80211 phy0: WM: ( 920.660889:93:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[  937.337048] ieee80211 phy0: WM: ( 930.604279:94:BSS-E)BSSInfo 0 Active
[  937.343724] ieee80211 phy0: WM: ( 930.604340:95:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[  937.353199] ieee80211 phy0: WM: ( 930.604706:96:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[  947.324193] ieee80211 phy0: WM: ( 940.589753:97:BSS-E)BSSInfo 0 Active
[  947.330881] ieee80211 phy0: WM: ( 940.589814:98:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[  947.340354] ieee80211 phy0: WM: ( 940.602723:99:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[  957.290994] ieee80211 phy0: WM: ( 950.554871:00:BSS-E)BSSInfo 0 Active
[  957.297677] ieee80211 phy0: WM: ( 950.554932:01:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[  957.307142] ieee80211 phy0: WM: ( 950.564087:02:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[  972.260568] ieee80211 phy0: WM: ( 965.521912:03:BSS-E)BSSInfo 0 Active
[  972.267241] ieee80211 phy0: WM: ( 965.522004:04:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[  972.276695] ieee80211 phy0: WM: ( 965.522339:05:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[  978.219516] ieee80211 phy0: WM: ( 971.479859:06:BSS-E)BSSInfo 0 Active
[  978.226199] ieee80211 phy0: WM: ( 971.479920:07:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[  978.235667] ieee80211 phy0: WM: ( 971.480286:08:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[  984.180871] ieee80211 phy0: WM: ( 977.440217:09:BSS-E)BSSInfo 0 Active
[  984.187554] ieee80211 phy0: WM: ( 977.440308:10:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[  984.197063] ieee80211 phy0: WM: ( 977.453248:11:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[ 1021.440139] device phy0-ap0 left promiscuous mode
[ 1021.444938] br-lan: port 5(phy0-ap0) entered disabled state
[ 1021.790883] ieee80211 phy0: WM: (1015.043915:12:BSS-E)bssDisableBcnOffloadFunc, BSS (0) Inactived
[ 1021.799836] ieee80211 phy0: WM: (1015.045044:13:CMD-E)Rx Hdr Translation enable=0, CheckBssid=0, InsVlan=0, RemVlan=0, QosTid=0, Mode=0
[ 1022.199875] ieee80211 phy0: WM: (1015.452851:14:CMD-E)Rx Hdr Translation enable=1, CheckBssid=0, InsVlan=0, RemVlan=0, QosTid=0, Mode=0
[ 1022.212119] ieee80211 phy0: WM: (1015.452912:15:CMD-E)Rx Hdr Translation BL Count=1
[ 1022.219898] ieee80211 phy0: WM: (1015.452942:16:CMD-E)Rx Hdr Translation BL setting idx=0, enable=1, ether-type=0x888e
[ 1022.224953] br-lan: port 5(phy0-ap0) entered blocking state
[ 1022.230744] ieee80211 phy0: WM: (1015.473755:17:CMD-S)ProtectCtrl!! Protection Index = 1, DBDC = 0
[ 1022.230805] ieee80211 phy0: WM: (1015.474915:18:BSS-E)BSSInfo 0 Active
[ 1022.236357] br-lan: port 5(phy0-ap0) entered disabled state
[ 1022.245356] ieee80211 phy0: WM: (1015.475007:19:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[ 1022.245401] ieee80211 phy0: WM: (1015.475342:20:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[ 1022.252272] device phy0-ap0 entered promiscuous mode
[ 1022.281596] br-lan: port 5(phy0-ap0) entered blocking state
[ 1022.287206] br-lan: port 5(phy0-ap0) entered forwarding state
[ 1022.800752] br-lan: port 5(phy0-ap0) entered disabled state
[ 1041.959616] ieee80211 phy0: WM: (1035.209229:21:RDM-E)[rdmRddStart] RDD HW address, g_au4PreviousHwAddr[0] = 0x416000
[ 1103.214438] ieee80211 phy0: WM: (1096.453766:22:CAL-S)DFS Cal Semaphore !!!
[ 1103.937259] ieee80211 phy0: WM: (1097.176484:23:RDM-E)[rdmRddStart] RDD HW address, g_au4PreviousHwAddr[0] = 0x416000
[ 1103.939828] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[ 1103.947942] ieee80211 phy0: WM: (1097.177277:24:BSS-E)BSSInfo 0 Active
[ 1103.954645] br-lan: port 5(phy0-ap0) entered blocking state
[ 1103.961254] ieee80211 phy0: WM: (1097.177338:25:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[ 1103.966868] br-lan: port 5(phy0-ap0) entered forwarding state
[ 1103.982094] ieee80211 phy0: WM: (1097.177674:26:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[ 1124.945800] mt7915e 0000:01:00.0: Message 00005aed (seq 6) timeout
```

ryderlee1110, 2022-12-13:
cat sys/kernel/debug/ieee80211/phy0/mt76/sys_recovery

graphine27, 2022-12-13:
I can't run any command over ssh/uart  when this happens, I have to remove power from the device so I can only provide that log when everything is ok
```
Please echo the correct value ...
0: grab firmware transient SER state
1: trigger system error L1 recovery
2: trigger system error L2 recovery
3: trigger system error L3 rx abort
4: trigger system error L3 tx abort
5: trigger system error L3 tx disable
6: trigger system error L3 bf recovery
7: trigger system error full recovery
8: trigger firmware crash

let's dump firmware SER statistics...
::E  R , SER_STATUS        = 0x00000000
::E  R , SER_PLE_ERR       = 0x00000000
::E  R , SER_PLE_ERR_1     = 0x00000000
::E  R , SER_PLE_ERR_AMSDU = 0x00000000
::E  R , SER_PSE_ERR       = 0x00000000
::E  R , SER_PSE_ERR_1     = 0x00000000
::E  R , SER_LMAC_WISR6_B0 = 0x00000000
::E  R , SER_LMAC_WISR6_B1 = 0x00000000
::E  R , SER_LMAC_WISR7_B0 = 0x00000000
::E  R , SER_LMAC_WISR7_B1 = 0x00000000

SYS_RESET_COUNT: WM 0, WA 0

```

graphine27, 2022-12-13:
It seems that with latest openwrt master I can't join wifi from my phone (ios and android) but it works from a macbook

Here's another log when the device got stuck

```
[  856.994054] ieee80211 phy0: WM: ( 850.286042:62:BSS-E)_whCapSetGeneric_Falcon DW0 = 0x0, DW1 = 0x0
[  857.003093] ieee80211 phy0: WM: ( 850.286438:63:TXC-E)heACtrlInitStaRec(): prOperMode=401270, u2OperMode=49!
[  857.013001] ieee80211 phy0: WM: ( 850.286469:64:BSS-E)muruAddStaRec, prRuStaRec->u1Bw = 1 eBand = 5000000 BSS BW = 1
[  923.838920] device phy0-ap0 left promiscuous mode
[  923.843718] br-lan: port 5(phy0-ap0) entered disabled state
[  924.171600] ieee80211 phy0: WM: ( 917.452424:65:BSS-E)bssDisableBcnOffloadFunc, BSS (0) Inactived
[  924.180549] ieee80211 phy0: WM: ( 917.453553:66:CMD-E)Rx Hdr Translation enable=0, CheckBssid=0, InsVlan=0, RemVlan=0, QosTid=0, Mode=0
[  924.658094] ieee80211 phy0: WM: ( 917.938843:67:CMD-E)Rx Hdr Translation enable=1, CheckBssid=0, InsVlan=0, RemVlan=0, QosTid=0, Mode=0
[  924.662116] br-lan: port 5(phy0-ap0) entered blocking state
[  924.670342] ieee80211 phy0: WM: ( 917.938904:68:CMD-E)Rx Hdr Translation BL Count=1
[  924.675953] br-lan: port 5(phy0-ap0) entered disabled state
[  924.683737] ieee80211 phy0: WM: ( 917.938935:69:CMD-E)Rx Hdr Translation BL setting idx=0, enable=1, ether-type=0x888e
[  924.689453] device phy0-ap0 entered promiscuous mode
[  924.700168] ieee80211 phy0: WM: ( 917.939118:70:CMD-S)ProtectCtrl!! Protection Index = 1, DBDC = 0
[  924.705220] br-lan: port 5(phy0-ap0) entered blocking state
[  924.714160] ieee80211 phy0: WM: ( 917.940308:71:BSS-E)BSSInfo 0 Active
[  924.714166] ieee80211 phy0: WM: ( 917.940369:72:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[  924.719762] br-lan: port 5(phy0-ap0) entered forwarding state
[  924.726410] ieee80211 phy0: WM: ( 917.940735:73:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[  925.181387] br-lan: port 5(phy0-ap0) entered disabled state
[  925.506828] ieee80211 phy0: WM: ( 918.787415:74:BSS-E)BSSInfo 0 Active
[  925.508586] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[  925.513505] ieee80211 phy0: WM: ( 918.787507:75:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[  925.513515] ieee80211 phy0: WM: ( 918.787842:76:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[  925.539026] br-lan: port 5(phy0-ap0) entered blocking state
[  925.544633] br-lan: port 5(phy0-ap0) entered forwarding state
[  927.433184] ieee80211 phy0: WM: ( 920.713471:77:BSS-E)_whCapSetGeneric_Falcon DW0 = 0x0, DW1 = 0x0
[  927.442204] ieee80211 phy0: WM: ( 920.713868:78:TXC-E)heACtrlInitStaRec(): prOperMode=401270, u2OperMode=51!
[  927.452093] ieee80211 phy0: WM: ( 920.713929:79:BSS-E)muruAddStaRec, prRuStaRec->u1Bw = 2 eBand = 5000000 BSS BW = 2
```

ryderlee1110, 2022-12-13:
echo 0 > sys/kernel/debug/ieee80211/phy0/mt76/sys_recovery
cat sys/kernel/debug/ieee80211/phy0/mt76/sys_recovery

graphine27, 2022-12-16:
The problem is I can't do anything or run any command with the device once it's stuck, not even over uart, all I can do is remove power.

graphine27, 2022-12-19:
Regarding the issue when I can't join the network from my phone, I noticed these logs. After the abort message I was able to join the network.
```
[ 5026.148527] ieee80211 phy0: WM: ( 723.738088:60:SER-E)BN0 TRB prt corruption!!
[ 5026.155815] ieee80211 phy0: WM: ( 723.738149:61:SER-W)[SER]SerL3RxAbort done code:0
[ 5041.107309] ieee80211 phy0: WM: ( 738.694387:62:SER-E)BN0 TRB prt corruption!!
[ 5041.114603] ieee80211 phy0: WM: ( 738.694448:63:SER-W)[SER]SerL3RxAbort done code:0
[ 5663.386421] ieee80211 phy0: WM: (1360.870168:64:BSS-E)_whCapSetGeneric_Falcon DW0 = 0x0, DW1 = 0x0
[ 5663.395765] ieee80211 phy0: WM: (1360.879537:65:TXC-E)heACtrlInitStaRec(): prOperMode=401270, u2OperMode=51!
```

ryderlee1110, 2022-12-19:
how to repro? Does it recover when you echo 1 or 7 > sys/kernel/debug/ieee80211/phy0/mt76/sys_recovery ?

graphine27, 2022-12-19:
Unfortunately I can not interact with the device as soon as it gets stuck.

So I'm facing 2 different issues right now.
First one, I can no longer join the network from iPhone or Android phone, however it works from a macbook. This likely happen after [these commits](https://github.com/openwrt/openwrt/commit/b1b29ba98769386f7f88ef66fabc88c79be56b33). @nbd168 any idea what might cause this? Trying to obtain some logs from android

Second issue, the rcu_sched stall. This happens pretty consistently, usually under traffic load. It can happen **way** more often if I plug another mt7915 DBDC card. I had this exact mt7915 card in a BPI-R64 and it works perfectly there. Maybe you can reproduce by using this card **not** with a Mediatek SoC. This one is based on Armada 8040

This is the config I used to build the image from this branch https://github.com/graphine27/openwrt/tree/clearfog-gt-8k
[configmvebu.txt](https://github.com/openwrt/mt76/files/10255135/configmvebu.txt)
This is the wireless config 
```
root@OpenWrt:/# uci show wireless
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='cp0/f2600000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
wireless.radio0.channel='36'
wireless.radio0.band='5g'
wireless.radio0.htmode='HE80'
wireless.radio0.cell_density='0'
wireless.radio0.country='RO'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.encryption='none'

```

graphine27, 2022-12-19:
Here's a log from android when trying to join and failing

```
12-19 03:33:54.464  5909  5909 I wpa_supplicant: wlan0: Trying to associate with SSID 'mt76'
12-19 03:33:54.464  5909  5909 I wpa_supplicant: WPA: Add samsung vsie for support Cisco Specific features
12-19 03:33:54.464  5909  5909 I wpa_supplicant: WPA: Set adaptive_11r support bit
12-19 03:33:54.467  1464  5773 D WifiClientModeImpl[wlan0]: ClientModeImpl: Leaving Connected state
12-19 03:33:54.467  1464  5773 D WifiClientModeImpl[wlan0]: Start Connecting Watchdog 25
12-19 03:33:54.468  1464  1994 D BatteryExternalStatsWorker: update stats : wifi-state (2)
12-19 03:33:54.468  1464  5773 I WifiClientModeImpl: Keep PMK cache for network disconnection reason 3
12-19 03:33:54.470  1464  5773 E WifiVendorHal: stopRssiMonitoring(l.2789) failed {.code = ERROR_NOT_AVAILABLE, .description = }
12-19 03:33:54.470  5909  5909 I wpa_sslib: Recv DRIVER command: BTCOEXMODE 2
12-19 03:33:54.471  1464  5867 D WifiConnectivityMonitor.WifiScoreMediator: WifiScoreMediator onStop with sessionId : 1152
12-19 03:33:54.471  6229  7286 D ApfFilter: (wlan0): shutting down
```

ryderlee1110, 2022-12-19:
I can only test it with MTK platform but can't repro.

graphine27, 2022-12-21:
Here's another stall, they all seem to be triggered by **Message 00005aed (seq 9) timeout**

```
root@OpenWrt:/# [88937.005139] mt7915e 0000:01:00.0: Message 00005aed (seq 9) ti                                                                                                                                                             meout
[88992.040892] rcu: INFO: rcu_sched detected expedited stalls on CPUs/tasks: { 0                                                                                                                                                             -... 2-... 3-... } 6368 jiffies s: 149 root: 0xd/.
[88992.052433] rcu: blocking rcu_node structures (internal RCU debug):
[88992.058725] Task dump for CPU 0:
[88992.061967] task:swapper/0       state:R  running task     stack:    0 pid:                                                                                                                                                                 0 ppid:     0 flags:0x0000000a
[88992.071929] Call trace:
[88992.074384]  __switch_to+0xd4/0x114
[88992.077893]  0xffffffc008ab3d60
[88992.081049] Task dump for CPU 2:
[88992.084288] task:swapper/2       state:R  running task     stack:    0 pid:                                                                                                                                                                 0 ppid:     1 flags:0x0000000a
[88992.094248] Call trace:
[88992.096702]  __switch_to+0xd4/0x114
[88992.100205]  0xffffffc008cd3de0
[88992.103359] Task dump for CPU 3:
[88992.106598] task:swapper/3       state:R  running task     stack:    0 pid:                                                                                                                                                                 0 ppid:     1 flags:0x0000000a
[88992.116557] Call trace:
[88992.119011]  __switch_to+0xd4/0x114
[88992.122515]  0xffffffc008cdbde0
[89075.564441] rcu: INFO: rcu_sched self-detected stall on CPU
[89075.570037] rcu:     0-....: (16127 ticks this GP) idle=67b/0/0x7 softirq=135                                                                                                                                                             504/135505 fqs=2783
[89075.578689]  (t=6000 jiffies g=237845 q=25)
[89075.582889] Task dump for CPU 0:
[89075.586127] task:swapper/0       state:R  running task     stack:    0 pid:                                                                                                                                                                 0 ppid:     0 flags:0x0000000a
[89075.596085] Call trace:
[89075.598539]  dump_backtrace+0x0/0x170
[89075.602218]  show_stack+0x14/0x30
[89075.605545]  sched_show_task+0x13c/0x170
[89075.609485]  dump_cpu_task+0x40/0x50
[89075.613075]  rcu_dump_cpu_stacks+0xe0/0x124
[89075.617275]  rcu_sched_clock_irq+0x67c/0x814
[89075.621562]  update_process_times+0x88/0xd0
[89075.625761]  tick_sched_timer+0x74/0x210
[89075.629700]  __hrtimer_run_queues+0x114/0x224
[89075.634074]  hrtimer_interrupt+0x10c/0x2e0
[89075.638187]  arch_timer_handler_phys+0x30/0x40
[89075.642649]  handle_percpu_devid_irq+0x80/0x130
[89075.647198]  handle_domain_irq+0x5c/0x90
[89075.651136]  gic_handle_irq+0x70/0x9c
[89075.654813]  do_interrupt_handler+0x30/0x54
[89075.659013]  el1_interrupt+0x2c/0x4c
[89075.662602]  el1h_64_irq_handler+0x14/0x20
[89075.666715]  el1h_64_irq+0x78/0x7c
[89075.670129]  queued_spin_lock_slowpath+0x198/0x2e0
[89075.674939]  dev_queue_xmit+0x10/0x20
[89075.678617]  br_dev_queue_push_xmit+0x70/0x154
[89075.683079]  br_forward_finish+0x90/0xac
[89075.687016]  __br_forward+0xcc/0x140
[89075.690605]  br_forward+0x124/0x144
[89075.694107]  br_dev_xmit+0x204/0x3f0
[89075.697697]  dev_hard_start_xmit+0xd0/0x150
[89075.701896]  __dev_queue_xmit+0x998/0xc50
[89075.705921]  dev_queue_xmit+0x10/0x20
[89075.709596]  ip6_finish_output2+0x250/0x5e0
[89075.713797]  __ip6_finish_output+0x148/0x260
[89075.718083]  ip6_finish_output+0x44/0xe0
[89075.722021]  ip6_output+0x5c/0x144
[89075.725435]  ip6_forward+0x4cc/0x934
[89075.729025]  ipv6_rcv+0xa4/0xdc
[89075.732178]  __netif_receive_skb_one_core+0x44/0x50
[89075.737075]  process_backlog+0xdc/0x230
[89075.740926]  __napi_poll+0x54/0x1b0
[89075.744427]  net_rx_action+0xe8/0x26c
[89075.748104]  _stext+0x10c/0x288
[89075.751256]  irq_exit+0x80/0xac
[89075.754410]  handle_domain_irq+0x60/0x90
[89075.758348]  gic_handle_irq+0x70/0x9c
[89075.762024]  call_on_irq_stack+0x28/0x44
[89075.765962]  do_interrupt_handler+0x4c/0x54
[89075.770161]  el1_interrupt+0x2c/0x4c
[89075.773750]  el1h_64_irq_handler+0x14/0x20
[89075.777863]  el1h_64_irq+0x78/0x7c
[89075.781277]  arch_cpu_idle+0x14/0x20
[89075.784867]  default_idle_call+0x1c/0x68
[89075.788806]  do_idle+0x1b0/0x1e0
[89075.792048]  cpu_startup_entry+0x20/0x60
[89075.795986]  rest_init+0xc4/0xd0
[89075.799225]  arch_call_rest_init+0xc/0x14
[89075.803249]  start_kernel+0x614/0x634
[89075.806925]  __primary_switched+0xa0/0xa8
[89075.810949] Task dump for CPU 2:
[89075.814188] task:swapper/2       state:R  running task     stack:    0 pid:                                                                                                                                                                 0 ppid:     1 flags:0x0000000a
[89075.824145] Call trace:
[89075.826599]  __switch_to+0xd4/0x114
[89075.830101]  0xffffffc008cd3de0
[89075.833254] Task dump for CPU 3:
[89075.836492] task:swapper/3       state:R  running task     stack:    0 pid:                                                                                                                                                                 0 ppid:     1 flags:0x0000000a
[89075.846449] Call trace:
[89075.848901]  __switch_to+0xd4/0x114
[89075.852404]  0xffffffc008cdbde0
```

graphine27, 2022-12-21:
I have found something interesting
When the router is started, I can not join the network from my phones, but it works from a macbook. It seems that after the macbook connects, the phones can also connect. Here are the logs

```
[   33.445435] ieee80211 phy0: WM: (  26.016633:48:RDM-E)[rdmRddStart] RDD HW address, g_au4PreviousHwAddr[0] = 0x416000
[   33.456132] ieee80211 phy0: WM: (  26.017090:49:RDM-E)[rdmRddStart] RDD HW address, g_au4PreviousHwAddr[1] = 0x416200
[   95.692252] ieee80211 phy0: WM: (  88.252808:50:CAL-S)DFS Cal Semaphore !!!
[   96.123454] ieee80211 phy0: WM: (  88.683930:51:CAL-S)halPhyCalReloadFdFi: Reload SX0-WF0/1 RXFD/FI !
[   96.132751] ieee80211 phy0: WM: (  88.683960:52:CAL-S)G[2] Sx[0] Ant[0] Freq[5210000] Bw[3]
[   96.302881] ieee80211 phy0: WM: (  88.863343:53:RDM-E)[rdmRddStart] RDD HW address, g_au4PreviousHwAddr[0] = 0x416000
[   96.305523] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[   96.313551] ieee80211 phy0: WM: (  88.863770:54:RDM-E)[rdmRddStart] RDD HW address, g_au4PreviousHwAddr[1] = 0x416200
[   96.313558] ieee80211 phy0: WM: (  88.864472:55:BSS-E)BSSInfo 0 Active
[   96.313566] ieee80211 phy0: WM: (  88.864533:56:BSS-E)ucDbdcIdx 0, ucOwnMacIdx 4, ucHetbRU26Disable 0
[   96.320298] br-lan: port 6(phy0-ap0) entered blocking state
[   96.330873] ieee80211 phy0: WM: (  88.864869:57:BSS-E)_whCapSetGeneric_Falcon DW0 = 0xe0000, DW1 = 0x0
[   96.361911] br-lan: port 6(phy0-ap0) entered forwarding state
```
**at this point I was trying to join from the phone but could not, nothing in logs**
**trying to join from macbook, I can see some logs**
```
[  182.841824] ieee80211 phy0: WM: ( 175.387482:58:BSS-E)_whCapSetGeneric_Falcon DW0 = 0x0, DW1 = 0x0
[  182.850866] ieee80211 phy0: WM: ( 175.387879:59:TXC-E)heACtrlInitStaRec(): prOperMode=401270, u2OperMode=51!
[  182.860775] ieee80211 phy0: WM: ( 175.387940:60:BSS-E)muruAddStaRec, prRuStaRec->u1Bw = 2 eBand = 5000000 BSS BW = 3
[  208.857380] ieee80211 phy0: WM: ( 201.398591:61:BSS-E)_whCapSetGeneric_Falcon DW0 = 0x0, DW1 = 0x0
```
**macbook joined**
**trying to join from phone**
```
[  208.866672] ieee80211 phy0: WM: ( 201.407898:62:TXC-E)heACtrlInitStaRec(): prOperMode=401308, u2OperMode=51!
```
**phone successfully joined the network**

@ryderlee1110 @nbd168 any idea why this happens on openwrt master?

ryderlee1110, 2022-12-21:
macbook works well ... what phone you're using?

graphine27, 2022-12-21:
Tried on iphone 8 and samsung a52 5g
How do I enable hostapd log? Thanks

ryderlee1110, 2022-12-21:
-d show more debug messages (-dd for even more)

Have you tested your BPI-R64?

graphine27, 2022-12-21:
R64 runs a old build, but I never had any issues with it such as rcu stalls

ryderlee1110, 2022-12-21:
Are u able to upgrade to the latest one for cross checking?

graphine27, 2022-12-23:
I will try
There was a power outage here and when it came back it suddenly works to join the network from the first try on my phone.
Maybe related to other wifi APs also getting rebooted and maybe switched channels.

So now only the rcu_sched issue is remaining

rany2, 2023-05-22:
After calling 7 in an attempt to recover without a reboot, a napi/phy0-9 is spinning non-stop using up an entire core and this is found in dmesg:

```
[  362.758261] mt7915e 0000:02:00.0: Message 000027ed (seq 2) timeout
[  382.158052] rcu: INFO: rcu_sched self-detected stall on CPU
[  382.163654] rcu: 	1-....: (1 GPs behind) idle=60b/1/0x40000002 softirq=4573/4574 fqs=2997 
[  382.171896] 	(t=6001 jiffies g=5809 q=17874)
[  382.176160] NMI backtrace for cpu 1
[  382.179633] CPU: 1 PID: 715 Comm: napi/phy0-9 Not tainted 5.15.112 #0
[  382.186055] Stack : 00000000 80085cd8 00000000 00000004 00000000 00000000 8140fd14 80a30000
[  382.194412]         80870000 8079db0c 81cd23d8 80874e03 00000000 00000001 8140fcc0 81450660
[  382.202771]         00000000 00000000 8079db0c 8140fb60 ffffefff 00000000 ffffffea 00000000
[  382.211130]         8140fb6c 0000088d 8087ab00 ffffffff 8079db0c 00000000 00000001 00000000
[  382.219489]         00000001 808722fc 80870000 80872300 00000018 80416fa8 00000004 80a30004
[  382.227849]         ...
[  382.230288] Call Trace:
[  382.232726] [<8000812c>] show_stack+0x28/0xf0
[  382.237098] [<80382034>] dump_stack_lvl+0x60/0x80
[  382.241802] [<80388e20>] nmi_cpu_backtrace+0x108/0x178
[  382.246931] [<80388fd8>] nmi_trigger_cpumask_backtrace+0x148/0x178
[  382.253089] [<800a09a4>] rcu_dump_cpu_stacks+0x158/0x1ac
[  382.258413] [<800a12e8>] rcu_sched_clock_irq+0x800/0x9d0
[  382.263712] [<800a79c0>] update_process_times+0xc8/0x124
[  382.269017] [<800bc5e0>] tick_handle_periodic+0x34/0xc8
[  382.274247] [<804e5c9c>] gic_compare_interrupt+0x7c/0x9c
[  382.279564] [<8008e6bc>] handle_percpu_devid_irq+0xbc/0x188
[  382.285136] [<80087d8c>] generic_handle_domain_irq+0x2c/0x44
[  382.290776] [<8039f2a0>] gic_handle_local_int+0xa4/0x110
[  382.296079] [<8039f31c>] gic_irq_dispatch+0x10/0x20
[  382.300941] [<80087cec>] handle_irq_desc+0x20/0x38
[  382.305718] [<806eeff8>] do_domain_IRQ+0x3c/0x50
[  382.310326] [<8039e5dc>] plat_irq_dispatch+0x98/0xcc
[  382.315277] [<80003568>] except_vec_vi_end+0xb8/0xc4
[  382.320230] [<82ff01a4>] mt76_mmio_init+0xf8/0x144 [mt76]
[  382.325647] [<830ae090>] mt7915_mac_wtbl_lmac_addr+0x1b0/0x820 [mt7915e]
[  382.332360] [<830b0000>] mt7915_mac_reset_work+0x274/0xc90 [mt7915e]

[  383.238298] mt7915e 0000:02:00.0: Message 000026ed (seq 3) timeout
[  403.718222] mt7915e 0000:02:00.0: Message 00005aed (seq 4) timeout
[  424.198463] mt7915e 0000:02:00.0: Message 00005aed (seq 5) timeout
[  424.204960] mt7915e 0000:02:00.0: phy0 SER recovery state: 0xc0000020
[  424.211443] mt7915e 0000:02:00.0: phy0 indicated firmware crash, attempting recovery
[  444.678527] mt7915e 0000:02:00.0: Message 000026ed (seq 7) timeout
[  465.158530] mt7915e 0000:02:00.0: Message 000036ed (seq 8) timeout
[  485.638533] mt7915e 0000:02:00.0: Message 000027ed (seq 9) timeout
[  506.118561] mt7915e 0000:02:00.0: Message 000026ed (seq 10) timeout
[  526.598361] mt7915e 0000:02:00.0: Message 00005aed (seq 11) timeout
[  547.078475] mt7915e 0000:02:00.0: Message 000036ed (seq 12) timeout
[  547.085105] ------------[ cut here ]------------
[  547.089930] WARNING: CPU: 0 PID: 32 at backports-6.1.24/net/mac80211/util.c:2301 ieee80211_parse_p2p_noa+0x1fc/0x358 [mac80211]
[  547.102596] Modules linked in: wireguard mt7915e(-) mt76_connac_lib mt76 mac80211 libchacha20poly1305 cfg80211 poly1305_mips libcurve25519_generic libcrc32c hwmon compat chacha_mips ledtrig_usbport ip_tunnel vxlan udp_tunnel ip6_udp_tunnel sha512_generic seqiv jitterentropy_rng drbg hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
[  547.140031] CPU: 0 PID: 32 Comm: kworker/0:1 Not tainted 5.15.112 #0
[  547.146513] Workqueue: events_freezable 0x83000208
[  547.151959] Stack : 846965ac 80085cd8 00000000 00000004 8079db0c 81513c74 814fbd00 8004ce34
[  547.160605]         80870000 81513cb5 00000001 81513ac0 83000208 00000001 81513c80 81452ca0
[  547.169241]         00000000 00000000 8079db0c 81513ac0 ffffefff 00000000 ffffffea 00000000
[  547.177865]         81513acc 000008b9 8087ab00 ffffffff 8079db0c 00000001 00000000 83046a2c
[  547.186483]         00000009 83087a68 82eb8a88 00000004 00000018 80416fa8 00000000 80a30000
[  547.195108]         ...
[  547.197631] Call Trace:
[  547.200106] [<8000812c>] show_stack+0x28/0xf0
[  547.204672] [<80382034>] dump_stack_lvl+0x60/0x80
[  547.209535] [<8002f668>] __warn+0x9c/0x124
[  547.213823] [<8002f74c>] warn_slowpath_fmt+0x5c/0xac
[  547.218897] [<83046a2c>] ieee80211_parse_p2p_noa+0x1fc/0x358 [mac80211]

[  547.228352] ---[ end trace 18e6e2d697907727 ]---
[  562.368149] rcu: INFO: rcu_sched self-detected stall on CPU
[  562.373753] rcu: 	1-....: (1 GPs behind) idle=60b/1/0x40000002 softirq=4573/4574 fqs=11990 
[  562.382080] 	(t=24022 jiffies g=5809 q=61593)
[... 23 lines cut ...]
[  562.517358] [<806eeff8>] do_domain_IRQ+0x3c/0x50
[  562.521965] [<8039e5dc>] plat_irq_dispatch+0x98/0xcc
[  562.526916] [<80003568>] except_vec_vi_end+0xb8/0xc4
[  562.531867] [<82ff01a4>] mt76_mmio_init+0xf8/0x144 [mt76]
[  562.537284] [<830ae090>] mt7915_mac_wtbl_lmac_addr+0x1b0/0x820 [mt7915e]
[  562.543996] [<830b0000>] mt7915_mac_reset_work+0x274/0xc90 [mt7915e]

[  607.238125] rcu: INFO: rcu_sched detected expedited stalls on CPUs/tasks: { 1-... } 6001 jiffies s: 141 root: 0x2/.
[  607.249073] rcu: blocking rcu_node structures (internal RCU debug):
[  607.255417] Task dump for CPU 1:
[... cut ...]
[  607.295451]         82d3de00 806e9a80 00000000 00000000 82d3de00 82d3d200 8155fad4 80053164
[  607.304317]         00000000 00000000 80053024 82d3de00 810033e0 8155fad4 80053024 82d3de00
[  607.313110]         ...
[  607.315658] Call Trace:
[  607.318234] [<806e9280>] __schedule+0x3f0/0x784
[  607.323033] [<806e9674>] schedule+0x60/0xd0
[  607.327305] [<805290f0>] napi_threaded_poll+0xb4/0x114
[  607.332702] [<80053164>] kthread+0x140/0x164
[  607.337165] [<80003058>] ret_from_kernel_thread+0x14/0x1c

```

lukasz1992, 2023-05-22:
this is because of irqlock not being released