openwrt/mt76 #718: MT7612E 5GHz LED not working

EddieTH, 2022-12-17:
Device Afoundry EW1200 SOC MediaTek MT7621A, WiFi MediaTek MT7603E, MediaTek MT7612E - 5GHz The led does not respond in any way.

System path is visible (/sys/class/leds/mt76-phy1/) but there is no response to 'echo 0 > brightness' or 'echo 1 > brightness'

The led should be active in the low state (at least the one from 2.4GHz works that way).

Of course, on a stock system it works.

dfateyev, 2022-12-17:
You can check commits in https://github.com/openwrt/openwrt/pull/11220. It looks like your device is similar to ZBT WE1326 discussed there (MT7603E+MT7612E).
The discussion there is mostly about MT7603E proper initialization, but commits provide DTS LED fixes for "kernel-5.15" from master branch.

EddieTH, 2022-12-17:
Thank you for the hint. 
In the meantime, I tried all versions of openwrt since my device is supported (from 18 to current) and it seems that this led **never worked**.
The radio itself works fine only the LED does not work.

dfateyev, 2022-12-17:
I suspect that LED configuration [mt7621_afoundry_ew1200.dts](https://github.com/openwrt/openwrt/blob/openwrt-22.03/target/linux/ramips/dts/mt7621_afoundry_ew1200.dts#L27) is broken — @DragonBluep could probably help

EddieTH, 2022-12-18:
I could not find any entries for LED wifi in this file. I suspect they are managed through the mt76 driver.

DragonBluep, 2022-12-29:
Usually we need to define the GPIO and polarity of the LED in the dts.
Please try to add these properties to `pcie1` node:
```
		led {
			led-sources = <2>;
			led-active-low;
		};
```

EddieTH, 2022-12-29:
Thank you @DragonBluep. Is it possible to modify a working system without compilation?

DragonBluep, 2022-12-30:
@EddieTH No, you have to recompile the source code.

EddieTH, 2022-12-30:
Ok, after many hours of compilation I have a working system but led still doesn't work :-(

DragonBluep, 2022-12-30:
Maybe you can try `led-sources = <0>;` or `led-sources = <1>;` or some other values.

EddieTH, 2022-12-30:
I noticed an interesting thing.... Adding the above entry broke the operation of the LED from the 2.4GHz radio! 
So it comes out that radio0 (mt76-phy0) is **pcie1** not **pcie0** and vice versa radio1 (mt76-phy1) is pcie0!
The final working configuration for this router is:
`&pcie0 {
	mt76@0,0 {
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x8000>;
		ieee80211-freq-limit = <5000000 6000000>;
		led {
		    led-sources = <2>;
		    led-active-low;
		    };
	};
};
&pcie1 {
	mt76@0,0 {
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x0000>;
		ieee80211-freq-limit = <2400000 2500000>;
		led {
		    led-active-low;
		    };
	};
};
`
Thank you @DragonBluep for your help, without you I wouldn't have gotten to this point! I think you need to update the dts file in the repository so that all owners of this hardware can fully enjoy it.

dfateyev, 2022-12-30:
@EddieTH have you tested these changes for kernel-5.15 from master? There can be some difference in DTS configuration between kernel-5.10 and kernel-5.15.
Also, https://github.com/openwrt/openwrt/issues/11530 is a duplicate, and can be closed.

EddieTH, 2022-12-30:
No, I compiled with kernel 5.10.146
I am a very beginner in matters of self compilation of openwrt and there are many things I don't know yet.

EddieTH, 2022-12-31:
Ok. I compiled the master system with kernel 5.15.85 and everything works fine.
Unfortunately, I completely don't know how to do a pull request properly. Compiling openwrt, git, etc is a total novelty for me. I'm attaching the modified dts file for the Afoundry EW1200 here for interested parties to use (I've added a .txt extension), and maybe some more knowledgeable person will include it in the project.

Thanks again @dfateyev and @DragonBluep  for your help!

[mt7621_afoundry_ew1200.dts.txt](https://github.com/openwrt/mt76/files/10327321/mt7621_afoundry_ew1200.dts.txt)

closed 2022-12-31 by EddieTH