openwrt/mt76 #720: 2022-11-23 firmware file changes break DBDC on mt7916

Hurricos, 2022-12-18:
Hi there,

Recent (Nov 2022) changes to MT7916 firmware files, e.g. in linux-firmware commit cdf9499cf6706b73935a139bb5e76777a87674fe ([link](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/?id=cdf9499cf6706b73935a139bb5e76777a87674fe)), break DBDC for me.

This manifests as being unable to transmit or receive from the 5/6GHz PHY on my AW7916-NPD (mt7916) through any manner. For example, stock hostapd is pointed at the card with this configuration:

```
config wifi-device 'radio1'     
        option type 'mac80211'
        option path 'ffe0a000.pcie/pcia000:02/a000:02:00.0/a000:03:00.0+1'
        option channel '36'                                               
        option band '5g'                                                  
        option htmode 'HE80'
        option disabled '0' 
                            
config wifi-iface 'default_radio1'
        option device 'radio1'    
        option network 'lan'      
        option mode 'ap'          
        option ssid 'AAAAAAAAAAAAAAAAA'
        option encryption 'psk2'       
        option key 'OpenWrtTest'   
```

... but despite not seeing any warnings when running `wifi`, no beacons are sent.

Tested on Extreme Networks AP3825i (mpc85xx) running OpenWrt current snapshot (r21523-4609228432).
Also tested using this board on OpenWrt 22.03.2, `kmod-mt76` package version `5.10.146+2022-09-06-d7054646-4`.

Reverting to firmware files from before linux-firmware's cdf9499cf6706b73935a139bb5e76777a87674fe fixes the issue.

ryderlee1110, 2022-12-19:
This is not an issue. We just limited it to 2+5GHz or 2+6GHz for new version. If you want to do 5+6GHz then you should stick to the previous version.

nbd168, 2022-12-19:
Is there any way to bring it back in newer firmware? Why was this limited? In my opinion this is a regression.

LorenzoBianconi, 2022-12-19:
Is there any way to bring it back in newer firmware? Why was this limited?

> In my opinion this is a regression.

Yes, I agree. This is definitely a regression.

Hurricos, 2022-12-19:
> This is not an issue. We just limited it to 2+5GHz or 2+6GHz for new version. If you want to do 5+6GHz then you should stick to the previous version. Are you using AisaRF device?

Yes, an AW7916-NPD.

To note, I was attempting to transmit on 2+5 or 2+6; I wasn't aware that the hardware was actually capable of 5+6, nor did I attempt it.

ryderlee1110, 2022-12-19:
Try this. I know the 0x191 of AsiaRF is wrong.

```
static void mt7915_eeprom_parse_band_config(struct mt7915_phy *phy)
{
          struct mt7915_dev *dev = phy->dev;
          u8 *eeprom = dev->mt76.eeprom.data;
          u8 band = phy->mt76->band_idx;
          u32 val;

          val = eeprom[MT_EE_WIFI_CONF + band];
          val = FIELD_GET(MT_EE_WIFI_CONF0_BAND_SEL, val);

          if (!is_mt7915(&dev->mt76)) {
                   switch (val) {
                   case MT_EE_V2_BAND_SEL_5GHZ:
                             phy->mt76->cap.has_5ghz = true;
                             return;
                   case MT_EE_V2_BAND_SEL_6GHZ:
                             phy->mt76->cap.has_6ghz = true;
                             return;
                   case MT_EE_V2_BAND_SEL_5GHZ_6GHZ:
                             phy->mt76->cap.has_5ghz = true;
                             phy->mt76->cap.has_6ghz = true;
                             return;
                   default:
                             phy->mt76->cap.has_2ghz = true;
                             return;
                   }
          } else if (val == MT_EE_BAND_SEL_DEFAULT && dev->dbdc_support) {

```

ryderlee1110, 2022-12-19:
> ted? In

business issue. hard to tell.

ryderlee1110, 2022-12-19:
> Is there any way to bring it back in newer firmware? Why was this limited? In my opinion this is a regression.

it's a business issue hard to tell.

NotANormalNerd, 2022-12-27:
Oh my god. I have been going crazy for a few days trying to setup hostapd with 2+5Ghz Wifi on a AW7916-NPD, because 5Ghz would just not work.

## 2.4 Ghz hostapd config
```
# the interface used by the AP
interface=wlan0
# "g" simply means 2.4GHz band
hw_mode=g
# the channel to use
channel=0
# limit the frequencies used to those allowed in the country
ieee80211d=1
# the country code
country_code=ES
# 802.11n support
ieee80211n=1
ht_capab=[LDPC][GF][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
# QoS support, also required for full speed on 802.11n/ac/ax
wmm_enabled=1
```

## 5 Ghz hostapd config
```
interface=wlan1
# "a" simply means 5GHz
hw_mode=a
# the channel to use, 0 means the AP will search for the channel with the least interferences (ACS)
channel=0
ieee80211d=1
country_code=ES

ieee80211h=1

ieee80211n=1
ht_capab=[LDPC][GF][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
wmm_enabled=1

ieee80211ac=1
vht_capab=[MAX-MPDU-11454][VHT160][RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
vht_oper_chwidth=2
```

Which starts up fine on 2.4Ghz. I can see the SSID on my Mac and it has the right channel width, but 5Ghz just does not send any beacon frames. I was going crazy trying to find a solution.

If anybody needs any debugging, please let me know what to do to try this.

trippleflux, 2022-12-31:
@NotANormalNerd 

Did you find solutions for not working 5 Ghz in AW7916-NPD?, I am also experiences it!
Hmm, my OpenWrt is loading different firmware : 
```
[   21.399246] mt7915e 0000:05:00.0: enabling device (0000 -> 0002)
[   21.417413] mt7915e 0000:05:00.0: disabling ASPM L0s L1
[   21.528122] mt7915e 0000:05:00.0: HW/SW Version: 0x8a108a10, Build Time: 20221031154652a
[   21.564585] mt7915e 0000:05:00.0: WM Firmware Version: ____000000, Build Time: 20221031154713
[   21.617025] mt7915e 0000:05:00.0: WA Firmware Version: DEV_000000, Build Time: 20221031154757

````

NotANormalNerd, 2022-12-31:
@trippleflux No, I am waiting on an answer from the maintainers of the mediatek driver. Since this seems to be either a driver problem or a hardware problem with the AsiaRF AW7916-NPD. If it is a hardware problem, then I am going to send the board back. If it is a driver problem I hope we will see a fix soon.
But since we have holidays we will have to wait a bit more.

Since I am not using OpenWRT but a vanilla Debian install, it is a bit harder for me to go back or forward in the repo here, but it is essentially the same code here and in the kernel and firmware repos.

trippleflux, 2022-12-31:
@NotANormalNerd 
For me it didn't need compiled code changes but only replacing firmware files with commit above.

On OpenWrt the firmware is located @ /lib/firmware/mediatek , I have just replaced the firmware files with mt7915_* , mt7916_* & mt7986_* and reboot, now I can uses and settings the 5 Ghz part of AW7916-NPD.

But I agree that newer or newest firmware files need to support AW7916-NPD

ryderlee1110, 2022-12-31:
did you guys try my change above?

trippleflux, 2023-01-01:
@ryderlee1110 

Just finished compiling & deploying latest OpenWrt snapshot for x86_64, based on your pointer, here is my patch : 
```
--- a/mt7915/eeprom.c
+++ b/mt7915/eeprom.c
@@ -152,7 +152,6 @@
 			return;
 		case MT_EE_V2_BAND_SEL_5GHZ_6GHZ:
 			phy->mt76->cap.has_5ghz = true;
-			phy->mt76->cap.has_6ghz = true;
 			return;
 		default:
 			phy->mt76->cap.has_2ghz = true;
```

But the ssid shown as "Hidden Network" on  Windows 11, the replacing with old firmware workaround is doesn't work anymore.

Client is Intel AX210

ryderlee1110, 2023-01-01:
@trippleflux So, is it working now?

trippleflux, 2023-01-01:
@ryderlee1110 

No, it was old firmware with your code, I was wrongly assumed that it was newest firmware, with newest firmware I am getting the non-working 5 Ghz.

ryderlee1110, 2023-01-01:
I think this is most likely device issue. I'll check around.

ryderlee1110, 2023-01-03:
Please report this issue to AsiaRF because I don't thin any issues on my devices.

klojsch, 2023-01-03:
Same problem here. I run a Unielec MT7623, MT7915 runs fine but the MT7916 wont do 5Ghz. Sys log looks normal, last thing i see in the log is that the DFS scan was successfull. According to luci IF is up and running. Snap build from yesterday.

nbd168, 2023-01-03:
@ryderlee1110 if this is an EEPROM issue and it's now affecting devices from multiple vendors, we will need a generic workaround in the driver. Especially since this shows up as a regression with newer firmware.

trippleflux, 2023-01-03:
@nbd168 @ryderlee1110 

Thanks for looking into these, any patches is welcome to test out.

ryderlee1110, 2023-01-03:
Try to use default bin to rule out eeprom issue.

csyuanc, 2023-01-06:
Please help to test if this patch works, thanks.

[https://gist.github.com/csyuanc/42f97e65c22e7dd21b99097fe1b96b14](https://gist.github.com/csyuanc/42f97e65c22e7dd21b99097fe1b96b14)

trippleflux, 2023-01-07:
@ryderlee1110 
I have tried your patch but thankfully @csyuanc patches is working fine for me 🥇 

Thanks @csyuanc !, great work

florisch, 2023-01-13:
Thanks, having the same issue. Applied [the patch](https://github.com/openwrt/mt76/issues/720#issuecomment-1373170468) 5GHz now work fine but it remove the 6GHz support.

How to get the 6GHz support back?

trippleflux, 2023-01-14:
I didn't test 6 Ghz before, only 5 Ghz, ping @csyuanc @ryderlee1110

trippleflux, 2023-01-22:
Testing the latest mediatek firmware files from [here](https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/28d0474d3b3758d782b607e1e45c339013699ddc/autobuild_mac80211_release/package/kernel/mt76/src/firmware), looks like it haven't been fixed yet. Probably we are 'doomed' to use those old firmware files.

ivankexko, 2023-01-26:
@NotANormalNerd 
Hello.
Did you manage to get MT7916 to work on Debian?
I also have Debian, and the MT7916 does not work even in 5GHz mode.

NotANormalNerd, 2023-01-26:
No sadly I haven't gotten it to work yet. I did have not yet have go at it, since there was no real change here.
Until Mediatek decides to support a broken customers eeprom, I think nothing much will happen.

trippleflux, 2023-01-29:
New patches by Carson has been posted in [here](https://patchwork.kernel.org/project/linux-wireless/patch/20230123175500.3375542-1-carson.vandegriffe@candelatech.com/), but I've seems cannot quite figuring it out to use it properly, the patches are originally targeting kernel 5.19.

My AsiaRF 7916 being detected as the following by hwinfo : 
```
36: PCI 500.0: 0280 Network controller
  [Created at pci.386]
  Unique ID: y9sn.2l6miJlj0V1
  Parent ID: Phdb.PRpwMa6hOY8
  SysFS ID: /devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:07.0/0000:05:00.0
  SysFS BusID: 0000:05:00.0
  Hardware Class: network
  Model: "MEDIATEK Network controller"
  Vendor: pci 0x14c3 "MEDIATEK Corp."
  Device: pci 0x7906 
  SubVendor: pci 0x14c3 "MEDIATEK Corp."
  SubDevice: pci 0x7906 
  Driver: "mt7915e"
  Driver Modules: "mt7915e"
  Device File: wlan0
  Memory Range: 0x6000000000-0x60000fffff (ro,non-prefetchable)
  Memory Range: 0x50700000-0x50707fff (rw,non-prefetchable)
  Memory Range: 0x6000100000-0x6000100fff (ro,non-prefetchable)
  IRQ: 174 (5520 events)
  HW Address: 00:0a:52:08:32:ac
  Permanent HW Address: 00:0a:52:08:32:ac
  Link detected: no
  Module Alias: "pci:v000014C3d00007906sv000014C3sd00007906bc02sc80i00"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #42 (PCI bridge)
```

From the patches notes and codes, since in OpenWrt the firmware files located in /lib/firmware/mediatek I need to create the following file in my /lib/firmware/mediatek/fwcfg-mmio-0000\:05\:00.0.txt or fwcfg-<bus>-<id>.txt, with content : 
```
high_band=5
```

Here is the patches for mt76 incase anyone wanted to test it. 
[0001-wifi-mt76-mt7916 Support per-radio configuration firmware file.patch](https://github.com/openwrt/mt76/files/10529629/0001-wifi-mt76-mt7916.Support.per-radio.configuration.firmware.file.patch)

trippleflux, 2023-01-30:
With above patches, it seems that 5G is working fine with newest firmware files but still 6G is still broken : 
This is being set high_band=6 : 
```
Wiphy phy1
        wiphy index: 1
        max # scan SSIDs: 4
        max scan IEs length: 2232 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 4:
                HE Iftypes: managed
                        HE MAC Capabilities (0x08011a000040):
                                +HTC HE Supported
                                Trigger Frame MAC Padding Duration: 2
                                OM Control
                                Maximum A-MPDU Length Exponent: 3
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x4c70ce126d00f3164e3f00):
                                HE40/HE80/5GHz
                                HE160/5GHz
                                242 tone RUs/5GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                HE SU PPDU with 1x HE-LTF and 0.8us GI
                                NDP with 4x HE-LTF and 3.2us GI
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
                                Full Bandwidth UL MU-MIMO
                                Partial Bandwidth UL MU-MIMO
                                DCM Max Constellation: 2
                                DCM Max Constellation Rx: 2
                                SU Beamformee
                                Beamformee STS <= 80Mhz: 3
                                Beamformee STS > 80Mhz: 3
                                Codebook Size SU Feedback
                                Codebook Size MU Feedback
                                Triggered CQI Feedback
                                Partial Bandwidth Extended Range
                                Partial Bandwidth DL MU-MIMO
                                PPE Threshold Present
                                Power Boost Factor ar
                                HE SU PPDU & HE PPDU 4x HE-LTF 0.8us GI
                                Max NC: 2
                                20MHz in 40MHz HE PPDU 2.4GHz
                                20MHz in 160/80+80MHz HE PPDU
                                80MHz in 160/80+80MHz HE PPDU
                                DCM Max BW: 1
                                Longer Than 16HE SIG-B OFDM Symbols
                                Non-Triggered CQI Feedback
                                TX 1024-QAM
                                RX 1024-QAM
                                RX Full BW SU Using HE MU PPDU with Compression SIGB
                                RX Full BW SU Using HE MU PPDU with Non-Compression SIGB
                        HE RX MCS and NSS set <= 80 MHz
[... 265 lines cut ...]
                * [ VHT_IBSS ]: VHT-IBSS
                * [ RRM ]: RRM
                * [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ ACK_SIGNAL_SUPPORT ]: ack signal level support
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ STA_TX_PWR ]: TX power control per station
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ DEL_IBSS_STA ]: deletion of IBSS station support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ FILS_DISCOVERY ]: FILS discovery frame transmission support
                * [ UNSOL_BCAST_PROBE_RESP ]: unsolicated broadcast probe response transmission support
                * [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
                * [ BSS_COLOR ]: BSS coloring support
                * [ RADAR_BACKGROUND ]: Radar background support
```

florisch, 2023-01-31:
@trippleflux that's surprising, I could manually apply the code changes from the patch found on patchworks on my kernel (a 6.2-rc4 from wireless-next) and successfully use the latest firmware from the linux-firmware repository in the 6GHz band.

trippleflux, 2023-01-31:
@florisch w00t Thanks for the confirmation, I am also doing manual merge and created that patch file for mt76, so I must be doing something wrong during that merge, would you mind if you share your applied code changes from mt7915/eeprom.c, I've meant the whole changed mt7915/eeprom.c file.

Also OpenWrt is currently using kernel 5.15, I am not really sure to which that I need to apply on the patch file, either this repo or the kernel.

csyuanc, 2023-02-01:
To use 6 GHz, please try this [patch](https://gist.github.com/csyuanc/f0037b2accd349dc2c85e6ff36d2c18a).
I've added a module param "enable_ipa_6ghz" for switching between 5 and 6 GHz.

trippleflux, 2023-02-01:
@csyuanc 
Thanks, Based on Your patch and Carson patch, I have modified my mt7915_eeprom_parse_band_config function as below : 
```
static void mt7915_eeprom_parse_band_config(struct mt7915_phy *phy)
{
	struct mt7915_dev *dev = phy->dev;
	u8 *eeprom = dev->mt76.eeprom.data;
	u8 band = phy->mt76->band_idx;
	u32 val;

	val = eeprom[MT_EE_WIFI_CONF + band];
	val = FIELD_GET(MT_EE_WIFI_CONF0_BAND_SEL, val);

	if (!is_mt7915(&dev->mt76)) {
		switch (val) {
		case MT_EE_V2_BAND_SEL_5GHZ:
			phy->mt76->cap.has_5ghz = true;
			return;
		case MT_EE_V2_BAND_SEL_6GHZ:
			phy->mt76->cap.has_6ghz = true;
			return;
		case MT_EE_V2_BAND_SEL_5GHZ_6GHZ:
		/* fwcfg intervention to set upper band to 5GHz or 6GHz */
			if ((dev->fwcfg.flags & MT7915_FWCFG_HIGH_BAND) &&
				val == MT_EE_V2_BAND_SEL_5GHZ_6GHZ) {
				dev_info(dev->mt76.dev, "FWCFG: Overriding 7916 high_band with %luGHz\n",
					(unsigned long)dev->fwcfg.high_band);

				if (dev->fwcfg.high_band == 5) {
					phy->mt76->cap.has_5ghz = true;
					u8p_replace_bits(&eeprom[MT_EE_WIFI_CONF + band],
							MT_EE_V2_BAND_SEL_5GHZ,
							MT_EE_WIFI_CONF0_BAND_SEL);
				}
				else if (dev->fwcfg.high_band == 6) {
					phy->mt76->cap.has_6ghz = true;
					u8p_replace_bits(&eeprom[MT_EE_WIFI_CONF + band],
							MT_EE_V2_BAND_SEL_6GHZ,
							MT_EE_WIFI_CONF0_BAND_SEL);
				}
			}
			/* force to buffer mode */
			dev->flash_mode = true;
			return;
		default:
			phy->mt76->cap.has_2ghz = true;
			return;
		}
	} else if (val == MT_EE_BAND_SEL_DEFAULT && dev->dbdc_support) {
		val = band ? MT_EE_BAND_SEL_5GHZ : MT_EE_BAND_SEL_2GHZ;
	}

	switch (val) {
	case MT_EE_BAND_SEL_5GHZ:
		phy->mt76->cap.has_5ghz = true;
		break;
	case MT_EE_BAND_SEL_2GHZ:
		phy->mt76->cap.has_2ghz = true;
		break;
	default:
		phy->mt76->cap.has_2ghz = true;
		phy->mt76->cap.has_5ghz = true;
		break;
	}
}
```

But luci eventough with [this ](https://github.com/openwrt/luci/pull/6104)patches 6 GHz still not properly detected on OpenWrt latest snapshot self-compiled.

Here is iw output for 6 GHz : 
[... 334 lines cut ...]
                * [ VHT_IBSS ]: VHT-IBSS
                * [ RRM ]: RRM
                * [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ ACK_SIGNAL_SUPPORT ]: ack signal level support
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ STA_TX_PWR ]: TX power control per station
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ DEL_IBSS_STA ]: deletion of IBSS station support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ FILS_DISCOVERY ]: FILS discovery frame transmission support
                * [ UNSOL_BCAST_PROBE_RESP ]: unsolicated broadcast probe response transmission support
                * [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
                * [ BSS_COLOR ]: BSS coloring support
                * [ RADAR_BACKGROUND ]: Radar background support
```

ivankexko, 2023-02-02:
This is right? (no IR)

trippleflux, 2023-02-02:
@ivankexko 
After changing country to Canada, I've get : 
```
                Frequencies:
                        * 5955 MHz [1] (12.0 dBm)
                        * 5975 MHz [5] (12.0 dBm)
                        * 5995 MHz [9] (12.0 dBm)
                        * 6015 MHz [13] (12.0 dBm)
                        * 6035 MHz [17] (12.0 dBm)
                        * 6055 MHz [21] (12.0 dBm)
                        * 6075 MHz [25] (12.0 dBm)
                        * 6095 MHz [29] (12.0 dBm)
                        * 6115 MHz [33] (12.0 dBm)
                        * 6135 MHz [37] (12.0 dBm)
                        * 6155 MHz [41] (12.0 dBm)
                        * 6175 MHz [45] (12.0 dBm)
                        * 6195 MHz [49] (12.0 dBm)
                        * 6215 MHz [53] (12.0 dBm)
                        * 6235 MHz [57] (12.0 dBm)
                        * 6255 MHz [61] (12.0 dBm)
                        * 6275 MHz [65] (12.0 dBm)
                        * 6295 MHz [69] (12.0 dBm)
                        * 6315 MHz [73] (12.0 dBm)
                        * 6335 MHz [77] (12.0 dBm)
                        * 6355 MHz [81] (12.0 dBm)
                        * 6375 MHz [85] (12.0 dBm)
                        * 6395 MHz [89] (12.0 dBm)
                        * 6415 MHz [93] (12.0 dBm)
                        * 6435 MHz [97] (12.0 dBm)
                        * 6455 MHz [101] (12.0 dBm)
                        * 6475 MHz [105] (12.0 dBm)
                        * 6495 MHz [109] (12.0 dBm)
                        * 6515 MHz [113] (12.0 dBm)
                        * 6535 MHz [117] (12.0 dBm)
                        * 6555 MHz [121] (12.0 dBm)
                        * 6575 MHz [125] (12.0 dBm)
                        * 6595 MHz [129] (12.0 dBm)
                        * 6615 MHz [133] (12.0 dBm)
                        * 6635 MHz [137] (12.0 dBm)
                        * 6655 MHz [141] (12.0 dBm)
                        * 6675 MHz [145] (12.0 dBm)
                        * 6695 MHz [149] (12.0 dBm)
                        * 6715 MHz [153] (12.0 dBm)
                        * 6735 MHz [157] (12.0 dBm)
                        * 6755 MHz [161] (12.0 dBm)
                        * 6775 MHz [165] (12.0 dBm)
                        * 6795 MHz [169] (12.0 dBm)
                        * 6815 MHz [173] (12.0 dBm)
                        * 6835 MHz [177] (12.0 dBm)
                        * 6855 MHz [181] (12.0 dBm)
                        * 6875 MHz [185] (12.0 dBm)
                        * 6895 MHz [189] (12.0 dBm)
                        * 6915 MHz [193] (12.0 dBm)
                        * 6935 MHz [197] (12.0 dBm)
                        * 6955 MHz [201] (12.0 dBm)
                        * 6975 MHz [205] (12.0 dBm)
                        * 6995 MHz [209] (12.0 dBm)
                        * 7015 MHz [213] (12.0 dBm)
                        * 7035 MHz [217] (12.0 dBm)
                        * 7055 MHz [221] (12.0 dBm)
                        * 7075 MHz [225] (12.0 dBm)
                        * 7095 MHz [229] (12.0 dBm)
                        * 7115 MHz [233] (12.0 dBm)
 ```

florisch, 2023-02-02:
@trippleflux as requested in https://github.com/openwrt/mt76/issues/720#issuecomment-1410154769

Here is the full [eeprom.c](https://github.com/openwrt/mt76/files/10567174/eeprom.txt) after applying the patch that allow me to use 6GHz. The lines you mentioned seems identical.

trippleflux, 2023-02-02:
@florisch thank you, indeed that's identical, probably it's only problematic with OpenWrt, still trying to narrow it down.

csyuanc, 2023-02-02:
Hi all,

Finally we've made some modifications on the calibration parts to let 5+6 work on band1.
Please try the firmware attached and let me know if it works as the previous firmware, thanks.
[mt7916_test_fw.zip](https://github.com/openwrt/mt76/files/10567955/mt7916_test_fw.zip)

Please note that if you want to do peak performance test, we still recommend modifying the eeprom band1 config to 5g or 6g only.

VA1DER, 2023-02-04:
Is it supposed to work without patches with the above-noted firmwares?  Because with today's snapshot (though with none of the above patches), I cannot get either an AW7915-NPD or an AW7916-NPD to work correctly on it's "second" radio no matter what firmware I use.  Both radios show up, and appear to be more or less correctly identified.  I say more or less because while the 7915 displays as expected, the 7916 appears as "Generic 802.11".  But with both cards the first 2.4ghz radio is identified as as 802.11axbgn, and the second 5ghz radio as 802.11acaxn.  However any time I try and scan with the card's "second" radio, it (like the first radio) displays only 2.4GHz stations.  In 22.03.3 scan works as expected on both radios.

With the latest snapshot I have used the firmware that came with it, along with this firmware:

> [EDIT] Reverted back into [4a27f23fc4f22510db81513680666be7e72ccb61 ](https://github.com/openwrt/mt76/tree/4a27f23fc4f22510db81513680666be7e72ccb61/firmware)fixed it for me.

and this firmware:

> Please try the firmware attached and let me know if it works as the previous firmware, thanks.
> [mt7916_test_fw.zip](https://github.com/openwrt/mt76/files/10567955/mt7916_test_fw.zip)

This is on an MT7622 device, which has its own built-in radio.  My /etc/config/wireless is:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/18000000.wmac'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option disabled '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option disabled '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio2'
        option type 'mac80211'
        option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0+1'
        option channel '36'
        option band '5g'
        option htmode 'HE80'
        option disabled '1'

config wifi-iface 'default_radio2'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
```

trippleflux, 2023-02-04:
@VA1DER 
Have you reboot the router after changing the firmware files?

VA1DER, 2023-02-04:
> reboot

Yes.  And verified in dmesg by the firmware date that they have been loaded.  For example, this firmware [mt7916_test_fw.zip](https://github.com/openwrt/mt76/files/10567955/mt7916_test_fw.zip) produces:

[    7.972978] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20230202145005
[    8.055203] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20230202143332

trippleflux, 2023-02-04:
@VA1DER 
Newest/later firmware required such patches above , try https://github.com/openwrt/mt76/tree/4a27f23fc4f22510db81513680666be7e72ccb61/firmware , did you using 160 MHz?, if yes, it's SSID requires few minutes before going up.

@csyuanc 
I am still having problems with 6GHz with patches above and vanilla mt76 firmware or your attached one, there is no SSID being shows up because of maximum transmit power listed as 255 dBm in OpenWrt Luci.

@florisch 
Sorry if bothering you again, wouldn't mind to share your "iw phy" command output for your working 6GHz?, and you are using AW7916-NPD ?, do you use any extra power supply for supplying power to the AW7916-NPD module?. If you don't mind please share with me your hostapd config for the currently working 6 GHz setup. What is the OS? Debian SID?.

VA1DER, 2023-02-04:
> https://github.com/openwrt/mt76/tree/4a27f23fc4f22510db81513680666be7e72ccb61/firmware 

Yes, I originally tried this firmware.  I can get it to work in AP mode, but not in client and scanning does not work on the second radio.

With the old firmware, does scanning on the second radio work for you?  How about with new firmware the patches?

csyuanc, 2023-02-08:
> @csyuanc I am still having problems with 6GHz with patches above and vanilla mt76 firmware files or your attached one, there is no SSID being shows up because of maximum transmit power listed as 255 dBm in OpenWrt Luci.

With my attached firmware, no additional patches are needed.
I've tested following cases:
- band1 scan shows 5g and 6g ssids
- band1 can connect to a 5g AP and ping
- band1 can connect to a 6g AP and ping

trippleflux, 2023-02-09:
Did you use OpenWrt latest Snapshot ? or Other OS?

Because for me, after selecting CA/Canada as the country and save apply, and selected 6G channel, Max Transmit Power listed as 255 dBm :
[image]

And 6G AP SSID doesn't shows up

Batchyx, 2023-02-14:
I have tested the proposed firmware (HW/SW Version: 0x8a108a10, Build Time: 20230202144915a) and I can't see any performance or functional difference between this new firmware and the v21111230 firmware from linux-firmware (i.e. the one before 2022-11-23).

I tested:
- AP and client mode on 5 GHz HE80 and 6 GHz HE160.
- encrypted mesh point in 6 GHz HE160.

I did not test 2.4 GHz or DFS channels. And i can't find any 255 dBm weirdness in iw phy/iw dev.

My setup is:
- Two 32bit ARM machines with AW7916-NPD modules, no heatsink yet. Temperature stays at 70°C.
- Upstream (not openwrt) Linux 6.0 with some of my mt76 patches in this repo that shouldn't affect mt7916. I do not have ee0863aec through...
- hostapd/wpasupplicant from git (3915e8834) with some patches to make 6 GHz mesh actually work.

VA1DER, 2023-03-19:
I still cannot get 6GHz to work.  At least not through LuCI.  Are LuCI patches required for proper detection of 6GHz?

Device: Banana Pi BPI-R64 v1.2
Card: AsiaRF AW7916-NPD
Firmware: [WM: 20230202145005, WA: 20230202143332](https://github.com/csyuanc/linux-firmware/tree/for-upstream/mediatek)

Have tried with and without this patch: https://gist.github.com/csyuanc/42f97e65c22e7dd21b99097fe1b96b14

What is needed to get 6GHz to appear in LuCI?

FallingSnow, 2023-04-01:
I can confirm that a fresh snapshot from 03-31-2023 (`OpenWrt SNAPSHOT r22456-75e78bcaab / LuCI Master git-23.074.82619-6ad6a24`) with the mt76 drivers/firmware installed, then replacing the files in `/lib/modules/mediatek` with @csyuanc's [mt7916_test_fw.zip](https://github.com/openwrt/mt76/files/10567955/mt7916_test_fw.zip) fixes 5Ghz AP.

I've tested 160Mhz on channel 100 with DFS and it works as expected with great performance. Below is the results of me setting 5 feet line of sight from the AP.

```sh
$ iperf -c 192.168.1.71
------------------------------------------------------------
Client connecting to 192.168.1.71, TCP port 5001
TCP window size: 16.0 KByte (default)
------------------------------------------------------------
[  1] local 192.168.1.53 port 44362 connected with 192.168.1.71 port 5001 (icwnd/mss/irtt=14/1448/1502)
[ ID] Interval       Transfer     Bandwidth
[  1] 0.0000-10.0337 sec  1.61 GBytes  1.38 Gbits/sec
```

trippleflux, 2023-07-11:
@csyuanc 

Hmm, found out new kind of error on latest OpenWrt Snapshot self compiled : 
```
[   18.964773] mt7915e 0000:05:00.0: enabling device (0000 -> 0002)
[   18.983076] mt7915e 0000:05:00.0: disabling ASPM L0s L1
[   19.091608] mt7915e 0000:05:00.0: HW/SW Version: 0x8a108a10, Build Time: 20211230135620a

[   19.127667] mt7915e 0000:05:00.0: WM Firmware Version: ____000000, Build Time: 20211230135720
[   19.181284] mt7915e 0000:05:00.0: WA Firmware Version: DEV_000000, Build Time: 20211230135808
[   21.230725] mt7915e 0000:05:00.0: Timeout for initializing firmware
```

Maybe because off applying this [PR](https://github.com/openwrt/openwrt/pull/13080) for latest linux-firmware updates?

ibrtuyima, 2023-07-19:
@Batchyx 
Could u please help and advise how did u manage to get MT7916 working in mesh mode using HE160 width.
Trying everything, never got HE160. only HE80

Batchyx, 2023-09-04:
@ibrtuyima I just used wpa_supplicant from git and set `max_oper_chwidth=2` inside the network block in the configuration file.

thyTwilightGoth, 2023-11-11:
Hi is there any solutions for the current 23.05 stable release?

jayjar291, 2024-03-12:
has there been any updates to this I am trying to get wifi 6 bands to work

NotANormalNerd, 2024-03-13:
@thyTwilightGoth @jayjar291 This has been fixed a long time ago. For me 6Ghz is working on a AW7916-NPD on OpenWRT 23.05.

You need to use WPA3 SAE (CCMP) & Management Frame protection. But then everything is working fine. Check your logs.

Shulyaka, 2024-03-13:
@NotANormalNerd What did you do to make it working? Is it detected as Generic MAC80211 or MediaTek device? Does DBDC works as well?

NotANormalNerd, 2024-03-13:
It doesn't matter if it is recognized as generic or not, that is just a txt file that maps the PCIe ID to the names. The chip you are using is not in that file yet, that is why it is generic.

DBDC works wothout problems for me on 23.05.

What does your log say?

Here is my setup: https://dennis.schmalacker.cloud/posts/openwrt-mikrotik-rbm11g-setup/

jayjar291, 2024-03-22:
@NotANormalNerd this is what I have for my wireless config
[image]
```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:10.0'
        option channel '1'
        option band '2g'
        option htmode 'HE20'
        option disabled '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:10.0+1'
        option channel 'auto'
        option band '6g'
        option cell_density '0'
        option country 'US'
        option htmode 'HE80'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Test6g'
        option encryption 'sae'
        option key 'TESTPa$$W0RD'
```

and it still does not show up on any wifi 6 or 6e devices
[image]

and I have not found any obvious errors in the system or kernel logs
[current.log](https://github.com/openwrt/mt76/files/14719250/current.log)

jayjar291, 2024-03-22:
also, this is a tri-band card is there any way to enable the third band? I am hoping that I can run 1 2.4g, 5g, and a 6g band

NotANormalNerd, 2024-03-22:
@jayjar291 Just grepping for hostapd:
[image]

Also the MT7916 is a dual band chip, but supports either 5 Ghz or 6 Ghz.
`opkg install kmod-mt7915e kmod-mt7916-firmware`

[7 more replies]