openwrt/mt76 #721: MT7915E，Frequent disconnection

small-white-rabbit, 2022-12-20:
Some devices will be disconnected frequently. The ping test lasts for about eight hours. However, the connection record of the device shows that the disconnection is extremely frequent. However, the ping is normal. When the device is disconnected, other devices are used normally, and the disconnection is not felt。

Tue Dec 20 09:21:42 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: authenticated
Tue Dec 20 09:21:42 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: associated (aid 1)
Tue Dec 20 09:21:42 2022 daemon.notice hostapd: wl1-ap0: AP-STA-CONNECTED 9c:b6:d0:60:c3:1d auth_alg=open
Tue Dec 20 09:21:42 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d WPA: pairwise key handshake completed (RSN)
Tue Dec 20 09:21:42 2022 daemon.notice hostapd: wl1-ap0: EAPOL-4WAY-HS-COMPLETED 9c:b6:d0:60:c3:1d
Tue Dec 20 09:21:42 2022 daemon.info dnsmasq-dhcp[7163]: DHCPREQUEST(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d
Tue Dec 20 09:21:42 2022 daemon.info dnsmasq-dhcp[7163]: DHCPACK(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d GT62
Tue Dec 20 09:21:44 2022 daemon.info dnsmasq-dhcp[7163]: read /etc/ethers - 0 addresses
Tue Dec 20 09:21:45 2022 daemon.info dnsmasq-dhcp[7163]: read /etc/ethers - 0 addresses
Tue Dec 20 12:48:39 2022 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED 9c:b6:d0:60:c3:1d
Tue Dec 20 12:48:39 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: authenticated
Tue Dec 20 12:48:39 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: associated (aid 1)
Tue Dec 20 12:48:39 2022 daemon.notice hostapd: wl1-ap0: AP-STA-CONNECTED 9c:b6:d0:60:c3:1d auth_alg=open
Tue Dec 20 12:48:39 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d WPA: pairwise key handshake completed (RSN)
Tue Dec 20 12:48:39 2022 daemon.notice hostapd: wl1-ap0: EAPOL-4WAY-HS-COMPLETED 9c:b6:d0:60:c3:1d
Tue Dec 20 12:48:39 2022 daemon.info dnsmasq-dhcp[7163]: DHCPREQUEST(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d
Tue Dec 20 12:48:39 2022 daemon.info dnsmasq-dhcp[7163]: DHCPACK(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d GT62
Tue Dec 20 12:51:49 2022 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED 9c:b6:d0:60:c3:1d
Tue Dec 20 12:51:49 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: authenticated
Tue Dec 20 12:51:49 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: associated (aid 1)
Tue Dec 20 12:51:49 2022 daemon.notice hostapd: wl1-ap0: AP-STA-CONNECTED 9c:b6:d0:60:c3:1d auth_alg=open
Tue Dec 20 12:51:49 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d WPA: pairwise key handshake completed (RSN)
Tue Dec 20 12:51:49 2022 daemon.notice hostapd: wl1-ap0: EAPOL-4WAY-HS-COMPLETED 9c:b6:d0:60:c3:1d
Tue Dec 20 12:51:49 2022 daemon.info dnsmasq-dhcp[7163]: DHCPREQUEST(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d
Tue Dec 20 12:51:49 2022 daemon.info dnsmasq-dhcp[7163]: DHCPACK(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d GT62
Tue Dec 20 12:58:47 2022 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED 9c:b6:d0:60:c3:1d
Tue Dec 20 12:58:47 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: authenticated
Tue Dec 20 12:58:47 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: associated (aid 1)
Tue Dec 20 12:58:47 2022 daemon.notice hostapd: wl1-ap0: AP-STA-CONNECTED 9c:b6:d0:60:c3:1d auth_alg=open
Tue Dec 20 12:58:47 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d WPA: pairwise key handshake completed (RSN)
Tue Dec 20 12:58:47 2022 daemon.notice hostapd: wl1-ap0: EAPOL-4WAY-HS-COMPLETED 9c:b6:d0:60:c3:1d
Tue Dec 20 12:58:47 2022 daemon.info dnsmasq-dhcp[7163]: DHCPREQUEST(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d
Tue Dec 20 12:58:47 2022 daemon.info dnsmasq-dhcp[7163]: DHCPACK(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d GT62
Tue Dec 20 13:01:57 2022 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED 9c:b6:d0:60:c3:1d
Tue Dec 20 13:01:57 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: authenticated
Tue Dec 20 13:01:57 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: associated (aid 1)
Tue Dec 20 13:01:57 2022 daemon.notice hostapd: wl1-ap0: AP-STA-CONNECTED 9c:b6:d0:60:c3:1d auth_alg=open
Tue Dec 20 13:01:57 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d WPA: pairwise key handshake completed (RSN)
Tue Dec 20 13:01:57 2022 daemon.notice hostapd: wl1-ap0: EAPOL-4WAY-HS-COMPLETED 9c:b6:d0:60:c3:1d
Tue Dec 20 13:01:57 2022 daemon.info dnsmasq-dhcp[7163]: DHCPREQUEST(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d
Tue Dec 20 13:01:57 2022 daemon.info dnsmasq-dhcp[7163]: DHCPACK(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d GT62

Tue Dec 20 13:29:18 2022 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED 9c:b6:d0:60:c3:1d
Tue Dec 20 13:29:18 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: authenticated
Tue Dec 20 13:29:18 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: associated (aid 1)
Tue Dec 20 13:29:18 2022 daemon.notice hostapd: wl1-ap0: AP-STA-CONNECTED 9c:b6:d0:60:c3:1d auth_alg=open
Tue Dec 20 13:29:18 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d WPA: pairwise key handshake completed (RSN)
Tue Dec 20 13:29:18 2022 daemon.info dnsmasq-dhcp[7163]: DHCPREQUEST(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d
Tue Dec 20 13:29:18 2022 daemon.notice hostapd: wl1-ap0: EAPOL-4WAY-HS-COMPLETED 9c:b6:d0:60:c3:1d
Tue Dec 20 13:29:18 2022 daemon.info dnsmasq-dhcp[7163]: DHCPACK(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d GT62
Tue Dec 20 13:32:28 2022 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED 9c:b6:d0:60:c3:1d
Tue Dec 20 13:32:28 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: authenticated
Tue Dec 20 13:32:28 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: associated (aid 1)
Tue Dec 20 13:32:28 2022 daemon.notice hostapd: wl1-ap0: AP-STA-CONNECTED 9c:b6:d0:60:c3:1d auth_alg=open
Tue Dec 20 13:32:28 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d WPA: pairwise key handshake completed (RSN)
Tue Dec 20 13:32:28 2022 daemon.notice hostapd: wl1-ap0: EAPOL-4WAY-HS-COMPLETED 9c:b6:d0:60:c3:1d
Tue Dec 20 13:32:28 2022 daemon.info dnsmasq-dhcp[7163]: DHCPREQUEST(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d
Tue Dec 20 13:32:28 2022 daemon.info dnsmasq-dhcp[7163]: DHCPACK(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d GT62
Tue Dec 20 13:47:31 2022 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED 9c:b6:d0:60:c3:1d
Tue Dec 20 13:47:31 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: authenticated
Tue Dec 20 13:47:31 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: associated (aid 1)
Tue Dec 20 13:47:31 2022 daemon.notice hostapd: wl1-ap0: AP-STA-CONNECTED 9c:b6:d0:60:c3:1d auth_alg=open
Tue Dec 20 13:47:31 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d WPA: pairwise key handshake completed (RSN)
Tue Dec 20 13:47:31 2022 daemon.notice hostapd: wl1-ap0: EAPOL-4WAY-HS-COMPLETED 9c:b6:d0:60:c3:1d
Tue Dec 20 13:47:33 2022 daemon.info dnsmasq-dhcp[7163]: DHCPREQUEST(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d
Tue Dec 20 13:47:33 2022 daemon.info dnsmasq-dhcp[7163]: DHCPACK(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d GT62
Tue Dec 20 13:50:40 2022 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED 9c:b6:d0:60:c3:1d
Tue Dec 20 13:50:40 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: authenticated
[... 73 lines cut ...]

Tue Dec 20 17:20:57 2022 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED 9c:b6:d0:60:c3:1d
Tue Dec 20 17:20:58 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: authenticated
Tue Dec 20 17:20:58 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d IEEE 802.11: associated (aid 1)
Tue Dec 20 17:20:58 2022 daemon.notice hostapd: wl1-ap0: AP-STA-CONNECTED 9c:b6:d0:60:c3:1d auth_alg=open
Tue Dec 20 17:20:58 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d WPA: pairwise key handshake completed (RSN)
Tue Dec 20 17:20:58 2022 daemon.notice hostapd: wl1-ap0: EAPOL-4WAY-HS-COMPLETED 9c:b6:d0:60:c3:1d
Tue Dec 20 17:20:58 2022 daemon.info dnsmasq-dhcp[7163]: DHCPREQUEST(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d
Tue Dec 20 17:20:58 2022 daemon.info dnsmasq-dhcp[7163]: DHCPACK(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d GT62
Tue Dec 20 17:24:07 2022 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED 9c:b6:d0:60:c3:1d
[... cut ...]
Tue Dec 20 17:24:39 2022 daemon.notice hostapd: wl1-ap0: AP-STA-CONNECTED 9c:b6:d0:60:c3:1d auth_alg=open
Tue Dec 20 17:24:39 2022 daemon.info hostapd: wl1-ap0: STA 9c:b6:d0:60:c3:1d WPA: pairwise key handshake completed (RSN)
Tue Dec 20 17:24:39 2022 daemon.notice hostapd: wl1-ap0: EAPOL-4WAY-HS-COMPLETED 9c:b6:d0:60:c3:1d
Tue Dec 20 17:24:39 2022 daemon.info dnsmasq-dhcp[7163]: DHCPREQUEST(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d
Tue Dec 20 17:24:39 2022 daemon.info dnsmasq-dhcp[7163]: DHCPACK(br-lan) 192.168.15.186 9c:b6:d0:60:c3:1d GT62

ryderlee1110, 2022-12-20:
Did you observe this issue on other connected devices?

small-white-rabbit, 2022-12-21:
> Did you observe this issue on other connected devices or older OpenWRT versions?

I can't feel the disconnection in use, but the log is the same

ryderlee1110, 2022-12-21:
I mean does this issue only happen on specific devices?

small-white-rabbit, 2022-12-21:
> I mean does this issue only happen on specific devices?

It may be that this device has been used for a long time, so I can see this phenomenon. If other devices have been used for a long time, it may be the same problem