openwrt/mt76 #724: Belkin RT3200/MT7915E: "bad VHT capabilities, disabling VHT" on client side w/ ax200

RSpliet, 2022-12-23:
I noticed on my Intel AX200-based client (kernel 6.0.13, Fedora) in the dmesg the following messages when connecting to my Belkin RT3200:
"[327897.958286] wlp3s0: authenticate with MA:CA:DD:RE:SS:SS
[327897.964049] wlp3s0: bad VHT capabilities, disabling VHT
[327897.964051] wlp3s0: Invalid HE elem, Disable HE
[327897.964053] wlp3s0: 80 MHz not supported, disabling VHT"

I am under the impression that this is a new issue that popped up after upgrading my router to OpenWrt SNAPSHOT r21452-1e240f60a5 ("kernel: modules/lib-lz4: add lz4hc_compress") . Could this be a regression in MT76?

RSpliet, 2023-01-08:
Can confirm that this is still an issue with a client on kernel 6.0.16, and the RT3200 at OpenWrt SNAPSHOT r21663-b069d4ac6d ("octeontx: add 5.15 testing kernel").

ThiloteE, 2023-01-09:
Client: Intel AX200 based client (driver 22.190.0 for Windows 10
Accesspoint: OpenWRT [D-Link DAP x1860](https://forum.openwrt.org/t/openwrt-support-for-d-link-dap-x1860/141538)

Can confirm VHE not being activated with -71 dBm. This is what I see in LuCI:

```
648.5 Mbit/s, 80 MHz, HE-MCS 6, HE-NSS 2
612.5 Mbit/s, 80 MHz, HE-MCS 6, HE-NSS 2, HE-GI 1
```

This is Channel 40 in 5 GHz being set to 80MHz, configured as repeater with relayd. Distance to client ca. 5 meters, going through 2 walls.

I tried two snapshot pre-releases from RolandoMagico. As far as I can see, there is no difference between snapshot from [2022-12-27](https://github.com/RolandoMagico/openwrt/releases/tag/Build_2022-12-27) and [2023-01-07](https://github.com/RolandoMagico/openwrt/releases/tag/Build_2023-01-07).

Noki, 2023-01-10:
I believe there is a general problem with negotiating tx/rx rates with MT7915E. I see similar problems with two MT7915E devices placed 2 meters apart in combination with 802.11s meshing.

ThiloteE, 2023-01-10:
@Noki apart from having configured my device as [extender using relayd](https://openwrt.org/docs/guide-user/network/wifi/relay_configuration), my config is vanilla. I have **not** consciously activated 802.11s, so what you experience **might** have different causes. My throughput is not the best, but i still can reach ca. 150 Mbits/sec - 350 Mbits/sec, going through 2 walls (measured via iperf3). I do not experience values as low as you do.

Indeed, `rx` was changing a lot on my side too, so i asked a question in the forum and [in this post](https://forum.openwrt.org/t/mt7621-mt7915e-multiple-streams-different-throughput-bandwidth-on-rx/147885/2) some user put forth the idea that it could be evidence for power saving mode.

ThiloteE, 2023-01-10:
Maybe relevant? - [Two EHT commits for suporting minimal EHT rate, in mac80211 and
iwlwifi](https://lore.kernel.org/linux-wireless/20230109110722.212758-1-gregory.greenman@intel.com/T/#t)

This part of the code also deals with VHT reporting on rx side.

Hope mt76, iwlwifi and mac80211 are compatible in that regard.

RSpliet, 2023-02-11:
I've played around a little bit, and I suspect that the VHT warning is benign. It seems to happen on the MAC address associated with the 2.4GHz radio (MT7622), which indeed does not support .ac. It's not shown when connected to 5G, and the router confirms that the client connects to the 5G radio with VHT speeds.
The "invalid HE elem" warning on the other hand seems like more of an issue, but with the recent firmware not supporting .ax/160MHz at all, I will have a hard time reproducing that one until we get a definite answer to #745 .

sandreas, 2024-02-09:
@RSpliet @ThiloteE @Noki 
Similar problem here OpenWRT 23.05.0 on Banana Pi BPI-R3.

```
[148924.790533] wlp61s0: authenticate with ......
[148924.790548] wlp61s0: 80 MHz not supported, disabling VHT
[148924.955325] wlp61s0: send auth to .... (try 1/3)
[148924.958057] wlp61s0: authenticated
[148924.959342] wlp61s0: associate with ... (try 1/3)
[148924.971486] wlp61s0: RX AssocResp from ... (capab=0x31 status=0 aid=1)
[148925.002619] wlp61s0: associated
```
The messages indicate that association worked fine, but I don't get a connection. 

Client OS: `Fedora 38 Workstation`
Client Wifi: `Intel 200AX`, `Mediatek RZ616` (both show similar behaviour)

I'm going to test a more recent Live CD soon - maybe I'm able to test Windows. If you need more info, feel free to ask.

ThiloteE, 2024-02-10:
Since I commented here, I learned a thing or two about wifi. Based on my current understanding, I assume VHT was not activated on my personal devices back then, because the connection was established via wifi 6.
`HE` stands for "High efficiency", which is the IEEE 802.11ax standard and as you can see, it shows up in my LuCi output. `VHT` stands for Very High Throughput and is the IEEE 802.11ac standard. The HE in my LuCi output was probably something to be expected, as my network topology was like this: router (fritzbox 7490) --> wifi 5 --> D-Link DAP x1860 --> wifi 6 --> Intel AX200.

As a consequence, I will unsubscribe from this thread, as I cannot reproduce and meanwhile I have moved flats, so doing any tests comparable to my old location would also require lots of effort.

Noki, 2024-02-10:
I think it is fixed. My MT7915E devices work all fine.

sandreas, 2024-02-10:
> As a consequence, I will unsubscribe from this thread

Ok, thanks for the feedback.

> I think it is fixed. My MT7915E devices work all fine.

@Noki 
Thank you for the quick feedback. What OpenWRT Builds are you using? Maybe an upgrade to `23.05.2` will fix the problem for me as well...?