openwrt/mt76 #727: Kernel panic (null pointer dereference) on linux-6.1 and warning on 6.2-rc2

patrakov, 2023-01-08:
I have a Mediatek mini-PCIe WiFi adapter in my desktop:

```
04:00.0 Network controller [0280]: MEDIATEK Corp. MT7612E 802.11acbgn PCI Express Wireless Network Adapter [14c3:7612]
```

With linux-6.0 or before, it worked acceptably. However, with linux-6.1 (I use Zen kernel on Arch Linux) it panics. So something needs to be sent to the stable team.

Here is the panic (captured over netconsole) and a few lines that precede it:

```
   22.143795] rfkill: input handler disabled
[   22.234236] systemd-journald[461]: Successfully sent stream file descriptor to service manager.
[   22.234395] systemd-journald[461]: Successfully sent stream file descriptor to service manager.
[   22.236424] systemd-journald[461]: Successfully sent stream file descriptor to service manager.
[   25.434800] wlp4s0: authenticate with 36:98:b5:19:b6:18
[   25.728955] wlp4s0: send auth to 36:98:b5:19:b6:18 (try 1/3)
[   25.732785] wlp4s0: authenticated
[   25.733752] wlp4s0: associate with 36:98:b5:19:b6:18 (try 1/3)
[   25.739076] wlp4s0: RX AssocResp from 36:98:b5:19:b6:18 (capab=0x1431 status=0 aid=1)
[   25.739137] wlp4s0: associated
[   25.780847] wlp4s0: Limiting TX power to 20 (20 - 0) dBm as advertised by 36:98:b5:19:b6:18
[   25.796949] IPv6: ADDRCONF(NETDEV_CHANGE): wlp4s0: link becomes ready
[   25.832219] kauditd_printk_skb: 30 callbacks suppressed
[   25.832223] audit: type=1130 audit(1673167695.026:94): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-wait-online comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   25.860211] systemd-journald[461]: Successfully sent stream file descriptor to service manager.
[   25.860501] audit: type=1130 audit(1673167695.054:95): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=clamav-freshclam comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   25.862150] systemd-journald[461]: Successfully sent stream file descriptor to service manager.
[   26.230675] Bridge firewalling registered
[   26.231783] audit: type=1325 audit(1673167695.425:96): table=nat family=2 entries=0 op=xt_register pid=1169 comm="iptables"
[   26.231793] audit: type=1300 audit(1673167695.425:96): arch=c000003e syscall=55 success=yes exit=0 a0=4 a1=0 a2=40 a3=7ffe72d938f0 items=0 ppid=1143 pid=1169 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/bin/xtables-legacy-multi" key=(null)
[   26.231796] audit: type=1327 audit(1673167695.425:96): proctitle=2F7573722F62696E2F69707461626C6573002D2D77616974002D74006E6174002D4C002D6E
[   26.238596] audit: type=1325 audit(1673167695.428:97): table=filter family=2 entries=0 op=xt_register pid=1170 comm="iptables"
[   26.238611] audit: type=1300 audit(1673167695.428:97): arch=c000003e syscall=55 success=yes exit=0 a0=4 a1=0 a2=40 a3=7ffc5fb3ec80 items=0 ppid=1143 pid=1170 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/bin/xtables-legacy-multi" key=(null)
[   26.238616] audit: type=1327 audit(1673167695.428:97): proctitle=2F7573722F62696E2F69707461626C6573002D2D77616974002D4C002D6E
[   26.258630] audit: type=1325 audit(1673167695.452:98): table=nat family=2 entries=5 op=xt_replace pid=1191 comm="iptables"
[   26.258658] audit: type=1300 audit(1673167695.452:98): arch=c000003e syscall=54 success=yes exit=0 a0=4 a1=0 a2=40 a3=5647d7c45a80 items=0 ppid=1143 pid=1191 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/bin/xtables-legacy-multi" key=(null)
[   27.868370] BUG: kernel NULL pointer dereference, address: 00000000000000a8
[   27.868379] #PF: supervisor write access in kernel mode
[   27.868382] #PF: error_code(0x0002) - not-present page
[   27.868384] PGD 0 P4D 0 
[   27.868387] Oops: 0002 [#1] PREEMPT SMP PTI
[   27.868400] CPU: 1 PID: 613 Comm: napi/phy0-323 Not tainted 6.1.3-zen1-1-zen #1 e84ae47f4a6d6b4d7d2701d8d72f471c5f76a3ff
[   27.868404] Hardware name: MSI MS-7851/Z87I (MS-7851), BIOS V1.9 05/15/2015
[   27.868406] RIP: 0010:ieee80211_deliver_skb+0x4c/0x1a0 [mac80211]
[   27.868465] Code: 00 00 8b 55 70 49 8b 85 50 05 00 00 65 48 03 05 0a 63 34 3e 48 01 50 08 48 ff 00 48 83 7b 28 00 74 11 48 8b 43 30 48 63 53 44 <48> 83 84 d0 a8 00 00 00 01 8b 87 e8 11 00 00 8d 50 fd 83 fa 01 0f
[   27.868468] RSP: 0018:ffffbbb3c0917ae8 EFLAGS: 00010286
[   27.868470] RAX: 0000000000000000 RBX: ffffbbb3c0917c90 RCX: 0000000000000008
[   27.868472] RDX: 0000000000000000 RSI: ffffa0eaaece064e RDI: ffffa0ea895109c0
[   27.868474] RBP: ffffa0eaabc4f000 R08: 000000000000003b R09: 0000000000000171
[   27.868476] R10: 0000000000000171 R11: 0000000000000000 R12: 0000000000004000
[   27.868488] R13: ffffa0ea89510000 R14: 0000000000004288 R15: ffffa0eaabc4f000
[   27.868490] FS:  0000000000000000(0000) GS:ffffa0ed4f640000(0000) knlGS:0000000000000000
[   27.868493] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   27.868548] CR2: 00000000000000a8 CR3: 00000002bf210006 CR4: 00000000001706e0
[   27.868550] Call Trace:
[   27.868554]  <TASK>
[   27.868557]  __ieee80211_rx_h_amsdu+0x22f/0x5b0 [mac80211 5aa547b8289f4567c72ef207cdf47dd6bfda0f0f]
[   27.868621]  ? ieee80211_prepare_and_rx_handle+0x1384/0x1cb0 [mac80211 5aa547b8289f4567c72ef207cdf47dd6bfda0f0f]
[   27.868671]  ieee80211_prepare_and_rx_handle+0x1384/0x1cb0 [mac80211 5aa547b8289f4567c72ef207cdf47dd6bfda0f0f]
[   27.868721]  ieee80211_rx_list+0xa42/0xd70 [mac80211 5aa547b8289f4567c72ef207cdf47dd6bfda0f0f]
[   27.868763]  mt76_rx_complete+0x207/0x2e0 [mt76 778938866a7f3ea5ea095213dfa0e7b90189f510]
[   27.868773]  mt76_rx_poll_complete+0x356/0x580 [mt76 778938866a7f3ea5ea095213dfa0e7b90189f510]
[   27.868781]  ? mt76_dma_rx_fill.isra.0+0x1a0/0x220 [mt76 778938866a7f3ea5ea095213dfa0e7b90189f510]
[   27.868788]  mt76_dma_rx_poll+0x397/0x500 [mt76 778938866a7f3ea5ea095213dfa0e7b90189f510]
[   27.868797]  ? __napi_poll+0x190/0x190
[   27.868801]  __napi_poll+0x2b/0x190
[   27.868805]  napi_threaded_poll+0x157/0x170
[   27.868810]  kthread+0xde/0x110
[   27.868814]  ? kthread_complete_and_exit+0x20/0x20
[... 87 lines cut ...]
[  168.774572] FS:  0000000000000000(0000) GS:ffff89564f680000(0000) knlGS:0000000000000000
[  168.774577] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  168.774579] CR2: 00007f3648a80088 CR3: 00000001f3610001 CR4: 00000000001706e0
[  168.774584] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  168.774586] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  168.774591] Call Trace:
[  168.774596]  <TASK>
[  168.774601]  ieee80211_do_stop+0x662/0x8a0 [mac80211 410b83d45043a4cb94322898869721d223f7a201]
[  168.774644]  ieee80211_stop+0x4d/0x180 [mac80211 410b83d45043a4cb94322898869721d223f7a201]
[  168.774685]  __dev_close_many+0x9e/0x110
[  168.774689]  dev_close_many+0x8f/0x150
[  168.774695]  ? ieee80211_handle_reconfig_failure+0x69/0x90 [mac80211 410b83d45043a4cb94322898869721d223f7a201]
[  168.774741]  dev_close+0x83/0xb0
[  168.774746]  cfg80211_shutdown_all_interfaces+0x4d/0xf0 [cfg80211 1605a2dada9c32fd4c9df2357ae678d7b25d2404]
[  168.774796]  ieee80211_restart_work+0x120/0x150 [mac80211 410b83d45043a4cb94322898869721d223f7a201]
[  168.774831]  process_one_work+0x1c8/0x380
[  168.774836]  worker_thread+0x51/0x390
[  168.774842]  ? __pfx_worker_thread+0x10/0x10
[  168.774845]  kthread+0xde/0x110
[  168.774851]  ? __pfx_kthread+0x10/0x10
[  168.774853]  ret_from_fork+0x2c/0x50
[  168.774861]  </TASK>
[  168.774863] ---[ end trace 0000000000000000 ]---
[  176.937776] audit: type=1131 audit(1673166451.075:165): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
```

patrakov, 2023-01-08:
I am not alone, people on Reddit complain about the same on USB cards: https://www.reddit.com/r/archlinux/comments/zufy3n/linuxzen_611zen11_crashes_while_trying_to_use_usb/

patrakov, 2023-01-08:
I have tested a combination of the latest linux git commit (9b43a525db125799df81e6fbef712a2ae50bfc5d) and the latest mt76 git commit (679254c50f279fe4f1e3ae1fb943f0d1ecdac4ab). This required one fixup for a forgotten `#include <net/netlink.h>`.

The result is identical to the one on 6.1:

```
[   23.881419] rfkill: input handler disabled
[   23.953331] systemd-journald[375]: Successfully sent stream file descriptor to service manager.
[   23.953423] systemd-journald[375]: Successfully sent stream file descriptor to service manager.
[   23.953633] systemd-journald[375]: Successfully sent stream file descriptor to service manager.
[   28.053771] wlp4s0: authenticate with 36:98:b5:19:b6:18
[   28.344302] wlp4s0: send auth to 36:98:b5:19:b6:18 (try 1/3)
[   28.349302] wlp4s0: authenticated
[   28.350847] wlp4s0: associate with 36:98:b5:19:b6:18 (try 1/3)
[   28.354457] wlp4s0: RX AssocResp from 36:98:b5:19:b6:18 (capab=0x1431 status=0 aid=1)
[   28.354522] wlp4s0: associated
[   28.355274] wlp4s0: Limiting TX power to 20 (20 - 0) dBm as advertised by 36:98:b5:19:b6:18
[   28.454278] IPv6: ADDRCONF(NETDEV_CHANGE): wlp4s0: link becomes ready
[   28.487883] kauditd_printk_skb: 61 callbacks suppressed
[   28.487886] audit: type=1130 audit(1673183191.632:94): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-wait-online comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   28.537785] systemd-journald[375]: Successfully sent stream file descriptor to service manager.
[   28.538011] audit: type=1130 audit(1673183191.682:95): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=clamav-freshclam comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   28.539533] systemd-journald[375]: Successfully sent stream file descriptor to service manager.
[   28.893914] Bridge firewalling registered
[   28.895084] audit: type=1325 audit(1673183192.039:96): table=nat family=2 entries=0 op=xt_register pid=1084 comm="iptables"
[   28.895094] audit: type=1300 audit(1673183192.039:96): arch=c000003e syscall=55 success=yes exit=0 a0=4 a1=0 a2=40 a3=7ffcd9874950 items=0 ppid=1055 pid=1084 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/bin/xtables-legacy-multi" key=(null)
[   28.895099] audit: type=1327 audit(1673183192.039:96): proctitle=2F7573722F62696E2F69707461626C6573002D2D77616974002D74006E6174002D4C002D6E
[   28.904058] audit: type=1325 audit(1673183192.039:97): table=filter family=2 entries=0 op=xt_register pid=1085 comm="iptables"
[   28.904069] audit: type=1300 audit(1673183192.039:97): arch=c000003e syscall=55 success=yes exit=0 a0=4 a1=0 a2=40 a3=7fff54363200 items=0 ppid=1055 pid=1085 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/bin/xtables-legacy-multi" key=(null)
[   28.904073] audit: type=1327 audit(1673183192.039:97): proctitle=2F7573722F62696E2F69707461626C6573002D2D77616974002D4C002D6E
[   28.925431] audit: type=1325 audit(1673183192.069:98): table=nat family=2 entries=5 op=xt_replace pid=1106 comm="iptables"
[   28.925445] audit: type=1300 audit(1673183192.069:98): arch=c000003e syscall=54 success=yes exit=0 a0=4 a1=0 a2=40 a3=5570c3f82a80 items=0 ppid=1055 pid=1106 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/bin/xtables-legacy-multi" key=(null)
[   29.379161] systemd-journald[375]: Successfully sent stream file descriptor to service manager.
[   29.379221] systemd-journald[375]: Successfully sent stream file descriptor to service manager.
[   33.799118] BUG: kernel NULL pointer dereference, address: 00000000000000a8
[   33.799132] #PF: supervisor write access in kernel mode
[   33.799136] #PF: error_code(0x0002) - not-present page
[   33.799140] PGD 0 P4D 0 
[   33.799145] Oops: 0002 [#1] PREEMPT SMP PTI
[   33.799149] CPU: 2 PID: 525 Comm: napi/phy0-323 Not tainted 6.2.0-rc2-bisect+ #1 0fc405c7a0b1ed546b76f0a5943280e49538b2e6
[   33.799154] Hardware name: MSI MS-7851/Z87I (MS-7851), BIOS V1.9 05/15/2015
[   33.799157] RIP: 0010:ieee80211_deliver_skb+0x4c/0x1a0 [mac80211]
[   33.799295] Code: 00 00 8b 55 70 49 8b 85 50 05 00 00 65 48 03 05 22 e5 68 3e 48 01 50 08 48 ff 00 48 83 7b 28 00 74 11 48 8b 43 30 48 63 53 44 <48> 83 84 d0 a8 00 00 00 01 8b 87 e8 11 00 00 8d 50 fd 83 fa 01 0f
[   33.799299] RSP: 0018:ffffa22900877ad8 EFLAGS: 00010286
[   33.799303] RAX: 0000000000000000 RBX: ffffa22900877c80 RCX: 0000000000008e88
[   33.799306] RDX: 0000000000000000 RSI: ffff8fa74812dc4e RDI: ffff8fa74853c9c0
[   33.799310] RBP: ffff8fa745ee6a00 R08: 000000ec0000010c R09: 0000000000000002
[   33.799313] R10: 000000000000070e R11: ffffce9ac0000000 R12: ffffa22900877b10
[   33.799316] R13: ffff8fa74853c000 R14: 0000000000000000 R15: ffff8fa74853c000
[   33.799319] FS:  0000000000000000(0000) GS:ffff8faa4f680000(0000) knlGS:0000000000000000
[   33.799323] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   33.799326] CR2: 00000000000000a8 CR3: 00000003bde10005 CR4: 00000000001706e0
[   33.799329] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[   33.799332] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[   33.799390] Call Trace:
[   33.799395]  <TASK>
[   33.799399]  __ieee80211_rx_h_amsdu+0x1a4/0x230 [mac80211 68489b873ca8fdf8477bee8e1a72c231e4b418cf]
[... cut ...]
[   33.800054] CR2: 00000000000000a8 CR3: 00000003bde10005 CR4: 00000000001706e0
[   33.800057] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[   33.800060] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[   33.800063] Kernel panic - not syncing: Fatal exception in interrupt
[   33.800402] Kernel Offset: 0x26800000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
[   33.800412] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

patrakov, 2023-01-08:
Bisected.

```
b320d6c456ff2aa43491654407d448bcfa58ac9f is the first bad commit
commit b320d6c456ff2aa43491654407d448bcfa58ac9f
Author: Benjamin Berg <benjamin.berg@intel.com>
Date:   Fri Sep 2 16:12:40 2022 +0200

    wifi: mac80211: use correct rx link_sta instead of default
    
    Use rx->link_sta everywhere instead of accessing the default link.
    
    Signed-off-by: Benjamin Berg <benjamin.berg@intel.com>
    Signed-off-by: Johannes Berg <johannes.berg@intel.com>

 net/mac80211/rx.c | 69 ++++++++++++++++++++++++++++---------------------------
 1 file changed, 35 insertions(+), 34 deletions(-)
```

morrownr, 2023-01-08:
@patrakov 

This problem is going to blow up soon as distros move to 6.1.

There is a patch that I have already tested that works. It is not a patch to mt76 but rather to mac80211:

```
Since the initialization of rx->link and rx->link_sta is rather convoluted
and duplicated in many places, clean it up by using a helper function to
set it.

Fixes: ccdde7c74ffd ("wifi: mac80211: properly implement MLO key handling")
Fixes: b320d6c456ff ("wifi: mac80211: use correct rx link_sta instead of default")

---
v5: fix sdata link assignment, fix sdata when receiving on multiple interfaces
v4: fix regression in handling mgmt frames with AP_VLAN
v3: include crash log
v2: fix uninitialized variable
 net/mac80211/rx.c | 222 +++++++++++++++++++++-------------------------
 1 file changed, 99 insertions(+), 123 deletions(-)

diff --git a/net/mac80211/rx.c b/net/mac80211/rx.c
index 7e3ab6e1b28f..6ccc487bad7f 100644
--- a/net/mac80211/rx.c
+++ b/net/mac80211/rx.c
@@ -4049,6 +4049,58 @@ static void ieee80211_invoke_rx_handlers(struct ieee80211_rx_data *rx)
 #undef CALL_RXH
 }

+static bool
+ieee80211_rx_is_valid_sta_link_id(struct ieee80211_sta *sta, u8 link_id)
+{
+       if (!sta->mlo)
+               return false;
+
+       return !!(sta->valid_links & BIT(link_id));
+}
+
+static bool ieee80211_rx_data_set_link(struct ieee80211_rx_data *rx,
+                                      u8 link_id)
+{
+       rx->link_id = link_id;
+       rx->link = rcu_dereference(rx->sdata->link[link_id]);
+
+       if (!rx->sta || !rx->sta->sta.mlo)
+               return rx->link;
+
+       if (!ieee80211_rx_is_valid_sta_link_id(&rx->sta->sta, link_id))
+               return false;
+
+       rx->link_sta = rcu_dereference(rx->sta->link[link_id]);
+
+       return rx->link && rx->link_sta;
+}
+
+static bool ieee80211_rx_data_set_sta(struct ieee80211_rx_data *rx,
+                                     struct ieee80211_sta *pubsta,
+                                     int link_id)
+{
+       struct sta_info *sta;
+
+       sta = container_of(pubsta, struct sta_info, sta);
+
+       rx->link_id = link_id;
+       rx->sta = sta;
+
+       if (sta) {
+               rx->local = sta->sdata->local;
+               if (!rx->sdata)
[... 320 lines cut ...]
-                       rx.sdata = prev_sta->sdata;
                        ieee80211_prepare_and_rx_handle(&rx, skb, false);

                        prev_sta = sta;
                }

                if (prev_sta) {
-                       if ((status->link_valid &&
-                            !ieee80211_rx_is_valid_sta_link_id(&prev_sta->sta,
-                                                               link_id)) ||
-                           (!status->link_valid && prev_sta->sta.mlo))
+                       rx.sdata = prev_sta->sdata;
+                       if (!ieee80211_rx_data_set_sta(&rx, &prev_sta->sta,
+                                                      link_id))
                                goto out;

-                       rx.link_id = status->link_valid ? link_id : -1;
-                       rx.sta = prev_sta;
-                       rx.sdata = prev_sta->sdata;
+                       if (!status->link_valid && prev_sta->sta.mlo)
+                               goto out;

                        if (ieee80211_prepare_and_rx_handle(&rx, skb, true))
                                return;
```

morrownr, 2023-01-08:
https://lore.kernel.org/all/20221230200747.19040-1-nbd@nbd.name/

patrakov, 2023-01-09:
Yes, the patch helps - thanks!

morrownr, 2023-01-09:
@patrakov 

What you could do that might help push things forward is add a follow on message to your report in Linux-wireless that shows the link to patch 5 above, then state that you have tested it and that it works with no apparent regressions.

jaccc, 2023-01-18:
MT7610U is also broken.

Slava-Nik, 2024-08-28:
@patrakov @morrownr Could you please tell me how can I use this patch you sent?
 I have the same null pointer dereference issue.
Kernel 6.8.11-arm64 (parallels virtual machine)
Chip: MT7610U

patrakov, 2024-08-28:
@Slava-Nik This patch is already included in your kernel. Therefore, the null pointer dereference that you see has a different cause.