openwrt/mt76 #728: MT7922 (RZ616) AP mode available?

aceboy50, 2023-01-11:
Hi, i wanna ask if i can use AMD RZ616 (rebranded MT7922) for AP mode yet? I'm using 22.03.3 with k5.10
Things i tried before posting: 
- Use firmware patch of MT7922 from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek. Pcie recognized, fw got patched in dmesg but cant scan, enable or add anything in wifi sections.
- Use fw patch of MT7921. Pcie recognized, fw got patch but shown up as generic wifi and inoperable.
I also read here (https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md) that i need to be at least on 5.19 to get AP mode working, the 7921 one was found working in many posts but 7922 nope, i wonder if there any specific step i need to do for it to work, is it still under WIP or got underated?

SerialVelocity, 2023-02-10:
Did you get this working? Just wondering if I should buy an RZ616 or RZ608

hiedaze, 2023-02-15:
Yes it's working, both RZ608 and RZ616. 
I made it works in all 2GHz, 5GHz and 6GHz, but only non-no-IR bands, which needs choosing suitable one. 
Since it's a non-dbdc NIC (till now), so you need to buy another one if you need dualband or more.

closed 2023-03-02 by hiedaze

cronop, 2023-03-07:
@hiedaze what did you have to do to make it work? Tx!

dellagd, 2023-04-29:
I would also like to know if there were specific steps that helped to make this functional! Running into issues which look driver related upon attempting to enable the interface in AP mode. Errors shown in `dmesg` like:

```
[   26.932967] mt7921e 0000:01:00.0: Message 00000010 (seq 14) timeout
[   26.933546] mt7921e 0000:01:00.0: Failed to release patch semaphore
[   27.020486] mt7921e 0000:01:00.0: HW/SW Version: 0x0, Build Time: 
[   27.024547] mt7921e 0000:01:00.0: WM Firmware Version: , Build Time: 
[   29.692744] mt7921e 0000:01:00.0: Timeout for initializing firmware
[   29.772967] mt7921e 0000:01:00.0: HW/SW Version: 0x0, Build Time: 
[   32.906292] mt7921e 0000:01:00.0: Message 00000010 (seq 6) timeout
[   32.906865] mt7921e 0000:01:00.0: Failed to release patch semaphore
[   32.990172] mt7921e 0000:01:00.0: HW/SW Version: 0x0, Build Time: 
[   32.994689] mt7921e 0000:01:00.0: WM Firmware Version: , Build Time: 
[   35.964864] mt7921e 0000:01:00.0: Timeout for initializing firmware
[   36.054019] mt7921e 0000:01:00.0: HW/SW Version: 0x0, Build Time: 
[   36.059178] mt7921e 0000:01:00.0: WM Firmware Version: , Build Time: 
[   39.056965] mt7921e 0000:01:00.0: Timeout for initializing firmware
[   39.139539] mt7921e 0000:01:00.0: HW/SW Version: 0x0, Build Time: 
[   39.144807] mt7921e 0000:01:00.0: WM Firmware Version: , Build Time: 
[   42.127921] mt7921e 0000:01:00.0: Timeout for initializing firmware
[   42.217301] mt7921e 0000:01:00.0: HW/SW Version: 0x0, Build Time: 
[   42.222453] mt7921e 0000:01:00.0: WM Firmware Version: , Build Time: 
[   45.225074] mt7921e 0000:01:00.0: Timeout for initializing firmware
[   45.311743] mt7921e 0000:01:00.0: hardware init failed
```

cronop, 2023-04-29:
@dellagd did you install the firmware? https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

dellagd, 2023-04-29:
Yep - I got the latest firmware (from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek) in the `/lib/firmware/mediatek` directory.

hiedaze, 2023-04-30:
> I would also like to know if there were specific steps that helped to make this functional! Running into issues which look driver related upon attempting to enable the interface in AP mode. Errors [...]

It indicates that there's a problem with your firmware. Which card do you own?

dellagd, 2023-04-30:
Facing a different problem now as it appears I was able to get the firmware to start loading properly. @hiedaze I filed a separate issue for it in https://github.com/openwrt/mt76/issues/775, where I also show what card I am using.

realJustinLee, 2023-07-13:
got this blog here: https://mans0n.github.io/2023/01/24/diy-6ghz-ap/
may be of help