openwrt/mt76 #734: mt7915 WED enabled router + Intel clients - strange behavior

levshutov, 2023-01-24:
Redmi AX6S (mt7622 soc with mt7915)
OpenWrt SNAPSHOT r21900-ac21dff5b6 (5.15.89 kernel)
mt76 @ git commit 6fe2c2383d3d9e537661b96f974408ddd719dff1
WAN - 1GbE DHCP
WLAN - AX 36ch 80MHz
Enabled HW Offloading and WED

Wan network has iperf3 server.

Wireless client have normal 500+ Mb/s iperf3 download speed for several seconds.
Then it drops less than 100 Mb/s for several seconds.
Then everything repeats.

In case client and router have direct traffic (pinging each other or refreshing LuCI in browser) during iperf3 test - no speed drops.

Disabling HW Offloading and/or WED - no speed drops.

Issue occurs on Intel cards AX200, AX1675, Advanced-N 6235 in Win11.

Issue does not occur on Realtek RTL8852AE in Win11.
Issue does not occur on Android/Iphone devices.

lukasz1992, 2023-01-25:
Have you tried to update drivers (especially AX200) to the most recent version?

levshutov, 2023-01-25:
> Have you tried to update drivers (especially AX200) to the most recent version?

Tried old and new drivers by Intel and Microsoft.

ryderlee1110, 2023-02-02:
Root cause  of this interoperability issue - there's no any keep alive packet being received by host stack => leads to mac80211 timer timeout => destroys BA session

ax series needs to maintain ba session for each 5s period to update timers.

```
--- a/net/mac80211/agg-tx.c
+++ b/net/mac80211/agg-tx.c
@@ -568,10 +568,9 @@ static void sta_tx_agg_session_timer_expired(struct timer_list *t)
 	}
 
 	timeout = tid_tx->last_tx + TU_TO_JIFFIES(tid_tx->timeout);
-	if (time_is_after_jiffies(timeout)) {
-		mod_timer(&tid_tx->session_timer, timeout);
-		return;
-	}
+	/* remove timerout handle for ax210 interoperability issue */
+	mod_timer(&tid_tx->session_timer, timeout);
+	return;
```

levshutov, 2023-02-02:
Thanks. It works.

ryderlee1110, 2023-02-02:
I think we can close this ticket.

@nbd168 just for your reference in case you have a better solution for upstream.

padavanonly, 2023-02-03:
Have you ever meet this issue with ax6s stock firmware ?

levshutov, 2023-02-07:
> Have you ever meet this issue with ax6s stock firmware ?

Didn't do much tests with stock firmware.

ryderlee1110, 2023-02-15:
@levshutov  could you help to test this patch https://pastebin.mozilla.org/SvDexZdv/raw ?

levshutov, 2023-02-19:
> @levshutov could you help to test this patch https://pastebin.mozilla.org/mWZ5tLU6/raw?

It works.

ryderlee1110, 2023-02-19:
We can close this ticket. I've posted to upstream. Maybe you can reply a acked-by or tested-by tag there.

closed 2023-02-19 by levshutov