openwrt/mt76 #735: ubus call iwinfo scan '{"device": "radio1"}' get wrong output

ptpt52, 2023-01-28:
`ubus call iwinfo scan '{"device": "radio1"}'` get all 2.4G wifi resaults
but `iwinfo phy1-ap0 scan` get the correct 5.8G wifi resaults.

tested on mt7915
device: wax202

also tested on RT-ACRH17 no issue

ThiloteE, 2023-01-29:
Not sure, if actually the same issue, but @RolandoMagico reported problems with scanning:

> Hi everyone,
> I'm currently using the snapshot from Friday and see the following issue:
> When scanning the wireless networks, independent which "Scan" button in Luci I use, I only see the 2.4GHz networks.
> Manually joining a 5GHz network works as expected.
> Can anybody confirm this issue.
> 
> Update: looks like it is still working in [https://github.com/RolandoMagico/openwrt/releases/tag/Build_2023-01-16 1](https://github.com/RolandoMagico/openwrt/releases/tag/Build_2023-01-16)
> 
> Thanks

and traced the regression window to

> f3d8de73988a79440d183050dc9711bb29d2a5e1 is the first bad commit
> commit f3d8de73988a79440d183050dc9711bb29d2a5e1
> Author: Christian Marangi <ansuelsmth@gmail.com>
> Date:   Sat Jan 21 01:23:22 2023 +0100
> 
> iwinfo: update to latest Git HEAD

    
    

Sources:

- https://forum.openwrt.org/t/openwrt-support-for-d-link-dap-x1860/141538/123?u=thilotee
- https://forum.openwrt.org/t/openwrt-support-for-d-link-dap-x1860/141538/131?u=thilotee
- https://forum.openwrt.org/t/openwrt-support-for-d-link-dap-x1860/141538/132?u=thilotee

dhewg, 2023-02-01:
My bad, see also https://github.com/openwrt/openwrt/issues/11902

Fix: http://patchwork.ozlabs.org/project/openwrt/patch/20230130185757.9512-1-a.heider@gmail.com/
Alternatively a patch for the OpenWrt tree it is in that PR