openwrt/mt76 #736: Belkin RT1800 snapshot r21950 - mt7915e fail to load

a-ser, 2023-01-29:
From dmesg:
[   11.717865] mt7915e_hif 0000:01:00.0: enabling device (0000 -> 0002)
[   11.735907] mt7915e 0000:02:00.0: enabling device (0000 -> 0002)
[   11.901753] mt7915e 0000:02:00.0: Direct firmware load for mediatek/mt7915_rom_patch.bin failed with error -2
[   11.911678] mt7915e 0000:02:00.0: Falling back to sysfs fallback for: mediatek/mt7915_rom_patch.bin
[   11.991635] mt7915e: probe of 0000:02:00.0 failed with error -12

Rolled back to r21599, where it works with the same config

ThiloteE, 2023-01-29:
cannot find commits r21950 and r21599. Please provide links to them

a-ser, 2023-01-29:
these are from nightly snapshots. should be the same as this:

[image]

ThiloteE, 2023-02-02:
is the device up to date and loads the correct firmwares? https://github.com/openwrt/openwrt/commit/9a07895729a9701a673951a31ceb71f16f0a6fad

andreagdipaolo, 2023-02-07:
I have the same problem on Belkin RT3200 r22000-6bc675c0be

```
[    6.832266] mt7915e 0000:01:00.0: assign IRQ: got 146
[    6.837430] mt7915e 0000:01:00.0: enabling device (0000 -> 0002)
[    6.843553] mt7915e 0000:01:00.0: enabling bus mastering
[    6.967378] mtk-pcie 1a143000.pcie: msi#0 address_hi 0x0 address_lo 0x44e2d0c0
[    6.987462] mt7915e 0000:01:00.0: Direct firmware load for mediatek/mt7915_rom_patch.bin failed with error -2
[    6.997419] mt7915e 0000:01:00.0: Falling back to sysfs fallback for: mediatek/mt7915_rom_patch.bin
[    7.147304] mt7915e: probe of 0000:01:00.0 failed with error -12
```

andreagdipaolo, 2023-02-14:
Solved by installing manually the relevant firmware packages.
AUC is not able to install them because it only reinstalls already installed packages (sort of).
I also needed to reinstall hostapd, it was gone, maybe because of the migration from wolfssl to mbedtls.