openwrt/mt76 #737: MAX RATE for mt7915 2.4G

myapple, 2023-01-31:
Hi all,
    I noticed that some wifi router like tplink wdr5620，using mtk 7615 , said they can support rate 400M/2.4G.
    but when I flash the device to openwrt 21.02 , the max link rate just is 300M.
    why is this, is it a bug of the driver ?

DragonBluep, 2023-01-31:
VHT rate support was rejected because no developers were interested. You can find the relevant patches in some other repositories like https://github.com/x-wrt/x-wrt.
https://github.com/openwrt/mt76/pull/333
https://github.com/openwrt/openwrt/pull/2522

myapple, 2023-01-31:
> VHT rate support was rejected because no developers were interested. You can find the relevant patches in some other repositories like https://github.com/x-wrt/x-wrt. #333 [openwrt/openwrt#2522](https://github.com/openwrt/openwrt/pull/2522)

thks

closed 2023-01-31 by myapple