openwrt/mt76 #745: Commit 98aa346 removed HE160 support from E8500 and RT3200 devices incorrectly.

andyleadbetter, 2023-02-10:
There devices work well with HE160, and this commit should not be merged for them.  Through put has halved on my device between r22036-1a145ccb0a and r20056-863288b49d

On r20056-863288b49d I can regularly transmit/recieve 800mbps 2m from the device HE160 on channel 100, on r22036-1a145ccb0a 2m from the device HE80 on channel 100 is only 450mpbs.

Why is this change deemed necessary, and what did it fix?

RSpliet, 2023-02-11:
Going through the manual for the RT3200, there is no mention of 160MHz support anywhere. That said, there is [at least one Wi-Fi certified certificate](https://api.cert.wi-fi.org/api/certificate/download/public?variantId=110362) that mentions an MT7915 with 160MHz support on the 5GHz band. It also mentions 6GHz support, which I don't expect from the MT7915E in my router - their product naming appears to be confusing and sadly opaque. That said, this certificate contradicts the claim made by the Mediatek employee in the mentioned patch that HE160 isn't supported on MT7915, although I can see that it might not be supported on _all_ MT7915.
The lack of communication makes me suspect that Mediatek has had to conclude recently that HW support for HE160 is broken on this gen of HW, and saw no way to fix it with driver and/or firmware workarounds. But that's of course just speculation, it'd be good to get a more definite answer on this question.

andyleadbetter, 2023-02-11:
I have seen online references suggesting the mimo config on rt3200 drops to 2+2 in 160mhz, instead of 4+4.  I have seen also on WiFi channel scanning that the device is definitely working at 160mhz on channels 64 and 100 in the UK.  

Empirically, I have had no issues and great performance with 160mhz bw on this device, whether officially supported or not.

Mushoz, 2023-02-16:
I have great performance on 160mhz on my rt3200 as well. Would be a shame if that is removed. It is true that it drops down to 2+2 instead of 4+4 spatial streams. But that's not such a big downside since not many clients are 4+4, nor do I experience high throughout from multiple 2+2 devices at the same time.

160mhz isn't strictly better, but in plenty of use cases it was better so I am sad to see it go

boecko, 2023-02-23:
Add me to the list. I had a rock solid 160MHz connection all the time. The iperf3-benchmark clocked at 1.34Gb/s.  

[image]
[image]

nbd168, 2023-03-01:
Please try this patch: https://nbd.name/p/d9c34f29

nbd168, 2023-03-02:
Support added back in f9ca70d6367a

closed 2023-03-02 by nbd168