openwrt/mt76 #746: MT7986 router - WED couldn't be enabled

levshutov, 2023-02-11:
Redmi AX6000 (MT7986 soc)
OpenWrt SNAPSHOT r22036-1a145ccb0a (5.15.92 kernel)
mt76 @ git commit 09c614734880865d1b4033b8cab4e9db86db4b15

Router hangs during boot after enabling WED
/etc/modules.conf:
`options mt7915e wed_enable=Y`

mt7915e module autoload can be disabled by removing /etc/modules.d/mt7915e
Running `modprobe mt7915e` hangs as well. It can't be killed.
Router keeps being accessible. Wifi doesn't up. Nothing interesting in logs.

ryderlee1110, 2023-02-14:
I can't repro this issue.

levshutov, 2023-02-14:
> I can't repro this issue.

What HW/SW do you have?

ryderlee1110, 2023-02-14:
mt7986  + latest openwrt.

levshutov, 2023-02-18:
Found the cause and solution.

**drivers/net/ethernet/mediatek/mtk_wed.c**

<details>
  <summary>mtk_wed_attach()</summary>

```
static int
mtk_wed_attach(struct mtk_wed_device *dev)
	__releases(RCU)
{
	struct mtk_wed_hw *hw;
	struct device *device;
	int ret = 0;

	RCU_LOCKDEP_WARN(!rcu_read_lock_held(),
			 "mtk_wed_attach without holding the RCU read lock");

	if ((dev->wlan.bus_type == MTK_WED_BUS_PCIE &&
	     pci_domain_nr(dev->wlan.pci_dev->bus) > 1) ||
	    !try_module_get(THIS_MODULE))
		ret = -ENODEV;

	rcu_read_unlock();

	if (ret)
		return ret;

	mutex_lock(&hw_lock);

	hw = mtk_wed_assign(dev);
	if (!hw) {
		module_put(THIS_MODULE);
		ret = -ENODEV;
		goto unlock;
	}

	device = dev->wlan.bus_type == MTK_WED_BUS_PCIE
		? &dev->wlan.pci_dev->dev
		: &dev->wlan.platform_dev->dev;
	dev_info(device, "attaching wed device %d version %d\n",
		 hw->index, hw->version);

	dev->hw = hw;
	dev->dev = hw->dev;
	dev->irq = hw->irq;
	dev->wdma_idx = hw->index;
	dev->version = hw->version;

	if (hw->eth->dma_dev == hw->eth->dev &&
	    of_dma_is_coherent(hw->eth->dev->of_node))
		mtk_eth_set_dma_device(hw->eth, hw->dev);

	ret = mtk_wed_tx_buffer_alloc(dev);
	if (ret)
		goto out;

	if (mtk_wed_get_rx_capa(dev)) {
		ret = mtk_wed_rro_alloc(dev);
		if (ret)
			goto out;
	}

	mtk_wed_hw_init_early(dev);
	if (hw->version == 1) {
		regmap_update_bits(hw->hifsys, HIFSYS_DMA_AG_MAP,
				   BIT(hw->index), 0);
	} else {
		dev->rev_id = wed_r32(dev, MTK_WED_REV_ID);
		ret = mtk_wed_wo_init(hw);
	}
out:
	if (ret)
		mtk_wed_detach(dev);
unlock:
	mutex_unlock(&hw_lock);

	return ret;
}
```
</details>

<details>
  <summary>mtk_wed_detach()</summary>

```
static void
mtk_wed_detach(struct mtk_wed_device *dev)
{
	struct mtk_wed_hw *hw = dev->hw;

	mutex_lock(&hw_lock);

	mtk_wed_deinit(dev);

	mtk_wdma_rx_reset(dev);
	mtk_wed_reset(dev, MTK_WED_RESET_WED);
	mtk_wed_free_tx_buffer(dev);
	mtk_wed_free_tx_rings(dev);

	if (mtk_wed_get_rx_capa(dev)) {
		mtk_wed_wo_reset(dev);
		mtk_wed_free_rx_rings(dev);
		mtk_wed_wo_deinit(hw);
	}

	if (dev->wlan.bus_type == MTK_WED_BUS_PCIE) {
		struct device_node *wlan_node;

		wlan_node = dev->wlan.pci_dev->dev.of_node;
		if (of_dma_is_coherent(wlan_node) && hw->hifsys)
			regmap_update_bits(hw->hifsys, HIFSYS_DMA_AG_MAP,
					   BIT(hw->index), BIT(hw->index));
	}

	if (!hw_list[!hw->index]->wed_dev &&
	    hw->eth->dma_dev != hw->eth->dev)
		mtk_eth_set_dma_device(hw->eth, hw->eth->dev);

	memset(dev, 0, sizeof(*dev));
	module_put(THIS_MODULE);

	hw->wed_dev = NULL;
	mutex_unlock(&hw_lock);
}
```
</details>

<details>
  <summary>mtk_wed_rro_alloc()</summary>

```
static int
mtk_wed_rro_alloc(struct mtk_wed_device *dev)
{
	struct reserved_mem *rmem;
	struct device_node *np;
	int index;

	index = of_property_match_string(dev->hw->node, "memory-region-names",
					 "wo-dlm");
	if (index < 0)
		return index;

	np = of_parse_phandle(dev->hw->node, "memory-region", index);
	if (!np)
		return -ENODEV;

	rmem = of_reserved_mem_lookup(np);
	of_node_put(np);

	if (!rmem)
		return -ENODEV;

	dev->rro.miod_phys = rmem->base;
	dev->rro.fdbk_phys = MTK_WED_MIOD_COUNT + dev->rro.miod_phys;

	return mtk_wed_rro_ring_alloc(dev, &dev->rro.ring,
				      MTK_WED_RRO_QUE_CNT);
}
```
</details>

Process hangs at `mutex_lock(&hw_lock);` in **mtk_wed_detach()**
Execution jumps to **mtk_wed_detach()** from **mtk_wed_attach()** before calling `mutex_unlock(&hw_lock);`
It jumps there because **mtk_wed_rro_alloc()** returns "error"
**mtk_wed_rro_alloc()** can't find "memory-region-names" DTS property
Some required MT7986 WED DTS properties are in patch https://www.spinics.net/lists/netdev/msg857681.html

Also **mt7986_wo_0.bin** firmware is required.

TX and RX offloading works now.

Summon @nbd168

levshutov, 2023-02-21:
> @levshutov I think this issue has been already fixed here: https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/commit/?id=587585e1bbebc250d46ebb0eeaf986d9c3c01c69
> 
> Is it available in your codebse?

It is not available in openwrt master branch.

nbd168, 2023-03-02:
I just added a package for the WO firmware and enabled it by default for the filogic subtarget

nbd168, 2023-03-05:
Added missing device tree entries in r22222-9f6d25551e71. I tested it and it seems to be working.

stoickish, 2023-03-05:
@nbd168 How did you test it?  I just made a build of r22223-2f8e90739a and set 'options mt7915e wed_enable=Y' in /etc/modules.conf and upon reboot the device just hangs.

levshutov, 2023-03-05:
> @nbd168 How did you test it? I just made a build of r22223-2f8e90739a and set 'options mt7915e wed_enable=Y' in /etc/modules.conf and upon reboot the device just hangs.
> 
> EDIT: Just to clarify, I'm using a Redmi AX6000

Try this patch https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/commit/?id=587585e1bbebc250d46ebb0eeaf986d9c3c01c69
It is not in openwrt yet.
This could possibly prevent hanging and see what breaking WED loading.
Also check /lib/firmware/mediatek/mt7986_wo_0.bin presence.

nbd168, 2023-03-05:
> @nbd168 How did you test it? I just made a build of r22223-2f8e90739a and set 'options mt7915e wed_enable=Y' in /etc/modules.conf and upon reboot the device just hangs.
> 
> EDIT: Just to clarify, I'm using a Redmi AX6000

Yes, that's how I enabled it. I also verified that traffic from ethernet to WLAN is actually offloaded.
Please make sure you enable the package ```mt7986-wo-firmware``` in your config.

stoickish, 2023-03-05:
> Yes, that's how I enabled it. I also verified that traffic from ethernet to WLAN is actually offloaded. Please make sure you enable the package `mt7986-wo-firmware` in your config.

Ah yeah, that was it.  Sorry about that.  Seems to be working now.

PussAzuki, 2023-03-07:
> Ah yeah, that was it. Sorry about that. Seems to be working now.

Have you tried the official daily build? I tried it and found that the computer connected through the network cable shows repeatedly disconnected and connected after startup, but I can enter the system normally again after deleting the dts part that WED needs.

I said bootloop in another issue because the behavior is somewhat similar...

Old official builds and X-WRT builds don't have this problem.

[my config](https://pastebin.com/e3xrAp0f)

PussAzuki, 2024-03-11:
[image]
[image]

Now does redmi ax6000 need to manually turn on WED again?