openwrt/mt76 #747: Belkin RT3200/MT7615E: WED + Intel AX200 = iwlwifi 0000:03:00.0: Unhandled alg: 0x703

RSpliet, 2023-02-11:
When enabling WED on my Belkin RT3200, the Intel AX200 occasionally shows the following message:
```iwlwifi 0000:03:00.0: Unhandled alg: 0x703```
I've monitored the same machine for a week with WED disabled and this issue does not appear. I suspect that occasional garbage packages reach the client. On older kernels I've seen this issue lead to firmware hangs/resets on the client side, but the log I'm currently looking at does not mention any such firmware resets. It may have gotten more robust with updates on either router- or client side.

Router specs:
- Belkin RT3200
- OpenWrt SNAPSHOT r21968-acd8e94d20 / LuCI Master git-23.013.73089-9634086 ("mt76: update PKG_SOURCE_HASH")
- Kernel 5.15.90
- WPA3 SAE
- 2.4GHz: .n, 40MHz
- 5GHz: .ac, 80MHz

Client specs:
- Fedora 37, kernel 6.1.8-200.fc37.x86_64
- Intel AX200, firmware version 72.daa05125.0 cc-a0-72.ucode op_mode iwlmvm

Note that this client-server combination also suffers from #724 , regardless of whether WED is enabled or disabled.

ryderlee1110, 2023-02-19:
Does this patch https://pastebin.mozilla.org/mWZ5tLU6/raw fix it?

RSpliet, 2023-02-21:
Thanks for this patch. Unfortunately I don't have a local kernel-build set-up for OpenWRT, I tend to always make use of the imagebuilder which pulls in a precompiled kernel. Sadly I don't have a lot of spare time to dive into creating one. Is there any other way I can test this change (e.g. a precompiled kernel I can obtain from elsewhere)?

RSpliet, 2023-03-03:
@ryderlee1110 Is the patch you proposed (which unfortunately already disappeared) equal or similar to b0721b96927b986f3682428e941ed4fb863d0737 ? If so: I just installed OpenWrt SNAPSHOT r22190-12a3c863d2 ("bmips: loader-lzma: drop unused RAMSIZE parameter"), which includes said commit. I re-enabled WED and I'll monitor my machine for the next few days to confirm whether there's improvement with this issue. With the caveat that I also re-enabled 160MHz now that it's available again.

RSpliet, 2023-03-03:
@ryderlee1110 Following up: I've had to downgrade back to the previous snapshot I had backed up because downspeed on 2.4GHz with the latest snapshot was abysmal. Didn't exceed 6Mbps either with or without WED. I'll open a separate bug report for that.

nbd168, 2023-03-03:
2.4 GHz throughput issue should be fixed in latest master

RSpliet, 2023-03-03:
@nbd168 Thanks, I'll give it a go again tomorrow when the snapshot builder had a go. And generally thanks (all!) for your hard work on the MT7622 platform; despite the occasional niggles this router on OpenWRT feels like a massive upgrade from ISP-issued routers and stock firmwares!

RSpliet, 2023-03-04:
Alas, the issue hasn't gone yet. I've started spotting these in my logs:
`[94357.662888] iwlwifi 0000:03:00.0: Unhandled alg: 0x707`
(No longer 703, but I'm not sure this signifies much!)

And also the ax200's firmware didn't like what's going on:
```
[97555.923479] iwlwifi 0000:03:00.0: Microcode SW error detected. Restarting 0x0.
[97555.923916] iwlwifi 0000:03:00.0: Start IWL Error Log Dump:
[97555.923919] iwlwifi 0000:03:00.0: Transport status: 0x0000004A, valid: 6
[97555.923923] iwlwifi 0000:03:00.0: Loaded firmware version: 72.daa05125.0 cc-a0-72.ucode
[97555.923927] iwlwifi 0000:03:00.0: 0x00000941 | ADVANCED_SYSASSERT          
[97555.923930] iwlwifi 0000:03:00.0: 0x00A022F0 | trm_hw_status0
[97555.923933] iwlwifi 0000:03:00.0: 0x00000000 | trm_hw_status1
[97555.923936] iwlwifi 0000:03:00.0: 0x004FBE16 | branchlink2
[97555.923938] iwlwifi 0000:03:00.0: 0x0000196C | interruptlink1
[97555.923940] iwlwifi 0000:03:00.0: 0x0000196C | interruptlink2
[97555.923943] iwlwifi 0000:03:00.0: 0x000001F3 | data1
[97555.923945] iwlwifi 0000:03:00.0: 0x00000000 | data2
[97555.923947] iwlwifi 0000:03:00.0: 0xFFFFFFFF | data3
[97555.923949] iwlwifi 0000:03:00.0: 0x00809DA6 | beacon time
[97555.923952] iwlwifi 0000:03:00.0: 0x0A34A281 | tsf low
[97555.923954] iwlwifi 0000:03:00.0: 0x0000000D | tsf hi
[97555.923956] iwlwifi 0000:03:00.0: 0x00000000 | time gp1
[97555.923959] iwlwifi 0000:03:00.0: 0xB5F056BC | time gp2
[97555.923961] iwlwifi 0000:03:00.0: 0x00000001 | uCode revision type
[97555.923963] iwlwifi 0000:03:00.0: 0x00000048 | uCode version major
[97555.923966] iwlwifi 0000:03:00.0: 0xDAA05125 | uCode version minor
[97555.923968] iwlwifi 0000:03:00.0: 0x00000340 | hw version
[97555.923970] iwlwifi 0000:03:00.0: 0x00C89000 | board version
[97555.923973] iwlwifi 0000:03:00.0: 0x80C1FD25 | hcmd
[97555.923975] iwlwifi 0000:03:00.0: 0x00021000 | isr0
[97555.923977] iwlwifi 0000:03:00.0: 0x00440000 | isr1
[97555.923979] iwlwifi 0000:03:00.0: 0x08F00002 | isr2
[97555.923982] iwlwifi 0000:03:00.0: 0x00C3680D | isr3
[97555.923984] iwlwifi 0000:03:00.0: 0x00000000 | isr4
[97555.923986] iwlwifi 0000:03:00.0: 0x0303001C | last cmd Id
[97555.923988] iwlwifi 0000:03:00.0: 0x00014DA6 | wait_event
[97555.923991] iwlwifi 0000:03:00.0: 0x000000D4 | l2p_control
[97555.923993] iwlwifi 0000:03:00.0: 0x00000020 | l2p_duration
[97555.923995] iwlwifi 0000:03:00.0: 0x00000007 | l2p_mhvalid
[97555.923998] iwlwifi 0000:03:00.0: 0x00000081 | l2p_addr_match
[97555.924000] iwlwifi 0000:03:00.0: 0x00000009 | lmpm_pmg_sel
[97555.924036] iwlwifi 0000:03:00.0: 0x00000000 | timestamp
[97555.924039] iwlwifi 0000:03:00.0: 0x0000A0A0 | flow_handler
[97555.924322] iwlwifi 0000:03:00.0: Start IWL Error Log Dump:
[97555.924324] iwlwifi 0000:03:00.0: Transport status: 0x0000004A, valid: 7
[97555.924327] iwlwifi 0000:03:00.0: 0x20000070 | NMI_INTERRUPT_LMAC_FATAL
[97555.924330] iwlwifi 0000:03:00.0: 0x00000000 | umac branchlink1
[97555.924332] iwlwifi 0000:03:00.0: 0x80455E3C | umac branchlink2
[97555.924334] iwlwifi 0000:03:00.0: 0x80472146 | umac interruptlink1
[97555.924337] iwlwifi 0000:03:00.0: 0x80472146 | umac interruptlink2
[97555.924339] iwlwifi 0000:03:00.0: 0x00000400 | umac data1
[97555.924341] iwlwifi 0000:03:00.0: 0x80472146 | umac data2
[97555.924343] iwlwifi 0000:03:00.0: 0x00000000 | umac data3
[97555.924346] iwlwifi 0000:03:00.0: 0x00000048 | umac major
[97555.924348] iwlwifi 0000:03:00.0: 0xDAA05125 | umac minor
[97555.924350] iwlwifi 0000:03:00.0: 0xB5F057B2 | frame pointer
[97555.924352] iwlwifi 0000:03:00.0: 0xC0886264 | stack pointer
[97555.924354] iwlwifi 0000:03:00.0: 0x00D10516 | last host cmd
[97555.924357] iwlwifi 0000:03:00.0: 0x00000000 | isr status reg
[97555.924621] iwlwifi 0000:03:00.0: IML/ROM dump:
[97555.924623] iwlwifi 0000:03:00.0: 0x00000003 | IML/ROM error/state
[97555.924978] iwlwifi 0000:03:00.0: 0x000061BF | IML/ROM data1
[97555.925259] iwlwifi 0000:03:00.0: 0x00000080 | IML/ROM WFPM_AUTH_KEY_0
[97555.925330] iwlwifi 0000:03:00.0: Fseq Registers:
[97555.925362] iwlwifi 0000:03:00.0: 0x60000000 | FSEQ_ERROR_CODE
[97555.925394] iwlwifi 0000:03:00.0: 0x80290021 | FSEQ_TOP_INIT_VERSION
[97555.925426] iwlwifi 0000:03:00.0: 0x00050008 | FSEQ_CNVIO_INIT_VERSION
[97555.925464] iwlwifi 0000:03:00.0: 0x0000A503 | FSEQ_OTP_VERSION
[97555.925496] iwlwifi 0000:03:00.0: 0x80000003 | FSEQ_TOP_CONTENT_VERSION
[97555.925529] iwlwifi 0000:03:00.0: 0x4552414E | FSEQ_ALIVE_TOKEN
[97555.925563] iwlwifi 0000:03:00.0: 0x00100530 | FSEQ_CNVI_ID
[97555.925594] iwlwifi 0000:03:00.0: 0x00000532 | FSEQ_CNVR_ID
[97555.925626] iwlwifi 0000:03:00.0: 0x00100530 | CNVI_AUX_MISC_CHIP
[97555.925659] iwlwifi 0000:03:00.0: 0x00000532 | CNVR_AUX_MISC_CHIP
[97555.925692] iwlwifi 0000:03:00.0: 0x05B0905B | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
[97555.925726] iwlwifi 0000:03:00.0: 0x0000025B | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR

```

So back to turning WED off I guess.