openwrt/mt76 #748: wifi: mt76: mt7915: BW160 and BW80+80 add support back!

CornelCotoara, 2023-02-13:
Please revert this removals regarding MT7915, MT7915 works just fine with HE160, been using this for over a month when I got this powerful client. Please see my tests in the comment below.
https://github.com/openwrt/mt76/commit/98aa346042bd21fe8c0b1594736edca16be436db#commitcomment-99775400
@nbd168 @MeiChia.Chiu

simontretter, 2023-02-19:
Same issue here. Have been using HE160 on mt7915e (AW7915-NP1 card) with 2gbit link speeds until support was taken out (now iw only reports HE80 capabilities).

DragonBluep, 2023-02-19:
HE160 has serious compatibility issues with Apple devices. Unless someone fix it, the 160 MHz won't come back.

CornelCotoara, 2023-02-19:
Wow, compatibility issues with Apple devices, that explains !all (mind the hairpin).
Btw, who's forcing you to even use Aple or HE160 if you have major issues with? 
I have done tests on my side,  HE160 has a huge gain against HE80 on my particular setup, so yes, for me it's a must have. The hardware supports HE160 so every owner  should be able to use that if needed.
Here's a small quote from OpenWrt repo itself,:
``
For developers, OpenWrt is the framework to build an application without having to build a complete firmware around it; for users this means the ability for full customization, to use the device in ways never envisioned.

Sunshine!
``

ThiloteE, 2023-02-19:
fix "what". Could you link to the issues?

CornelCotoara, 2023-02-19:
I can link the tests performed, they are placed as comments to the HE160 cutoff. 
https://github.com/openwrt/mt76/commit/98aa346042bd21fe8c0b1594736edca16be436db#commitcomment-100460501

DragonBluep, 2023-02-19:
> fix "what". Could you link to the issues?

https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466

ThiloteE, 2023-02-19:
> > fix "what". Could you link to the issues?
> 
> https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466

Thank you for your answer.

Just trying to summarize a little what we know here. People report 

1. 👍 The first comment mentioning that the new snapshot fixed "802-11ax-worse-than-802-11ac-with-mt76-driver" for rt3200 was https://github.com/openwrt/mt76/issues/709#issuecomment-1414438364 and since then multiple people have voiced that this update fixed rt3200 for them

   This was right after [mt76: update to the latest version, import WED related mtk_eth_soc patches](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=521efb62ebb55dcc67dc618757aa2e3ccbdf3774) was merged. This update of the mt76 driver indeed contained the commit "98aa346042bd wifi: mt76: mt7915: remove BW160 and BW80+80 support".

2. 👎  Unfortunately, it is not quite as clear, since this update contained A LOT of other commits as well...

3. 👍 👎  Also, there are [comments](https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/182) like these, which show that 160MHz can improve performance in certain situations, but decrease performance in others:
      
      > Is the slow down observed happening to just Apple devices? [@dsouza](https://forum.openwrt.org/u/dsouza)
      > 
      > I recently, took a spare RT3200 and used it as a [client only](https://openwrt.org/docs/guide-user/network/wifi/connect_client_wifi) just like a phone or a laptop. When I placed it behind the couch several walls away from the Redmi AX6000 AP, I could not reproduce the single digit iperf3 values that my iPhone 13 was reporting.
      > 
      > I am wondering if the AX slow down is only affecting Apple hardware.
      > 
      > Example 1. Redmi AX6000 on channel 149/80MHz width.
      > 
      > iphone 13: 6 Mbps
      > RT3200 client: 604 Mbps
      > 
      > Example 2. Redmi AX6000 on channel 100/160MHz width.
      > 
      > iphone 13: 515 Mbps
      > RT3200 client: 153 Mbps
      > 
      > I switched channels/width a total of 3 times. The results are consistent.
      > 

4. 👍 👎  [This report](https://github.com/openwrt/mt76/issues/745#issuecomment-1433315378) points to other cause why performance might deteroriate with 160 MHz in certain circumstances, but reports better performance with 160 MHz overall.

      > 
      > I have great performance on 160mhz on my rt3200 as well. Would be a shame if that is removed. It is true that it drops down to 2+2 instead of 4+4 spatial streams. But that's not such a big downside since not many clients are 4+4, nor do I experience high throughout from multiple 2+2 devices at the same time.
      > 
      > 160mhz isn't strictly better, but in plenty of use cases it was better so I am sad to see it go
      
To sum it all up: it probably should be investigated if it was really the removal of the 160 MHz that fixed things for some people, or if it was one of the other commits...

CornelCotoara, 2023-02-20:
It should definitely be investigated, as you mentioned there were many code changes in this regard included in the build of the openwrt snapshot image.
I'm no programming expert, but I find it hard to believe that butchering the ability to use HE160 for MT7915 (and related) miraculously solved the "AX" speed problems.
Personally, I don't have and haven't encountered these problems using the AX in HE160 mode (I've been using it this way for about 2 months already).
Correct me if I'm wrong, but having these capabilities available does not mean that they are used in default mode (does /etc/config/wireless have a default, right?), the choice is definitely up to the user.

riKirsti, 2023-02-20:
Based on my research those codes came from [wifi: mt76: mt7915: remove BW80+80 support](https://patchwork.kernel.org/project/linux-wireless/patch/20221229021131.9361-1-MeiChia.Chiu@mediatek.com/) and [[v2] wifi: mt76: mt7915: remove BW160 and BW80+80 support](https://patchwork.kernel.org/project/linux-mediatek/patch/20230104090352.31938-1-MeiChia.Chiu@mediatek.com/) by MeiChia Chiu (邱美嘉) <MeiChia.Chiu@mediatek.com>  and Money Wang (王信安) <Money.Wang@mediatek.com>. They seem to be the employees of MediaTek.

On the first path, MeiChia Chiu answered why.
> On 12/28/22 6:11 PM, MeiChia Chiu wrote:
> **mt7915 doesn't support bw80+80.**
>  You are removing a bunch of 160Mhz support flags as well, was that on
purpose?
>  From what I can tell, 7915 does support 160Mhz in 2x2 mode.
>  Thanks,
>  Ben

While on the second patch, MeiChia Chiu made some changes. 

> MeiChia Chiu <MeiChia.Chiu@mediatek.com> writes:
> **Remove BW160 and BW80+80 capability in mt7915.**
> This patch is already applied, so too late to change it, but I comment anyway as this is important:
> The commit log should _always_ answer to the question "why?" and explain
> the reason behind the patch. For example, I have no clue why you are
> removing these features from mt7915.

Since this is an open-source driver for OpenWRT. We don't need to obey the rules from MediaTek.

nbd168, 2023-02-20:
I will look into adding back support for 160 MHz, possibly guarded by a module parameter

nbd168, 2023-02-20:
Please test this patch: https://nbd.name/p/7945d62a

joaochainho, 2023-02-21:
Hi @nbd168 , many thanks for providing the patch.
I've tested it on my RT3200 and M1 Mac, and I didn't notice any significant decrease in performance with HE80  (base channel = 36). However, if I select HE160, there's a noticeable decrease in performance, namely upload, but not as bad as before.
Note: the M1 Mac only supports HE80.

Hope it helps.

nbd168, 2023-02-21:
It's normal that performance decreases in HE160 mode if your client doesn't support it, since that limits the available number of tx/rx chains to 2. 
I'd like to get some test feedback from people using HE160 with clients that support it.

CornelCotoara, 2023-02-21:
I need a hint about how the apply the patch , I've successfully setup a build container using alpine linux, but the image i build is missing the patch. 
```
wget https://nbd.name/p/7945d62a -O patches/01_restore_HE160.patch
quilt import patches/01_restore_HE160.patch
quilt push
Applying patch 01_restore_HE160.patch
can't find file to patch at input line 5
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/mt7915/init.c b/mt7915/init.c
|index ec297e047fe8..4f9188d825e1 100644
|--- a/mt7915/init.c
|+++ b/mt7915/init.c
--------------------------
No file to patch.  Skipping patch.
4 out of 4 hunks ignored
```

nbd168, 2023-02-21:
You could place the patch as package/kernel/mt76/patches/100-restore_HE160.patch and rebuild.

CornelCotoara, 2023-02-21:
I applied the patch, I can select HE160 but my client get only 300/300Mbps!
```
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r22124-bfa5e4e4eb
 -----------------------------------------------------
root@OpenWrt:~# iwinfo wl1-ap0 htmodelist
HT20 HT40 VHT20 VHT40 VHT80 HE20 HE40 HE80 HE160
```
```
netsh wlan show all
There is 1 interface on the system:

    Name                   : Wi-Fi
    Description            : Intel(R) Wi-Fi 6E AX211 160MHz
    GUID                   : f2066b41-7c39-4064-80b7-fff1bxxxxxxx
    Physical address       : f4:ce:23:xx:xx:xx
    State                  : connected
    SSID                   : OpenWrt_5G
    BSSID                  : 5c:02:14:b0:25:d9
    Network type           : Infrastructure
    Radio type             : 802.11n
    Authentication         : WPA3-Personal  (H2E)
    Cipher                 : CCMP
    Connection mode        : Profile
    Channel                : 100
    Receive rate (Mbps)    : 300
    Transmit rate (Mbps)   : 300
    Signal                 : 93%
    Profile                : OpenWrt_5G
```

ThiloteE, 2023-02-21:
> Radio type             : 802.11n

@CornelCotoara I assume this is your client. For some reason, it is configured to run with 802.11n mode, instead of 802.11ax...

CornelCotoara, 2023-02-21:
It's definitely configured as AX client.

CornelCotoara, 2023-02-21:
No luck, rebuilding de image from scratch did not help, the wifi is still capped at 300Mbps both ways like it shows in the above post.

nbd168, 2023-03-01:
Please try this patch: https://nbd.name/p/d9c34f29

CornelCotoara, 2023-03-01:
The patch is working! 
```
 OpenWrt SNAPSHOT, r22171-fdd1af9a44
 -----------------------------------------------------
root@OpenWrt:~# iwinfo wl1-ap0 htmodelist
HT20 HT40 VHT20 VHT40 VHT80 VHT160 HE20 HE40 HE80 HE160
```
```
 Name                   : Wi-Fi
    Description            : Intel(R) Wi-Fi 6E AX211 160MHz
    GUID                   : f2066b41-7c39-4064-80b7-ffffffffffff
    Physical address       : f4:ce:23:xx:xx:xx
    State                  : connected
    SSID                   : OpenWrt_5G
    BSSID                  : 5c:02:14:xx:xx:xx
    Network type           : Infrastructure
    Radio type             : 802.11ax
    Authentication         : WPA3-Personal  (H2E)
    Cipher                 : CCMP
    Connection mode        : Profile
    Channel                : 100
    Receive rate (Mbps)    : 2162
    Transmit rate (Mbps)   : 1729
    Signal                 : 92%
    Profile                : OpenWrt_5G
```
Iperf summary:
```
 .\iperf3.exe -c docker -u -b 2G -P 4 -V
iperf 3.1.3
CYGWIN_NT-10.0 ccornel-2ZB81T3 2.5.1(0.297/5/3) 2016-04-21 22:14 x86_64
Time: Wed, 01 Mar 2023 15:17:22 GMT
Connecting to host docker, port 5201
      Cookie: ccornel-2ZB81T3.1677683842.711089.0a
[truncated blah blah content]
[SUM]   0.00-10.00  sec  1.24 GBytes  1.06 Gbits/sec  0.000 ms  0/0 (0%)
CPU Utilization: local/sender 79.3% (14.7%u/64.6%s), remote/receiver 0.0% (0.0%u/0.0%s)
```

PS. I only need to add bridger, but i cannot find the package in make menuconfig, there's no "services" under network. Any clues on this? I can share the build if someone wanna test but i warn you that's a very basic build, no GUI included.

nbd168, 2023-03-01:
To enable bridger, you need to enable LLVM toolchain support. You can download the precompiled llvm toolchain if you want to and extract it to your openwrt source directory.

wired-dev, 2023-03-02:
ndb168; ty for fixing this with https://github.com/openwrt/mt76/commit/f9ca70d6367a16fecd80b93fec60fb0d4e3b9fd0

nbd168, 2023-03-02:
Added in f9ca70d6367a

closed 2023-03-02 by nbd168

Anteus, 2023-03-07:
This specific commit broke something on the rt3200 80MHz with some obstacles. Iperf went from around 400Mbit/s to 40Mbit/s.

csharper2005, 2023-07-09:
> HE160 has serious compatibility issues with Apple devices. Unless someone fix it, the 160 MHz won't come back.

AX 160MHz works perfectly with `kmod-7915e` on Mediatek filogic devices (mt7986).

Can't apple users disable 160 MHz on their APs?
[image]

darmach, 2023-10-27:
> Wow, compatibility issues with Apple devices, that explains !all (mind the hairpin). Btw, who's forcing you to even use Aple or HE160 if you have major issues with? I have done tests on my side, [...]

Right? If some specific group has problems, it's not like they can't disable it. Unlike enabling it when it's no longer supported - and for VR Meta Quest 3 Airlink 160MHz AX is the prerequisite, tried at 80MHz and it was unusable.