openwrt/mt76 #749: mt7915e driver with WED cause kernel error

levshutov, 2023-02-13:
Redmi AX6S (mt7622 soc with mt7915)
OpenWrt SNAPSHOT r21995-059263dd6e (5.15.91 kernel)
mt76 @ git commit b2360d59747c6fed2b65bc1c3563c10593c83f3e

Wifi dropped after several days uptime.
<details>
  <summary>dmesg</summary>

```
[772064.764675] ------------[ cut here ]------------
[772064.769402] Timeout waiting for MCU reset state 8
[772064.774283] WARNING: CPU: 0 PID: 19410 at mt7915_mcu_rf_regval+0x408/0x700 [mt7915e]
[772064.782170] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_CT wireguard pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_amanda nf_nat nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_recent xt_physdev xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_bpf xt_TCPMSS xt_NFQUEUE xt_LOG xt_HL xt_DSCP xt_CLASSIFY ts_kmp ts_fsm ts_bm slhc poly1305_neon
[772064.782697]  nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_objref nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_counter nft_compat nfnetlink_queue nf_tables nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables hwmon fou crc_ccitt compat br_netfilter asn1_decoder xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip_gre gre ip6_udp_tunnel udp_tunnel ip_tunnel tun seqiv leds_gpio gpio_button_hotplug
[772064.945896] CPU: 0 PID: 19410 Comm: kworker/u4:12 Not tainted 5.15.91 #0
[772064.952693] Hardware name: Xiaomi Redmi Router AX6S (DT)
[772064.958093] Workqueue: mt76 mt7915_mac_reset_work [mt7915e]
[772064.963787] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[772064.970844] pc : mt7915_mcu_rf_regval+0x408/0x700 [mt7915e]
[772064.976529] lr : mt7915_mcu_rf_regval+0x408/0x700 [mt7915e]
[772064.982211] sp : ffffffc00aa9bc60
[772064.985611] x29: ffffffc00aa9bc60 x28: ffffff8002ad9e28 x27: 0000000000000000
[772064.992854] x26: ffffff8002ad1e28 x25: 0000000000000000 x24: ffffff8002ad2020
[772065.000095] x23: 0000000000000000 x22: ffffff8002ad6020 x21: 0000000000000008
[772065.007334] x20: ffffff8002ad9e48 x19: 0000000000000000 x18: ffffffc008afa3a0
[772065.014575] x17: 0000000000000000 x16: 0000000000000000 x15: 000000000000052b
[772065.021814] x14: 00000000000011b9 x13: ffffffc00aa9b988 x12: ffffffc008b523a0
[772065.029056] x11: fffffffffffeab60 x10: fffffffffffeaaa0 x9 : 0000000000002958
[772065.036297] x8 : ffffffc008afa350 x7 : ffffffc008afa3a0 x6 : 0000000000000001
[772065.043537] x5 : ffffff800ff246f8 x4 : 0000000000000000 x3 : 0000000000000027
[772065.050777] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000025
[772065.058016] Call trace:
[772065.060549]  mt7915_mcu_rf_regval+0x408/0x700 [mt7915e]
[772065.065887]  mt7915_mac_reset_work+0x1f0/0xc6c [mt7915e]
[772065.071311]  process_one_work+0x210/0x3b0
[772065.075420]  worker_thread+0x170/0x4d0
[772065.079265]  kthread+0x11c/0x130
[772065.082588]  ret_from_fork+0x10/0x20
[772065.086259] ---[ end trace b00b5af8e158c26f ]---
[772095.484582] ------------[ cut here ]------------
[772095.489298] Timeout waiting for MCU reset state 20
[772095.494210] WARNING: CPU: 0 PID: 19410 at mt7915_mcu_rf_regval+0x408/0x700 [mt7915e]
[772095.502062] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_CT wireguard pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_amanda nf_nat nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_recent xt_physdev xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_bpf xt_TCPMSS xt_NFQUEUE xt_LOG xt_HL xt_DSCP xt_CLASSIFY ts_kmp ts_fsm ts_bm slhc poly1305_neon
[772095.502241]  nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_objref nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_counter nft_compat nfnetlink_queue nf_tables nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables hwmon fou crc_ccitt compat br_netfilter asn1_decoder xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip_gre gre ip6_udp_tunnel udp_tunnel ip_tunnel tun seqiv leds_gpio gpio_button_hotplug
[772095.665155] CPU: 0 PID: 19410 Comm: kworker/u4:12 Tainted: G        W         5.15.91 #0
[772095.673326] Hardware name: Xiaomi Redmi Router AX6S (DT)
[772095.678716] Workqueue: mt76 mt7915_mac_reset_work [mt7915e]
[772095.684380] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[772095.691421] pc : mt7915_mcu_rf_regval+0x408/0x700 [mt7915e]
[772095.697078] lr : mt7915_mcu_rf_regval+0x408/0x700 [mt7915e]
[772095.702735] sp : ffffffc00aa9bc60
[772095.706126] x29: ffffffc00aa9bc60 x28: ffffff8002ad9e28 x27: 0000000000000000
[772095.713343] x26: ffffff8002ad1e28 x25: 0000000000000000 x24: ffffff8002ad2020
[772095.720561] x23: 0000000000000000 x22: ffffff8002ad6020 x21: 0000000000000020
[772095.727779] x20: ffffff8002ad9e48 x19: 0000000000000000 x18: ffffffc008afa3a0
[772095.734997] x17: ffffffc00745a000 x16: ffffffc008004000 x15: 0000000000000585
[772095.742214] x14: 00000000000011d7 x13: ffffffc00aa9b988 x12: ffffffc008b523a0
[772095.749431] x11: fffffffffffec2b8 x10: fffffffffffec1f0 x9 : 0000000000002c28
[772095.756649] x8 : ffffffc008afa350 x7 : ffffffc008afa3a0 x6 : 0000000000000001
[772095.763867] x5 : ffffff800ff246f8 x4 : 0000000000000000 x3 : 0000000000000027
[772095.771083] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000026
[772095.778300] Call trace:
[772095.780825]  mt7915_mcu_rf_regval+0x408/0x700 [mt7915e]
[772095.786135]  mt7915_mac_reset_work+0x314/0xc6c [mt7915e]
[772095.791532]  process_one_work+0x210/0x3b0
[772095.795626]  worker_thread+0x170/0x4d0
[772095.799454]  kthread+0x11c/0x130
[772095.802762]  ret_from_fork+0x10/0x20
[772095.806419] ---[ end trace b00b5af8e158c270 ]---
[772115.964549] mt7915e 0000:01:00.0: Message 000026ed (seq 12) timeout
[... 99 lines cut ...]
[337717.841171] mt7915e 0000:01:00.0: Message 0000aded (seq 4) timeout
[337738.321102] mt7915e 0000:01:00.0: Message 00005aed (seq 5) timeout
[337758.801035] mt7915e 0000:01:00.0: Message 0000aded (seq 6) timeout
[337779.281001] mt7915e 0000:01:00.0: Message 00005aed (seq 7) timeout
[337799.760942] mt7915e 0000:01:00.0: Message 0000aded (seq 8) timeout
[337820.240887] mt7915e 0000:01:00.0: Message 00005aed (seq 9) timeout
[337840.720839] mt7915e 0000:01:00.0: Message 0000aded (seq 10) timeout
[337861.200782] mt7915e 0000:01:00.0: Message 00005aed (seq 11) timeout
[337881.680728] mt7915e 0000:01:00.0: Message 0000aded (seq 12) timeout
[337902.160680] mt7915e 0000:01:00.0: Message 00005aed (seq 13) timeout
[... cut ...]
[338045.520320] mt7915e 0000:01:00.0: Message 00005aed (seq 5) timeout
[338066.010286] mt7915e 0000:01:00.0: Message 00005aed (seq 6) timeout
[338086.480244] mt7915e 0000:01:00.0: Message 000025ed (seq 7) timeout
[338086.486564] wl1-ap0: HW problem - can not stop rx aggregation for 22:cd:9d:9d:fb:5d tid 0
[338106.960166] mt7915e 0000:01:00.0: Message 000025ed (seq 8) timeout
[338106.966496] wl1-ap0: HW problem - can not stop rx aggregation for 22:cd:9d:9d:fb:5d tid 4
[338127.440117] mt7915e 0000:01:00.0: Message 00005aed (seq 9) timeout
```
</details>

levshutov, 2023-02-19:
The issue can be triggered with 
`echo 1 > /sys/kernel/debug/ieee80211/wl1/mt76/sys_recovery`

Enabled WED is required to reproduce.

ryderlee1110, 2023-02-19:
https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/fd8e1153eec03fb5a76c7d83b10d4b3150c49de4

Can you check mtk repo and apply wed changes as we don't see the issue? I think it's most likely SER related thing.

levshutov, 2023-02-25:
> https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/fd8e1153eec03fb5a76c7d83b10d4b3150c49de4
> 
> Can you check mtk repo and apply wed changes as we don't see the issue? I think it's most likely SER related thing.

Thanks.
I don't see any new SER related code that is not presented in mt76 and openwrt repos.
Also the issue is not reproduced on mt7986 based device (Redmi AX6000) which share wifi/wed/ethernet code with mt7622.

Djfe, 2024-02-09:
Is this issue still present? @levshutov

dashernln, 2024-03-03:
@Djfe i think this is actual issue on current stable 23.05.2, i have 2 redmi ax6s/ax3200, on both periodically i can see in dmesg similar messages (with WED enabled) and dead 5GHz wifi.

Djfe, 2024-03-04:
good to know because the last post in this issue was a year old.