openwrt/mt76 #754: mt7915e: Belkin RT3200 - Timeout waiting for MCU reset state

Fail-Safe, 2023-02-27:
I just experienced a series of mt7915e crashes. Hopefully the provided stack traces provide enough info to identify the issue.

Build info:
```
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r22133-eaf5c85fd3"
OPENWRT_BOARD="mediatek/mt7622"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r22133-eaf5c85fd3"
```

Stack traces:
```
[215557.481510] ------------[ cut here ]------------
[215557.486224] Timeout waiting for MCU reset state 8
[215557.491043] WARNING: CPU: 1 PID: 29183 at mt7915_mcu_rf_regval+0x408/0x6f4 [mt7915e]
[215557.498888] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 cfg80211 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon compat cls_flower act_vlan cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev autofs4 seqiv authencesn authenc leds_gpio gpio_button_hotplug
[215557.569268] CPU: 1 PID: 29183 Comm: kworker/u4:1 Tainted: G S                5.15.94 #0
[215557.577351] Hardware name: Linksys E8450 (UBI) (DT)
[215557.582308] Workqueue: mt76 mt7915_mac_reset_work [mt7915e]
[215557.587968] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[215557.595009] pc : mt7915_mcu_rf_regval+0x408/0x6f4 [mt7915e]
[215557.600665] lr : mt7915_mcu_rf_regval+0x408/0x6f4 [mt7915e]
[215557.606321] sp : ffffffc00ee73c50
[215557.609713] x29: ffffffc00ee73c50 x28: ffffff8002864968 x27: ffffff80028625d0
[215557.616931] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff8002862020
[215557.624149] x23: 0000000000000000 x22: ffffff8002866020 x21: 0000000000000008
[215557.631366] x20: ffffff8002869d88 x19: 0000000000000000 x18: ffffffc008aba320
[215557.638583] x17: 0000000000ff1740 x16: 2d1a2d0b01036c60 x15: 0000000000000471
[215557.645800] x14: 000000000000017b x13: ffffffc00ee73978 x12: ffffffc008b12320
[215557.653017] x11: 4d20726f6620676e x10: ffffffc008b12320 x9 : 0000000000000000
[215557.660234] x8 : ffffffc008aba2d0 x7 : ffffffc008aba320 x6 : 0000000000000001
[215557.667451] x5 : ffffff801feb46f8 x4 : 0000000000000000 x3 : 0000000000000027
[215557.674668] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000025
[215557.681886] Call trace:
[215557.684409]  mt7915_mcu_rf_regval+0x408/0x6f4 [mt7915e]
[215557.689719]  mt7915_mac_reset_work+0x218/0xd0c [mt7915e]
[215557.695115]  process_one_work+0x210/0x3b0
[215557.699209]  worker_thread+0x170/0x4d0
[215557.703037]  kthread+0x11c/0x130
[215557.706346]  ret_from_fork+0x10/0x20
[215557.710002] ---[ end trace 24de3e00bec6051a ]---
[215588.201120] ------------[ cut here ]------------
```

```
[215588.201120] ------------[ cut here ]------------
[215588.205834] Timeout waiting for MCU reset state 20
[215588.210740] WARNING: CPU: 1 PID: 29183 at mt7915_mcu_rf_regval+0x408/0x6f4 [mt7915e]
[215588.218585] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 cfg80211 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon compat cls_flower act_vlan cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev autofs4 seqiv authencesn authenc leds_gpio gpio_button_hotplug
[215588.288966] CPU: 1 PID: 29183 Comm: kworker/u4:1 Tainted: G S      W         5.15.94 #0
[215588.297049] Hardware name: Linksys E8450 (UBI) (DT)
[215588.302006] Workqueue: mt76 mt7915_mac_reset_work [mt7915e]
[215588.307666] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[215588.314707] pc : mt7915_mcu_rf_regval+0x408/0x6f4 [mt7915e]
[215588.320363] lr : mt7915_mcu_rf_regval+0x408/0x6f4 [mt7915e]
[215588.326020] sp : ffffffc00ee73c50
[... 86 lines cut ...]
[220798.558867] Call trace:
[220798.561391]  ___ieee80211_stop_tx_ba_session+0x348/0x3ac [mac80211]
[220798.567762]  ieee80211_sta_tear_down_BA_sessions+0x74/0x130 [mac80211]
[220798.574387]  ieee80211_find_sta_by_link_addrs+0x370/0x540 [mac80211]
[220798.580839]  sta_info_destroy_addr_bss+0x38/0x70 [mac80211]
[220798.586510]  ieee80211_color_change_finish+0x1278/0x1500 [mac80211]
[220798.592876]  cfg80211_check_station_change+0x1384/0x4720 [cfg80211]
[220798.599238]  genl_family_rcv_msg_doit+0xb4/0x110
[220798.603940]  genl_rcv_msg+0xd0/0x1c0
[220798.607596]  netlink_rcv_skb+0x58/0x120
[220798.611512]  genl_rcv+0x34/0x50
[220798.614733]  netlink_unicast+0x1f0/0x2ec
[220798.618735]  netlink_sendmsg+0x19c/0x3d0
[220798.622738]  ____sys_sendmsg+0x258/0x2a0
[220798.626744]  ___sys_sendmsg+0x78/0xc0
[220798.630485]  __sys_sendmsg+0x54/0xb0
[220798.634139]  __arm64_sys_sendmsg+0x20/0x30
[220798.638313]  invoke_syscall+0x44/0x110
[220798.642143]  el0_svc_common.constprop.0+0x48/0xf0
[220798.646926]  do_el0_svc+0x18/0x20
[220798.650319]  el0_svc+0x14/0x50
[220798.653454]  el0t_64_sync_handler+0xe0/0x110
[220798.657802]  el0t_64_sync+0x158/0x15c
[220798.661545] ---[ end trace 24de3e00bec60528 ]---
```

levshutov, 2023-03-01:
https://github.com/openwrt/mt76/issues/749

Fail-Safe, 2023-03-02:
Additional info, and admittedly this could be unrelated, but I only started having this issue when I switched my `scaling_governor` to `performance`. i.e. `echo performance > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor`

Since opening this issue, I have switched back to `ondemand` governor and have not yet had a reoccurrence of this issue. Will update here if that changes.

CharlesMengCA, 2023-07-10:
Same here, luci lost response.

r23545_5.15.120

```
Mon Jul 10 11:41:14 2023 kern.warn kernel: [126247.750004] ------------[ cut here ]------------
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.754752] Timeout waiting for MCU reset state 8
Mon Jul 10 11:41:14 2023 kern.warn kernel: [126247.759629] WARNING: CPU: 0 PID: 11439 at 0xffffffc000b27288 [mt7915e@00000000bbaddc9d+0x1f000]
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.768450] Modules linked in: ksmbd nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 cfg80211 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon compat cls_flower act_vlan ntfs3 cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact asn1_decoder oid_registry nls_utf8 seqiv md5 md4 des_generic libdes nls_iso8859_1 nls_cp437 usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug vfat fat exfat usbcore usb_common [last unloaded: ksmbd]
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.853380] CPU: 0 PID: 11439 Comm: kworker/u4:2 Tainted: G S                5.15.120 #0
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.861567] Hardware name: Linksys E8450 (UBI) (DT)
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.866533] Workqueue: mt76 0xffffffc000b29344 [mt7915e@00000000bbaddc9d+0x1f000]
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.874126] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.881183] pc : 0xffffffc000b27288 [mt7915e@00000000bbaddc9d+0x1f000]
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.887810] lr : 0xffffffc000b27288 [mt7915e@00000000bbaddc9d+0x1f000]
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.894436] sp : ffffffc00a91bc50
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.897835] x29: ffffffc00a91bc50 x28: ffffff8002911ec0 x27: ffffff80029125f8
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.905077] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff8002912020
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.912317] x23: 0000000000000000 x22: ffffff8002916020 x21: 0000000000000008
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.919557] x20: ffffff8002919ee0 x19: 0000000000000000 x18: ffffffc008a3a338
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.926797] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000528
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.934036] x14: 00000000000001b8 x13: ffffffc00a91b978 x12: ffffffc008a92338
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.941276] x11: 4d20726f6620676e x10: ffffffc008a92338 x9 : 0000000000000000
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.948517] x8 : ffffffc008a3a2e8 x7 : ffffffc008a3a338 x6 : 0000000000000001
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.955756] x5 : ffffff801fea26f8 x4 : 0000000000000000 x3 : 0000000000000027
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.962995] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000025
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.970234] Call trace:
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.972768]  0xffffffc000b27288 [mt7915e@00000000bbaddc9d+0x1f000]
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.979048]  0xffffffc000b29524 [mt7915e@00000000bbaddc9d+0x1f000]
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.985327]  0xffffffc008050980
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.988556]  0xffffffc008050c90
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.991785]  0xffffffc008058e5c
Mon Jul 10 11:41:14 2023 kern.debug kernel: [126247.995014]  0xffffffc00801581c
Mon Jul 10 11:41:14 2023 kern.warn kernel: [126247.998244] ---[ end trace 4b71a22a675d885b ]---
Mon Jul 10 11:41:44 2023 kern.warn kernel: [126278.469723] ------------[ cut here ]------------
Mon Jul 10 11:41:44 2023 kern.debug kernel: [126278.474481] Timeout waiting for MCU reset state 20
Mon Jul 10 11:41:44 2023 kern.warn kernel: [126278.479436] WARNING: CPU: 0 PID: 11439 at 0xffffffc000b27288 [mt7915e@00000000bbaddc9d+0x1f000]
Mon Jul 10 11:41:44 2023 kern.debug kernel: [126278.488249] Modules linked in: ksmbd nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 cfg80211 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon compat cls_flower act_vlan ntfs3 cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact asn1_decoder oid_registry nls_utf8 seqiv md5 md4 des_generic libdes nls_iso8859_1 nls_cp437 usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug vfat fat e [... cut ...]
Mon Jul 10 12:00:51 2023 kern.debug kernel: [127425.638559]  0xffffffc0080115cc
Mon Jul 10 12:00:51 2023 kern.warn kernel: [127425.641779] ---[ end trace 4b71a22a675d885d ]---
Mon Jul 10 12:01:11 2023 kern.err kernel: [127445.829525] mt7915e 0000:01:00.0: Message 00005aed (seq 4) timeout
Mon Jul 10 12:01:32 2023 kern.err kernel: [127466.299340] mt7915e 0000:01:00.0: Message 000025ed (seq 5) timeout
Mon Jul 10 12:01:32 2023 kern.debug kernel: [127466.305766] ------------[ cut here ]------------

```

lukasz1992, 2023-07-11:
which openwrt vesion do you have?

CharlesMengCA, 2023-07-11:
Kernel 5.15.120 snapshot - r23530 - wolfssl: update to 5.6.3

I think it could be related to memory leak somewhere.  

After reboot, everything start working for a while and above error happens again and the luci lost response.

quarkysg, 2023-07-20:
My E8450 mt7913e crashed as well with the following log:

```
[574799.857830] ------------[ cut here ]------------
[574799.862573] Timeout waiting for MCU reset state 8
[574799.867445] WARNING: CPU: 0 PID: 30082 at mt7915_mcu_rf_regval+0x1f8/0x4e0 [mt7915e]
[574799.875322] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD wireguard pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT ebtable_nat ebtable_filter ebtable_broute chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbnet slhc poly1305_neon nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_objref nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_counter nft_compat nf_tables nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libcrc32c libchacha iptable_mangle iptable_filter ipt_ECN
[574799.875851]  ipheth ip_tables hwmon ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat cls_flower act_vlan cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_gre ip_gre gre ip6_udp_tunnel udp_tunnel sit ipip ip6_tunnel tunnel6 tunnel4 ip_tunnel tun seqiv usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common mii
[574800.045334] CPU: 0 PID: 30082 Comm: kworker/u4:5 Tainted: G S                5.15.120 #0
[574800.053510] Hardware name: Linksys E8450 (UBI) (DT)
[574800.058470] Workqueue: mt76 mt7915_mac_reset_work [mt7915e]
[574800.064148] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[574800.071194] pc : mt7915_mcu_rf_regval+0x1f8/0x4e0 [mt7915e]
[574800.076859] lr : mt7915_mcu_rf_regval+0x1f8/0x4e0 [mt7915e]
[574800.082523] sp : ffffffc00a91bc50
[574800.085916] x29: ffffffc00a91bc50 x28: ffffff8003524a00 x27: ffffff80035225f8
[574800.093142] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff8003522020
[574800.100366] x23: 0000000000000000 x22: ffffff8003526020 x21: 0000000000000008
[574800.107589] x20: ffffff8003529e20 x19: 0000000000000000 x18: ffffffc008aba2b8
[574800.114813] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000dce
[574800.122037] x14: 000000000000049a x13: ffffffc00a91b978 x12: ffffffc008b122b8
[574800.129262] x11: 4d20726f6620676e x10: ffffffc008b122b8 x9 : 0000000000000000
[574800.136486] x8 : ffffffc008aba268 x7 : ffffffc008aba2b8 x6 : 0000000000000001
[574800.143710] x5 : ffffff801fea46f8 x4 : 0000000000000000 x3 : 0000000000000027
[574800.150934] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000025
[574800.158158] Call trace:
[574800.160684]  mt7915_mcu_rf_regval+0x1f8/0x4e0 [mt7915e]
[574800.166002]  mt7915_mac_reset_work+0x200/0xd70 [mt7915e]
[574800.171406]  process_one_work+0x210/0x3b0
[574800.175503]  worker_thread+0x170/0x4d0
[574800.179336]  kthread+0x11c/0x130
[574800.182653]  ret_from_fork+0x10/0x20
[574800.186314] ---[ end trace 6329951489e774e4 ]---
[574830.577434] ------------[ cut here ]------------
[574830.582164] Timeout waiting for MCU reset state 20
[574830.587067] WARNING: CPU: 0 PID: 30082 at mt7915_mcu_rf_regval+0x1f8/0x4e0 [mt7915e]
[574830.594912] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD wireguard pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT ebtable_nat ebtable_filter ebtable_broute chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbnet slhc poly1305_neon nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_objref nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_counter nft_compat nf_tables nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libcrc32c libchacha iptable_mangle iptable_filter ipt_ECN
[574830.595097]  ipheth ip_tables hwmon ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat cls_flower act_vlan cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_ [... cut ...]
[574912.497482] mt7915e 0000:01:00.0: Message 0000aded (seq 3) timeout
[574932.976082] mt7915e 0000:01:00.0: Message 000800c4 (seq 4) timeout
[574953.455875] mt7915e 0000:01:00.0: Message 00005aed (seq 5) timeout
[574973.935544] mt7915e 0000:01:00.0: Message 0000aded (seq 6) timeout
[574994.415271] mt7915e 0000:01:00.0: Message 000800c4 (seq 7) timeout
[575014.894981] mt7915e 0000:01:00.0: Message 00005aed (seq 8) timeout
[575035.374716] mt7915e 0000:01:00.0: Message 0000aded (seq 9) timeout
```

darekxan, 2023-07-25:
```
[482057.966563] Timeout waiting for MCU reset state 8
[482057.971573] WARNING: CPU: 0 PID: 20992 at mt7915_mcu_rf_regval+0x1f8/0x4f0 [mt7915e]
[482057.979469] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount xt_state xt_helper xt_conntrack xt_connmark xt_connbytes xt_CT pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbnet slhc nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_objref nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_counter nf_tables nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ipt_ECN ipheth ip_tables hwmon crc_ccitt compat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet
[482057.980016]  ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun seqiv leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common mii
[482058.105587] CPU: 0 PID: 20992 Comm: kworker/u4:2 Tainted: G S                5.15.120 #0
[482058.113775] Hardware name: Linksys E8450 (UBI) (DT)
[482058.118743] Workqueue: mt76 mt7915_mac_reset_work [mt7915e]
[482058.124444] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[482058.131503] pc : mt7915_mcu_rf_regval+0x1f8/0x4f0 [mt7915e]
[482058.137187] lr : mt7915_mcu_rf_regval+0x1f8/0x4f0 [mt7915e]
[482058.142869] sp : ffffffc00c603c50
[482058.146270] x29: ffffffc00c603c50 x28: ffffff8002a51ec0 x27: ffffff8002a525f8
[482058.153511] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff8002a52020
[482058.160751] x23: 0000000000000000 x22: ffffff8002a56020 x21: 0000000000000008
[482058.167990] x20: ffffff8002a59ee0 x19: 0000000000000000 x18: ffffffc008b0a338
[482058.175230] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000405
[482058.182469] x14: 0000000000000157 x13: ffffffc00c603978 x12: ffffffc008b62338
[482058.189709] x11: 4d20726f6620676e x10: ffffffc008b62338 x9 : 0000000000000000
[482058.196948] x8 : ffffffc008b0a2e8 x7 : ffffffc008b0a338 x6 : 0000000000000001
[482058.204186] x5 : ffffff801fea26f8 x4 : 0000000000000000 x3 : 0000000000000027
[482058.211425] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000025
[482058.218663] Call trace:
[482058.221197]  mt7915_mcu_rf_regval+0x1f8/0x4f0 [mt7915e]
[482058.226536]  mt7915_mac_reset_work+0x1e0/0xcdc [mt7915e]
[482058.231959]  process_one_work+0x210/0x3b0
[482058.236067]  worker_thread+0x170/0x4d0
[482058.239910]  kthread+0x11c/0x130
[482058.243240]  ret_from_fork+0x10/0x20
[482058.246912] ---[ end trace 7263ee33ab80cd84 ]---
```

```
482088.681325] ------------[ cut here ]------------
[482088.686067] Timeout waiting for MCU reset state 20
[482088.691004] WARNING: CPU: 0 PID: 20992 at mt7915_mcu_rf_regval+0x1f8/0x4f0 [mt7915e]
[482088.698859] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount xt_state xt_helper xt_conntrack xt_connmark xt_connbytes xt_CT pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbnet slhc nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_objref nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_counter nf_tables nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ipt_ECN ipheth ip_tables hwmon crc_ccitt compat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet
[482088.699052]  ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun seqiv leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common mii
[482088.824481] CPU: 0 PID: 20992 Comm: kworker/u4:2 Tainted: G S      W         5.15.120 #0
[482088.832653] Hardware name: Linksys E8450 (UBI) (DT)
[482088.837610] Workqueue: mt76 mt7915_mac_reset_work [mt7915e]
[482088.843279] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[482088.850321] pc : mt7915_mcu_rf_regval+0x1f8/0x4f0 [mt7915e]
[482088.855979] lr : mt7915_mcu_rf_regval+0x1f8/0x4f0 [mt7915e]
[482088.861635] sp : ffffffc00c603c50
[482088.865027] x29: ffffffc00c603c50 x28: ffffff8002a51ec0 x27: ffffff8002a525f8
[... cut ...]
[482088.922763] x5 : ffffff801fea26f8 x4 : 0000000000000000 x3 : 0000000000000027
[482088.929980] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000026
[482088.937198] Call trace:
[482088.939722]  mt7915_mcu_rf_regval+0x1f8/0x4f0 [mt7915e]
[482088.945035]  mt7915_mac_reset_work+0x210/0xcdc [mt7915e]
[482088.950433]  process_one_work+0x210/0x3b0
[482088.954526]  worker_thread+0x170/0x4d0
[482088.958355]  kthread+0x11c/0x130
[482088.961666]  ret_from_fork+0x10/0x20
[482088.965323] ---[ end trace 7263ee33ab80cd85 ]---
```

darekxan, 2023-07-25:
```
[484566.747162] WARNING: CPU: 1 PID: 2065 at ___ieee80211_stop_tx_ba_session+0x34c/0x3b0 [mac80211]
[484566.756061] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount xt_state xt_helper xt_conntrack xt_connmark xt_connbytes xt_CT pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbnet slhc nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_objref nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_counter nf_tables nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ipt_ECN ipheth ip_tables hwmon crc_ccitt compat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet
[484566.756605]  ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun seqiv leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common mii
[484566.882175] CPU: 1 PID: 2065 Comm: hostapd Tainted: G S      W         5.15.120 #0
[484566.889840] Hardware name: Linksys E8450 (UBI) (DT)
[484566.894805] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[484566.901862] pc : ___ieee80211_stop_tx_ba_session+0x34c/0x3b0 [mac80211]
[484566.908653] lr : ___ieee80211_stop_tx_ba_session+0x214/0x3b0 [mac80211]
[484566.915433] sp : ffffffc00a243780
[484566.918833] x29: ffffffc00a243780 x28: ffffff8002696780 x27: ffffffc00a243dd0
[484566.926076] x26: ffffff8000066080 x25: ffffff8002a50880 x24: ffffffc008bd1000
[484566.933316] x23: ffffffc000c217a4 x22: ffffff80036320e8 x21: 0000000000000003
[484566.940557] x20: ffffff8006369a00 x19: ffffff8003630000 x18: ffffffc008b0a338
[484566.947797] x17: 0000000000000000 x16: 0000000000000000 x15: 00000000000008d0
[484566.955037] x14: 00000000000002f0 x13: 000b2fda304c46a0 x12: ffffffc008847678
[484566.962276] x11: 0000088d2b52f28b x10: ffffff8000ad45c0 x9 : 0000000000000002
[484566.969516] x8 : 0000000000569190 x7 : 0000000000000001 x6 : 0000000000000000
[484566.976754] x5 : 0000000000000000 x4 : ffffff801feb8380 x3 : 0000000000000000
[484566.983993] x2 : 0000000000000001 x1 : 0000000000000002 x0 : 00000000ffffff92
[484566.991234] Call trace:
[484566.993767]  ___ieee80211_stop_tx_ba_session+0x34c/0x3b0 [mac80211]
[484567.000202]  ieee80211_sta_tear_down_BA_sessions+0x74/0x130 [mac80211]
[484567.006896]  ieee80211_find_sta_by_link_addrs+0x390/0x580 [mac80211]
[484567.013415]  sta_info_destroy_addr_bss+0x38/0x70 [mac80211]
[484567.019152]  ieee80211_color_change_finish+0x1c18/0x1e80 [mac80211]
[484567.025584]  cfg80211_check_station_change+0x1384/0x4700 [cfg80211]
[484567.031984]  genl_family_rcv_msg_doit+0xb4/0x110
[484567.036703]  genl_rcv_msg+0xd0/0x1c0
[484567.040374]  netlink_rcv_skb+0x58/0x120
[484567.044304]  genl_rcv+0x34/0x50
[484567.047538]  netlink_unicast+0x1f0/0x2ec
[484567.051554]  netlink_sendmsg+0x19c/0x3d0
[484567.055569]  ____sys_sendmsg+0x27c/0x2b0
[484567.059585]  ___sys_sendmsg+0x80/0xf0
[484567.063341]  __sys_sendmsg+0x44/0xa0
[484567.067011]  __arm64_sys_sendmsg+0x20/0x30
[484567.071201]  invoke_syscall.constprop.0+0x4c/0xe0
[484567.076000]  do_el0_svc+0x40/0xd0
[484567.079408]  el0_svc+0x14/0x50
[484567.082564]  el0t_64_sync_handler+0xe0/0x110
[484567.086935]  el0t_64_sync+0x15c/0x160
[484567.090691] ---[ end trace 7263ee33ab80cd8b ]---
[484587.215173] mt7915e 0000:01:00.0: Message 00005aed (seq 6) timeout
[484607.694917] mt7915e 0000:01:00.0: Message 000025ed (seq 7) timeout
[484628.174746] mt7915e 0000:01:00.0: Message 0000aded (seq 8) timeout
[484648.654515] mt7915e 0000:01:00.0: Message 00005aed (seq 9) timeout
[484669.134296] mt7915e 0000:01:00.0: Message 000800c4 (seq 10) timeout
[484689.614084] mt7915e 0000:01:00.0: Message 00005aed (seq 11) timeout
[484710.093865] mt7915e 0000:01:00.0: Message 000025ed (seq 12) timeout
[484710.100292] wl1-ap0: HW problem - can not stop rx aggregation for 2e:4f:69:71:1b:86 tid 0
[484730.573661] mt7915e 0000:01:00.0: Message 00005aed (seq 13) timeout
[484751.053430] mt7915e 0000:01:00.0: Message 000025ed (seq 14) timeout
[484751.059857] wl1-ap0: HW problem - can not stop rx aggregation for 2e:4f:69:71:1b:86 tid 1
[484771.533233] mt7915e 0000:01:00.0: Message 00005aed (seq 15) timeout
[484792.013016] mt7915e 0000:01:00.0: Message 000025ed (seq 1) timeout
[484792.019347] wl1-ap0: HW problem - can not stop rx aggregation for 2e:4f:69:71:1b:86 tid 5
[484812.492813] mt7915e 0000:01:00.0: Message 00005aed (seq 2) timeout
[484832.972581] mt7915e 0000:01:00.0: Message 000025ed (seq 3) timeout
[484832.978956] wl1-ap0: HW problem - can not stop rx aggregation for 2e:4f:69:71:1b:86 tid 6
[484853.452453] mt7915e 0000:01:00.0: Message 00005aed (seq 4) timeout
[484873.932162] mt7915e 0000:01:00.0: Message 000025ed (seq 5) timeout
[484873.938591] ------------[ cut here ]------------
[... cut ...]
[484874.255707]  ___sys_sendmsg+0x80/0xf0
[484874.259463]  __sys_sendmsg+0x44/0xa0
[484874.263132]  __arm64_sys_sendmsg+0x20/0x30
[484874.267323]  invoke_syscall.constprop.0+0x4c/0xe0
[484874.272122]  do_el0_svc+0x40/0xd0
[484874.275529]  el0_svc+0x14/0x50
[484874.278685]  el0t_64_sync_handler+0xe0/0x110
[484874.283055]  el0t_64_sync+0x15c/0x160
[484874.286810] ---[ end trace 7263ee33ab80cd8c ]---
```

Not sure which part is relevant, 5G radio becomes unavailable with that log after few (3-5) days

quarkysg, 2023-08-03:
Would like to chime in that since using the mt7915_wa.bin firmware found by @rany2 from this [post](https://github.com/openwrt/mt76/issues/690#issuecomment-1557036194), my Linksys E8450 did not encounter anymore timeout issue (yet?) and has been up more than 6 days now.

Could be a lucky co-incidence and the crash may be coming soon.

Will continue to monitor.

andyleadbetter, 2023-08-03:
Is it possible to use the bin file without a rebuild? I dropped it in lib/modules but it did not survive reboot

Fail-Safe, 2023-08-03:
Should be placed in `/lib/firmware/mediatek`, I believe. Can't speak to whether that survives reboots, but I think it should.

andyleadbetter, 2023-08-04:
Confirmed too, not sure what happened first time, suspect it was an unsynced reboot on overlay.

quarkysg, 2023-08-10:
> Would like to chime in that since using the mt7915_wa.bin firmware found by @rany2 from this https://github.com/openwrt/mt76/issues/690#issuecomment-1557036194, my Linksys E8450 did not encounter anymore timeout issue (yet?) and has been up more than 6 days now.
> 
> Could be a lucky co-incidence and the crash may be coming soon.
> 
> Will continue to monitor.

Bad news folks.  My E8450 mt7915e driver crashed again with timeout today, after 13 days of uptime.  So looks like it's back to square one.

Fail-Safe, 2023-08-10:
@quarkysg Have you tried this yet? 

https://github.com/rany2/mt76/commit/4108cca1fac04d58581d6840a2e43b1e42d03c31.patch

I’m testing it now (for about three days) and haven’t hit my daily crash I typically experience.

quarkysg, 2023-08-11:
@Fail-Safe yup, testing it now.  @rany2 said there's another patch that's required tho:

https://github.com/rany2/mt76/commit/16eae0c9ee29981b695abca45a9907938e1bc95e?diff=unified

Am testing both patches now, in addition to the mt7915_wa.bin firmware as well.

Fail-Safe, 2023-08-11:
Were you using these two patches when you hit your last crash after 13 days?

quarkysg, 2023-08-11:
@Fail-Safe no.  When my E8450, I only had the mt7915_wa.bin replaced.

Now I’m testing the .bin file and both patches suggested by @rany2.  Unfortunately I do not know what will trigger the timeout, so I’ll have to monitor the router.  Only 14 hours uptime, so it’s early days yet.

Fail-Safe, 2023-08-18:
I hate to admit it, but I am resorting to this for now:

`/root/check-for-crash.sh`:
```sh
#!/bin/sh

dmesg | grep -Fq -e '------------[ cut here ]------------'

if [ $? -eq 0 ]; then
  now_date=`date +"%Y%m%d-%H%M%S"`
  dmesg > "/root/crash-log_${now_date}.log"
  reboot
fi
```

`/etc/crontabs/root`:
```sh
*/5 * * * * /root/check-for-crash.sh
```

So every five minutes, if I find a "------------[ cut here ]------------" in my kernel log, I just dump the kernel log out to a dated file and reboot the darn AP. At least that gets my 5Ghz radio back online. Otherwise, I often go for hours (or days) before I think to go check if the crash has occurred.

Not elegant, but until there is a proper solution available this will at least increase quality of ~life~ bandwidth for now.

quarkysg, 2023-08-20:
@Fail-Safe I’m also using a cron job running every 10mins to detect mt7915e timeout and force a kernel panic if any is detected in syslog.

The good news is that my E8450 is running without issues (touch wood) for more than 9 days with the two suggested patches by @rany2.  Will continue to monitor.

quarkysg, 2023-08-24:
@Fail-Safe my E8450 still crashed with the timeout issue even with the suggested patches from @rany2.

Back to square one.

Fail-Safe, 2023-08-25:
I'm sorry, @quarkysg. Mine have also crashed multiple times even with the same patches. Not sure where to go next.

andyleadbetter, 2023-08-25:
Same here, just had 7915e failure Timeout waiting for MCU reset state 20
[217085.929629] Timeout waiting for MCU reset state 20
[217085.934608] WARNING: CPU: 1 PID: 16218 at 0xffffffc000a40288 [mt7915e@0000000051da8509+0x1f000]
[217085.943436] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 chacha_neon cfg80211 slhc poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha hwmon crc_ccitt compat ip6_udp_tunnel udp_tunnel seqiv leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common
[217086.016694] CPU: 1 PID: 16218 Comm: kworker/u4:0 Tainted: G S      W         5.15.127 #0
[217086.024864] Hardware name: Linksys E8450 (UBI) (DT)
[217086.029821] Workqueue: mt76 0xffffffc000a42344 [mt7915e@0000000051da8509+0x1f000]
[217086.037394] pstate: 20400005 (nzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[217086.044435] pc : 0xffffffc000a40288 [mt7915e@0000000051da8509+0x1f000]
[217086.051043] lr : 0xffffffc000a40288 [mt7915e@0000000051da8509+0x1f000]
[217086.057650] sp : ffffffc0091d3c50
[217086.061041] x29: ffffffc0091d3c50 x28: ffffff8002771ec0 x27: ffffff80027725d0
[217086.068258] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff8002772020
[217086.075475] x23: 0000000000000000 x22: ffffff8002776020 x21: 0000000000000020
[217086.082693] x20: ffffff8002779ee0 x19: 0000000000000000 x18: ffffffc008a3a338
[217086.089910] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000525
[217086.097128] x14: 00000000000001b7 x13: ffffffc0091d3978 x12: ffffffc008a92338
[217086.104344] x11: 434d20726f662067 x10: ffffffc008a92338 x9 : 0000000000000000
[217086.111562] x8 : ffffffc008a3a2e8 x7 : ffffffc008a3a338 x6 : 0000000000000001
[217086.118778] x5 : ffffff801feb36f8 x4 : 0000000000000000 x3 : 0000000000000027
[217086.125996] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000026
[217086.133213] Call trace:
[217086.135737]  0xffffffc000a40288 [mt7915e@0000000051da8509+0x1f000]
[217086.141998]  0xffffffc000a42554 [mt7915e@0000000051da8509+0x1f000]
[217086.148259]  0xffffffc008050930
[217086.151477]  0xffffffc008050c40
[217086.154695]  0xffffffc008058e0c
[217086.157914]  0xffffffc00801581c

Fail-Safe, 2023-08-25:
@andyleadbetter && @quarkysg Did you see this commit? Somehow I missed it.

I'm going to give this a try and see what happens: 

https://github.com/rany2/mt76/commit/5565bfed100e1835eb825092c300cfcab2d2e3a8

Fail-Safe, 2023-08-25:
> Same here, just had 7915e failure Timeout waiting for MCU reset state 20 [217085.929629] Timeout waiting for MCU reset state 20 [217085.934608] WARNING: CPU: 1 PID: 16218 at 0xffffffc000a40288 [...]

What firmware versions were you running at the time?

quarkysg, 2023-08-28:
Not sure if I've stumble across something here that may be the root cause?

This [commit](https://github.com/openwrt/mt76/commit/4046b59183234a503e8fb017549c6d66576e941d) changed the way that mt7915e beacon capability was configured, but this [commit](https://github.com/openwrt/mt76/commit/c50be0b54cddba864637bc0a5b13be2d89494103) reverted it.  But the reversion of the logic seem to miss out the `ht_ldpc` flag.  The way that the flags are set seems to be different as well.  The later code is just a copy from existing config.

Could this have somehow confused the firmware and causing the MCU timeout for mt7915e?

Wonder if we just revert the behaviour to what is was before both changes would fix the timeout issue.

Anyone familiar with 80211 protocols and mt7915e can comment?

rany2, 2023-08-30:
@nbd168 I'll test it on clean mt76 without any modifications. Just a note, I still do see "MCU timeout" with my changes after a week of use but they are not fatal and SER kicks in successfully.

rany2, 2023-08-30:
It doesn't help with this issue: https://github.com/openwrt/mt76/issues/776

levshutov, 2023-08-30:
I also experience "MCU timeout" issue when WED is enabled. It occurs when SER (System Error Recovery) process starts.

I use debugfs to trigger SER for debuging/testing
`echo 1 > /sys/kernel/debug/ieee80211/wl1/mt76/sys_recovery`

I tryed @nbd168 mt76 repo + patch and @rany2 mt76 repo. I got "MCU timeout" everywhere.

According to @ryderlee1110 comment https://github.com/openwrt/mt76/issues/749#issuecomment-1436072703 MTK builds don't have this issue.

rany2, 2023-08-30:
> I got "MCU timeout" everywhere.

It's OK if you see the "MCU timeout" in your log as long as the chip recovers. If it recovers after a minute then it should be fine it's a different story if the chip/driver is stuck in this state and hangs.

levshutov, 2023-08-30:
> It's OK if you see the "MCU timeout" in your log as long as the chip recovers. If it recovers after a minute then it should be fine it's a different story if the chip/driver is stuck in this state and hangs.

Every time I see "Timeout waiting for MCU reset state" in kernel log SER never stops. More and more error messages in infinite loop. It always stucks for me. System load more than 1. `reboot` doesn't work.

Have you ever tryed to trigger SER manualy?

ryderlee1110, 2023-08-30:
> Every time I see "Timeout waiting for MCU reset state" in kernel log SER never stops. More and more error messages in infinite loop. It always stucks for me. System load more than 1. `reboot` [...]

How did you trigger SER manually?

rany2, 2023-08-30:
> Can you please test if this patch improves stability? https://nbd.name/p/aebc85e5

No idea if it improved it, but it didn't hurt.  I still have the same issue with openwrt mt76 + your patch though.

levshutov, 2023-08-30:
> How did you trigger SER manually?

`echo 1 > /sys/kernel/debug/ieee80211/wl1/mt76/sys_recovery`

ryderlee1110, 2023-08-30:
> > How did you trigger SER manually?
> 
> `echo 1 > /sys/kernel/debug/ieee80211/wl1/mt76/sys_recovery`

can't  it recover back?

levshutov, 2023-08-30:
> can't it recover back?

SER can't finish successfully.

quarkysg, 2023-08-31:
> > Can you please test if this patch improves stability? https://nbd.name/p/aebc85e5
> 
> No idea if it improved it, but it didn't hurt. I still have the same issue with openwrt mt76 + your patch though.
I’m starting to suspect the timeout issue is caused by firmware rather than driver.  Most likely the firmware for mt7915 (for me at least) that was changed sometime around Nov ‘22.  Looks like firmware is returning a NULL skb data structure to mt7915 driver and that gets reported as a timeout event.

The main problem is that there’s no deterministic way to trigger it, so cannot really tell if a fix works.  Can only tell if it does not work and that really depends on whether the bug gets triggered.

Still waiting for the next timeout for me.  This time tho. I’ve changed one of my E8450 to use channel 149 instead of auto, to rule out issue with radar scanning.  Still running after 6 days.  I have another E8450 that’s still running after 19 days, so it’s a mystery what triggers the timeout, and it’s on auto channel 104 at the moment.

ryderlee1110, 2023-08-31:
> 

At least SER works on our boards. We can manually trigger it without issues ... we don’t have WED enabled by default

ryderlee1110, 2023-08-31:
> > can't it recover back?
> 
> SER can't finish successfully.

At least SER works on our boards. We can manually trigger it without issues ... we don’t have WED enabled by default

quarkysg, 2023-08-31:
@ryderlee1110 do you think this [commit ](https://github.com/openwrt/mt76/commit/dd682eead016a49f1efe24d361c5fbc149339a52) has anything to do with the timeout issues?

Increasing the message max size likely have to match what the firmware is expecting?  Message size increased by 256 bytes.

levshutov, 2023-08-31:
> The main problem is that there’s no deterministic way to trigger it, so cannot really tell if a fix works.

It is
`echo 1 > /sys/kernel/debug/ieee80211/wl1/mt76/sys_recovery`

> At least SER works on our boards. We can manually trigger it without issues ... we don’t have WED enabled by default

The issue occurs only when WED is enabled.

quarkysg, 2023-08-31:
> > The main problem is that there’s no deterministic way to trigger it, so cannot really tell if a fix works.
> 
> It is `echo 1 > /sys/kernel/debug/ieee80211/wl1/mt76/sys_recovery`
> 

Whoa ... thanks for this tip!  My E8450 is throwing timeout immediately after this!

Now to find the fix ... hopefully.

Fail-Safe, 2023-08-31:
> Turning off `wed` seems to not trigger the timeout. So looks like `wed` is the cause or part of the root cause. With `wed` enabled, 100% able to trigger the timeout (edit: even with no client connected to the 5G interface, i.e. fresh from boot.)

Is this true even with the patch @nbd168 provided for testing [here](https://github.com/openwrt/mt76/issues/754#issuecomment-1698680854)?

rany2, 2023-08-31:
> The issue occurs only when WED is enabled.

This is not true with my MT7905D. I do not have WED enabled yet I still get the MCU timeout + hang.

rany2, 2023-08-31:
> Turning off `wed` seems to not trigger the timeout. So looks like `wed` is the cause or part of the root cause. With `wed` enabled, 100% able to trigger the timeout (edit: even with no client connected to the 5G interface, i.e. fresh from boot.)

I think it might be that it accelerates it but it's not the cause as I don't have WED enabled. When I'm using mt76 from this repo for it keeps crashing especially when I have DAWN enabled (which sends a lot of 802.11k beacon requests (every 20 seconds by default) to all clients connected to the AP). 

If I don't have DAWN running, it takes weeks for a crash to occur.

As to whether it is a firmware issue or not, I don't think it is entirely. I have found that using [this firmware](https://github.com/rany2/mt76/commit/e8927b64e07520aba769988551caad70ae20df81) only helped me with the UDP flood issue. However I also needed https://github.com/rany2/mt76/commit/a73a7c18586866c7aa1e67ef152860680453806d.  I confirmed that these two individually somehow fixed the issue on my chip, even after https://github.com/openwrt/mt76/commit/1f0fbbee359c1ec4d2ed0235225ffb676f4c90bf got added into openwrt mt76 (it wouldn't apply to my chip anyway as this is mt7916 issue only)

Fail-Safe, 2023-08-31:
> I think it might be that it accelerates it but it's not the cause as I don't have WED enabled. When I'm using mt76 from this repo for it keeps crashing especially when I have DAWN enabled (which [...]

Interesting observations! I do have WED enabled (per my OP), but that alone doesn't seem to trigger the issue.

In my scenario, I have three RT3200s each ethernet connected back to a core switch. Each RT3200 is running as a basic WAP with three VLANs/SSIDs.

However, I run [Usteer](https://git.openwrt.org/?p=project/usteer.git;a=summary) on each AP. I have long suspected this may also be contributing to the speed and frequency at which the issue presents. I seem to be fairly commonly able to trigger the issue under actual usage by taking my particular M1 MacBook toward my living room AP (thus triggering a roaming condition). This is the AP that crashes far more often than either of the other two. It is also the AP with the highest number of connected STAs, typically.

FWIW, I have reverted back to this firmware for now and am running a fresh build with the patch @nbd168 put out for us a few days back:
```sh
[    9.281864] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20220929104145
[    9.334697] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20220929104205
```

levshutov, 2023-08-31:
> This is not true with my MT7905D. I do not have WED enabled yet I still get the MCU timeout + hang.

This is also not true for MT7986. It doesn't have this issue even with WED enabled.

quarkysg, 2023-08-31:
> Is this true even with the patch @nbd168 provided for testing [here](https://github.com/openwrt/mt76/issues/754#issuecomment-1698680854)?

I think @rany2 tested it and said the patch did not help with the timeout issue?  I'll give it a go as well.

> This is also not true for MT7986. It doesn't have this issue even with WED enabled.

Now it seems to be clear to me that I started seeing the timeout issues right around the time I started playing with `wed`.  So I guess for `mt7915e` + `mt7622`, `wed` is maybe playing a big role in the issue.

ryderlee1110, 2023-09-01:
@levshutov  Does SER work when you revert this change https://github.com/openwrt/mt76/commit/d152c8688c1437e2d026192bd1f0cad25faec572 ?

ryderlee1110, 2023-09-01:
> > This is not true with my MT7905D. I do not have WED enabled yet I still get the MCU timeout + hang.
> 
> This is also not true for MT7986. It doesn't have this issue even with WED enabled.

Are you using the up-to-date mt76 version?

levshutov, 2023-09-01:
> Are you using the up-to-date mt76 version?

I am using Openwrt build r22571-00a240e77f at 2023-04-14 on MT7986 based Redmi AX6000.

ryderlee1110, 2023-09-01:
> > Are you using the up-to-date mt76 version?
> 
> I am using Openwrt build r22571-00a240e77f at 2023-04-14 on MT7986 based Redmi AX6000.

What about your MT7622+MT7915? My MT7986 + MT7915 wed works fine.

levshutov, 2023-09-01:
> What about your MT7622+MT7915? My MT7986 + MT7915 wed works fine.

My MT7622+MT7915 (Redmi AX6S, Xiaomi AX3200, Linksys E8450) have the issue on latest build and had it 7 months ago.
My MT7986+MT7915 (Redmi AX6000) is always fine.

ryderlee1110, 2023-09-02:
@nbd168 I don't have mt7622 for checking right now, but 7986 works fine. Could you please test if mt7622 works after reverting this commithttps://github.com/openwrt/mt76/commit/d152c8688c1437e2d026192bd1f0cad25faec572 ? I guess at this point https://github.com/openwrt/mt76/commit/f83958376085908e3298350293fc3d9ba469e097 it should work at least.

nbd168, 2023-09-02:
Already tried reverting d152c8688c1437e2d026192bd1f0cad25faec572 - doesn't work. Switching to commit f83958376085908e3298350293fc3d9ba469e097 also doesn't work, so there probably is a regression in the WED code as well.

quarkysg, 2023-10-11:
@nbd168 @ryderlee1110, I was looking thru the codes to try to identify potential cause of the timeout, and not sure if this could be of interest to both of you.

In the function [mt7915_mac_reset_work()](https://github.com/openwrt/mt76/blob/4a0f839da0f108837baad908ed12b550f493b912/mt7915/mac.c#L1483), early on in the code it checks if `wed `is active and if it is, disabled the wireless device.

Code subsequent disable `DMA`, and a lot of other stuffs and wait for timeout.

It looks to me that the device may have been disabled too early, causing the timeout?

rany2, 2023-12-23:
Removing this bit of code here: https://github.com/openwrt/mt76/blob/bebd9cffc2aeb2cecb40aadbb8c6eab3bdf7971b/mt7915/mac.c#L1523-L1527 makes it so that the "Timeout waiting for MCU reset state" error is not printed anymore and L1 SER completes. HOWEVER, while this makes it so that L1 SER completes "successfully" the radio doesn't _actually_ work. This is on mt7622+mt7915. Maybe this is helpful to figure out the root of the issue, @nbd168

rany2, 2024-01-19:
Seems like https://github.com/openwrt/openwrt/commit/2c5b3bee38eb0d76f5afe90f9969b73e19d28685 might fix the issue, but I haven't tested it yet.

rany2, 2024-01-19:
Confirmed that it is resolved for me. L1 recovery works again with WED enabled.

quarkysg, 2024-01-26:
Six days running the latest master build (Linksys E8450) with WED and H/W offload enabled.  Looking good so far.

Djfe, 2024-02-09:
@Fail-Safe you might want to close this issue then? :)

[5 more replies]

closed 2024-02-10 by Fail-Safe