openwrt/mt76 #755: mt7915e: joining ac device crashes complete mesh

PolynomialDivision, 2023-02-28:
I have following mesh setup:
```
       [TP-Link EAP 225 Outdoor (QCA9886 ath10k)]
         / \                             / \
          |                               |
         \ /                             \ /
[Netgear WAX202 (mt7915e)]  <-------> [ZyXEL NWA55AXE (mt7915e)]
```
Configuration:
- Netgear WAX202: HE20
- ZyXEL NWA55AXE: HE20
- TP-Link EAP 225 Outdoor: VHT20

If I start only the Netgear WAX202 and ZyXEL NWA55AXE everything is fine. Mesh is working and also the bitrate is correct:
```
$ iw dev wlan5-mesh station dump && iw dev wlan2-mesh station dump) | grep -E '(Station|(tx|rx) bitrate)'
Station 34:98:b5:0f:37:62 (on wlan5-mesh)
	tx bitrate:	32.4 MBit/s HE-MCS 1 HE-NSS 2 HE-GI 1 HE-DCM 0
	rx bitrate:	68.8 MBit/s HE-MCS 3 HE-NSS 2 HE-GI 0 HE-DCM 0
```

However, when I start the TP-Link EAP 225 nothing works anymore. Sometimes the complete mesh communication breaks down and sometimes only partial. In all cases, the communication between the two ax devices does not work at all. Only the communication to the TP-Link EAP 225 might be possible.

```
Station b8:ec:a3:e1:22:05 (on wlan5-mesh)
	tx bitrate:	6.0 MBit/s
	rx bitrate:	68.8 MBit/s HE-MCS 3 HE-NSS 2 HE-GI 0 HE-DCM 0
Station d8:47:32:3f:70:19 (on wlan5-mesh)
	tx bitrate:	28.9 MBit/s VHT-MCS 1 short GI VHT-NSS 2
	rx bitrate:	86.7 MBit/s VHT-MCS 4 short GI VHT-NSS 2

```

PolynomialDivision, 2023-02-28:
Surprisingly, when I set everything to HT20, it seems to work.

PolynomialDivision, 2023-03-02:
Updated to latest openwrt master. It does not help. :/

DragonBluep, 2023-03-03:
It seems like that low tx bitrate issue is related to https://github.com/openwrt/openwrt/commit/2f96580c5282cd138d2bb858f7f27f539c8a63e1

PolynomialDivision, 2023-03-15:
After some time the mesh between two ax station again broke. Now it is in a weird state. If I do a ping `ff02::1%wlan5-mesh`, the request arrive at the other router, but they don't respond to that echo request.

PolynomialDivision, 2023-03-18:
Okay, if I enable mesh_fwding on the mesh interface, it does not crash anymore.

ryderlee1110, 2023-04-01:
Do you still observe mt7915 mesh related issue?

PolynomialDivision, 2023-04-01:
> Do you still observe mt7915 mesh related issue?

Yes. Mesh communication still freezing after some time.

ryderlee1110, 2023-04-01:
Could you please upgrade to the latest openwrt?

PolynomialDivision, 2023-04-07:
> Could you please upgrade to the latest openwrt?

Did not help. :/

ryderlee1110, 2023-04-07:
Only happened on AC device?

PolynomialDivision, 2023-04-07:
> Only happened on AC device?

All devices are freezing. I am not sure if it would also crash if I have 3 ax devices, or some other combination. I need to organize me some other wireless equipment to test a bit further. But especially this 2 x mt7915e and the ath10k is always freezing after some time.

stweil, 2023-05-02:
I can report crashes in a wireless mesh with 3 NWA55AX. Related issue with more information: https://github.com/freifunk-gluon/gluon/issues/2822.

PolynomialDivision, 2023-07-05:
Seems like latest OpenWrt main is fixing the issue. Not sure which commit exactly fixed it. I guess it's one of @nbd168 commits:
- https://github.com/openwrt/openwrt/commit/86ebaef5d427c5732847168f6ceeda12bf528414
- https://github.com/openwrt/openwrt/commit/c8963580aaf6a1207367c809d1350557d6e075e9

Djfe, 2024-02-09:
@PolynomialDivision This should be closed by now. I'm fairy certain, that this was fixed in/before release OpenWrt 23.05