openwrt/mt76 #756: mt7915e crashing on key removal

dangowrt, 2023-03-01:
I just walked out of the house with a wifi connected device in my pocket. Seems like mt7915e didn't like that.

```
Wed Mar  1 16:20:53 2023 daemon.notice hostapd: wl1-ap0: AP-STA-DISCONNECTED 34:13:e8:xx:xx:xx
Wed Mar  1 16:21:13 2023 kern.err kernel: [669534.752997] mt7915e 0000:01:00.0: Message 000025ed (seq 1) timeout
Wed Mar  1 16:21:13 2023 kern.err kernel: [669534.759299] wl1-ap0: failed to remove key (0, 34:13:e8:xx:xx:xx) from hardware (-110)
Wed Mar  1 16:21:33 2023 kern.err kernel: [669555.233031] mt7915e 0000:01:00.0: Message 00005aed (seq 2) timeout
Wed Mar  1 16:21:54 2023 kern.err kernel: [669575.712521] mt7915e 0000:01:00.0: Message 00005aed (seq 3) timeout
Wed Mar  1 16:22:14 2023 kern.err kernel: [669596.192243] mt7915e 0000:01:00.0: Message 00005aed (seq 4) timeout
Wed Mar  1 16:22:35 2023 kern.err kernel: [669616.672010] mt7915e 0000:01:00.0: Message 000025ed (seq 5) timeout
Wed Mar  1 16:22:35 2023 kern.info kernel: [669616.678324] wl1-ap0: HW problem - can not stop rx aggregation for 34:13:e8:xx:xx:xx tid 0
Wed Mar  1 16:22:55 2023 kern.err kernel: [669637.151781] mt7915e 0000:01:00.0: Message 00005aed (seq 6) timeout
Wed Mar  1 16:23:16 2023 kern.err kernel: [669657.631618] mt7915e 0000:01:00.0: Message 000025ed (seq 7) timeout
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.638021] ------------[ cut here ]------------
Wed Mar  1 16:23:16 2023 kern.warn kernel: [669657.642725] WARNING: CPU: 0 PID: 1921 at ___ieee80211_stop_tx_ba_session+0x348/0x3ac [mac80211]
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.651623] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 cfg80211 slhc sch_cake nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat cls_flower act_vlan cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ifb seqiv usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.730982] CPU: 0 PID: 1921 Comm: hostapd Tainted: G S                5.15.94 #0
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.738546] Hardware name: Linksys E8450 (UBI) (DT)
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.743502] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.750543] pc : ___ieee80211_stop_tx_ba_session+0x348/0x3ac [mac80211]
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.757267] lr : ___ieee80211_stop_tx_ba_session+0x210/0x3ac [mac80211]
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.763982] sp : ffffffc0090c37a0
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.767375] x29: ffffffc0090c37a0 x28: ffffff8002419600 x27: ffffffc0090c3db0
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.774593] x26: ffffff8000066880 x25: ffffff8002880880 x24: ffffffc008bd1000
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.781811] x23: ffffffc000a51604 x22: ffffff800a1fc0e8 x21: 0000000000000003
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.789028] x20: ffffff8005980b00 x19: ffffff800a1f8000 x18: ffffffc008b0a3a0
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.796245] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000846
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.803462] x14: 00000000000002c2 x13: ffffffc0090c32a8 x12: ffffffc008b623a0
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.810680] x11: 6465353230303030 x10: ffffffc008b623a0 x9 : 0000000000000000
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.817897] x8 : ffffffc008b0a350 x7 : ffffffc008b0a3a0 x6 : 000007999a644355
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.825114] x5 : 0000000000000000 x4 : ffffffc008af8000 x3 : 0000000000000000
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.832331] x2 : 0000000000000001 x1 : 0000000000000002 x0 : 00000000ffffff92
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.839548] Call trace:
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.842072]  ___ieee80211_stop_tx_ba_session+0x348/0x3ac [mac80211]
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.848443]  ieee80211_sta_tear_down_BA_sessions+0x74/0x130 [mac80211]
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.855073]  ieee80211_find_sta_by_link_addrs+0x390/0x560 [mac80211]
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.861528]  sta_info_destroy_addr_bss+0x38/0x70 [mac80211]
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.867203]  ieee80211_color_change_finish+0x1bf8/0x1e80 [mac80211]
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.873573]  cfg80211_check_station_change+0x1384/0x4720 [cfg80211]
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.879936]  genl_family_rcv_msg_doit+0xb4/0x110
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.884637]  genl_rcv_msg+0xd0/0x1c0
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.888291]  netlink_rcv_skb+0x58/0x120
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.892207]  genl_rcv+0x34/0x50
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.895427]  netlink_unicast+0x1f0/0x2ec
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.899429]  netlink_sendmsg+0x19c/0x3d0
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.903430]  ____sys_sendmsg+0x258/0x2a0
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.907434]  ___sys_sendmsg+0x78/0xc0
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.911176]  __sys_sendmsg+0x54/0xb0
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.914831]  __arm64_sys_sendmsg+0x20/0x30
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.919006]  invoke_syscall+0x44/0x110
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.922837]  el0_svc_common.constprop.0+0x48/0xf0
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.927619]  do_el0_svc+0x18/0x20
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.931013]  el0_svc+0x14/0x50
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.934151]  el0t_64_sync_handler+0xe0/0x110
Wed Mar  1 16:23:16 2023 kern.debug kernel: [669657.938501]  el0t_64_sync+0x158/0x15c
Wed Mar  1 16:23:16 2023 kern.warn kernel: [669657.942244] ---[ end trace 39718feb58817c72 ]---
```

configuration
```
config wifi-device 'radio1'
	option type 'mac80211'
	option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option band '5g'
	option htmode 'HE80'
	option channel 'auto'
	option country 'PT'
	option cell_density '1'
	option txpower '17'
	option local_pwr_constraint '3'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'xxxxxxxx'
	option encryption 'psk-mixed'
	option key 'xxxxxxxx'
	option ieee80211w '1'
	option wpa_disable_eapol_key_retries '1'
	option ieee80211r '1'
	option ft_over_ds '0'
	option ft_psk_generate_local '1'
```

dangowrt, 2023-03-01:
No hardware flow offloading or WED what-so-ever btw.

dangowrt, 2023-03-01:
```
ubus call system board
{
	"kernel": "5.15.94",
	"hostname": "belkin-rt3200",
	"system": "ARMv8 Processor rev 4",
	"model": "Linksys E8450 (UBI)",
	"board_name": "linksys,e8450-ubi",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "SNAPSHOT",
		"revision": "r22124-bfa5e4e4eb",
		"target": "mediatek/mt7622",
		"description": "OpenWrt SNAPSHOT r22124-bfa5e4e4eb"
	}
}
```

rany2, 2023-05-09:
Just to keep this up-to-date, but I had a similar issue here: https://github.com/openwrt/mt76/issues/690#issuecomment-1540946390. It occurred after I enabled WPA3 and PMF.

Fail-Safe, 2024-06-03:
Just a little more info that may, or may not, be beneficial:

```sh
[   54.874089] mt798x-wmac 18000000.wifi: phy0 SER recovery state: 0x00000004
[   54.880981] mt798x-wmac 18000000.wifi:
[   54.880981] phy0 L1 SER recovery start.
[   54.888622] mt798x-wmac 18000000.wifi: Message 000025ed (seq 11) timeout
[   54.889422] mt798x-wmac 18000000.wifi: phy0 SER recovery state: 0x00000008
[   55.131284] mt798x-wmac 18000000.wifi: phy0 SER recovery state: 0x00000010
[   55.138214] mt798x-wmac 18000000.wifi: phy0 SER recovery state: 0x00000020
[   55.395327] mt798x-wmac 18000000.wifi: send message 000025ed timeout, try again(1).
[   55.404744] mt798x-wmac 18000000.wifi:
[   55.404744] phy0 L1 SER recovery completed.
```