openwrt/mt76 #758: mt7603 eeprom.c:27 error

siwind, 2023-03-04:
```
#dmesg
[   42.124873] ------------[ cut here ]------------
[   42.134057] WARNING: CPU: 3 PID: 645 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2023-03-01-c32d6d84/mt7603/eeprom.c:27 mt7603_eeprom_init+0x3b8/0x5ec [mt7603e]
[   42.164071] Modules linked in: mt7603e(+) mt76 mac80211 cfg80211 usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat ledtrig_usbport sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd uhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
[   42.271597] CPU: 3 PID: 645 Comm: kmodloader Not tainted 5.15.96 #0
[   42.284070] Stack : 00000001 80085c18 00000000 00000004 00000000 00000000 829abb24 80a30000
[   42.300723]         80870000 8079c9a8 8296bc18 80870e83 00000000 00000001 829abad0 80c73840
[   42.317377]         00000000 00000000 8079c9a8 829ab970 ffffefff 00000000 ffffffea 00000000
[   42.334028]         829ab97c 00000115 80876b30 ffffffff 8079c9a8 00000001 00000000 82a07d38
[   42.350681]         00000009 82bc8220 000f0000 82a10000 00000018 8041fcec 0000000c 80a3000c
[   42.367331]         ...
[   42.372195] Call Trace:
[   42.377045] [<8000812c>] show_stack+0x28/0xf0
[   42.385741] [<80395638>] dump_stack_lvl+0x60/0x80
[   42.395104] [<8002f7dc>] __warn+0x9c/0x124
[   42.403272] [<8002f8c0>] warn_slowpath_fmt+0x5c/0xac
[   42.413147] [<82a07d38>] mt7603_eeprom_init+0x3b8/0x5ec [mt7603e]
[   42.425307] [<82a0200c>] mt7603_register_device+0x110/0xb90 [mt7603e]
[   42.438152] [<82a00168>] init_module+0xba168/0xbb354 [mt7603e]
[   42.449789] 
[   42.452877] ---[ end trace aa22e38b5aae1e16 ]---
[   44.505960] mt7603e 0000:01:00.0: Firmware Version: _e1_drv_tv
[   44.517662] mt7603e 0000:01:00.0: Build Time: 20140313215548
[   45.544745] mt7603e 0000:01:00.0: Timeout waiting for ROM code to become ready
[   45.559247] mt7603e: probe of 0000:01:00.0 failed with error -5
[   45.580193] pci 0000:00:01.0: enabling device (0000 -> 0003)

```

with latest master snapshot version:
```
root@OpenWrt:~# uname -a
Linux OpenWrt 5.15.96 #0 SMP Sat Mar 4 17:55:20 2023 mips GNU/Linux
root@OpenWrt:~# opkg list-installed | grep mt76
kmod-mt76-core - 5.15.96+2023-03-01-c32d6d84-5
kmod-mt7603 - 5.15.96+2023-03-01-c32d6d84-5
kmod-mt76x02-common - 5.15.96+2023-03-01-c32d6d84-5
kmod-mt76x2 - 5.15.96+2023-03-01-c32d6d84-5
kmod-mt76x2-common - 5.15.96+2023-03-01-c32d6d84-5
root@OpenWrt:~# 
```

lukasz1992, 2023-03-06:
what device are you using?

lukasz1992, 2023-04-12:
Looks like this driver supports 7603e2 version only.