openwrt/mt76 #762: TL-WR850N MT7603

silverfox0000, 2023-03-17:
WiFi 2.4Ghz crashes on high load, as logs i can provide only screenshot
[image]

captain-n3m0, 2023-04-18:
The warning message mentions "nft_flow_offload" and "nf_flow_table" modules, which are related to network flow offloading. You can try disabling these features by running the following commands.

# modprobe -r nft_flow_offload
# modprobe -r nf_flow_table_ipv6
# modprobe -r nf_flow_table_ipv4
# modprobe -r nf_flow_table_inet

captain-n3m0, 2023-04-18:
This is a kernel level issue.

lukasz1992, 2023-04-18:
It is not related to flow offlowad, but DMA hang.

captain-n3m0, 2023-04-19:
The call trace shows the sequence of function calls that led to the warning message. It starts with the show_stack() function, which captures the current call stack at the point where the warning was triggered. The call trace then shows several functions from the mt7603e driver, including mt7603_filter_tx(), mt7603_wtbl_set_ps(), and mt7603_sta_ps(), which are related to managing the power-saving (PS) mode for a station (STA) in the mt7603e driver. The trace also shows functions from the mt76 driver, including mt76_rx_poll_complete() and mt76_dma_rx_poll(), which are involved in handling received packets in the mt76 driver. Finally, the trace includes functions from the network polling subsystem, such as __napi_poll() and napi_threaded_poll(), which are responsible for polling the network interface for incoming packets.

Based on this log, it appears that there may be an issue with the power-saving mode for a station in the mt7603e driver, which could be triggering a warning in the kernel.

captain-n3m0, 2023-04-19:
maybe i am wrong.

captain-n3m0, 2023-04-19:
Also pardon my english i am not very good at it.

lukasz1992, 2023-04-19:
> maybe i am wrong.

https://github.com/openwrt/mt76/blob/master/mt7603/mac.c#L208

dfateyev, 2023-08-24:
There are recent changes in master regarding mt7603 stability.
@silverfox0000 you are encouraged to re-test and give a feedback.