openwrt/mt76 #766: MT7915 BPSK 1/2 and Wave 2 Wi-Fi 6 features

yguel, 2023-03-27:
I would like to implement communication in BPSK 1/2 modulation using MU-MIMO UL/DL in 2.4GHz with MT7915.
The objective is to obtain long range communication with good link budget thanks to the MU-MIMO.
It is not clear to me if the MT7915 can do that (from a hardware perspective), but if its follows the Wi-Fi 6 specifications it should be.
Are there ressources somewhere that would help me see if the chip can operate in these modes and how to implement them in openwrt ?

lukasz1992, 2023-03-28:
Why don't you want auto mode for negotiating speed (in case of better conditions you have better speed)?

But anyway I think you can almost do it, the job is to configure /etc/config/wireless.
Read https://en.wikipedia.org/wiki/Wi-Fi_6 descriptions of rates, read https://openwrt.org/docs/guide-user/network/wifi/basic and test.

I'd be for setting:
require_vht        1 (no setting for forcing he yet AFAIK)
rate_list              8000 or 8600 (don't know exactly, but depends on channel width, guard interval), don't know if it works
basic_rate_list    8000 (?)
htmode             HEsomething

Two wifi6 devices should use HE automatically, including BPSK 1/2 if needed and MU-MIMO UL/DL

Djfe, 2023-04-25:
sounds like long-range dcm could be interesting to you
https://forum.huawei.com/enterprise/en/huawei/m/ViewThread.html?tid=733129&lang=en

Not sure if that's just supported by mt7915
or also by mt7905.

yguel, 2023-04-26:
Hi Djfe,
That is very interesting, I do not know how to check if mt7915 or mt7905 support it, but I will look for it, because it is a really clever system to increase range.

Djfe, 2023-04-26:
I only know about it, because it was mentioned here as one of its features:
https://www.mediatek.com/blog/mediatek-mt7915-wi-fi-6-wave-1-chipset-builds-in-a-range-of-industry-firsts
> Some Wi-Fi 6 Wave 1+ features are only optional and not supported by competitor alternatives, but MediaTek MT7915 ticks the boxes on all of them, such as 1024QAM, UL/DL MU-MIMO user#, spatial reuse and **_long-range DCM_**.

jackdondy, 2024-03-09:
> Hi Djfe, That is very interesting, I do not know how to check if mt7915 or mt7905 support it, but I will look for it, because it is a really clever system to increase range.
> 
> Thank you very much for sharing it.

Hi YGUEL, have you confirm that MT7915 supports DL/UL MU-MIMO? I am working on how to make MT7986 (which is based on MT7915) transmit HE-MU packets, but I never receive these multi-user packets. I guess the router just don't send HE-MU packets because the channel is not busy enough. Do you have any clues on this?