openwrt/mt76 #774: mt7622 not initializing on ax6s

vladie, 2023-04-27:
I have two redmi ax6s routers. Both are the same model and series (serial number is different for only two last digits). I've flashed both to openwrt. One works fine, second have a problem with mt7622 driver. I see in logs:
[   29.286174] mt7622-wmac 18000000.wmac: Message 00000010 (seq 1) timeout
[   29.292810] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   29.516425] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   29.736309] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   29.956281] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   30.176289] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   30.396317] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   30.616376] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   30.836325] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   31.056280] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   31.276277] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   31.496279] mt7622-wmac 18000000.wmac: Failed to get patch semaphore

After this i can't up wlan0 interface - got i/o error.
But on the second router mt7622 initializes fine:
[    8.552796] mt7622-wmac 18000000.wmac: HW/SW Version: 0x8a108a10, Build Time: 20190801210006a
[    8.669194] mt7622-wmac 18000000.wmac: N9 Firmware Version: _reserved_, Build Time: 20220630094834

On the stock redmi firmware both routers works fine with 2.4ghz.
I've tried different builds of openwrt - but no luck.

vladie, 2023-04-27:
just tried to load stock mtk drivers with openwrt, they also failed:

[  133.142513] wb_sys 18000000.wmac: Direct firmware load for mt7622_patch_e2_hdr.bin failed with error -2
[  133.157400] wb_sys 18000000.wmac: Falling back to sysfs fallback for: mt7622_patch_e2_hdr.bin
[  136.246986] /home/john/build/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/mt7622_mt_wifi/mt_wifi_ap/../mt_wifi/embedded/mcu/andes_core.c:891 assert FALSEfailed
[  136.263512] CPU: 1 PID: 2759 Comm: ifconfig Not tainted 5.10.161 #0
[  136.269769] Hardware name: Xiaomi Redmi Router AX6S (DT)
[  136.275070] Call trace:
[  136.277509]  0xffffffc01000a310
[  136.280640]  0xffffffc01000a4d4
[  136.283771]  0xffffffc0107ff5e0
[  136.286922]  0xffffffc008f0fa78 [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.293791]  0xffffffc008ee4fc8 [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.300656]  0xffffffc008f19b7c [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.307523]  0xffffffc008f1d530 [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.314389]  0xffffffc008f1e45c [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.321255]  0xffffffc008f2190c [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.328121]  0xffffffc008e92664 [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.334988]  0xffffffc008e92bc8 [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.341854]  0xffffffc008eeaf50 [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.348720]  0xffffffc008f5efac [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.355586]  0xffffffc008f5f064 [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.362452]  0xffffffc008e974bc [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.369318]  0xffffffc008f5ea30 [mt7622_mt_wifi@000000003036e28b+0x1c5000]
[  136.376182]  0xffffffc010661c64
[  136.379314]  0xffffffc010662034
[  136.382446]  0xffffffc0106620b0
[  136.385577]  0xffffffc010727584
[  136.388707]  0xffffffc01072aebc
[  136.391839]  0xffffffc010634134
[  136.394970]  0xffffffc010634558
[  136.398101]  0xffffffc01019c0ac
[  136.401232]  0xffffffc010010ee0
[  136.404363]  0xffffffc010010fac
[  136.407494]  0xffffffc0108054e4
[  136.410625]  0xffffffc010805900
[  136.413756]  0xffffffc0100025c8
[  138.074761] efuse_probe: efuse = 10000012
[  184.825425] efuse_probe: efuse = 10000012

so it's quite strange how stock firmware works....

nshopik, 2023-10-26:
Had same issue recent snapshot `r24194-e1aaa1defd` fixed it for me