openwrt/mt76 #775: MT7922 Unable to Start in AP Mode with Encryption

dellagd, 2023-04-30:
Attempting to use a MT7922 M.2 card (picture below) to stand up an access point in 6G band, which the `iw info` for the device claims to support. As I understand it, the MT7916 is more targeted at access point use, but I could only find one mPCIe device from AsiaRF with that chipset, not in an M.2 form factor. The system I am hoping to put this device in only has an M.2 slot and USB ports, so the mPCIe option is both more expensive and doesn't even fit. Hoping to make the MT7922 option work as I don't need particularly fancy AP features, I just need it to be in the 6G band.

I am running on linux kernel 6.1.26 using the latest mt76 (https://github.com/openwrt/mt76/commit/cddbd796bbc649debe457f7b34b200bbda4fcc53)

[image]

Relevant `dmesg` output. The system boots fine, and at 890 seconds I attempt to enable the AP, which proceeds to hang the luci web interface, though I can still interact with the device over the command line. The looping of chip reset and timeouts continues indefinitely (log cut short here).

It appears whatever issue is happening is related to the use of encryption. I'm able to start up an open AP without issue, but setting any type of encryption produces the following logs. 

```
[   14.335064] mt7921e 0000:01:00.0: enabling device (0000 -> 0002)
[   14.351616] mt7921e 0000:01:00.0: ASIC revision: 79220010
[   14.433931] mt7921e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20230328143229a
[   14.816193] mt7921e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20230328143313
[  890.195329] br-lan: port 2(phy0-ap0) entered blocking state
[  890.195852] br-lan: port 2(phy0-ap0) entered disabled state
[  890.196980] device phy0-ap0 entered promiscuous mode
[  890.197825] br-lan: port 2(phy0-ap0) entered blocking state
[  890.198342] br-lan: port 2(phy0-ap0) entered forwarding state
[  890.199520] br-lan: port 2(phy0-ap0) entered disabled state
[  893.302924] mt7921e 0000:01:00.0: Message 00020003 (seq 14) timeout
[  893.303541] phy0-ap0: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-110)
[  896.502989] mt7921e 0000:01:00.0: Message 0004008a (seq 15) timeout
[  899.703118] mt7921e 0000:01:00.0: Message 0004008a (seq 1) timeout
[  902.903191] mt7921e 0000:01:00.0: Message 0004008a (seq 2) timeout
[  906.103261] mt7921e 0000:01:00.0: Message 0004008a (seq 3) timeout
[  909.303383] mt7921e 0000:01:00.0: Message 00020003 (seq 4) timeout
[  909.304000] phy0-ap0: failed to set key (4, ff:ff:ff:ff:ff:ff) to hardware (-110)
[  912.503466] mt7921e 0000:01:00.0: Message 0004008a (seq 5) timeout
[  915.703516] mt7921e 0000:01:00.0: Message 0004008a (seq 6) timeout
[  918.903610] mt7921e 0000:01:00.0: Message 0004008a (seq 7) timeout
[  922.103675] mt7921e 0000:01:00.0: Message 0004008a (seq 8) timeout
[  925.303772] mt7921e 0000:01:00.0: Message 0004008a (seq 9) timeout
[  928.503822] mt7921e 0000:01:00.0: Message 0004008a (seq 10) timeout
[  928.504490] mt7921e 0000:01:00.0: chip reset failed
[  931.703902] mt7921e 0000:01:00.0: Message 00020001 (seq 11) timeout
[  934.903956] mt7921e 0000:01:00.0: Message 00020002 (seq 14) timeout
[  938.104021] mt7921e 0000:01:00.0: Message 00020003 (seq 15) timeout
[  941.304074] mt7921e 0000:01:00.0: Message 00020002 (seq 1) timeout
[  944.504180] mt7921e 0000:01:00.0: Message 00020002 (seq 2) timeout
[  947.704235] mt7921e 0000:01:00.0: Message 0004008a (seq 3) timeout
[  950.904313] mt7921e 0000:01:00.0: Message 0004008a (seq 4) timeout
[  954.104351] mt7921e 0000:01:00.0: Message 0004008a (seq 5) timeout
[  957.304444] mt7921e 0000:01:00.0: Message 0004008a (seq 6) timeout
[  960.504579] mt7921e 0000:01:00.0: Message 0004008a (seq 7) timeout
[  963.704578] mt7921e 0000:01:00.0: Message 0004008a (seq 8) timeout
[  966.904638] mt7921e 0000:01:00.0: Message 0004008a (seq 9) timeout
[  970.104707] mt7921e 0000:01:00.0: Message 0004008a (seq 10) timeout
[  973.304787] mt7921e 0000:01:00.0: Message 0004008a (seq 11) timeout
[  976.504853] mt7921e 0000:01:00.0: Message 0004008a (seq 12) timeout
[  976.505524] mt7921e 0000:01:00.0: chip reset failed
[  979.704950] mt7921e 0000:01:00.0: Message 00020001 (seq 13) timeout
[  982.905124] mt7921e 0000:01:00.0: Message 00020002 (seq 2) timeout
[  986.105267] mt7921e 0000:01:00.0: Message 00020003 (seq 3) timeout
[  989.305366] mt7921e 0000:01:00.0: Message 0004008a (seq 5) timeout
[  992.505497] mt7921e 0000:01:00.0: Message 0004008a (seq 6) timeout
[  995.705637] mt7921e 0000:01:00.0: Message 0004008a (seq 7) timeout
[  998.905762] mt7921e 0000:01:00.0: Message 0004008a (seq 8) timeout
[ 1002.105879] mt7921e 0000:01:00.0: Message 0004008a (seq 11) timeout
[ 1005.306024] mt7921e 0000:01:00.0: Message 0004008a (seq 12) timeout
[ 1008.506146] mt7921e 0000:01:00.0: Message 0004008a (seq 15) timeout
[ 1011.706250] mt7921e 0000:01:00.0: Message 0004008a (seq 1) timeout
[ 1014.906343] mt7921e 0000:01:00.0: Message 0004008a (seq 2) timeout
[ 1018.106482] mt7921e 0000:01:00.0: Message 0004008a (seq 3) timeout
[ 1018.107144] mt7921e 0000:01:00.0: chip reset failed
[ 1021.306592] mt7921e 0000:01:00.0: Message 00020001 (seq 6) timeout
[ 1024.506718] mt7921e 0000:01:00.0: Message 00020002 (seq 9) timeout
[ 1027.706816] mt7921e 0000:01:00.0: Message 00020003 (seq 10) timeout
[ 1030.907061] mt7921e 0000:01:00.0: Message 00020002 (seq 11) timeout
[ 1034.107183] mt7921e 0000:01:00.0: Message 00020002 (seq 12) timeout
[ 1037.307117] mt7921e 0000:01:00.0: Message 0004008a (seq 13) timeout
[ 1040.507212] mt7921e 0000:01:00.0: Message 0004008a (seq 14) timeout
[ 1043.707324] mt7921e 0000:01:00.0: Message 0004008a (seq 15) timeout
[ 1046.907420] mt7921e 0000:01:00.0: Message 0004008a (seq 1) timeout
[ 1050.107513] mt7921e 0000:01:00.0: Message 0004008a (seq 2) timeout
[ 1053.307653] mt7921e 0000:01:00.0: Message 0004008a (seq 3) timeout
[ 1056.507720] mt7921e 0000:01:00.0: Message 0004008a (seq 4) timeout
[ 1059.707783] mt7921e 0000:01:00.0: Message 0004008a (seq 5) timeout
[ 1062.907864] mt7921e 0000:01:00.0: Message 0004008a (seq 6) timeout
[ 1066.107961] mt7921e 0000:01:00.0: Message 0004008a (seq 7) timeout
[ 1066.108618] mt7921e 0000:01:00.0: chip reset failed
```

dellagd, 2023-05-01:
In attempting to debug this, one thing I tried was patching [main.c](https://github.com/openwrt/mt76/blob/master/mt7921/main.c) to use `MCU_EXT_CMD` instead of `MCU_UNI_CMD` for the set_key process. The idea here was that from the mt7915, where the same process was executed with `MCU_EXT_CMD`.

https://github.com/openwrt/mt76/blob/cddbd796bbc649debe457f7b34b200bbda4fcc53/mt7915/main.c#L422 

https://github.com/openwrt/mt76/blob/cddbd796bbc649debe457f7b34b200bbda4fcc53/mt7921/main.c#L570

This resulted in the following patch:

```
--- a/mt7921/main.c
+++ b/mt7921/main.c
@@ -558,7 +560,7 @@ static int mt7921_set_key(struct ieee80211_hw *hw, enum set_key_cmd cmd,
 
 	mt76_wcid_key_setup(&dev->mt76, wcid, key);
 	err = mt76_connac_mcu_add_key(&dev->mt76, vif, &msta->bip,
-				      key, MCU_UNI_CMD(STA_REC_UPDATE),
+				      key, MCU_EXT_CMD(STA_REC_UPDATE),
 				      &msta->wcid, cmd);
 	if (err)
 		goto out;
@@ -567,7 +569,7 @@ static int mt7921_set_key(struct ieee80211_hw *hw, enum set_key_cmd cmd,
 	    key->cipher == WLAN_CIPHER_SUITE_WEP40)
 		err = mt76_connac_mcu_add_key(&dev->mt76, vif,
 					      &mvif->wep_sta->bip,
-					      key, MCU_UNI_CMD(STA_REC_UPDATE),
+					      key, MCU_EXT_CMD(STA_REC_UPDATE),
 					      &mvif->wep_sta->wcid, cmd);
 out:
 	mt7921_mutex_release(dev);
```

To my surprise, this seemed to have an effect, and I was suddenly able to start the access point with encryption and associate a station! I have not encountered any crashes so far, and I was also able to start an access point in the 6G band!

However, after this, it seems there is still some sort of issue. When a client associates to the AP, it seems there is some issue with data transmission. I see the following repeating continuously in logs:

```
Mon May  1 03:21:57 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 12:34:56:78:90:AB
Mon May  1 03:21:57 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.116 12:34:56:78:90:AB
Mon May  1 03:22:06 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 12:34:56:78:90:AB
Mon May  1 03:22:06 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.116 12:34:56:78:90:AB
Mon May  1 03:22:13 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 12:34:56:78:90:AB
Mon May  1 03:22:13 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.116 12:34:56:78:90:AB
Mon May  1 03:22:14 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 12:34:56:78:90:AB
Mon May  1 03:22:14 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.116 12:34:56:78:90:AB
Mon May  1 03:22:16 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 12:34:56:78:90:AB
Mon May  1 03:22:16 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.116 12:34:56:78:90:AB
Mon May  1 03:22:21 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 12:34:56:78:90:AB
Mon May  1 03:22:21 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.116 12:34:56:78:90:AB
Mon May  1 03:22:29 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 12:34:56:78:90:AB
Mon May  1 03:22:29 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.116 12:34:56:78:90:AB
```

This doesn't happen if the AP is running with no encryption, where a client successfully gets a DHCP lease and can communicate. I think this DHCP issue is just an indicator of some sort of general data transmission issue, because if I manually set an IP such that DHCP shouldn't be necessary, I appears I still can't communicate over the link (ping the router).

Any ideas?

morrownr, 2023-05-07:
Hi @dellagd 

> This patch adds firmware files v211120221208 for MT7981 WiFi. 

Are you sure that is the correct firmware?

See section 1:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

morrownr, 2023-05-07:
linux-firmware: update firmware for MT7922 WiFi device
Update binary firmware for MT7922 WiFi devices

File: firmware/WIFI_MT7922_patch_mcu_1_1_hdr.bin
Version: 20230328143229a
File: firmware/WIFI_RAM_CODE_MT7922_1.bin
Version: 20230328143313

dellagd, 2023-05-07:
Hey @morrownr, I think the logs I posted above show those firmware versions called out. I did double check I grabbed the newest available firmware.

morrownr, 2023-05-07:
> I am running on linux kernel 6.1.26 using the latest mt76 (cddbd79)

The link in the above says:

inux-firmware: add firmware for MT7981
This patch adds firmware files v211120221208 for MT7981 WiFi.

So, you said you are running the mt7981 firmware.

The mt7981 firmware is not the right firmware for a mt7922 chipset. I pointed you to the right fimrware.

That may or may not have anything to do with you problem.

dellagd, 2023-05-07:
Oh, sorry for the confusion. I was just trying to call out the commit hash for the version of openwrt/mt76 I am running. That specific commit does indeed seem to add the firmware files for the mt7981, but I don't think their presence in the filesystem means that the driver would try to load them onto this chipset. The mt7922 firmware files are also present on my filesystem, and I think mt76 is able to ID the chipset and load it with the correct firmware file (correct me if I am wrong).

Thank you for taking a look at this with me!

morrownr, 2023-05-07:
> I don't think their presence in the filesystem means that the driver would try to load them onto this chipset. 

Let me see if I can further explain things. We have been seeing something that is not common with the firmware for the mt7921 and mt7922 chipsets. For many months we have been seeing new firmware monthly for the mt7921 and the mt7922 is not far behind.

My point being that it may be very important for you to make sure you have the very latest firmware and keep it up to date while working your problem. As Dorthey used to say, Toto, we aren't in Kansas anymore. These WiFi 6e drivers and their firmware are incredibly complicated and it appears that regular releases for now are required to keep fixing problems and adding features. It might be a good idea for you to also consider updating to kernel 6.2 or 6.3. You are on the bleeding edge with a product that does not have widespread use yet. I should have a PCIe card with the mt7922 chipset within a couple of weeks so I will see what the state of the driver and firmware is myself soon.

morrownr, 2023-05-07:
FYI: You are welcome to stop by my site. It is primarily used for USB-WiFi but we talk about cards sometimes as well. The site is:

https://github.com/morrownr/USB-WiFi

We use issues for news and discussions. I recently posted news about a new ASUS card with the mt7922 chipset and I ordered one. I'll be posting progress reports in the issue as I bring it online in a couple of weeks.

Regards

dellagd, 2023-05-07:
> For many months we have been seeing new firmware monthly for the mt7921 and the mt7922 is not far behind.

Yeah, I have been looking through the mediatek firmware releases and was surprised to see so many. That combined with the amount of issues filed regarding these chips lead me to believe things probably just aren't stable yet. I actually even tried loading on an earlier version of firmware in my testing but got the same sort of errors in the logs.

> I should have a PCIe card with the mt7922 chipset within a couple of weeks so I will see what the state of the driver and firmware is myself soon.

Ooh, excited to hear it! Will those reports go into a specific issue thread on your repository that I should follow?

I also ordered [one of these (AW7916-NPD-1) from AsiaRF](https://www.asiarf.com/shop/wifi-wlan/wifi_mini_pcie/wifi6e-3000-802-11ax-3t3r-dbdc-mpcie/ ) which should supposedly also support 6E. We will see what the drivers are like there - I think this chipset is primarily targeted at AP usage at least.

[image]

Just wanted to confirm though, I see the following logs printed out on boot:

```
[   14.433931] mt7921e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20230328143229a
[   14.816193] mt7921e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20230328143313
```

Those strings match the versions you called out for the latest firmware I should be using, so I think I have the latest firmware going on right now. Please let me know if I should try something different here. I'll see if I can get on a later kernel.

> File: firmware/WIFI_MT7922_patch_mcu_1_1_hdr.bin
> Version: 20230328143229a
> File: firmware/WIFI_RAM_CODE_MT7922_1.bin
> Version: 20230328143313

morrownr, 2023-05-08:
> Will those reports go into a specific issue thread on your repository that I should follow?

Yes

https://github.com/morrownr/USB-WiFi/issues/261

You are welcome to make a similar new issue about your little card so we can follow along. I would be interested in a link where one can be ordered.

The mt7916 is interesting. It is mostly aimed at routers I assume?

dellagd, 2023-05-08:
That's my understanding - we will see how the software support pans out though.

corecell, 2023-05-16:
I have same problem.

**Board:** NanoPI R5C
**Kernel:** Linux nanopi 6.4-rc1
**OpenWRT:** [unofficial build](https://github.com/mj22226/openwrt/) r0-803e80d
**mt76:** cddbd796bbc649debe457f7b34b200bbda4fcc53
**Firmwares builtin:**
```
mt7921e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20230328143229a
mt7921e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20230328143313
```
```
8628325955ef1b26960327a774e38d5f  BT_RAM_CODE_MT7922_1_1_hdr.bin
9ba39f2ad888d63df937650af8ae3126  WIFI_MT7922_patch_mcu_1_1_hdr.bin
14eab07f220ded220f4424cb96c5cc28  WIFI_RAM_CODE_MT7922_1.bin
```
Checked with last firmwares from kernel.org:
```
4001b4ceb0bad72c439e2143f87ae47e  BT_RAM_CODE_MT7922_1_1_hdr.bin
de29098ddf1773051601de2e9ab60657  WIFI_MT7922_patch_mcu_1_1_hdr.bin
14c08c8298b639ee52409b5e9711a083  WIFI_RAM_CODE_MT7922_1.bin
```

I found that, when you set cipher in wireless security to:

- **Auto** appears this problem, heartbeat led start blinking more frequently, luci interface stucks.
- **Force CCMP (AES)** same problem
- **Force TKIP** wireless card starts in N and AC mode, but performance very poor, about 5-10 Mbps in 5GHz, a bit more in 2.4GHz.
- **Force TKIP and CCMP (AES)** wireless card starts fine in N and AC mode, performance good (about 200Mbps in AC 5GHz)

Working config:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option path '3c0000000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel 'auto'
	option band '5g'
	option htmode 'VHT40'
	option cell_density '0'
	option txpower '5'
	option country 'UA'

config wifi-iface 'wifinet0'
	option device 'radio0'
	option mode 'ap'
	option ssid 'myssid'
	option encryption 'psk2+tkip+ccmp'
	option key 'mypassword'
	option network 'lan'
```
The wireless card works fine most of the time (not tested for heavy load, used only for my smartphone), but I still had two freezes (same issue). I didn't save dmesg log for the first time stuck, but as far as I remember the problem was something like this:
`pcieport 0000:00:00.0: AER: Multiple Corrected error received`

fhteagle, 2023-05-18:
Interesting find. Two questions for those that have one:

1. Does it do dual band dual connect when operating as AP? As in, can it simultaneously make a 2.4ghz and a 5ghz AP, and connect clients on both at the same time?
2. Is the heatsink mounted with glue (removable with mild heat?), screwed on, or hard connected on?

dellagd, 2023-05-18:
> Does it do dual band dual connect when operating as AP? As in, can it simultaneously make a 2.4ghz and a 5ghz AP, and connect clients on both at the same time?

Right now I am using it as a 6GHz AP and a 2.4GHz client. It looks like the secondary radio in the DBDC card is only 2.4GHz. It lists AP as a capability for that radio but I haven't tried it yet.

> Is the heatsink mounted with glue (removable with mild heat?), screwed on, or hard connected on?

I'm not sure - I think you could reach out to AsiaRF if you wanted to know, they were response to me over email.

exkc, 2023-11-03:
> Right now I am using it as a 6GHz AP and a 2.4GHz client. It looks like the secondary radio in the DBDC card is only 2.4GHz. It lists AP as a capability for that radio but I haven't tried it yet.

secondary radio dont work for me . i tried to set up two wifi network (one is 2.4 one is 5g) in luci.

i am using openwrt 65380dc41cc39d5f9e73f509848d327d60470ea2 with whatever m76 it shipped in that commit (as well as mtk wifi firmware from artix)

antnyfls, 2023-11-03:
> secondary radio dont work for me . i tried to set up two wifi network (one is 2.4 one is 5g) in luci and it didnt work.

@exkc mt7922 wifi card is single band for dual band you need one like this https://asiarf.com/product/wi-fi-6e-m-2-ae-key-module-mt7916-aw7916-aed/

exkc, 2023-11-04:
oh sry yeah I got confused between the two.I dont really need dual band. i could have only one 2.4g wifi . 5g wifi isnt that necessary for me.Anyway since I have cheap second hand tp link ar9545 card (TP-LINK TL-WN781ND v2) , so i can  use MT7922 for 5g wifi only and cheap second hand tp link ar9545 card (TP-LINK TL-WN781ND v2) for 2g wifi.