openwrt/mt76 #778: mt7601u active monitor mode is not working as expected

ZerBea, 2023-05-09:
Running monitor mode with active monitor mode flag, only frames addressed to BRADCAST MAC (ff:ff:ff:ff:ff:ff:ff) or addressed to the device MAC are received.

tested mt7601u devices (both showing the same behavor):
ID 7392:7710 Edimax Technology Co., Ltd Edimax Wi-Fi
ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter

tested kernels:
5.15,
6.1
6.3

To reproduce:
run latest git head hcxdumptool (https://github.com/ZerBea/hcxdumptool)
run Wireshark in parallel and set filter option to "eapol"
Wireshark only show EAPOL M1 frames addressed to the device MAC

deactive active monitor mode by option -p
run latest git head hcxdumptool -p 
run Wireshark in parallel and set filter option to "eapol"
Wireshark only show all EAPOL frames addressed to the device MAC

Please also notice my last commit to diable active monitor mode on mt7601u devices (as a workaround):
https://github.com/ZerBea/hcxdumptool/commit/afd007347eeabfd2f1ccfa4f0564ab8e6acdd5df
and the changelog:
https://github.com/ZerBea/hcxdumptool/blob/master/changelog

Other mt76 devices, e.g.
ID 0b05:17d1 ASUSTek Computer, Inc. AC51 802.11a/b/g/n/ac Wireless Adapter [Mediatek MT7610U]
are not affected. Active monitor mode is working as expected (confirmed by Wireshark running in parallel on the same device).

At the moment, I'm not sure if the problem is related to the driver or to the firmware.

lukasz1992, 2023-05-19:
mt76 driver does not support mt7601u chipset

ZerBea, 2023-05-19:
Good point. Thanks.

closed 2023-05-19 by ZerBea