openwrt/mt76 #780: Doing the following on will cause some clients to have longer latency or skipped seq during ping

thelinuxdude, 2023-05-11:
On a mt7615 we have ODBC enabled using phy1 as the 2.4 radio we have an AP setup on wlan1.  We setup another interface on the same phy and do a site survey scan and after clients connected to the AP are slower to respond with ping.  Latency increases and we see skipped seq.

```
iw phy1 interface add wlan99 type station
iwinfo wlan99 scan
iw wlan99 del

```

Any thoughts on why this happens? 
We have a reason for setting up a second iface and we have done this successfully on a MT7620/7612 setup.  Our setup is with a MT7621/7615

rany2, 2023-05-15:
I believe this is to be expected for most drivers/hardware, as long as the driver doesn't crash I don't think there is cause for alarm.

thelinuxdude, 2023-05-19:
On other radios like the rt2800 and the mt7612 we have not observed this behavior, only on the mt7615 in ODBC.

rany2, 2023-05-26:
P.S. you don't really need to make a new interface, you could just do `iwinfo phy0 scan`. It doesn't help you with your problem, but thought to let you know..