openwrt/mt76 #781: FCC test mode for MT7612 and MT7615 Radio modules

MaxRF77, 2023-05-12:
Hello Fellow MediaTek & Openwrt Users, 

We need to perform FCC test of a device that contains both the MT7612 and MT7615 Radio modules.
We have searched extensively (both on github and other resources) for a way to exercise/stimulate the mini PCIe modules in a Stand-Alone mode on a Linux / ARM platform.  We have not yet found anything that we can use.

For FCC test, we need to be able to:
select:

    - channels
    - data rates
    - modulation schemes 
    - power levels

A code module and a CLI with instructions would simply be awesome! (If there is such a thing.)

I apologize in advance if this is unintended usage of this user driven & provided online resource but if someone can point us in the right direction i suspect that we should not take up too many resources.

BTW, has anyone had any success with testing a Linux compatible equivalent to the Windows-based Qualcomm QSPR?

Best Regards,
Max

nachalni, 2023-06-07:
Of course cfg80211 testmode support has to be enabled and mt76-test tool has to be built

Djfe, 2023-07-27:
did this answer your question or did you run into further issues? @MaxRF77