openwrt/mt76 #783: Alfa Network AWUS036AXML: TX power limited to 3 dBm

patrakov, 2023-05-28:
I bought a new USB WiFi adapter, Alfa Network AWUS036AXML, and tried to use it on my desktop with Arch Linux. The kernel version is 6.3.3, the linux-firmware package version is 20230404.2e92a49f-1.

The adapter is recognized, but in the 5 GHz band, the TX power is limited to 3 dBm. I am in the Philippines, and the regulatory domain is set correctly. The bad received signal strength is also confirmed by my router.

As this is a USB card, there is no such thing as a "factory partition" from where the calibration data can be obtained.

```
# iw dev
phy#2
	Interface wlp0s20u4i3
		ifindex 34
		wdev 0x200000001
		addr 00:c0:ca:b3:c7:3c
		ssid Archer
		type managed
		channel 153 (5765 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
phy#0  <-- onboard WiFi, please ignore
	Interface wlp4s0
		ifindex 4
		wdev 0x1
		addr f2:b5:c2:8a:89:b9
		type managed
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0

```

patrakov, 2023-05-28:
```
[389839.043749] usb 3-4: new high-speed USB device number 14 using xhci_hcd
[389839.899850] usb 4-4: new SuperSpeed USB device number 4 using xhci_hcd
[389839.913461] usb 4-4: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[389839.913467] usb 4-4: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[389839.913469] usb 4-4: Product: Wireless_Device
[389839.913471] usb 4-4: Manufacturer: MediaTek Inc.
[389839.913473] usb 4-4: SerialNumber: 000000000
[389843.461174] Bluetooth: hci0: Device setup in 3458344 usecs
[389843.461188] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[389845.522024] Bluetooth: hci0: Opcode 0x c03 failed: -110
[389847.570104] Bluetooth: hci0: Failed to read MSFT supported features (-110)
[389849.618187] Bluetooth: hci0: AOSP get vendor capabilities (-110)
[389849.732595] usb 4-4: reset SuperSpeed USB device number 4 using xhci_hcd
[389849.783694] mt7921u 4-4:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230302150916a

[389849.900223] Bluetooth: hci0: Device setup in 149656 usecs
[389849.900230] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[389849.970167] Bluetooth: hci0: AOSP extensions version v1.00
[389849.970171] Bluetooth: hci0: AOSP quality report is supported
[389850.060740] mt7921u 4-4:1.3: WM Firmware Version: ____010000, Build Time: 20230302150956
[389851.953094] mt7921u 4-4:1.3 wlp0s20u4i3: renamed from wlan0
[389904.364191] wlp0s20u4i3: authenticate with e8:9f:80:d4:9e:c6
[389904.922280] wlp0s20u4i3: send auth to e8:9f:80:d4:9e:c6 (try 1/3)
[389904.927584] wlp0s20u4i3: authenticate with e8:9f:80:d4:9e:c6
[389904.936713] wlp0s20u4i3: send auth to e8:9f:80:d4:9e:c6 (try 1/3)
[389904.939690] wlp0s20u4i3: authenticated
[389904.940446] wlp0s20u4i3: associate with e8:9f:80:d4:9e:c6 (try 1/3)
[389904.953527] wlp0s20u4i3: RX AssocResp from e8:9f:80:d4:9e:c6 (capab=0x1011 status=0 aid=3)
[389904.987585] wlp0s20u4i3: associated
[389904.987719] wlp0s20u4i3: Limiting TX power to 30 (30 - 0) dBm as advertised by e8:9f:80:d4:9e:c6
[389905.990537] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s20u4i3: link becomes ready
```

But still the actual TX power is 3 dBm.

rany2, 2023-05-29:
I wouldn't expect it to be able to connect with 3dBm as TX power. Maybe it's an issue with it reporting incorrect TX power?

patrakov, 2023-05-29:
I have deliberately placed the adapter in a bad spot. Well, sorry for the misleading report - Linux connects at a higher speed than Windows.

Arch Linux:

The AP (Linksys E8450) reports a signal strength of -79 or -80 dBm. iperf3 speed test yields 334 Mbits/sec download (from the tablet's perspective) and 109 Mbits/sec download.

Windows 11:

The AP reports a signal strength of -87 dBm. iperf3 speed test yields 244 Mbits/sec download (from the tablet's perspective) and 33.7 Mbits/sec download.

rany2, 2023-05-29:
I think it's worth keeping that issue open but with "inaccurate TX power report" as the title.

patrakov, 2023-05-29:
I still think that the TX power is wrong. I have placed an Alfa Network AWUS036ACM adapter into the same bad spot, and the AP now reports a much higher signal level, and the speed test is returning better results.

Linux:

The AP reports a signal strength of -69 dBm, iperf3 reports 237 Mbits/sec download and 179 Mbits/sec upload

Windows 11:

The AP reports a signal strength of -69 dBm. iperf3 reports 221 Mbits/sec download and 193 Mbits/sec upload.

Now compare this with a -80 or -87 dBm from AWUS036AXML.

patrakov, 2023-05-29:
Another hypothesis is that the AX mode on Linksys E8450 (with OpenWrt) is somehow cursed. There were reports of horrible throughput with iPhones, maybe it's the same problem? Let's test by disabling AX on the AP and retesting.

Linux:

AP reports -77 dBm, the adapter itself reports -67 dBm, iperf3 reports 353 Mbits/sec download and 131 Mbits/sec upload (from the tablet viewpoint)

Windows:

AP reports -82 dBm, iperf3 reports 297 Mbits/sec download and 54.4 Mbits/sec upload (from the tablet viewpoint)

So I think the bug is real, and not Linux-specific. Might be just that I should not recommend this AWUS036AXML hardware to anyone.

rany2, 2023-05-29:
@patrakov if it's mt7915 try to disable bw160 with this patch: https://github.com/rany2/mt76/commit/2e31de46bc1ff7505975bae4272989a5e0b93c2c

patrakov, 2023-05-29:
It is mt7921u, not mt7915.

rany2, 2023-05-30:
@patrakov If you like, I could provide you with the sysupgrade package. Just tell me which packages to include, though it does require some faith in my binaries so to speak ;)

patrakov, 2023-05-30:
The router contains some unofficial stuff, so I do not want a full sysupgrade image. I would rather get a compatible package that includes just a rebuilt kernel module.

The router is running OpenWrt 22.03.5, r20134-5f15225c1e.

jopdyke, 2023-05-30:
It's an incorrect reporting. That is 2 milliwats and the connection won't take place if that was the actual power. This is known to impact AWUS036ACHM in Linux also, which is the same chip as AWUS036ACM. We have reported it to ALFA before and will follow up.

deutrino, 2023-06-03:
See also:

  * https://github.com/openwrt/mt76/issues/633
  * https://github.com/morrownr/USB-WiFi/issues/34

patrakov, 2023-07-11:
I understand that I have not responded to the request to test the driver patch that disables 160 MHz AC support on the OpenWrt AP. I have just tested this fix instead: https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/562

It makes the upload speed much better under Linux, but hurts even more under Windows.

31337-4554551n, 2023-09-05:
Jumping in on this. Ordered a new ALFA AWUS036AXML and getting terrible reception. Found this thread, confirming 3dBm, compared to an older ALFA rtl8812au, and an older still 2.4ghz, that both show up as 20dBm.
I am in Australia, but ordered the unit from amazon.com.au.
Returned it, and got a new one assuming a physical fault, but still getting 3dBm and terrible range.

jopdyke, 2023-09-05:
3 dbm is not accurately reported as previously noted in this thread, the WiFi adapter could not make a connection in the same home if it were. When you say terrible range, can you be more specific and also post what kernel you have?

31337-4554551n, 2023-09-05:
[image]
Ok, so more than 5, but the last few have no signal so barely count lol
Here you can see the difference between the xml, and the 8812au

jopdyke, 2023-09-05:
What model product is the 8812AU?

[difference]<https://user-images.githubusercontent.com/26422914/265735744-252610cf-91de-459f-95b0-38aa2dbe1a71.jpg>

31337-4554551n, 2023-09-05:
ALFA AWUS036ACH

jopdyke, 2023-09-05:
That is a high power adapter with built-in amplifier. ALFA does not have a high power tri-band model yet. AWUS036XML range wont equal or exceed AWUS036ACH, but adds 6 Ghz. We have not seen a huge range disparity or had one reported, but ACH model is higher power. If your disparity is very large in all environments try both a USB 2 and USB 3 port and let us know if you see any difference, as it could be related to power draw on that specific device.

31337-4554551n, 2023-09-05:
I am connecting both to the same USB3 port.

Can you please suggest an alternative that would give me the power of AWUS036ACH but with the drivers built into the linux kernel?

31337-4554551n, 2023-09-05:
[image]

jopdyke, 2023-09-05:
So you do not need 6 Ghz capability? You can try AWUS036ACM which is native in Linux with Mediatek chip (2.4 and 5 Ghz only). But the AWUS036ACH has the best overall range/power of ALFA adapters currently.
You can try connecting your AWUS036AXML to a USB 2 port so see if there is any reception difference. Most likely there should not be, but the test will help isolate if a power input limitation.

31337-4554551n, 2023-09-05:
So AWUS036ACM is the closest to the power with native drivers?
And ACH doesn't have native drivers because...? Blobs?

And now we're getting a little off topic, but can you recommend somewhere I can get this kind of detail without poking kind strangers on github threads? I kinda just found a list of natively supported new cards, and picked the ALFA. Reviews and stuff don't have the kind of information you're giving me :(

lukaszzyla, 2024-01-10:
same issue here with AWUS036AXML bad reception.
txpower reported as 3db and honestly - both my other much cheaper adapters (asus ac51 and tplink archer 2u plus) have better reception.
I am really surprised about it - according to David Bombal this should be one of the best adapters on the market at the moment.
I am using kali linux on rpi5.
I have a strange feeling it used to work better when connected for the first time.
Than i did connect it to windows once for a moment.
I feel like it's worst of my adapters, while the other 2 cost less together than AWUS.
Any ideas?
What could be done to check it?
I would be gratefull for your input - maybe some idead what to try?
I am more than happy to post whatever you might need to help diagnose the situation.

SeaX, 2024-10-30:
Hi I found this thread after searching and testing but without any possibility to increase the tx power. 

Is there anything we could do ?  I agree that at least another adapter has a better reception - I tested another alpha (the ACH) and had great result.

I wan expecting the same results with Wifi6 as bonus - snif :'(

ninjape, 2025-01-10:
any changes so far? I cannot change the txpower at all

sidiqsanjaya, 2025-01-31:
> any changes so far? I cannot change the txpower at all

well on model like MT7921E still have same problem

```
Interface phy1-ap0
        ifindex 12
        wdev 0x100000003
        addr xx:xx:xx:xx:xx
        ssid xxxxx
        type AP
        wiphy 1
        channel 149 (5745 MHz), width: 40 MHz, center1: 5755 MHz
        txpower 3.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                0       0       358     0       0       0       0       82306           401

root@LoliskWrt:~#  iwinfo phy1-ap0 txpower
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
*  3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
root@LoliskWrt:~# iw dev phy1-ap0 get power_save
Power save: off
root@LoliskWrt:~# ls /lib/firmware/mediatek/
WIFI_MT7922_patch_mcu_1_1_hdr.bin  WIFI_RAM_CODE_MT7922_1.bin
WIFI_MT7961_patch_mcu_1_2_hdr.bin  WIFI_RAM_CODE_MT7961_1.bin
root@LoliskWrt:~# lsmod | grep mt7921
cfg80211              724992  6 mt7921_common,ath10k_core,ath,mt76_connac_lib,mt76,mac80211
mac80211              933888  5 mt7921_common,ath10k_core,mt792x_lib,mt76_connac_lib,mt76
mt76                   81920  4 mt7921e,mt7921_common,mt792x_lib,mt76_connac_lib
mt76_connac_lib        49152  3 mt7921e,mt7921_common,mt792x_lib
mt7921_common          57344  1 mt7921e
mt7921e                16384  0
mt792x_lib             40960  2 mt7921e,mt7921_common

```

[image]

razvanphp, 2025-02-07:
I tracked the code for this and it seems that for this chipset particularly the `mt76_phy->txpower_cur` is never initialised, like in the case of other chipsets like mt7603, mt7615, mt76x0, mt7915, etc.

[image]

The implementation seems to be pretty complicated and involves talking with the mcu.
https://github.com/openwrt/mt76/blob/a22d59e4ad50c89326342a0736cd2c1ba32e8a0b/mt7915/mcu.c#L3362-L3404

razvanphp, 2025-02-08:
I have fixed this bug and tested locally with my mt7921u device (Alfa).

I will send the patch to the linux-kernel wireless list.

If anybody wants to test, this diff is based on 23.05 branch:

```diff
diff --git a/mac80211.c b/mac80211.c
index e7b763ba..f0c92818 100644
--- a/mac80211.c
+++ b/mac80211.c
@@ -1544,6 +1544,17 @@ void mt76_wcid_cleanup(struct mt76_dev *dev, struct mt76_wcid *wcid)
 }
 EXPORT_SYMBOL_GPL(mt76_wcid_cleanup);
 
+s8 mt76_get_power_bound(struct mt76_phy *phy, s8 txpower)
+{
+	int n_chains = hweight16(phy->chainmask);
+
+	txpower = mt76_get_sar_power(phy, phy->chandef.chan, txpower * 2);
+	txpower -= mt76_tx_power_nss_delta(n_chains);
+
+	return txpower;
+}
+EXPORT_SYMBOL_GPL(mt76_get_power_bound);
+
 int mt76_get_txpower(struct ieee80211_hw *hw, struct ieee80211_vif *vif,
 		     int *dbm)
 {
diff --git a/mt76.h b/mt76.h
index 2cbea731..2d826c67 100644
--- a/mt76.h
+++ b/mt76.h
@@ -1413,6 +1413,8 @@ void mt76_sta_pre_rcu_remove(struct ieee80211_hw *hw, struct ieee80211_vif *vif,
 
 int mt76_get_min_avg_rssi(struct mt76_dev *dev, bool ext_phy);
 
+s8 mt76_get_power_bound(struct mt76_phy *phy, s8 txpower);
+
 int mt76_get_txpower(struct ieee80211_hw *hw, struct ieee80211_vif *vif,
 		     int *dbm);
 int mt76_init_sar_power(struct ieee80211_hw *hw,
diff --git a/mt7915/debugfs.c b/mt7915/debugfs.c
index 57801388..db8560f7 100644
--- a/mt7915/debugfs.c
+++ b/mt7915/debugfs.c
@@ -1084,13 +1084,13 @@ mt7915_rate_txpower_set(struct file *file, const char __user *user_buf,
 		return -EINVAL;
 
 	if (pwr160)
-		pwr160 = mt7915_get_power_bound(phy, pwr160);
+		pwr160 = mt76_get_power_bound(mphy, pwr160);
 	if (pwr80)
-		pwr80 = mt7915_get_power_bound(phy, pwr80);
+		pwr80 = mt76_get_power_bound(mphy, pwr80);
 	if (pwr40)
-		pwr40 = mt7915_get_power_bound(phy, pwr40);
+		pwr40 = mt76_get_power_bound(mphy, pwr40);
 	if (pwr20)
-		pwr20 = mt7915_get_power_bound(phy, pwr20);
+		pwr20 = mt76_get_power_bound(mphy, pwr20);
 
 	if (pwr160 < 0 || pwr80 < 0 || pwr40 < 0 || pwr20 < 0)
 		return -EINVAL;
diff --git a/mt7915/mcu.c b/mt7915/mcu.c
index fe54a2f4..4ceb10c1 100644
--- a/mt7915/mcu.c
+++ b/mt7915/mcu.c
@@ -3281,7 +3281,7 @@ int mt7915_mcu_set_txpower_frame(struct mt7915_phy *phy,
 	if (ret)
 		return ret;
 
-	txpower = mt7915_get_power_bound(phy, txpower);
+	txpower = mt76_get_power_bound(mphy, txpower);
 	if (txpower > mphy->txpower_cur || txpower < 0)
 		return -EINVAL;
 
@@ -3331,7 +3331,7 @@ int mt7915_mcu_set_txpower_sku(struct mt7915_phy *phy)
 	int i, idx;
 	int tx_power;
 
-	tx_power = mt7915_get_power_bound(phy, hw->conf.power_level);
+	tx_power = mt76_get_power_bound(mphy, hw->conf.power_level);
 	tx_power = mt76_get_rate_power_limits(mphy, mphy->chandef.chan,
 					      &limits_array, tx_power);
 	mphy->txpower_cur = tx_power;
diff --git a/mt7915/mcu.h b/mt7915/mcu.h
index b41ac4aa..44e39feb 100644
--- a/mt7915/mcu.h
+++ b/mt7915/mcu.h
@@ -515,16 +515,4 @@ enum {
 					 sizeof(struct bss_info_bmc_rate) +\
 					 sizeof(struct bss_info_ext_bss))
 
-static inline s8
-mt7915_get_power_bound(struct mt7915_phy *phy, s8 txpower)
-{
-	struct mt76_phy *mphy = phy->mt76;
-	int n_chains = hweight16(mphy->chainmask);
-
-	txpower = mt76_get_sar_power(mphy, mphy->chandef.chan, txpower * 2);
-	txpower -= mt76_tx_power_nss_delta(n_chains);
-
-	return txpower;
-}
-
 #endif
diff --git a/mt7921/main.c b/mt7921/main.c
index 35f8997e..135988ae 100644
--- a/mt7921/main.c
+++ b/mt7921/main.c
@@ -1213,6 +1213,8 @@ int mt7921_set_tx_sar_pwr(struct ieee80211_hw *hw,
 			  const struct cfg80211_sar_specs *sar)
 {
 	struct mt76_phy *mphy = hw->priv;
+	int tx_power;
+	struct mt76_power_limits limits_array;
 
 	if (sar) {
 		int err = mt76_init_sar_power(hw, sar);
@@ -1222,6 +1224,11 @@ int mt7921_set_tx_sar_pwr(struct ieee80211_hw *hw,
 	}
 	mt792x_init_acpi_sar_power(mt792x_hw_phy(hw), !sar);
 
+	tx_power = mt76_get_power_bound(mphy, hw->conf.power_level);
+	tx_power = mt76_get_rate_power_limits(mphy, mphy->chandef.chan,
+					      &limits_array, tx_power);
+	mphy->txpower_cur = tx_power;
+
 	return mt76_connac_mcu_set_rate_txpower(mphy);
 }
 
diff --git a/mt7925/main.c b/mt7925/main.c
index 1f07ec5a..e2081ec9 100644
--- a/mt7925/main.c
+++ b/mt7925/main.c
@@ -1231,6 +1231,8 @@ int mt7925_set_tx_sar_pwr(struct ieee80211_hw *hw,
 			  const struct cfg80211_sar_specs *sar)
 {
 	struct mt76_phy *mphy = hw->priv;
+	int tx_power;
+	struct mt76_power_limits limits_array;
 
 	if (sar) {
 		int err = mt76_init_sar_power(hw, sar);
@@ -1240,6 +1242,11 @@ int mt7925_set_tx_sar_pwr(struct ieee80211_hw *hw,
 	}
 	mt792x_init_acpi_sar_power(mt792x_hw_phy(hw), !sar);
 
+	tx_power = mt76_get_power_bound(mphy, hw->conf.power_level);
+	tx_power = mt76_get_rate_power_limits(mphy, mphy->chandef.chan,
+					      &limits_array, tx_power);
+	mphy->txpower_cur = tx_power;
+
 	return mt7925_mcu_set_rate_txpower(mphy);
 }
```

CoelacanthusHex, 2025-03-06:
> I will send the patch to the linux-kernel wireless list.

I don't know why, but if I apply your patchset (https://lore.kernel.org/linux-wireless/20250211081247.5892-1-razvan.grigore@vampirebyte.ro/) (without the mt7996 one to avoid conflict) to the 6.13.5 kernel on my Arch Linux with mt7925, I get an even worse situation: `iw dev wlan0 info` returns `txpower 3.00 dBm`.

How can I debug it? Or could you provide a patch that will generate useful debug log?

And I don't know whether related, I also applied the patchset from https://lore.kernel.org/linux-wireless/20250226025647.102904-1-sean.wang@kernel.org/ to resolve another connectivity issue.

everything411, 2025-03-07:
after applying the patches `iw wlan0 info` gets `txpower 0.00 dBm` on my mt7925

add some debug print, it shows that
```
[ 6146.321090] (NULL device *): hw->conf.power_level is 0
[ 6146.321099] (NULL device *): mt76_get_power_bound() returns -6
[ 6146.321104] (NULL device *): mt76_get_rate_power_limits() returns -6
[ 6146.485930] (NULL device *): hw->conf.power_level is 0
[ 6146.485935] (NULL device *): mt76_get_power_bound() returns -6
[ 6146.485938] (NULL device *): mt76_get_rate_power_limits() returns -6
```

everything411, 2025-03-07:
according to https://elixir.bootlin.com/linux/v6.13.5/source/drivers/net/wireless/mediatek/mt76/mt7925/mcu.c#L3399 when `hw->conf.power_level` is 0, the txpower is set to 127, so `3dbm` is a false value

everything411, 2025-03-07:
mt7921 set txpower by calling [mt76_connac_mcu_rate_txpower_band](https://elixir.bootlin.com/linux/v6.13.5/C/ident/mt76_connac_mcu_rate_txpower_band) and also has the `if (tx_power) tx_power = 127;` fallback. please log print the `hw->conf.power_level` on the affected adapter to see why the fallback does not work

razvanphp, 2025-03-07:
I only tested the fix on my mt7921, waiting for the mt7925 to arrive and I can retest with that one.

PS: before the patch, what txpower did you see? 3 dBm?

everything411, 2025-03-07:
yes, 3dbm, in sta mode. i have not tested in ap mode and i'm not sure if hw->conf.power_level is always `0` in sta mode.

zimuchen5, 2025-03-07:
huhh, im pretty new in here but how exactly is the patch file? and how do i implement it?

zimuchen5, 2025-03-09:
oh nvm i found out after some research

nbd168, 2025-03-12:
I think the problem is the fact that you can't rely on hw->conf.power_level on drivers that use chanctx. On those drivers, tx power is expected to be configured on the vif. I will revert the mt7921 and mt7925 patches on mt76.git for now until a proper fix emerges.

razvanphp, 2025-03-12:
But what use-case fails? Since it definitely fixed it for me on mt7921.

razvanphp, 2025-03-12:
I did not test on openwrt, I compiled the module on nvidia jetson, kernel 5.15. I can now see the 23dB as set by the region and when changing the txpower manually to a smaller value, it is correctly displayed by `iw`.

nbd168, 2025-03-12:
@razvanphp did you use the 5.15 version of the driver, or did you use backports?
The 5.15 version is very old and does not have chanctx support, so it would make sense that you don't see the issue there.

CoelacanthusHex, 2025-03-25:
> I did not test on openwrt, I compiled the module on nvidia jetson, kernel 5.15.

Please test the patch on torvalds branch or mt76-next if you want send it to linux mainline. Kernel will change many things between minor versions.

Nexarian, 2025-07-28:
What are the next steps to move this issue/fix forward?