openwrt/mt76 #784: mt76x2e: unaligned accesses when loading ROM patch

LGA1150, 2023-06-01:
On ramips/mt7621, set `/sys/kernel/debug/mips/unaligned_action` to 2 and load mt76x2e module, then the dmesg is spammed with stack traces:

```
[   19.144380] mt76x2e 0000:02:00.0: enabling device (0000 -> 0002)
[   19.150616] mt76x2e 0000:02:00.0: ASIC revision: 76120044
[   19.934352] mt76x2e 0000:02:00.0: ROM patch build: 20141115060606a
[   19.946417] CPU: 2 PID: 723 Comm: kmodloader Not tainted 5.15.114 #0
[   19.952801] $ 0   : 00000000 00000001 00000001 000066d3
[   19.958043] $ 4   : c1810000 c1701022 c17076ee 000066d0
[   19.963285] $ 8   : ffffefff 00000000 ffffffea 00000000
[   19.968514] $12   : 815c19f4 00000134 808acb18 ffffffff
[   19.973759] $16   : 832f97a0 00000000 00007612 00000001
[   19.978990] $20   : 00000708 00010000 00010001 00402698
[   19.984233] $24   : 00000001 80426160
[   19.989463] $28   : 815c0000 815c1c40 00000000 8335158c
[   19.994708] Hi    : 0c8efead
[   19.997581] Lo    : 61f8a500
[   20.000446] epc   : 8034cc94 __iowrite32_copy+0x18/0x38
[   20.005684] ra    : 8335158c mt76x2_mcu_init+0x280/0x5c4 [mt76x2e]
[   20.011878] Status: 1100fc03 KERNEL EXL IE
[   20.016090] Cause : 40800010 (ExcCode 04)
[   20.020088] BadVA : c170101e
[   20.022973] PrId  : 0001992f (MIPS 1004Kc)
[   20.027057] Modules linked in: mt76x2e(+) mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 cfg80211 ts_kmp ts_fsm ts_bm tcp_bbr slhc poly1305_mips nfnetlink nf_tproxy_ipv6 nf_tproxy_ipv4 nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c fou6 fou crc_ccitt compat cls_flower chacha_mips act_vlan ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ksmbd asn1_decoder ifb ip6_udp_tunnel udp_tunnel sit ipip ip6_tunnel oid_registry netlink_diag tunnel6 tunnel4 ip_tunnel tun nls_utf8 nls_iso8859_1 nls_cp936 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac ecb des_generic libdes cmac authencesn authenc usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd sd_mod scsi_mod scsi_common ramoops
[   20.027570]  reed_solomon pstore gpio_button_hotplug vfat fat ext4 mbcache jbd2 exfat usbcore nls_base usb_common crc32c_generic
[   20.125649] Process kmodloader (pid: 723, threadinfo=8a4f634f, task=5fcd1be6, tls=77e80dfc)
[   20.134010] Stack : 99f36391 00000000 c1701000 00000001 8335265c 838b4ed4 00000800 832cb780
[   20.142376]         832f97a0 00000000 81534c00 8335265c 00000010 83350f90 83352478 832f97a0
[   20.150751]         fffffff4 832f97a0 00000000 8335265c 00000010 00000000 832f97a0 83350188
[   20.159137]         81534c00 8335265c 76120044 000016e5 00000080 83352478 832f97a0 00000000
[   20.167521]         81515a00 81534c78 83352620 81534c00 8335265c 803e293c 00000010 80436b50
[   20.175916]         ...
[   20.178366] Call Trace:
[   20.180798] [<8034cc94>] __iowrite32_copy+0x18/0x38
[   20.185692] [<8335158c>] mt76x2_mcu_init+0x280/0x5c4 [mt76x2e]
[   20.191532] [<83350f90>] mt76x2_register_device+0x104/0x480 [mt76x2e]
[   20.197989] [<83350188>] init_module+0x362188/0x362870 [mt76x2e]
[   20.204240]
[   20.205730] Code: 10400008  00000000  24a50004 <8ca7fffc> 0000000f  00a6182b  ac870000  1460fffa  24840004
```

lukasz1992, 2023-06-07:
Do not worry about it

LGA1150, 2023-06-12:
> Do not worry about it

It inarguably slows the loading down