openwrt/mt76 #787: Alfa AWUS036ACHM (MT7610U) incorrect txpower (set & displayed)

deutrino, 2023-06-03:
There are a number of issues with this adapter. Some related discussion can be found at

  * https://github.com/openwrt/mt76/issues/633
  * https://github.com/morrownr/USB-WiFi/issues/34

Please understand that while I have development experience, I'm not super familiar with hacking on OpenWRT yet.

Issues visible on SNAPSHOT as of this writing:

  1. `txpower` may be set too low, see also above discussion
  2. `txpower` may also be _reported_ too low as reported above
  3. the maximum `txpower` one can set in Luci is 14dBm
  4. the dongle is detected as a "Sabrent NTWLAC 802.11ac/b/g/n"

I believe some of the issues come from the fact that the Alfa AWUS036ACHM has a separate RF amplifier, so max settable `txpower` and displayed `txpower` are not being properly scaled somehow.

The very low maximum `txpower` which may be observed in e.g. 2.4GHz band is likely related to the above-mentioned issues. I have tried hex editing `mt76x0-common.ko` as described at [this comment](https://github.com/openwrt/mt76/issues/633#issuecomment-1445294226) but it did not improve issues for me.

deutrino, 2023-06-03:
I'm not familiar with OpenWRT development yet, if somebody can give me a link or a list of useful commands to dump info I will be happy to do so for this adapter as long as I have a spare device available with which to do it.

morrownr, 2023-06-12:
> txpower may be set too low, see also above discussion

mt7921 and mt7922 show the same `txpower` low reported number.

RolandoMagico, 2023-07-14:
Hello everyone,
not sure if I face the same issue but I tried to get OpenWrt running on a D-Link M32 Mesh system. The 2.4GHz WiFi doesn't work as expected. Scanning available WLANs is working, but connecting to it doesn't work. If I configure the 2.4GHz Wifi as AP, the SSID isn't visible on most of the clients. A Wifi monitor tells me that the signal is quite weak (about 10-20%) even if the client is directly placed next to the router. I tried it on two devices, same problem on both. And the problem doesn't happen with OEM firmware. With OEM firmware, I have a quite good WiFi signal on 2.4GHz.
Any help would be appreciated