openwrt/mt76 #788: LED initialization issue

hldc0091, 2023-06-03:
I am building for a MT768 eval board.
And I saw that it automatically initialize the LED "MT76-phy0" through the flag CONFIG_MT76_LEDS.

Is there a way to disable this properly?
I tried removing the -DCONFIG_MT76_LEDS flag from the Makefile, and then test it via GPIO control (value 0 and 1),
but it did not work.

rany2, 2023-06-03:
You have to remove it from:

- https://github.com/openwrt/mt76/blob/master/mt7603/Makefile
- https://github.com/openwrt/mt76/blob/master/Makefile

hldc0091, 2023-06-03:
@rany2 Thank you good sir for your comment. I was able to disable the LED from your suggestion. 
Now, actually I want it to define it manually to my dts file, which is why I wanted to disable it in the first place. 

But before that I have to know which GPIO it is. I know the GPIO number, but to be sure I tested all the GPIOs using the [gpio_test.sh](https://openwrt.org/docs/guide-developer/add.new.device#gpio_leds). However, it never turned off or on during the test.

Have I made a mistake from your suggestion? Or am I missing something else?

hldc0091, 2023-06-05:
I was able to fix the issue I am having. 
If I want to initialize the LED through dts, I need to include the "wled_an" like below.

&state_default {
	gpio {
		groups = "i2c", "jtag", "wled_an";
		function = "gpio";
	};
};

closed 2023-06-05 by hldc0091