openwrt/mt76 #790: MT7916 probe failed with error -12

Sev5000, 2023-06-11:
Hi! 
I'm trying to get a MT7916 working under OpenWRT on a Raspberry Pi CM4 on the official IO-Board. So far I've not gotten further than error -12 on the latest snapshot and -22 on stable. From a fresh image I'm installing kmod-mt7916-firmware with all the dependencies that come with it. How can I dig further?

Heres every pci related dmesg line:

...
[    0.175411] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.175446] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    0.175526] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    0.175572] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    0.176255] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    0.176278] pci_bus 0000:00: root bus resource [bus 00-ff]
[    0.176299] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    0.176350] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    0.176432] pci 0000:00:00.0: PME# supported from D0 D3hot
[    0.179949] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    0.180076] pci_bus 0000:01: supply vpcie3v3 not found, using dummy regulator
[    0.180203] pci_bus 0000:01: supply vpcie3v3aux not found, using dummy regulator
[    0.180257] pci_bus 0000:01: supply vpcie12v not found, using dummy regulator
[    0.237564] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    0.237628] pci 0000:01:00.0: [14c3:7906] type 00 class 0x028000
[    0.237678] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit pref]
[    0.237719] pci 0000:01:00.0: reg 0x18: [mem 0x00000000-0x00007fff 64bit]
[    0.237755] pci 0000:01:00.0: reg 0x20: [mem 0x00000000-0x00000fff 64bit pref]
[    0.237896] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    0.237977] pci 0000:01:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0000:00:00.0 (capable of 8.000 Gb/s with 5.0 GT/s PCIe x2 link)
[    0.238219] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    0.238260] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    0.238283] pci 0000:00:00.0: BAR 9: assigned [mem 0x600100000-0x6002fffff 64bit pref]
[    0.238308] pci 0000:01:00.0: BAR 0: assigned [mem 0x600100000-0x6001fffff 64bit pref]
[    0.238344] pci 0000:01:00.0: BAR 2: assigned [mem 0x600000000-0x600007fff 64bit]
[    0.238377] pci 0000:01:00.0: BAR 4: assigned [mem 0x600200000-0x600200fff 64bit pref]
[    0.238411] pci 0000:00:00.0: PCI bridge to [bus 01]
[    0.238428] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    0.238445] pci 0000:00:00.0:   bridge window [mem 0x600100000-0x6002fffff 64bit pref]
[    0.238615] pcieport 0000:00:00.0: enabling device (0000 -> 0002)
[    0.238757] pcieport 0000:00:00.0: PME: Signaling with IRQ 26
[    0.238985] pcieport 0000:00:00.0: AER: enabled with IRQ 26
...
[    5.060671] mt7915e 0000:01:00.0: enabling device (0000 -> 0002)
[    5.166050] mt7915e: probe of 0000:01:00.0 failed with error -12
...

And lspci -v:

01:00.0 Network controller: MEDIATEK Corp. Device 7906
	Subsystem: MEDIATEK Corp. Device 7906
	Flags: fast devsel, IRQ 26
	Memory at 600100000 (64-bit, prefetchable) [size=1M]
	Memory at 600000000 (64-bit, non-prefetchable) [size=32K]
	Memory at 600200000 (64-bit, prefetchable) [size=4K]
	Capabilities: [80] Express Endpoint, MSI 00
	Capabilities: [e0] MSI: Enable- Count=1/32 Maskable+ 64bit+
	Capabilities: [f8] Power Management version 3
	Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [108] Latency Tolerance Reporting
	Capabilities: [110] L1 PM Substates
	Capabilities: [200] Advanced Error Reporting

lts-rad, 2023-06-13:
@sev5000 these cards draw a lot of current, are you providing enough power? Would be curious if anyone has gotten this to work with a pi

Sev5000, 2023-06-18:
Yes, the IO-Board can do 3.3A on the 3.3V line. But I'd guess that's only needed for full throttle tx and not for init, right?
Looking at this: https://twitter.com/will_whang/status/1599637385477705729 it seems at least to be possible (scroll up).
@will127534 what's your secret? :D

Sev5000, 2023-06-18:
reopen -  didnt mean to

will127534, 2023-06-18:
I think it was `dtoverlay=pcie-32bit-dma` that makes it work.

Sev5000, 2023-06-18:
Thank you! That really did fix it

closed 2023-06-18 by Sev5000