openwrt/mt76 #792: mt7981/mt7986 sta connection to AP with AX mode fail

ptpt52, 2023-06-29:
```
7981 client STA(wds)----(broken connection)----7981 AP(wds) with AX
```

ptpt52, 2023-06-29:
@nbd168 
confirmed this issue
and if use AP with AC mode the connection is good:

good:
```
7981 client STA(wds)----(broken connection)----7981 AP(wds) with AC
```

soxrok2212, 2023-06-29:
Same issue as the following:

https://github.com/openwrt/openwrt/issues/12905
https://forum.openwrt.org/t/wireless-client-mode-broken-on-wax206-post-r22866/162406
https://forum.openwrt.org/t/mt7621-wds-with-channel-width-20-mhz-and-ax-not-working-in-newer-snapshots/161942

cmonroe, 2023-06-29:
I think this [patch](https://protect-de.mimecast.com/s/iT8qCmqKZQuEGGoF9l_H6?domain=urldefense.com) should work around the problem you are running into.

Alternately, you could revert [4046b591](https://github.com/openwrt/mt76/commit/4046b59183234a503e8fb017549c6d66576e941d) for now.

The fix above has not made it into mt76 just yet.. it sounds like there's some discussion as to whether the proposed patch above is the ideal way to fix the issue or if it should be handled in mac80211.

ptpt52, 2023-06-29:
@cmonroe  I would like to have this a try.

zhaojh329, 2023-06-30:
> I think this [patch](https://pastebin.mozilla.org/qib2rh8Y/raw) should work around the problem you are running into.
>
> Alternately, you could revert [...]

Good. I tested. It works fine.

cryobry, 2023-07-02:
I'm still having issues after latest merge with stability on WDS client and 802.11ax. Server seems to function fine, WDS on client works fine using 802.11ac 160mhz but with 802.11ax 160mhz it repeatedly drops the connection and gets locked into lower rates, etc. This is using a matched pair of AX6000's, MT7986.

ptpt52, 2023-07-02:
@cryobry  I tested the mt76 latest code, and it looks good.

soxrok2212, 2023-07-02:
Did either of you test with a VAP on the same radio? I'm still getting poor throughput when a radio has a VAP and a mesh node. May/may not be the same with WDS.

cmonroe, 2023-07-02:
@cryobry I have a pair of 2G/6G MT7986 boards using 4mac client mode on the 6G interface of one on a 160MHz channel.. no issues with that. I will try a 5G 4mac client on MT7986 shortly.

@soxrok2212 I thought I saw an issue on a unit which has a 4mac STA (wireless uplink) and 4mac AP (downlink to another mesh node) but have not been able to reproduce it today. Does the problem change for you after reboot or power cycle?

soxrok2212, 2023-07-02:
@cmonroe negative. Poor performance immediately after it boots.

soxrok2212, 2023-07-05:
@nbd168 Seems like when 802.11s is enabled with other VAPs on the same radio, TX on the router gets limited to 20MHz for wireless clients only. Speeds reflect that:
```
xx:xx:xx:xx:xx:xx  -64 dBm / -92 dBm (SNR 28)  0 ms ago
	RX: 907.4 MBit/s, HE-MCS 9, 80MHz, HE-NSS 2, HE-GI 1, HE-DCM 0   1304942 Pkts.
	TX: 286.7 MBit/s, HE-MCS 11, 20MHz, HE-NSS 2, HE-GI 0, HE-DCM 0   1410962 Pkts.
	expected throughput: unknown
```

RX direction is fine, and this doesn't appear to affect the mesh node, only wireless clients.

cryobry, 2023-07-05:
@ptpt52 Indeed, on latest snapshot WDS client has been steady for several hours for me on 160mhz AX. I will report back if there are issues. Thank you.

Djfe, 2025-08-07:
this has likely been fixed by now, has it?

knuxyl, 2025-11-01:
No I am still not able to get STA/client working with this SoC on AX. I didn't know it was related to it being AX but this is certainly still a problem

knuxyl, 2026-02-02:
> > Edit on Jan 29 2026 Still doesn't work with OpenWRT 25.12 RC3. WiFi AC, AX, N, or any bandwidth, all the same result. It 99% of time sits at 6 Mbit/s receive at 20 MHz. Testing on a Banana Pi R3 Mini.
> 
> [@knuxyl](https://github.com/knuxyl) Run a speedtest or iperf

It doesn't communicate at all. I can't communicate with the Access Point, so a speed test will do nothing.

knuxyl, 2026-02-10:
it might be a little bit before i can get to testing this. too much going on atm and my build system is wack right now (gentoo). im keeping this in the back of my mind though, I want this to work because I have 2 useless routers that I can benefit from so I'll come back to this later.