openwrt/mt76 #795: Is a device crashing MT76 driver?

sinancetinkaya, 2023-07-02:
I have an old tablet connected to a Mi Router 4A Gigabit Edition. I see this stack trace time to time when the tablet is connected (Masked the MAC adress)

```
Mon Jul  3 00:43:42 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 18:4f:32:0e:xx:xx
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36443.885465] ------------[ cut here ]------------
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36443.890121] WARNING: CPU: 0 PID: 1330 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2023-05-13-969b7b5e/mt7603/mac.c:208 0x82b946d8 [mt7603e@85af787c+0x9640]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36443.904772] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat sha512_generic seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36443.965229] CPU: 0 PID: 1330 Comm: hostapd Not tainted 5.15.118 #0
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36443.971400] Stack : 00003a98 80000000 00000001 8008599c 00000000 00000000 00000000 00000000
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36443.979761]         00000000 00000000 00000000 00000000 00000000 00000001 8313b928 80c45940
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36443.988112]         8313b9c0 00000000 00000000 8313b7d0 00000038 803a4e04 ffffffea 00000000
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36443.996465]         8313b7dc 0000013a 80824b90 ffffffff 807430d0 8313b908 00000000 82b946d8
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.004818]         00000009 00000001 00003a98 80000000 00000018 80416fec 00000000 809e0000
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.013168]         ...
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.015606] Call Trace:
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.015615] [<8008599c>] 0x8008599c
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.021540] [<803a4e04>] 0x803a4e04
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.025021] [<82b946d8>] 0x82b946d8 [mt7603e@85af787c+0x9640]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.030755] [<80416fec>] 0x80416fec
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.034234] [<80007b00>] 0x80007b00
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.037707] [<80007b08>] 0x80007b08
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.041181] [<82b946d8>] 0x82b946d8 [mt7603e@85af787c+0x9640]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.046910] [<80388e90>] 0x80388e90
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.050384] [<8002f674>] 0x8002f674
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.053859] [<82b946d8>] 0x82b946d8 [mt7603e@85af787c+0x9640]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.059588] [<8002f758>] 0x8002f758
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.063069] [<82b946d8>] 0x82b946d8 [mt7603e@85af787c+0x9640]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.068800] [<82b90000>] 0x82b90000 [mt7603e@85af787c+0x9640]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.074542] [<82b91a24>] 0x82b91a24 [mt7603e@85af787c+0x9640]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.080278] [<829f3a6c>] 0x829f3a6c [mt76@4b0fb6de+0xbb80]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.085751] [<82b4b4f4>] 0x82b4b4f4 [mac80211@4971439f+0x8ca60]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.091655] [<8016f698>] 0x8016f698
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.095135] [<829f3bcc>] 0x829f3bcc [mt76@4b0fb6de+0xbb80]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.100613] [<82b0b598>] 0x82b0b598 [mac80211@4971439f+0x8ca60]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.106517] [<8009b5f4>] 0x8009b5f4
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.109993] [<82b0b8c0>] 0x82b0b8c0 [mac80211@4971439f+0x8ca60]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.115900] [<82a13e70>] 0x82a13e70 [cfg80211@8acf7c5b+0x44950]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.121810] [<805abacc>] 0x805abacc
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.125285] [<82a13d9c>] 0x82a13d9c [cfg80211@8acf7c5b+0x44950]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.131197] [<805ab978>] 0x805ab978
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.134671] [<805aa95c>] 0x805aa95c
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.138150] [<805ab310>] 0x805ab310
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.141625] [<805a9f0c>] 0x805a9f0c
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.145097] [<801bea60>] 0x801bea60
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.148578] [<805aa3b8>] 0x805aa3b8
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.152068] [<805aa048>] 0x805aa048
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.155539] [<804fb664>] 0x804fb664
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.159013] [<8032e640>] 0x8032e640
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.162506] [<804fd5b0>] 0x804fd5b0
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.165990] [<8032e640>] 0x8032e640
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.169486] [<8006534c>] 0x8006534c
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.172958] [<80064c28>] 0x80064c28
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.176452] [<804fd72c>] 0x804fd72c
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.179926] [<804fa1ac>] 0x804fa1ac
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.183408] [<80014550>] 0x80014550
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.186895]
Mon Jul  3 00:43:42 2023 kern.warn kernel: [36444.188655] ---[ end trace 7ddc04dfb3a28668 ]---
Mon Jul  3 00:43:42 2023 daemon.notice hostapd: phy0-ap0: STA 18:4f:32:0e:xx:xx IEEE 802.11: did not acknowledge authentication response
Mon Jul  3 00:45:34 2023 daemon.info hostapd: phy0-ap0: STA 18:4f:32:0e:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

```
config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option band '2g'
	option htmode 'HT20'
	option cell_density '0'
	option country 'TR'
	option channel '13'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option key 'XXXXXXXXXX'
	option encryption 'psk2'
	option ssid 'XXXXXXX'

```

lukasz1992, 2023-07-03:
Sure it can

rany2, 2023-07-03:
It's not a fatal error, this line raises a **warning** when there is still something stuck still being transmitted: https://github.com/openwrt/mt76/blob/969b7b5e/mt7603/mac.c#L208. If everything keeps working as usual, I don't think there is a real cause for concern; it's not a crash.

sinancetinkaya, 2023-07-03:
Actually the problem was worse than that. Previously I was using 2.4ghz for `client` connection in between routers. Wireless was going down for a few minutes.
```
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.602027] pci 0000:00:01.0: BAR 8: assigned [mem 0x60200000-0x602fffff]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.608816] pci 0000:00:01.0: BAR 9: assigned [mem 0x60300000-0x603fffff pref]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.616034] pci 0000:00:00.0: BAR 1: assigned [mem 0x60400000-0x6040ffff]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.622798] pci 0000:00:01.0: BAR 1: assigned [mem 0x60410000-0x6041ffff]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.629589] pci 0000:00:00.0: BAR 7: assigned [io  0x1e160000-0x1e160fff]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.636374] pci 0000:00:01.0: BAR 7: assigned [io  0x1e161000-0x1e161fff]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.643182] pci 0000:01:00.0: BAR 0: assigned [mem 0x60000000-0x600fffff 64bit]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.650472] pci 0000:01:00.0: BAR 6: assigned [mem 0x60100000-0x6010ffff pref]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.657685] pci 0000:00:00.0: PCI bridge to [bus 01]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.662630] pci 0000:00:00.0:   bridge window [io  0x1e160000-0x1e160fff]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.669416] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.676199] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.683427] pci 0000:02:00.0: BAR 0: assigned [mem 0x60200000-0x602fffff]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.690188] pci 0000:00:01.0: PCI bridge to [bus 02]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.695159] pci 0000:00:01.0:   bridge window [io  0x1e161000-0x1e161fff]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.701918] pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.708713] pci 0000:00:01.0:   bridge window [mem 0x60300000-0x603fffff pref]
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.722406] mt7530 mdio-bus:1f: Link is Up - 1Gbps/Full - flow control rx/tx
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.730680] VFS: Mounted root (squashfs filesystem) readonly on device 31:9.
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.741801] Freeing unused kernel memory: 1252K
Sat Jul  1 16:47:07 2023 kern.warn kernel: [    1.746388] This architecture does not have kernel memory protection.
Sat Jul  1 16:47:07 2023 kern.info kernel: [    1.752818] Run /sbin/init as init process
Sat Jul  1 16:47:07 2023 kern.debug kernel: [    1.757006]   with arguments:
Sat Jul  1 16:47:07 2023 kern.debug kernel: [    1.757014]     /sbin/init
Sat Jul  1 16:47:07 2023 kern.debug kernel: [    1.757019]   with environment:
Sat Jul  1 16:47:07 2023 kern.debug kernel: [    1.757024]     HOME=/
Sat Jul  1 16:47:07 2023 kern.debug kernel: [    1.757029]     TERM=linux
Sat Jul  1 16:47:07 2023 user.info kernel: [    2.310143] init: Console is alive
Sat Jul  1 16:47:07 2023 user.info kernel: [    2.314013] init: - watchdog -
Sat Jul  1 16:47:07 2023 user.info kernel: [    3.005167] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Sat Jul  1 16:47:07 2023 user.info kernel: [    3.125082] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Sat Jul  1 16:47:07 2023 user.info kernel: [    3.133318] init: - preinit -
Sat Jul  1 16:47:07 2023 kern.notice kernel: [    3.973963] random: jshn: uninitialized urandom read (4 bytes read)
Sat Jul  1 16:47:07 2023 kern.notice kernel: [    4.124006] random: jshn: uninitialized urandom read (4 bytes read)
Sat Jul  1 16:47:07 2023 kern.notice kernel: [    4.161121] random: jshn: uninitialized urandom read (4 bytes read)
Sat Jul  1 16:47:07 2023 kern.info kernel: [    4.402166] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
Sat Jul  1 16:47:07 2023 kern.info kernel: [    4.411081] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Sat Jul  1 16:47:07 2023 kern.info kernel: [    4.413018] mt7530 mdio-bus:1f lan1: configuring for phy/gmii link mode
Sat Jul  1 16:47:07 2023 kern.info kernel: [    4.426511] 8021q: adding VLAN 0 to HW filter on device lan1
Sat Jul  1 16:47:07 2023 kern.info kernel: [    4.434811] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sat Jul  1 16:47:07 2023 kern.notice kernel: [    8.757162] jffs2: notice: (564) jffs2_build_xattr_subsystem: complete building xattr subsystem, 53 of xdatum (38 unchecked, 15 orphan) and 75 of xref (16 dead, 0 orphan) found.
Sat Jul  1 16:47:07 2023 user.info kernel: [    8.777686] mount_root: switching to jffs2 overlay
Sat Jul  1 16:47:07 2023 kern.warn kernel: [    8.788100] overlayfs: upper fs does not support tmpfile.
Sat Jul  1 16:47:07 2023 user.warn kernel: [    8.816570] urandom-seed: Seeding with /etc/urandom.seed
Sat Jul  1 16:47:07 2023 user.info kernel: [    9.009421] procd: - early -
Sat Jul  1 16:47:07 2023 user.info kernel: [    9.012528] procd: - watchdog -
Sat Jul  1 16:47:07 2023 user.info kernel: [    9.663058] procd: - watchdog -
Sat Jul  1 16:47:07 2023 user.info kernel: [    9.667379] procd: - ubus -
Sat Jul  1 16:47:07 2023 kern.notice kernel: [    9.750867] random: ubusd: uninitialized urandom read (4 bytes read)
[... cut ...]
Sat Jul  1 18:08:48 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.4.239 xx:xx:xx:xx:xx
Sat Jul  1 18:08:48 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) xx:xx:xx:xx:xx
Sat Jul  1 18:08:48 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.4.239 xx:xx:xx:xx:xx
Sat Jul  1 18:08:48 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.4.239 xx:xx:xx:xx:xx
Sat Jul  1 18:08:48 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.4.239 xx:xx:xx:xx:xx CoreELEC
Sat Jul  1 18:09:17 2023 user.info : luci: accepted login on / for root from 192.168.4.239

```

rany2, 2023-07-04:
@sinancetinkaya You could try out newer firmware, mt7603 in particular has out of date firmware in mt76 tree (not because of a fault from the developers, but because MediaTek hasn't provided up-to-date firmware blobs for Linux). However people have extracted some up-to-date firmware from OEM firmware with the latest one extracted is available here: https://github.com/openwrt/mt76/pull/295#issuecomment-968864364.

sinancetinkaya, 2023-07-04:
@rany2 but there are two firmware files in OpenWRT, which one should I replace? 
[image]

rany2, 2023-07-04:
@sinancetinkaya You will only need to replace `mt7603_e2.bin`

closed 2023-07-04 by sinancetinkaya

rany2, 2023-07-04:
@sinancetinkaya Did my suggestion fix it for you?