openwrt/mt76 #796: RX rate reverts to 6mbit/20MHz periodically or semi-permanently

VA1DER, 2023-07-07:
Devices: mt7915, mt7986
Modes: 2.4GHz & 5GHz AC and AX modes
Symptoms: RX speeds in either AP or STA modes will periodically (once per minute) revert to 6mbit/20MHz, regardless of SNR or signal strength.

Typically once every 60 seconds the RX speed will revert to 6mbit/20MHz, stay there for 4 or 5 seconds, then increase back to an expected rate.  In some instances the RX rate will remain at 6mbit/20MHz for extended periods, only briefly increasing to expected rates.

[image]

This happens on both 2.4GHz and 5GHz connections on affected devices.  So far affected devices are mt7915 and mt7986. 

Originally believed to be associated with openwrt/openwrt#13003 but now appears separate.

danpawlik, 2023-07-10:
Is it not something related to the WNM? What wpad version (basic/full) you have installed?

jww-cw, 2023-07-12:
I am only seeing this on some devices on the 2.4GHz MT7622

Samsung A33 5G  has the issue.

Xiaomi Mi Lite 5G does not have the issue.

@VA1DER have you tested with multiple 2.4GHZ devices.

fightforlife, 2023-07-21:
I just found the same issue. Client is a Xiaomi Poco F1 with latest Lineage os 20. I also have a camera which shows the same behaviour. Dcs6100

Model GL-MT300N-V2
Architecture MediaTek MT7628AN ver:1 eco:2
Target Platform ramips/mt76x8
Firmware Version OpenWrt SNAPSHOT r23589-08607da37c / LuCI Master git-23.158.78004-23a246e
Kernel Version 5.15.120

gca31, 2023-07-23:
I've got the same issue with a BPI-R3 (mt7986, snapshot r23589-08607da37c) configured as AP and various STA clients (netgear WAX206 with mt7622 and mt7975 running OEM firmware, apple devices, windows devices and ESP32 devices) whatever the Wifi (N,AC,AX).  I've got also a variant with TX at 6mbits/20MHz and another variant with RX at 24mbits/20MHz
All pictures captured from the BPI-R3 AP

RX rate at 6Mbits/20MHz - Wifi N
[image]

TX rate at 6Mbits/20MHz - Wifi N
[image]

Variant RX rate at 24Mbits/20MHz - Wifi AC
[image]

RX rate at 6Mbits/20MHz - Wifi AX
[image]

fightforlife, 2023-07-23:
I think this might be power saving initiated by the clients. 
I can reproduce it by putting my Android phone into standby. Lock and leave for 15secs.
When I unlock it again and open the browser the speed goes up again.

lukasz1992, 2023-07-24:
Yes, it is power saving and not the issue.

Device and driver can only report what was received - if device wishes to transmit frames at lowest possible speed, you see 6mbit.

gca31, 2023-07-24:
Ok, thanks for the explanation.
Therefore, if we all agree on this, does it make sense this issue has to be closed ? @VA1DER ?

Djfe, 2023-07-27:
This is also likely what happens, when you don't run your wireless adapter on "maximum performance" on windows.
I noticed decreased performance to an AP (not openwrt), but this setting lowers power consumption (great for battery but at the cost of performance)
On Android you can't set this manually.
I agree, this is a non issue and intentional behaviour.

VA1DER, 2023-07-28:
Consensus is intended behaviour.  Clogins.

closed 2023-07-28 by VA1DER