openwrt/mt76 #800: MT7610U Crash hostpad

Gabriele121, 2023-07-15:
I have an ALFA AWUS036ACHM with MT7610U chipset, the issue is that when I try to open an AP with hostapd the terminal crash and I cannot execute Network related command. Also I have to restart and then everything works fine. I tried to search online but nothing. Any ideas? Sorry for my English.

cat hostapd.conf
interface=wlan0mon
driver=nl80211
ssid=test
hw_mode=g
channel=8
macaddr_acl=0
ignore_broadcast_ssid=0

sudo dmesg | grep mt
[    9.537924] mt76x0u 1-9:1.0: ASIC revision: 76100002 MAC revision: 76502000
[   12.285066] mt76x0u 1-9:1.0: EEPROM ver:02 fae:04
[   12.545680] usbcore: registered new interface driver mt76x0u
[   12.550040] mt76x0u 1-9:1.0 wlx00c0cab10aab: renamed from wlan0
[  311.469035] mt76x0u 1-9:1.0: EEPROM ver:02 fae:04
[ 2558.701290] mt76x0u 1-9:1.0: MAC error detected
[ 2560.814899] mt76x0u 1-9:1.0: ASIC revision: 76100002 MAC revision: 76502000
[ 2564.390613] mt76x0u 1-9:1.0: EEPROM ver:02 fae:04
[ 2564.683967] mt76x0u 1-9:1.0 wlx00c0cab10aab: renamed from wlan0

sudo lshw -C network

*-network
       description: Wireless interface
       physical id: 4
       bus info: usb@1:9
       logical name: wlx00c0cab10aab
       serial: 00:c0:ca:b1:0a:ab
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=mt76x0u driverversion=5.19.0-46-generic firmware=N/A link=no multicast=yes wireless=IEEE 802.11

hcombee, 2023-07-17:
Hi,

I see some similar errors in my logging.

I am running OpenWRT 23.05.0-rc2 on a Netgear R7800 and I am trying to use an Alfa Tube-U on the USB3 port of the Netgear. When I try to connect this router as a client using this Alfa (which also has the MT7610u) the connection crashes.
This also happened when I was using OpenWRT22.03.

[  299.103117] usb 3-1: USB disconnect, device number 3
[  299.154502] mt76x0u 3-1:1.0: MAC error detected
[  299.366477] mt76x0u 3-1:1.0: MAC stop failed
[  299.755468] mt76x0u 3-1:1.0: MAC error detected
[  299.970784] mt76x0u 3-1:1.0: MAC stop failed
[  300.404775] usb 3-1: new high-speed USB device number 4 using xhci-hcd
[  300.734836] usb 3-1: reset high-speed USB device number 4 using xhci-hcd
[  300.919138] mt76x0u 3-1:1.0: ASIC revision: 76100002 MAC revision: 76502000
[  301.248283] mt76x0u 3-1:1.0: EEPROM ver:02 fae:04
[  301.305974] ieee80211 phy4: Selected rate control algorithm 'minstrel_ht'
[  308.511904] usb 3-1: USB disconnect, device number 4
[  308.564604] mt76x0u 3-1:1.0: MAC error detected
[  308.789336] mt76x0u 3-1:1.0: MAC stop failed
[  309.217436] mt76x0u 3-1:1.0: MAC error detected
[  309.440096] mt76x0u 3-1:1.0: MAC stop failed
[  309.864287] usb 3-1: new high-speed USB device number 5 using xhci-hcd
[  310.205007] usb 3-1: reset high-speed USB device number 5 using xhci-hcd
[  310.388138] mt76x0u 3-1:1.0: ASIC revision: 76100002 MAC revision: 76502000
[  310.689918] mt76x0u 3-1:1.0: EEPROM ver:02 fae:04

brada4, 2023-08-03:
Looks more like USB (?power?) issue? Try changing USB2/USB3, and avoid any extra cabling and unpowered hubs in the middle.
```
[ 308.511904] usb 3-1: USB disconnect, device number 4
[ 309.864287] usb 3-1: new high-speed USB device number 5 using xhci-hcd
[ 310.205007] usb 3-1: reset high-speed USB device number 5 using xhci-hcd
```

hcombee, 2023-08-04:
Thanks for pointing this out @brada4. I will look further into this direction.