openwrt/mt76 #801: New firmware testing

szyper-sakrona, 2023-07-26:
Could you include/test new firmware?

https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/refs/heads/master/autobuild_mac80211_release/package/kernel/mt76/src/firmware

this is mt7915 version is:
mt7915e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20230418151304a
mt7915e 0000:02:00.0: WM Firmware Version: ____000000, Build Time: 20230418151336
mt7915e 0000:02:00.0: WA Firmware Version: DEV_000000, Build Time: 20230418151358

Thanks!
So fart so good on my mt7915 machines!

Fail-Safe, 2024-01-06:
There is even newer firmware for many mt76 chips here: https://svn.dd-wrt.com/browser/src/router/mac80211/drivers/net/wireless/mediatek/mt76/firmware

szyper-sakrona, 2024-03-04:
Recently this firmware (20231018...) give many problems with timouts and wifi not working properly. Reverting back to 20230418... gives normal work of my router with mt7915e chipset.

lukasz1992, 2024-03-04:
You can also check newer firmwares: https://github.com/cmonroe/feed-wifi-master/tree/smartrg-master/mt76/files/firmware

I use them and do not have any problems.