openwrt/mt76 #803: need help in fixing: Cudy WR1300 v2 5GHz wifi led not working - mt7613

Djfe, 2023-07-30:
I'm trying to get the WR1300 v2 LED (5GHz) to work but haven't had any success so far. (the 2.4GHz led of the mt7603 is working fine) 

I tried setting the led_pin in `/sys/kernel/debug/ieee80211/phyx/mt76` and I also tried libgpiod/gpiod-tools for testing GPIOs (sadly only character device version 1 is supported so far)
gpiod-tools didn't make any of the LEDs light up, I hope that's just cause other gpios are already claimed/in use by the kernel (because they were set in the dts) and not because this method doesn't work

other mt7613 define their led outside of the pcie node as a normal led node (or not at all). are these just special cases or is the mt7615/mt7663 firmware not able to control mt7613 LEDs, yet? (because mt7615 already has some led code, that I'm sure of, but maybe just for dbdc or regular mt7615 and not mt7613?)

DragonBluep, 2023-08-18:
Maybe you can have a look. https://github.com/openwrt/mt76/pull/651

DragonBluep, 2023-08-18:
The WiFi LED setting in the stock firmware:
```
disable_mt7663e() {
	iwpriv rai0 set led_setting=1-0-0-0-2-0-0-1
}

enable_mt7663e() {
		[ "$ifname" = rai0 ] && iwpriv rai0 set led_setting=1-0-0-0-2-0-0-2
}
```
[radio1-mt7663e.zip](https://github.com/openwrt/mt76/files/12379905/radio1-mt7663e.zip)

Djfe, 2023-08-18:
thanks for taking a closer look. much appreciated.
The stuff you linked, was never merged, right?
Do you expect this to need a change in mt76 (by patch in openwrt or as a patch to send to the mailing list later on)

Djfe, 2023-08-18:
I surely could address your feedback on #651 and test whether it works this way (defining via dts)

DragonBluep, 2023-08-18:
Before that, you can replace `asus,rt-ac1200-v2` in the patch with `cudy,wr1300-v2` to see if it works for you.

Djfe, 2023-08-18:
that's a great idea thank you