openwrt/mt76 #804: OpenWrt on MT7621/MT7615N devices with 5GHz problems

Rexadev, 2023-07-30:
https://forum.openwrt.org/t/openwrt-on-mt7621-mt7615n-devices-with-5ghz-problems/107392

# Slow speed over D-Link Dir 2640 OpenWrt 22.03.5

ONT

Hardware TP-Link XC220-G3v Firmware :1.2.0 0.8.0 v6062.0

***Over 5Ghz 80Mhz Wifi -*** Download 146Mbps Upload 109Mbps

D-Link Dir 2640

OpenWrt 22.03.5

***Over 5Ghz 80Mhz Wifi -*** Download 61.2Mbps Upload 25.2Mbps

with Qosify disabled - Download 96.6Mbps Upload 107Mbps

Packages build using firmware selector

```
luci
luci-app-wireguard luci-app-pbr
stubby qosify
kmod-ipt-geoip 
netdata 
usbutils kmod-usb-storage-uas block-mount kmod-fs-ext4 kmod-usb-storage kmod-usb3 kmod-usb2 kmod-usb-core
```

Speedtest using https://speed.cloudflare.com/

danpawlik, 2023-08-16:
Maybe better share the `/etc/config/wireless` configuration?

DragonBluep, 2023-09-21:
This patch fixes the issue for me on an unofficial supported device with external 5 GHz PA.

For the external PA devices, it seems that MT_EE_EXT_PA_{2,5}G_TARGET_POWER (0xf2 0xf3) eeprom value will be sent to the MCU as DAC output strength reference. I think it's wrong because it contains the PA gain. This leads to PA working in the nonlinear amplification region.

Some devices even don't have a valid MT_EE_EXT_PA_{2,5}G_TARGET_POWER value. For example, on raisecom msg1500 it's `0x00` which caused the 6dbm low tx power issue https://github.com/openwrt/mt76/issues/574.

Forum thread:
https://forum.openwrt.org/t/openwrt-on-mt7621-mt7615n-devices-with-5ghz-problems/107392/288

MT7615 EEPROM content:
[MT7615 EEPROM Content.pdf](https://github.com/openwrt/mt76/files/12684454/MT7615.EEPROM.Content.pdf)
[image]
[image]

By the way, the mtk vendor driver seems only use a SKU file to control the tx power.
```
Subject: [PATCH] mt7615: fix exPA TX target power value
---
 mt7615/eeprom.c | 9 ---------
 mt7615/init.c   | 2 +-
 2 files changed, 1 insertion(+), 10 deletions(-)

--- a/mt7615/eeprom.c
+++ b/mt7615/eeprom.c
@@ -204,15 +204,6 @@ int mt7615_eeprom_get_target_power_index
 	if (chain_idx > 3)
 		return -EINVAL;
 
-	/* TSSI disabled */
-	if (mt7615_ext_pa_enabled(dev, chan->band)) {
-		if (chan->band == NL80211_BAND_2GHZ)
-			return MT_EE_EXT_PA_2G_TARGET_POWER;
-		else
-			return MT_EE_EXT_PA_5G_TARGET_POWER;
-	}
-
-	/* TSSI enabled */
 	if (chan->band == NL80211_BAND_2GHZ) {
 		index = MT_EE_TX0_2G_TARGET_POWER + chain_idx * 6;
 	} else {
--- a/mt7615/init.c
+++ b/mt7615/init.c
@@ -285,7 +285,7 @@ void mt7615_init_txpower(struct mt7615_d
 	    (MT_EE_RATE_POWER_EN | MT_EE_RATE_POWER_SIGN))
 		delta += rate_val & MT_EE_RATE_POWER_MASK;
 
-	if (!is_mt7663(&dev->mt76) && mt7615_ext_pa_enabled(dev, band))
+	if (!is_mt7663(&dev->mt76) && !mt7615_ext_pa_enabled(dev, band))
 		target_chains = 1;
 	else
 		target_chains = n_chains;
```
Friendly ping @nbd168

DragonBluep, 2023-09-22:
I found even mtk eeprom example doesn't set the EEPROM MT_EE_EXT_PA_{2,5}G_TARGET_POWER (0xf2 0xf3), so I believe reading target power here is wrong.
https://github.com/hanwckf/padavan-4.4/blob/ff77acc51a4ad5d88660f0ac25a34a1f6f9060c6/trunk/linux-4.4.x/drivers/net/wireless/mediatek/mt7615/embedded/include/eeprom/mt7615_e2p_ePAeLNA.h#L21C9-L21C9

LZeugirdor, 2023-11-03:
I think I may be having this issue and over long periods of time it's causing disconnects. 
`
[   30.664031] mt7530 mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   35.410154] wlan1: authenticate with 00:00:00:00:00:00
[   35.768139] wlan1: send auth to 00:00:00:00:00:00 (try 1/3)
[   35.793226] wlan1: authenticated
[   35.808967] wlan1: associate with 00:00:00:00:00:00 (try 1/3)
[   35.825881] wlan1: RX AssocResp from 00:00:00:00:00:00 (capab=0x1011 status=0 aid=1)
[   35.841877] wlan1: associated
[   35.984431] wlan1: Limiting TX power to 30 (30 - 0) dBm as advertised by 00:00:00:00:00:00
[   37.099589] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  168.520086] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  168.531109] mt76x2e 0000:01:00.0: Build: 1
[  168.539518] mt76x2e 0000:01:00.0: Build Time: 201607111443____
[  168.568918] mt76x2e 0000:01:00.0: Firmware running!
[  168.589022] ieee80211 phy1: Hardware restart was requested
[  463.405380] wlan1: deauthenticating from 00:00:00:00:00:00 by local choice (Reason: 3=DEAUTH_LEAVING)
[  470.445148] wlan1: authenticate with 00:00:00:00:00:00
[  470.807920] wlan1: send auth to 00:00:00:00:00:00 (try 1/3)
[  470.819872] wlan1: authenticated
[  470.829945] wlan1: associate with 00:00:00:00:00:00 (try 1/3)
[  470.845192] wlan1: RX AssocResp from 00:00:00:00:00:00 (capab=0x1011 status=0 aid=1)
[  470.861100] wlan1: associated
[  470.878827] wlan1: Limiting TX power to 30 (30 - 0) dBm as advertised by 00:00:00:00:00:00
[  470.980916] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  964.630297] wlan1: deauthenticating from 00:00:00:00:00:00 by local choice (Reason: 3=DEAUTH_LEAVING)
[  972.708422] wlan1: authenticate with 00:00:00:00:00:00
[  973.071013] wlan1: send auth to 00:00:00:00:00:00 (try 1/3)
[  973.083156] wlan1: authenticated
[  973.092976] wlan1: associate with 00:00:00:00:00:00 (try 1/3)
[  973.108320] wlan1: RX AssocResp from 00:00:00:00:00:00 (capab=0x1011 status=0 aid=1)
[  973.124285] wlan1: associated
[  973.234024] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  973.247487] wlan1: Limiting TX power to 30 (30 - 0) dBm as advertised by 00:00:00:00:00:00
[ 1375.434853] wlan1: deauthenticating from 00:00:00:00:00:00 by local choice (Reason: 3=DEAUTH_LEAVING)
[ 1381.048705] wlan1: authenticate with 00:00:00:00:00:00
[ 1381.729709] wlan1: send auth to 00:00:00:00:00:00 (try 1/3)
[ 1381.742274] wlan1: authenticated
[ 1381.751930] wlan1: associate with 00:00:00:00:00:00 (try 1/3)
[ 1381.766967] wlan1: RX AssocResp from 00:00:00:00:00:00 (capab=0x1011 status=0 aid=1)
[ 1381.782987] wlan1: associated
[ 1381.852308] wlan1: Limiting TX power to 30 (30 - 0) dBm as advertised by 00:00:00:00:00:00
[ 1381.904231] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[ 1498.942651] wlan1: cannot understand ECSA IE operating class, 22, ignoring
`
all mac addresses filtered for privacy