openwrt/mt76 #808: mt76x0e 0000:01:00.0: MAC error detected - Netgear R6220

Camis, 2023-08-15:
OpenWrt SNAPSHOT r23727-e6f8b69918

Log:

`Mon Aug 14 18:25:18 2023 kern.err kernel: [10768.226872] mt76x2e 0000:01:00.0: MAC error detected
Mon Aug 14 18:25:19 2023 kern.err kernel: [10769.466845] mt76x2e 0000:01:00.0: MAC error detected
Mon Aug 14 18:25:21 2023 kern.err kernel: [10770.706909] mt76x2e 0000:01:00.0: MAC error detected
Mon Aug 14 18:25:22 2023 kern.err kernel: [10772.056800] mt76x2e 0000:01:00.0: MAC error detected
Mon Aug 14 18:25:23 2023 kern.err kernel: [10773.296777] mt76x2e 0000:01:00.0: MAC error detected
...
`

`config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
	option channel '3'
	option band '2g'
	option htmode 'HT20'
	option cell_density '0'
	option country 'PL'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel '149'
	option band '5g'
	option htmode 'VHT80'
	option country 'PL'
	option cell_density '0'

config wifi-iface 'wifinet0'
	option device 'radio0'
	option mode 'ap'
	option ssid 'AP-****'
	option encryption 'sae'
	option key '*******'
	option ieee80211r '1'
	option mobility_domain '6F16'
	option ft_over_ds '0'
	option ft_psk_generate_local '1'
	option network 'lan'

config wifi-iface 'wifinet1'
	option device 'radio1'
	option mode 'ap'
	option ssid 'AP-****'
	option encryption 'sae'
	option key '*******'
	option ieee80211r '1'
	option mobility_domain '6F16'
	option ft_over_ds '0'
	option ft_psk_generate_local '1'
	option network 'lan'
`

quantumbeat, 2024-01-04:
same here
`[   47.963090] mt76x2e 0000:01:00.0: MAC error detected
[   49.223048] mt76x2e 0000:01:00.0: MAC error detected
[   68.413129] mt76x2e 0000:01:00.0: MAC error detected
[  221.152773] mt76x2e 0000:01:00.0: MAC error detected
[  280.142235] mt76x2e 0000:01:00.0: MAC error detected
[  288.992114] mt76x2e 0000:01:00.0: MAC error detected
[  295.312057] mt76x2e 0000:01:00.0: MAC error detected
[  415.261105] mt76x2e 0000:01:00.0: MAC error detected
[  450.960871] mt76x2e 0000:01:00.0: MAC error detected
[  465.090803] mt76x2e 0000:01:00.0: MAC error detected
[  466.900730] mt76x2e 0000:01:00.0: MAC error detected
[  477.290648] mt76x2e 0000:01:00.0: MAC error detected
[ 1090.004303] mt76x2e 0000:01:00.0: MAC error detected
[ 1094.674239] mt76x2e 0000:01:00.0: MAC error detected
[ 1108.254373] mt76x2e 0000:01:00.0: MAC error detected`

serj-47, 2024-08-27:
I had a similar problem, "mac error detected" appeared periodically in the logs, especially during the high utilization of the 5g channel. Next to my router was another router with 5g disabled, but to which an external usb hdd was connected. At first I had suspicions that this was interference from the usb bus from a neighboring router and I moved the second router with hdd away from the first one, I also tried to change txpower, but all this did not help and the mac error still haunted me. As a result, I just switched to another channel 36=>52, and there have been no problems for a couple of months now.

```
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.path='1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
wireless.radio1.band='5g'
wireless.radio1.disabled='0'
wireless.radio1.cell_density='0'
wireless.radio1.country='RU'
wireless.radio1.htmode='VHT80'
wireless.radio1.txpower='20'
wireless.radio1.channel='52'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid=
wireless.default_radio1.encryption='psk2+ccmp'
wireless.default_radio1.key=
```