openwrt/mt76 #815: WED not working on mt7986

tkso1997, 2023-08-27:
I am running a self compiled OpenWRT (kernen 6.1.x) on my ASUS RT-AX59U which uses mt7986 as wireless chip.The system runs fine and both 2.4 and 5GHz are working nicely. For some reason it is not possible to activate WED. 

When adding "options mt7915e wed_enable=Y" to /etc/modules.conf wife still works normal and a "cat /sys/module/mt7915e/parameters/wed_enable gives me a "Y" which implies it is activated, but the CPU load is the same as before when WIFI is under load. In addition there is no information saying that the offload firmware was successfully loaded in dmesg.

```
[ 12.196554] mt798x-wmac 18000000.wifi: HW/SW Version: 0x8a108a10, Build Time: 20221012174648a
[ 12.196554]
[ 12.548727] mt798x-wmac 18000000.wifi: WM Firmware Version: ____000000, Build Time: 20221012174725
[ 12.710033] mt798x-wmac 18000000.wifi: WA Firmware Version: DEV_000000, Build Time: 20221012174937
[ 12.857014] mt798x-wmac 18000000.wifi: eeprom load fail, use default bin
[ 15.788977] PPP generic driver version 2.4.2
[ 15.794019] NET: Registered PF_PPPOX protocol family
```

I have all firmware files in /lib/firmware/mediatek
`ls
mt7986_eeprom_mt7975.bin	
mt7986_rom_patch_mt7975.bin
mt7986_eeprom_mt7975_dual.bin	
mt7986_wa.bin
mt7986_eeprom_mt7976.bin	
mt7986_wm.bin
mt7986_eeprom_mt7976_dbdc.bin	
mt7986_wm_mt7975.bin
mt7986_eeprom_mt7976_dual.bin	
mt7986_wo_0.bin
mt7986_rom_patch.bin		
mt7986_wo_1.bin`

Is there anything a can to to figure out why its not working/debug it?

lukasz1992, 2023-08-27:
maybe dts misses something to enable it?

zhaojh329, 2023-08-31:
https://openwrt.org/toh/linksys/e8450#wireless_ethernet_dispatch_wed_hw_acceleration_for_wireless_clients

lukasz1992, 2023-08-31:
> I have this commit to fix wed on mt7981
> 
> ```diff
> diff --git a/mt7915/main.c b/mt7915/main.c
> index 9b79e915..fc9ec9df 100644
> --- a/mt7915/main.c
> +++ b/mt7915/main.c
> @@ -1632,7 +1632,7 @@ mt7915_net_fill_forward_path(struct ieee80211_hw *hw,
>         path->dev = ctx->dev;
>         path->mtk_wdma.wdma_idx = wed->wdma_idx;
>         path->mtk_wdma.bss = mvif->mt76.idx;
> -       path->mtk_wdma.wcid = is_mt7915(&dev->mt76) ? msta->wcid.idx : 0x3ff;
> +       path->mtk_wdma.wcid = (is_mt7915(&dev->mt76) || is_mt7981(&dev->mt76)) ? msta->wcid.idx : 0x3ff;
>         path->mtk_wdma.queue = phy != &dev->phy;
>  
>         ctx->dev = NULL;
> ```

@ptpt52 do you have any special use case? For me it looks like it works without this patch; I use 23.05 branch

ptpt52, 2023-08-31:
@lukasz1992 
wed could not work without this patch for my 7981 device.
is your device 7981 ?

nicefile, 2023-09-05:
@ptpt52 @lukasz1992 
I can also confirm WED works for me using 23.05-rc3 on Cudy WRC3000 mt7981 with mt7981_wo.bin added. 
On the other side not all traffic is offloaded and shows in `/sys/kernel/debug/ppe0/bind` . Bridger restart  usually help for a while

ptpt52, 2023-09-20:
very strange I can confirm that WED cannot work without my patch.

JiaY-shi, 2024-01-05:
@tkso1997 hi buddy, have you solved this problem? I have the same problem on gl-mt6000, with the 6.1 kernel. After activate WED, I tested the wifi and the CPU had a high load.

gssjshark, 2024-03-02:
> @tkso1997 hi buddy, have you solved this problem? I have the same problem on gl-mt6000, with the 6.1 kernel. After activate WED, I tested the wifi and the CPU had a high load.

I’m having the same issue. Did you get this resolved?

rodsmar, 2024-06-21:
In my AX3000T (mt7981 cpu) WED works flawless with this patch.

> I have this commit to fix wed on mt7981
>
> ```diff
> diff --git a/mt7915/main.c b/mt7915/main.c
> index 9b79e915..fc9ec9df 100644
> --- a/mt7915/main.c
> +++ b/mt7915/main.c
> @@ -1632,7 +1632,7 @@ [...]

Why not submit an PR to merge this in mt7915 code?

gssjshark, 2024-06-21:
> In my AX3000T (mt7981 cpu) WED works flawless with this patch.
>
>
> Why not submit an PR to merge this in mt7915 code?

Were you having similar issues as above before applying this patch?

pcslide, 2024-07-15:
@ptpt52 
On ex5700(mt7986+mt7916 device), wed works on mt7916, however, it does not work on mt7986.

may be we should use instead
` path->mtk_wdma.wcid = (is_mt7915(&dev->mt76) || is_mt798x(&dev->mt76)) ? msta->wcid.idx : 0x3ff;`
Any idea?

Headcrabed, 2024-07-15:
> In my AX3000T (mt7981 cpu) WED works flawless with this patch.
>
>
> Why not submit an PR to merge this in mt7915 code?

@rodsmar I'm porting this downstream patch to mainline. It would fix this problem. If you can test this patch, can you give me a tested-by?
https://patchwork.kernel.org/project/linux-wireless/patch/OS3P286MB259748F6FA6BE628C3295D7498D92@OS3P286MB2597.JPNP286.PROD.OUTLOOK.COM/